arendst/Tasmota #19244: Incorrect power values and units (TELEINFO)

jojolll, 2023-08-02:
### PROBLEM DESCRIPTION
When using tasmota's TELEINFO feature with a Linky meter in France, the power values are displayed in the wrong units and some don't make sense. This applies to "Standard mode (9600 baud)". I discussed this with Charles Hallard some time ago, but it seems that the problem lies with Tasmota and not its library.

In history mode, only active power is displayed by Tasmota. The problem is the same (wrong unit), but less false information is displayed.

### REQUESTED INFORMATION

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): ESP32-S2
- [X] Tasmota binary firmware version number used: 13.0.0.3(teleinfo)
  - [X] Pre-compiled
  - [ ] Self-compiled
- [X] Flashing tools used: Tasmota official web flasher
- [X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
10:31:41.104 MQT: stat/tasmota_EAF356/RESULT = {"NAME":"Wemos Teleinfo","GPIO":[1,1,1,1,1,1,1,1376,1,1,1,5632,1,1,1,1,1,1,1,1,1,1,640,1,608,1,0,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":1}
10:31:41.326 MQT: stat/tasmota_EAF356/RESULT = {"Module":{"0":"Wemos Teleinfo"}}
10:31:41.581 MQT: stat/tasmota_EAF356/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO6":{"0":"None"},"GPIO7":{"1376":"WS28121"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"5632":"TInfo Rx"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"0":"None"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"0":"None"},"GPIO33":{"640":"I2C SDA1"},"GPIO34":{"0":"None"},"GPIO35":{"608":"I2C SCL1"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"},"GPIO40":{"0":"None"},"GPIO41":{"0":"None"},"GPIO42":{"0":"None"},"GPIO43":{"0":"None"},"GPIO44":{"0":"None"},"GPIO45":{"0":"None"},"GPIO46":{"0":"None"}}
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [X] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
10:32:22.559 MQT: stat/tasmota_EAF356/STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_EAF356","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
10:32:22.564 MQT: stat/tasmota_EAF356/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"https://github.com/tasmota/install/raw/main/firmware/unofficial/tasmota32-teleinfo.bin","RestartReason":"Software reset digital core","Uptime":"0T20:55:10","StartupUTC":"2023-08-01T12:37:12","Sleep":50,"CfgHolder":4617,"BootCount":7,"BCResetTime":"2023-08-01T13:19:13","SaveCount":19}}
10:32:22.568 MQT: stat/tasmota_EAF356/STATUS2 = {"StatusFWR":{"Version":"13.0.0.3(teleinfo)","BuildDateTime":"2023-07-31T16:09:58","Core":"2_0_11","SDK":"4.4.5.230722","CpuFrequency":240,"Hardware":"ESP32-S2FNR2","CR":"456/699"}}
10:32:22.572 MQT: stat/tasmota_EAF356/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Livebox-17E6",""],"TelePeriod":300,"Resolution":"568180C0","SetOption":["00008009","2805C80001000600003C5A00192800000000","00400080","00006000","00004000","00000000"]}}
10:32:22.581 MQT: stat/tasmota_EAF356/STATUS4 = {"StatusMEM":{"ProgramSize":1350,"Free":1529,"Heap":108,"StackLowMark":4,"PsrMax":2045,"PsrFree":2025,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"164020","FlashFrequency":80,"FlashMode":"DIO","Features":["00000809","8FB087CF","001093A1","14BF17CF","41001331","583A0D51","000400A0","00200000","5400002C","00000080"],"Drivers":"1,2,3,4,7,8,9,10,12,13,16,20,21,24,27,29,38,50,52,62","Sensors":"1,2,3,5,6,7,8,9,10,13,14,15,16,17,18,20,29,54,55,57,58,59,63,127","I2CDriver":"3,4,5,6,7,8,9,10,11,14,15,16,22,35,36,40,41,42,43,47"}}
10:32:22.586 MQT: stat/tasmota_EAF356/STATUS5 = {"StatusNET":{"Hostname":"tasmota-EAF356-4950","IPAddress":"192.168.1.196","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.1.4","DNSServer2":"192.168.1.12","Mac":"84:F7:03:EA:F3:56","IP6Global":"","IP6Local":"fe80::86f7:3ff:feea:f356","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":15.0}}
10:32:22.590 MQT: stat/tasmota_EAF356/STATUS6 = {"StatusMQT":{"MqttHost":"homeassistant.lan","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_EAF356","MqttUser":"teleinfo","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
10:32:22.595 MQT: stat/tasmota_EAF356/STATUS7 = {"StatusTIM":{"UTC":"2023-08-02T09:32:22","Local":"2023-08-02T10:32:22","StartDST":"2023-03-26T02:00:00","EndDST":"2023-10-29T03:00:00","Timezone":"+01:00","Sunrise":"05:24","Sunset":"20:27"}}
10:32:22.600 MQT: stat/tasmota_EAF356/STATUS9 = {"StatusPTH":{"PowerDelta":[0,0,0,0,0,0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
10:32:22.660 MQT: stat/tasmota_EAF356/STATUS10 = {"StatusSNS":{"Time":"2023-08-02T10:32:22","ENERGY":{"TotalStartTime":"2023-08-01T13:20:12","Total":11274.61426,"Yesterday":3.20507,"Today":1.24218,"Power":457,"ApparentPower":464,"ReactivePower":80,"Factor":0.98,"Voltage":232,"Current":2.000,"Load":6},"TIC":{"ADSC":"811775540966","VTIC":2,"NGTF":"TEMPO","LTARF":"HP  BLEU","EAST":11274614,"EASF01":10709704,"EASF02":498982,"EASF03":17077,"EASF04":48851,"EASF05":0,"EASF06":0,"EASF07":0,"EASF08":0,"EASF09":0,"EASF10":0,"EASD01":11274614,"EASD02":0,"EASD03":0,"EASD04":0,"IRMS1":2,"URMS1":232,"PREF":6,"PCOUP":6,"SINSTS":457,"SMAXSN":460,"SMAXSN-1":360,"UMOY1":230,"STGE":"013A0401","PRM":2147483647,"RELAIS":0,"NTARF":2,"NJOURF":0,"NJOURF+1":0}}}
10:32:22.671 MQT: stat/tasmota_EAF356/STATUS11 = {"StatusSTS":{"Time":"2023-08-02T10:32:22","Uptime":"0T20:55:10","UptimeSec":75310,"Heap":108,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Berry":{"HeapUsed":3,"Objects":35},"POWER":"OFF","Dimmer":10,"Color":"1A1A1A","HSBColor":"0,0,10","Channel":[10,10,10],"Scheme":0,"Width":1,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"Livebox-17E6","BSSId":"50:39:2F:CA:73:D0","Channel":6,"Mode":"11n","RSSI":64,"Signal":-68,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
This is not applicable, the problem is permanent

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
- Install Tasmota on an ESP (versions 12 and 13 tested) with the Teleinfo option enabled
- Use a Linky meter supported by Teleinfo
- Wrong information displayed on ESP dashboard

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Due to its protocol, the Linky connected meter does not return the following information: 
- Active power
- Reactive power
- Power factor

However, these data are displayed by Tasmota, using a calculation that I don't understand. 

The only instantaneous consumption information returned by Linky is apparent power, maximum apparent power (in the last 10 minutes), voltage and current rounded to the unit (so without any precision, the current is not usable for calculations).

Tasmota's display should therefore only show apparent power in VA, with the correct value. 
A second problem is that, at present, the power reported by Linky as apparent power (SINSTS in VA) is interpreted by Tasmota as active power (in W). The apparent power currently displayed by Tasmota makes no sense and seems to be calculated, but I can't figure it out (I couldn't find the calculation in the code).

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
Here's an example at time T in Standard mode. Tasmota retourne ces informations : 
[image]

On the Linky meter screen, the actual information is : 
- Apparent Power = 406 VA (SINSTS)
- Max Apparent Power = 460 VA (SMAXSN)
and that's it for the instantaneous values. 

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
Linky documentation (French) : https://www.enedis.fr/media/2035/download

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2023-08-02:
@hallard Seems the values are swapped.

scrouillou, 2023-08-10:
Hi,

seems to me that the issue comes from [tasmota](https://github.com/arendst/Tasmota/tree/d85557711d176100b293d6e203a8ccfa3286426f/tasmota)/[tasmota_xnrg_energy/xnrg_15_teleinfo.ino](https://github.com/arendst/Tasmota/blob/d85557711d176100b293d6e203a8ccfa3286426f/tasmota/tasmota_xnrg_energy/xnrg_15_teleinfo.ino#L334) Lines 328-340 :
```
 else if (ilabel == LABEL_PAPP || ilabel == LABEL_SINSTS || ilabel == LABEL_SINSTS1 || ilabel == LABEL_SINSTS2 || ilabel == LABEL_SINSTS3)
        {
           [...]

            if (ilabel == LABEL_PAPP || ilabel == LABEL_SINSTS1 || (ilabel == LABEL_SINSTS && Energy->phase_count == 1)) {
                Energy->active_power[0] = power;
            } else if (ilabel == LABEL_SINSTS2) {
                Energy->active_power[1] = power;
            } else if (ilabel == LABEL_SINSTS3) {
                Energy->active_power[2] = power;
            }
        }
```

Where `active_power` is filled instead of `apparent_power`

I do not know where and how the other powers are computed from the `active_power`, so it is unknown to me if changing those lines to `apparent_power` will fix the issue or if another action is necessary to disable/change the computation of the other data.

Cheers.

closed 2023-08-25 by arendst

hallard, 2023-08-25:
Just to leave a note on this one, all values displayed above the progress bar are values calculated and filled by Energy Module (Tasmota core) but with values filled by Teleinfo.
Anyway I also added Apparent Power below the progress bar (this one direct from SmartMeter) just to see if any difference (sounds like not) and this will also display the 3 phases power in case of 3 phase contract below the progress bar.
But now I've filled apparent power to Energy Module this one report Active Power as 0 (because we don't have Active Power from SmartMeter). Don't know if we can remove this value on WEB UI @arendst or if any hints to makes things better on my side.

jojolll, 2023-08-25:
Thanks for the fix ! 

I just flashed my module, the information is now where it should be.

However, I have a small display error in STANDARD mode : reactive power is equal to active power. This may be an automatic calculation since there are zero watts (Q² = S²-P²).

I couldn't test the HISTORY mode because of my counter settings.

PR #19381 (2023-08-25): Teleinfo power fix
## Description:

Teleinfo Fix Apparent Power Value / Unit Display

**Related issue :** fixes #19244

## Checklist:
  - [x] The pull request is done against the latest development branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR and the code change compiles without warnings
  - [x] The code change is tested and works with Tasmota core ESP8266 V.2.7.4.9
  - [x] The code change is tested and works with Tasmota core ESP32 V.2.0.11
  - [x] I accept the [CLA](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md#contributo [... cut]