arendst/Tasmota #19262: Shelly EM do not report energy consumption except if the relay is set to ON

JanickGers85, 2023-08-05:
### PROBLEM DESCRIPTION

Shelly EM do not report energy consumption except if the relay is set to ON.

### REQUESTED INFORMATION

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Shelly EM
- [ ] Tasmota binary firmware version number used: 13.0.0
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: [MgOS to Tasmota](https://github.com/arendst/mgos-to-tasmota), then flashed the tasmota-IT via OTA
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
12:01:19.604 MQT: stat/shelly-em-garage/RESULT = {"NAME":"Shelly EM","GPIO":[0,0,0,0,0,0,0,0,640,3456,608,224,0,1],"FLAG":0,"BASE":18}
12:01:19.820 MQT: stat/shelly-em-garage/RESULT = {"Module":{"0":"Shelly EM"}}
12:01:20.027 MQT: stat/shelly-em-garage/RESULT = {"GPIO0":{"0":"Nessuno"},"GPIO1":{"0":"Nessuno"},"GPIO2":{"0":"Nessuno"},"GPIO3":{"0":"Nessuno"},"GPIO4":{"0":"Nessuno"},"GPIO5":{"0":"Nessuno"},"GPIO9":{"0":"Nessuno"},"GPIO10":{"0":"Nessuno"},"GPIO12":{"640":"I2C - SDA"},"GPIO13":{"3456":"ADE7953 - IRQ1"},"GPIO14":{"608":"I2C - SCL"},"GPIO15":{"224":"Relè1"},"GPIO16":{"0":"Nessuno"},"GPIO17":{"0":"Nessuno"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
No rules
```
- [x] Provide the output of this command: `Status 0`:
```lua
12:02:39.585 MQT: stat/shelly-em-garage/STATUS = {"Status":{"Module":0,"DeviceName":"Shelly EM - Garage","FriendlyName":["Shelly EM - Garage"],"Topic":"shelly-em-garage","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
12:02:39.593 MQT: stat/shelly-em-garage/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota-IT.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:23:22","StartupUTC":"2023-08-05T09:39:17","Sleep":50,"CfgHolder":4617,"BootCount":11,"BCResetTime":"2023-08-04T22:50:57","SaveCount":46,"SaveAddress":"FB000"}}
12:02:39.598 MQT: stat/shelly-em-garage/STATUS2 = {"StatusFWR":{"Version":"13.0.0(tasmota)","BuildDateTime":"2023-06-26T10:59:43","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"409/699"}}
12:02:39.605 MQT: stat/shelly-em-garage/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Dino-HOME",""],"TelePeriod":10,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A192800000000","00000080","00006000","00004000","00000000"]}}
12:02:39.620 MQT: stat/shelly-em-garage/STATUS4 = {"StatusMEM":{"ProgramSize":635,"Free":368,"Heap":23,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1620C2","FlashFrequency":40,"FlashMode":"DOUT","Features":["00000410","8F9AC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","54000020","00000080"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45,62,68","Sensors":"1,2,3,4,5,6","I2CDriver":"7"}}
12:02:39.628 MQT: stat/shelly-em-garage/STATUS5 = {"StatusNET":{"Hostname":"shelly-em-garage-4958","IPAddress":"192.168.1.96","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.1.1","DNSServer2":"0.0.0.0","Mac":"E8:68:E7:D2:B3:5E","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
12:02:39.634 MQT: stat/shelly-em-garage/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.217","MqttPort":1883,"MqttClientMask":"shelly-em-garage","MqttClient":"shelly-em-garage","MqttUser":"mosquitto","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
12:02:39.643 MQT: stat/shelly-em-garage/STATUS7 = {"StatusTIM":{"UTC":"2023-08-05T10:02:39","Local":"2023-08-05T12:02:39","StartDST":"2023-03-26T02:00:00","EndDST":"2023-10-29T03:00:00","Timezone":99,"Sunrise":"06:05","Sunset":"20:30"}}
12:02:39.650 MQT: stat/shelly-em-garage/STATUS9 = {"StatusPTH":{"PowerDelta":[0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
12:02:39.664 MQT: stat/shelly-em-garage/STATUS10 = {"StatusSNS":{"Time":"2023-08-05T12:02:39","ENERGY":{"TotalStartTime":"2023-08-04T23:23:36","Total":0.001,"Yesterday":0.000,"Today":0.001,"Power":[0,0],"ApparentPower":[0,0],"ReactivePower":[0,0],"Factor":[0.00,0.00],"Frequency":0,"Voltage":0,"Current":[0.000,0.000]}}}
12:02:39.673 MQT: stat/shelly-em-garage/STATUS11 = {"StatusSTS":{"Time":"2023-08-05T12:02:39","Uptime":"0T00:23:22","UptimeSec":1402,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Dino-HOME","BSSId":"DC:39:6F:EF:E0:B5","Channel":1,"Mode":"11n","RSSI":58,"Signal":-71,"LinkCount":1,"Downtime":"0T00:00:03"}}}
```
- [x] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
Not applicable.
```

### TO REPRODUCE
Plug a log to one of the channel (any of the two).
With relay off there no consumption
Switching on power consuption is read.

### EXPECTED BEHAVIOUR
Misuring power consumption with relay off.

### SCREENSHOTS
With relay off and load on channel.
[image]

With relay on and load on channel.
[image]
### ADDITIONAL CONTEXT
Shelly EM is connected to MQTT, data reported is coerent with the ones shown on the UI.

arendst, 2023-08-05:
Try ``SO21 1``

JanickGers85, 2023-08-05:
That was quick... didn't spot it on the documentation.
Fastest issue on GitHub Award, closing 😆

closed 2023-08-05 by JanickGers85