arendst/Tasmota #19271: WeMos D1 mini ADC temperature wrong

Duvio, 2023-08-06:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
When connecting an NTC thermistor to A0 with an external voltage divider on a WeMos D1 mini and selecting ADC temp for that pin in the module configuration, the calculated temperature is wrong. I suspect this is because this board already has an internal voltage divider.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [x] Tasmota binary firmware version number used: 13.0.0
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: esptool
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
3.3v ---- NTC thermistor ---- A0 ---- resistor ---- GND
The resistor is 10kO and the thermistor is 10kO at 25°C and b=3435.
In module configuration set the module to generic and afterwards A0 to ADC temp.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
The temperature to be displayed correctly.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2023-08-06:
If the calculated temperature is not what you'd want, it is up to you to use AdcParam to get it to be better. You may be right about the voltage divider, but that's not something you should expect Tasmota to fix for you.

Duvio, 2023-08-06:
> If the calculated temperature is not what you'd want, it is up to you to use AdcParam to get it to be better. You may be right about the voltage divider, but that's not something you should expect Tasmota to fix for you.

I already set the values using AdcParam correctly, the value shown is still wrong. Do you mean i have to manually factor in the voltage dividers and tweak the AdcParam accordingly?

barbudor, 2023-08-06:
The Tasmota NTC is meant for a NTC connected exactly like in the schematic shown in the code of the analog driver (sorry, it's in the ino code as comment, not in the docs, PR welcomed) which how it is connected on the Shelly 1

Any other connection circuit may need adaptation.

lalo-uy, 2023-08-06:
There is a problem in adc term code with wemos D1.
The code for ESP8266 asume the adc input range of 1V and 3.3v for ESP32.
That do not takes into account the wemos divider.

I reported some months ago and patched my own code.

In line 529 remove the ANALOG_V33

Rt = (double)Adc[idx].param1 * (ANALOG_RANGE * ANALOG_V33 - (double)adc) / (
double)adc;