arendst/Tasmota #19291: Tasmota-ir is compiled with the `USE_DEEPSLEEP` feature enabled by default but deepsleep does not join build

LockeKK, 2023-08-10:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

Tasmota-ir is compiled with the `USE_DEEPSLEEP` feature enabled by default. However, the deepsleep driver `xdrv_29_deepsleep.ino `is not included in the build. As a result, both tasmota-ir and tasmota32-ir do not have the deepsleep function. This poses a challenge if we require a battery-powered IR controller.

The standard versions of tasmota and tasmota32 support IRSend, IRReceive, and Deepsleep, but they do not support IRHVAC. I have been unable to find a complete build that meets my requirements. I kindly request that future ir builds include the deepsleep feature and that the standard build supports IRHVAC.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): esp32_devkitc_v4
- [ ] Tasmota binary firmware version number used: 13.0.0.3(ir)-2_0_11
  - [x ] Pre-compiled
  - [x] Self-compiled
- [x] Flashing tools used: PlatformIO_VS code
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
N/A

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
N/A

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
N/A

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
N/A

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Build Tasmota-ir or Tasmota32-ir, check `USE_DEEPSLEEP` is defined by default, but build out firmware does not have DeepSleepTime command.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Tasmota-ir or Tasmota32-ir should have deep sleep support.
Or
Tasmota or Tasmota32 should support IRHVAC

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

closed 2023-08-10 by Jason2866

LockeKK, 2023-08-10:
Thanks for the clearify.
It turns out my `#define USE_DEEPSLEEP` in user_config_override.h was ignored in ir build.
So finally I forced xdrv_29_deepsleep.ino to join ir build then I got deep sleep enabled.
Thanks.

barbudor, 2023-08-10:
Don't forget rule number 1
[image]
https://tasmota.github.io/docs/Compile-your-build/#customize-your-build

LockeKK, 2023-08-11:
> Don't forget rule number 1 [image] https://tasmota.github.io/docs/Compile-your-build/#customize-your-build

Thanks. I did not follow this rule while enable deepsleep into ir build.

I also tried to enable `USE_IR_REMOTE_FULL` in user_config_override.h as I need IRHVAC to control my AC but there is link issue(I did delete all .cache files).

`#ifndef USE_IR_REMOTE_FULL`
`#define USE_IR_REMOTE_FULL`
`#endif  // USE_IR_REMOTE_FULL`

But never mind deepsleep in my customized ir build is working.
I am happy to close this ticket and great thanks to your support.