arendst/Tasmota #19296: ESP32-S2 AP and web server not working correctly after flashing with dfu-util

vinibali, 2023-08-11:
### PROBLEM DESCRIPTION
The AP has a weird name and after connecting to that it's not showing the web UI.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): Wemos S2 mini
- [X] Tasmota binary firmware version number used: 13.0.0.4
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [X] Flashing tools used: dfu-util
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
N/A
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
N/A
```
- [ ] Provide the output of this command: `Status 0`:
```lua
N/A
```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
N/A
```

### TO REPRODUCE
* flash any of the tasmota32s2 binaries (like tasmota32s2.factory.bin or tasmota32s2cdc.factory.bin) with dfu-util
* the AP will have some scratchy name and the web server will not work on that

### EXPECTED BEHAVIOUR
* flash any of the tasmota32s2 binaries (like tasmota32s2.factory.bin or tasmota32s2cdc.factory.bin) with dfu-util
* the AP will have the default tasmota-like name and the web server will work on that

### SCREENSHOTS

### ADDITIONAL CONTEXT
Used command, none of the binaries worked.
```
sudo dfu-util -d 303a:0002 --download tasmota32s2.factory.bin -R
sudo dfu-util -d 303a:0002 --download tasmota32s2cdc.factory.bin -R
```
kernel message shows some real information:
```
Aug 11 05:50:22 PC kernel: usb 4-1: new full-speed USB device number 113 using xhci_hcd                                                                            
Aug 11 05:50:22 PC kernel: usb 4-1: New USB device found, idVendor=303a, idProduct=0002, bcdDevice= 1.00                                                           
Aug 11 05:50:22 PC kernel: usb 4-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3                                                                       
Aug 11 05:50:22 PC kernel: usb 4-1: Product: Espressif Generic ESP32-S2 >= 4M Flash PSRAM, Tasmota 2880k Code/OTA, 320k FS   
Aug 11 05:50:22 PC kernel: usb 4-1: Manufacturer: Espressif Systems                                                                                                
Aug 11 05:50:22 PC kernel: usb 4-1: SerialNumber: 0                                                                                                                
Aug 11 05:50:22 PC kernel: cdc_acm 4-1:1.0: ttyACM0: USB ACM device
```
Flasing ESPEasy the same way works as expected

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2023-08-11:
If it does not work for you when not using a method recommended for flashing Tasmota, better use a recommended method instead. That includes esptool.py for command-line operation.

vinibali, 2023-08-11:
I didn't know that the /dev/ttyACM0 exist this time. I was searching for something like /dev/ttyCDC*.
Thanks for that!
/closed

closed 2023-08-11 by vinibali