arendst/Tasmota #19323: command: Webbutton<x>

Mark-Muc, 2023-08-16:
### Discussed in https://github.com/arendst/Tasmota/discussions/19274

<div type='discussions-op-text'>

<sup>Originally posted by **Mark-Muc** August  7, 2023</sup>
Hello .. just made a W32 board with port expander .. working fine  and stable.
I've added 25 relays and 12 physical buttons .. so far so good..
I'am using the command: `webbutton<x>` to display some information/status on the button .. 
now i found out .. it only works until <x> <= 16 .. 

`webbutton17 Hallo!!`
in the console. Result:

```
08:14:21.091 CMD: webbutton17 Hallo!!
08:14:21.099 MQT: stat/LCN_M5_12_Port/RESULT = {"Command":"Error"}
```

x untill 16 is working!! 
Due to that u can choose Relays untill 28 .. so the command `webbutton `should be enhanced!!
Thx and Regards
Mark_Muc

Any News on this??
</div>

sfromis, 2023-08-16:
Do no open new threads, and do not use "Issues" when you want enhancements. Instead the proper place is Discussions/Ideas.

Mark-Muc, 2023-08-16:
ok i will .. sorry
Mark_Muc