arendst/Tasmota #19328: ESP32 Shutterposition RESULT Inverted

RedBlackThunder, 2023-08-16:
### PROBLEM DESCRIPTION

Tasmota reports ShutterPosition on ESP32 always Inverted.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Sonoff SwitchMan 2C
- [ ] Tasmota binary firmware version number used: 13.1.0
  - [ ] Pre-compiled
  - [x] Self-compiled
- [x] Flashing tools used: OTA
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [x] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
16:17:52.842 MQT: tasmota/EG/SZ/Shutter/stat/RESULT = {"Shutter1":{"Position":70,"Direction":0,"Target":70,"Tilt":0}}
16:18:40.715 CMD: ShutterPosition 70
16:18:40.730 MQT: tasmota/EG/SZ/Shutter/stat/RESULT = {"POWER2":"ON"}
16:18:40.733 MQT: tasmota/EG/SZ/Shutter/stat/POWER2 = ON
16:18:40.737 MQT: tasmota/EG/SZ/Shutter/stat/RESULT = {"Shutter1":{"Position":70,"Direction":-1,"Target":30,"Tilt":0}}
16:18:40.914 MQT: tasmota/EG/SZ/Shutter/stat/RESULT = {"Shutter1":{"Position":69,"Direction":-1,"Target":30,"Tilt":0}}
16:18:41.917 MQT: tasmota/EG/SZ/Shutter/stat/RESULT = {"Shutter1":{"Position":61,"Direction":-1,"Target":30,"Tilt":0}}
16:18:42.916 MQT: tasmota/EG/SZ/Shutter/stat/RESULT = {"Shutter1":{"Position":53,"Direction":-1,"Target":30,"Tilt":0}}
16:18:43.867 MQT: tasmota/EG/SZ/Shutter/stat/RESULT = {"Shutter1":{"Position":46,"Direction":-1,"Target":30,"Tilt":0}}
16:18:44.916 MQT: tasmota/EG/SZ/Shutter/stat/RESULT = {"Shutter1":{"Position":37,"Direction":-1,"Target":30,"Tilt":0}}
16:18:45.822 MQT: tasmota/EG/SZ/Shutter/stat/RESULT = {"POWER2":"OFF"}
16:18:45.832 MQT: tasmota/EG/SZ/Shutter/stat/POWER2 = OFF
16:18:45.841 MQT: tasmota/EG/SZ/Shutter/stat/SHUTTER1 = 30
16:18:45.850 MQT: tasmota/EG/SZ/Shutter/stat/RESULT = {"Shutter1":{"Position":30,"Direction":0,"Target":30,"Tilt":0}}
16:18:45.928 SHT: About to save settings to file /.drvset027
16:21:00.215 CMD: ShutterInvert
16:21:00.224 MQT: tasmota/EG/SZ/Shutter/stat/RESULT = {"ShutterInvert1":0}

```

### TO REPRODUCE
Set ShutterPosition to $value (doesn't matter if ShutterInverted 0/1)

ShutterPosition is interpreted as inverted value when set.

### EXPECTED BEHAVIOUR

Reported ShutterPosition same as set ShutterPosition

**(Please, remember to close the issue when the problem has been addressed)**

stefanbode, 2023-08-17:
As this was tested to work very strange. Can you share some logs pls

stefanbode, 2023-08-18:
I badly messed around with || and && junction in an IF statement. No Idea why I did not see this in the test; or the test went well and this was a last minute change....

SteWers, 2023-08-18:
Looks good for me. 👍 Thanks!

sternegugger, 2023-08-20:
I have a similar problem: **`ShutterInvert1 1` does not invert the reported position** as described in [Documentation SutterInvert table](https://tasmota.github.io/docs/Blinds-and-Shutters/#operation)
<code>
19:42:39.015 CMD: **shutterinvert1 1**
19:42:39.026 MQT: Rolladen/stat/RESULT = {"ShutterInvert1":1}
19:42:45.764 CMD: shutteropen1
19:42:45.795 MQT: Rolladen/stat/RESULT = {"POWER2":"OFF"}
19:42:45.802 MQT: Rolladen/stat/POWER2 = OFF
19:42:45.927 MQT: Rolladen/stat/RESULT = {"POWER1":"ON"}
19:42:45.936 MQT: Rolladen/stat/POWER1 = ON
19:42:45.946 MQT: Rolladen/stat/RESULT = {"Shutter1":{"Position":90,"Direction":1,"Target":100,"Tilt":0}}
19:42:47.008 MQT: Rolladen/stat/RESULT = {"Shutter1":{"Position":93,"Direction":1,"Target":100,"Tilt":0},"Shutter2":{"Position":100,"Direction":0,"Target":100,"Tilt":0},"Shutter3":{"Position":100,"Direction":0,"Target":100,"Tilt":0},"Shutter4":{"Position":100,"Direction":0,"Target":100,"Tilt":0},"Shutter5":{"Position":48,"Direction":0,"Target":48,"Tilt":0},"Shutter6":{"Position":100,"Direction":0,"Target":100,"Tilt":0},"Shutter7":{"Position":100,"Direction":0,"Target":100,"Tilt":0},"Shutter8":{"Position":100,"Direction":0,"Target":100,"Tilt":0},"Shutter9":{"Position":0,"Direction":0,"Target":0,"Tilt":0},"Shutter10":{"Position":100,"Direction":0,"Target":100,"Tilt":0},"Shutter11":{"Position":100,"Direction":0,"Target":100,"Tilt":0},"Shutter12":{"Position":0,"Direction":0,"Target":0,"Tilt":0}}
19:42:48.009 MQT: Rolladen/stat/RESULT = {"Shutter1":{"Position":97,"Direction":1,"Target":100,"Tilt":0},"Shutter2":{"Position":100,"Direction":0,"Target":100,"Tilt":0},"Shutter3":{"Position":100,"Direction":0,"Target":100,"Tilt":0},"Shutter4":{"Position":100,"Direction":0,"Target":100,"Tilt":0},"Shutter5":{"Position":48,"Direction":0,"Target":48,"Tilt":0},"Shutter6":{"Position":100,"Direction":0,"Target":100,"Tilt":0},"Shutter7":{"Position":100,"Direction":0,"Target":100,"Tilt":0},"Shutter8":{"Position":100,"Direction":0,"Target":100,"Tilt":0},"Shutter9":{"Position":0,"Direction":0,"Target":0,"Tilt":0},"Shutter10":{"Position":100,"Direction":0,"Target":100,"Tilt":0},"Shutter11":{"Position":100,"Direction":0,"Target":100,"Tilt":0},"Shutter12":{"Position":0,"Direction":0,"Target":0,"Tilt":0}}
19:42:48.999 MQT: Rolladen/stat/RESULT = {"Shutter1":{"Position":100,"Direction":1,"Target":100,"Tilt":0},"Shutter2":{"Position":100,"Direction":0,"Target":100,"Tilt":0},"Shutter3":{"Position":100,"Direction":0,"Target":100,"Tilt":0},"Shutter4":{"Position":100,"Direction":0,"Target":100,"Tilt":0},"Shutter5":{"Position":48,"Direction":0,"Target":48,"Tilt":0},"Shutter6":{"Position":100,"Direction":0,"Target":100,"Tilt":0},"Shutter7":{"Position":100,"Direction":0,"Target":100,"Tilt":0},"Shutter8":{"Position":100,"Direction":0,"Target":100,"Tilt":0},"Shutter9":{"Position":0,"Direction":0,"Target":0,"Tilt":0},"Shutter10":{"Position":100,"Direction":0,"Target":100,"Tilt":0},"Shutter11":{"Position":100,"Direction":0,"Target":100,"Tilt":0},"Shutter12":{"Position":0,"Direction":0,"Target":0,"Tilt":0}}
19:42:49.146 MQT: Rolladen/stat/RESULT = {"POWER1":"OFF"}
19:42:49.155 MQT: Rolladen/stat/POWER1 = OFF
19:42:49.163 MQT: Rolladen/stat/**SHUTTER1 = 100**
19:42:49.174 MQT: Rolladen/stat/RESULT = {"Shutter1":{"Position":100,"Direction":0,"Target":100,"Tilt":0}}
</code>

**Expected:** on `ShutterInvert1 1`, `ShutterOpen1` should result in `SHUTTER1 = 0`

Tested with Firmware from latest Development Branch (20. August 2023 tasmota32-DE) with ''#define USE_MCP23XXX_DRV''.
I am using an ESP32 with 3 MCP23017 to control 24 relays and 24 switches for up to 12 shutters.

stefanbode, 2023-08-22:
Yes it was a ESP32 only issue. Fix above should solve this one. the shutterbutton bug was also found == instead of =. Small change, high impact. This is tested and ESP32 and should also work now

sternegugger, 2023-08-22:
@stefanbode additionally to my last comment:
Your fix works for shutter1 up to shutter9. :-)
Beginning with shutter9 the position is allways invers as expected. :-(

first 9 shutters:
shutterInvert 0 -> open=100% close = 0%
shutterInvert 1 -> open=0% close=100%

from shutter10 and higher (I tested it up tho shutter12):
shutterInvert 0 -> open=0% close=100%
shutterInvert 1 -> open=100% close = 0%

Example for shutter 11, which shows the position in the opposite way as expected:
<code>
16:15:46.534 CMD: **shutterinvert11 0**
16:15:46.542 MQT: Rolladen/stat/RESULT = {"ShutterInvert11":0}
16:15:52.868 CMD: **shutteropen11**
16:15:52.878 MQT: Rolladen/stat/RESULT = {"Shutter11":{**"Position":0,**"Direction":0,"Target":0,"Tilt":0}}
16:16:26.587 CMD: **shutterinvert11 1**
16:16:26.595 MQT: Rolladen/stat/RESULT = {"ShutterInvert11":1}
16:16:27.341 SHT: About to save settings to file /.drvset027
16:16:33.024 CMD: **shutteropen11**
16:16:33.033 MQT: Rolladen/stat/RESULT = {"Shutter11":{**"Position":100**,"Direction":0,"Target":100,"Tilt":0}}
</code>

stefanbode, 2023-08-22:
Now you bring me into trouble with testing.... looks like it hicks up with the double digit of the shutter 10++
I can say for sure this is NOT tested :-)

stefanbode, 2023-08-22:
Same version different result...
```
16:31:18.904 RSL: RESULT = {"Shutter1":{"Position":0,"Direction":0,"Target":0,"Tilt":0}"Shutter2":{"Position":0,"Direction":0,"Target":0,"Tilt":0}"Shutter3":{"Position":0,"Direction":0,"Target":0,"Tilt":0}"Shutter4":{"Position":0,"Direction":0,"Target":0,"Tilt":0}"Shutter5":{"Position":0,"Direction":0,"Target":0,"Tilt":0}"Shutter6":{"Position":0,"Direction":0,"Target":0,"Tilt":0}"Shutter7":{"Position":0,"Direction":0,"Target":0,"Tilt":0}"Shutter8":{"Position":0,"Direction":0,"Target":0,"Tilt":0}"Shutter9":{"Position":0,"Direction":0,"Target":0,"Tilt":0}"Shutter10":{"Position":0,"Direction":0,"Target":0,"Tilt":0}"Shutter11":{"Position":75,"Direction":1,"Target":100,"Tilt":0}}
16:31:19.911 RSL: RESULT = {"Shutter1":{"Position":0,"Direction":0,"Target":0,"Tilt":0}"Shutter2":{"Position":0,"Direction":0,"Target":0,"Tilt":0}"Shutter3":{"Position":0,"Direction":0,"Target":0,"Tilt":0}"Shutter4":{"Position":0,"Direction":0,"Target":0,"Tilt":0}"Shutter5":{"Position":0,"Direction":0,"Target":0,"Tilt":0}"Shutter6":{"Position":0,"Direction":0,"Target":0,"Tilt":0}"Shutter7":{"Position":0,"Direction":0,"Target":0,"Tilt":0}"Shutter8":{"Position":0,"Direction":0,"Target":0,"Tilt":0}"Shutter9":{"Position":0,"Direction":0,"Target":0,"Tilt":0}"Shutter10":{"Position":0,"Direction":0,"Target":0,"Tilt":0}"Shutter11":{"Position":85,"Direction":1,"Target":100,"Tilt":0}}
16:31:20.898 RSL: RESULT = {"Shutter1":{"Position":0,"Direction":0,"Target":0,"Tilt":0}"Shutter2":{"Position":0,"Direction":0,"Target":0,"Tilt":0}"Shutter3":{"Position":0,"Direction":0,"Target":0,"Tilt":0}"Shutter4":{"Position":0,"Direction":0,"Target":0,"Tilt":0}"Shutter5":{"Position":0,"Direction":0,"Target":0,"Tilt":0}"Shutter6":{"Position":0,"Direction":0,"Target":0,"Tilt":0}"Shutter7":{"Position":0,"Direction":0,"Target":0,"Tilt":0}"Shutter8":{"Position":0,"Direction":0,"Target":0,"Tilt":0}"Shutter9":{"Position":0,"Direction":0,"Target":0,"Tilt":0}"Shutter10":{"Position":0,"Direction":0,"Target":0,"Tilt":0}"Shutter11":{"Position":95,"Direction":1,"Target":100,"Tilt":0}}
16:31:21.398 RSL: RESULT = {"POWER21":"OFF"}
16:31:21.399 RSL: POWER21 = OFF
16:31:21.411 RSL: SHUTTER11 = 100
16:31:21.413 RSL: RESULT = {"Shutter11":{"Position":100,"Direction":0,"Target":100,"Tilt":0}}
16:31:21.892 SHT: About to save settings to file /.drvset027
16:31:29.064 CMD: shutteropen11
16:31:29.090 RSL: RESULT = {"Shutter11":{"Position":100,"Direction":0,"Target":100,"Tilt":0}}
16:32:02.229 CMD: shutterinvert11 1
16:32:02.245 RSL: RESULT = {"ShutterInvert11":1}
16:32:02.891 SHT: About to save settings to file /.drvset027
16:32:17.159 CMD: shutteropen11
16:32:17.184 RSL: RESULT = {"Shutter11":{"Position":0,"Direction":0,"Target":0,"Tilt":0}}
```

stefanbode, 2023-08-22:
I checked a bit forward and backward and all shutters 1.. 11 behave as they should be. I run out of gpio for more shutters.... maxbe you can send on shutter11 and shutterinvert11 1 a logfile level4. start at shutterposition 50 and execute shutteropen11. Lets see what internally goes on.

stefanbode, 2023-08-22:
if already in shutteropen... then change the invert and this is what loglevel4 should look like:
```
16:45:38.749 CMD: Grp 0, Cmd 'SHUTTERINVERT', Idx 11, Len 1, Pld 1, Data '1'
16:45:38.751 RSL: RESULT = {"ShutterInvert11":1}
16:45:39.653 SHT: About to save settings to file /.drvset027
16:45:42.733 CMD: shutteropen11
16:45:42.735 SRC: Serial
16:45:42.735 CMD: Grp 0, Cmd 'SHUTTEROPEN', Idx 11, Len 0, Pld -99, Data ''
16:45:42.737 SHT: Pos. payload <> (0), payload 100, idx 11 (1), src 7
16:45:42.748 SHT: lastsource 7:, real 200000, target 200000, tiltreal: 0, tilttarget: 0, payload 100
16:45:42.759 RSL: RESULT = {"Shutter11":{"Position":0,"Direction":0,"Target":0,"Tilt":0}}
```

stefanbode, 2023-08-23:
shutteropen0 show all relays are correctly triggered and all invert and not invert are correct. Tested with 11 shutters:
[image]

sternegugger, 2023-08-23:
> I checked a bit forward and backward and all shutters 1.. 11 behave as they should be. I run out of gpio for more shutters.... maxbe you can send on shutter11 and shutterinvert11 1 a logfile level4. start at shutterposition 50 and execute shutteropen11. Lets see what internally goes on.

Here is the LogLevel4 Trace for

- shutterinvert11 1
- shutterposition11 50
- shutteropen11

You see at the end Position:100 (for open in inverted mode 0 is expected)

```lua
12:48:17.257 CMD: shutterinvert11
12:48:17.259 SRC: WebConsole from 192.168.2.48
12:48:17.261 CMD: Grp 0, Cmd 'SHUTTERINVERT', Idx 11, Len 0, Pld -99, Data ''
12:48:17.269 MQT: Rolladen/stat/RESULT = {"ShutterInvert11":1}
12:48:21.979 WIF: Prüfe Verbindung...

12:48:27.819 CMD: shutterposition11 50
[...]
12:48:42.702 MQT: Rolladen/stat/RESULT = {"POWER21":"OFF"}
12:48:42.713 MQT: Rolladen/stat/POWER21 = OFF
12:48:42.715 SHT: Shtr11 Real 292000, Start 292000, Stop 292000, Dir 0, Delay 0, Rtc 14.65 [s], Freq 0, PWM 0, Tilt 0
12:48:42.719 MQT: Rolladen/stat/SHUTTER11 = 50
12:48:42.727 MQT: Rolladen/stat/RESULT = {"Shutter11":{"Position":50,"Direction":0,"Target":50,"Tilt":0}}
12:48:42.771 SHT: About to save settings to file /.drvset027
12:49:02.003 WIF: Prüfe Verbindung...

12:49:09.975 CMD: shutteropen11
12:49:09.977 SRC: WebConsole from 192.168.2.48
12:49:09.979 CMD: Grp 0, Cmd 'SHUTTEROPEN', Idx 11, Len 0, Pld -99, Data ''
12:49:09.982 SHT: Pos. payload <> (0), payload 100, idx 11 (1), src 7
12:49:09.983 SHT: lastsource 7:, real 292000, target 0, tiltreal: 0, tilttarget: 0, payload 0
12:49:09.985 SHT: Delay Start? var11 <99>=<>, max10s?
12:49:09.986 SRC: Shutter
12:49:10.007 MQT: Rolladen/stat/RESULT = {"POWER22":"ON"}
12:49:10.010 MQT: Rolladen/stat/POWER22 = ON
12:49:10.111 SRC: Shutter
12:49:10.133 MQT: Rolladen/stat/RESULT = {"POWER21":"ON"}
12:49:10.141 MQT: Rolladen/stat/POWER21 = ON
12:49:10.144 SHT: Shtr11 Real 292000, Start 292000, Stop 0, Dir -1, Delay 0, Rtc 0.00 [s], Freq 0, PWM 0, Tilt 0
12:49:10.156 MQT: Rolladen/stat/RESULT = {"Shutter11":{"Position":50,"Direction":-1,"Target":100,"Tilt":0}}
12:49:11.117 SHT: Shtr11 Time 19(2), cStop 0, cVelo 0, mVelo 0, aVelo 1000, mRun 0, aPos 0, aPos2 274000, nStop 0, Trgt 0, mVelo 0, Dir -1, Tilt 0, TrgtTilt: 0, Tiltmove: 0
12:49:11.216 SHT: Shtr11 Time 21(1), cStop 0, cVelo 0, mVelo 0, aVelo 1000, mRun 0, aPos 0, aPos2 272000, nStop 0, Trgt 0, mVelo 0, Dir -1, Tilt 0, TrgtTilt: 0, Tiltmove: 0
12:49:11.266 SHT: Shtr11 Time 22(0), cStop 0, cVelo 0, mVelo 0, aVelo 1000, mRun 0, aPos 0, aPos2 271000, nStop 0, Trgt 0, mVelo 0, Dir -1, Tilt 0, TrgtTilt: 0, Tiltmove: 0
12:49:11.317 SHT: Shtr11 Time 23(0), cStop 0, cVelo 0, mVelo 0, aVelo 1000, mRun 0, aPos 0, aPos2 270000, nStop 0, Trgt 0, mVelo 0, Dir -1, Tilt 0, TrgtTilt: 0, Tiltmove: 0
12:49:11.367 SHT: Shtr11 Time 24(0), cStop 0, cVelo 0, mVelo 0, aVelo 1000, mRun 0, aPos 0, aPos2 269000, nStop 0, Trgt 0, mVelo 0, Dir -1, Tilt 0, TrgtTilt: 0, Tiltmove: 0
12:49:11.417 SHT: Shtr11 Time 25(0), cStop 0, cVelo 0, mVelo 0, aVelo 1000, mRun 0, aPos 0, aPos2 268000, nStop 0, Trgt 0, mVelo 0, Dir -1, Tilt 0, TrgtTilt: 0, Tiltmove: 0
12:49:11.468 SHT: Shtr11 Time 26(0), cStop 0, cVelo 0, mVelo 0, aVelo 1000, mRun 0, aPos 0, aPos2 267000, nStop 0, Trgt 0, mVelo 0, Dir -1, Tilt 0, TrgtTilt: 0, Tiltmove: 0
12:49:11.519 SHT: Shtr11 Time 27(0), cStop 0, cVelo 0, mVelo 0, aVelo 1000, mRun 0, aPos 0, aPos2 266000, nStop 0, Trgt 0, mVelo 0, Dir -1, Tilt 0, TrgtTilt: 0, Tiltmove: 0
12:49:11.569 SHT: Shtr11 Time 28(0), cStop 0, cVelo 0, mVelo 0, aVelo 1000, mRun 0, aPos 0, aPos2 265000, nStop 0, Trgt 0, mVelo 0, Dir -1, Tilt 0, TrgtTilt: 0, Tiltmove: 0
12:49:11.619 SHT: Shtr11 Time 29(0), cStop 0, cVelo 0, mVelo 0, aVelo 1000, mRun 0, aPos 0, aPos2 264000, nStop 0, Trgt 0, mVelo 0, Dir -1, Tilt 0, TrgtTilt: 0, Tiltmove: 0
12:49:11.670 SHT: Shtr11 Time 30(0), cStop 0, cVelo 0, mVelo 0, aVelo 1000, mRun 0, aPos 0, aPos2 263000, nStop 0, Trgt 0, mVelo 0, Dir -1, Tilt 0, TrgtTilt: 0, Tiltmove: 0
12:49:11.720 SHT: Shtr11 Time 31(0), cStop 0, cVelo 0, mVelo 0, aVelo 1000, mRun 0, aPos 0, aPos2 262000, nStop 0, Trgt 0, mVelo 0, Dir -1, Tilt 0, TrgtTilt: 0, Tiltmove: 0
12:49:11.770 SHT: Shtr11 Time 32(0), cStop 0, cVelo 0, mVelo 0, aVelo 1000, mRun 0, aPos 0, aPos2 261000, nStop 0, Trgt 0, mVelo 0, Dir -1, Tilt 0, TrgtTilt: 0, Tiltmove: 0
12:49:11.820 SHT: Shtr11 Time 33(0), cStop 0, cVelo 0, mVelo 0, aVelo 1000, mRun 0, aPos 0, aPos2 260000, nStop 0, Trgt 0, mVelo 0, Dir -1, Tilt 0, TrgtTilt: 0, Tiltmove: 0
12:49:11.870 SHT: Shtr11 Time 34(0), cStop 0, cVelo 0, mVelo 0, aVelo 1000, mRun 0, aPos 0, aPos2 259000, nStop 0, Trgt 0, mVelo 0, Dir -1, Tilt 0, TrgtTilt: 0, Tiltmove: 0
12:49:11.921 SHT: Shtr11 Time 35(0), cStop 0, cVelo 0, mVelo 0, aVelo 1000, mRun 0, aPos 0, aPos2 258000, nStop 0, Trgt 0, mVelo 0, Dir -1, Tilt 0, TrgtTilt: 0, Tiltmove: 0
12:49:11.972 SHT: Shtr11 Time 36(0), cStop 0, cVelo 0, mVelo 0, aVelo 1000, mRun 0, aPos 0, aPos2 257000, nStop 0, Trgt 0, mVelo 0, Dir -1, Tilt 0, TrgtTilt: 0, Tiltmove: 0
12:49:11.992 SHT: Shtr11 Real 256000, Start 292000, Stop 0, Dir -1, Delay 0, Rtc 1.85 [s], Freq 0, PWM 0, Tilt 0
12:49:12.003 MQT: Rolladen/stat/RESULT = {"Shutter1":{"Position":0,"Direction":0,"Target":0,"Tilt":0}"Shutter2":{"Position":0,"Direction":0,"Target":0,"Tilt":0}"Shutter3":{"Position":0,"Direction":0,"Target":0,"Tilt":0}"Shutter4":{"Position":0,"Direction":0,"Target":0,"Tilt":0}"Shutter5":{"Position":100,"Direction":0,"Target":100,"Tilt":0}"Shut [... cut ...]
12:49:24.842 MQT: Rolladen/stat/POWER21 = OFF
12:49:24.845 SRC: Shutter
12:49:24.865 MQT: Rolladen/stat/RESULT = {"POWER22":"OFF"}
12:49:24.873 MQT: Rolladen/stat/POWER22 = OFF
12:49:24.876 SHT: Shtr11 Real 0, Start 0, Stop 0, Dir 0, Delay 0, Rtc 14.65 [s], Freq 0, PWM 0, Tilt 0
12:49:24.883 MQT: Rolladen/stat/SHUTTER11 = 0
12:49:24.887 MQT: Rolladen/stat/RESULT = {"Shutter11":{"Position":100,"Direction":0,"Target":100,"Tilt":0}}
12:49:25.249 SHT: About to save settings to file /.drvset027

```

stefanbode, 2023-08-24:
I found the bug and just testing why the mess was not bigger. accessed a 4 shutter esp8266 array... Normally this should not work in any case...

stefanbode, 2023-08-24:
Indeed more than visible on the first view. Pls. give it a try.

closed 2023-08-24 by arendst

sternegugger, 2023-08-24:
Thank you for fix #19374.
Now all shutters (including 10 and higher) work as designed, in invert and normal mode.
Test with ESP32 and 3x MCP23017.

stefanbode, 2023-08-24:
Maybe you can send a photo of the build. I assume it looks impressive. For sure the record of most shutters on one esp32 is yours.

sternegugger, 2023-08-24:
> Maybe you can send a photo of the build. I assume it looks impressive. For sure the record of most shutters on one esp32 is yours.

On the left you can see the 250V part. One 16 relay module and on 8 relay module. I am using shutter mode 2 (one relay on/off, the other up/down). On the right are 3x MCP23017, connected to the lines coming from the switches (24 switch inputs and 24 outputs for the relays). In between is a ESP32 WROOM Dev-Kit. Below, left is a 5V 3A power adapter.

My next steps are to reduce the interferences to the ESP caused by the switching of the relays...

[image]

stefanbode, 2023-08-25:
Hi Michael, awesome construction. Thanks for sharing. If I zoom in I already see some resistors at the MCP. I assume this was your first attempt the get away from the accidental switches.. Anyhow this will not help on the small spikes you might get though the magnetic field. Typically you use some R-C combination. I would do a first try and solder in parallel to the resistor a 20nonofarad capacitor (203 or 204). That might be capable to take the spike. You can try with one or two capacitors on switches where you see problems and then move on in case of success.

PR #19374 (2023-08-24): fix inverted shutter inconsistencies
## Description:
inverted shutter move to wrong position
inverted shutter behave wrong with tilt at end positions
**Related issue (if applicable):** fixes #<Tasmota issue number goes here>
fix #19328
## Checklist:
  - [x] The pull request is done against the latest development branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR and the code change compiles without warnings
  - [x] The code change is tested and works with Tasmota core ESP8266 V.2.7.4.9
  - [x] The code change is tested and works with Tasmota core ESP3 [... cut]