arendst/Tasmota #19340: esp32s3 firmware.map parsing error

soosp, 2023-08-17:
### PROBLEM DESCRIPTION
I tried to build a custom firmware for my esp32s3 board. The compilation is OK, but firmware map parsing produces an error:
```
C:\Users\sp\.platformio\penv\Scripts\python.exe -m tasmota_metrics "C:\Users\sp\Documents\Dev\Tasmota-13.1.0\.pio\build\tasmota32c3-l10c\firmware.map"
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "C:\Users\sp\.platformio\penv\Lib\site-packages\tasmota_metrics\__main__.py", line 4, in <module>
    main()
  File "C:\Users\sp\.platformio\penv\Lib\site-packages\tasmota_metrics\core.py", line 507, in main
    detected_target, segments, sections = load_map_data(args.map_file)
                                          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\sp\.platformio\penv\Lib\site-packages\tasmota_metrics\core.py", line 236, in load_map_data
    detected_chip = detect_target_chip(map_file)
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\sp\.platformio\penv\Lib\site-packages\tasmota_metrics\core.py", line 306, in detect_target_chip
    raise RuntimeError('Target not detected')
RuntimeError: Target not detected
*** Error 1
```
It seems to be a Windows relates issue, because the relevant lines in the firmware.map file looks like this:
```
LOAD C:/Users/sp/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32c3/lib\libesp_ringbuf.a
```
The last separator is a backslash, but the `.platformio\penv\Lib\site-packages\tasmota_metrics\core.py` searches for pattern
```
'^LOAD .*?/framework-arduinoespressif32/tools/sdk/([^-]+)/lib/'
```
at line 279.
If you remove the tailing slash the pattern matches, and the target is found.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): ESP32-C3 SuperMini Development Board
- [X] Tasmota binary firmware version number used: 13.1.0
  - [ ] Pre-compiled
  - [X] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
I'm using Visual Studio Code 1.81.0 with PlatformIO 3.3.1

**(Please, remember to close the issue when the problem has been addressed)**

barbudor, 2023-08-17:
Why doesn't this happens on build default tasmota32s3 ?
I have hte same line in firmware.map but no error on executing tasmota-metrics
```
LOAD .pio/build/tasmota32s3/libFrameworkArduino.a
LOAD C:/Users/xxxxxxxxx/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32s3/lib\libesp_ringbuf.a
LOAD C:/Users/xxxxxxxxx/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32s3/lib\libefuse.a
LOAD C:/Users/xxxxxxxxx/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32s3/lib\libesp_ipc.a
LOAD C:/Users/xxxxxxxxx/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32s3/lib\libdriver.a
LOAD C:/Users/xxxxxxxxx/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32s3/lib\libesp_pm.a
```
```
C:\Users\xxxxxxxxx\.platformio\penv\Scripts\python.exe -m tasmota_metrics "D:\Projets\repos\Tasmota\.pio\build\tasmota32s3\firmware.map"       
Total sizes:
Used static IRAM:   77886 bytes ( 284354 remain, 21.5% used)
      .text size:   76859 bytes
   .vectors size:    1027 bytes
Used stat D/IRAM:   66129 bytes ( 279727 remain, 19.1% used)
      .data size:   18065 bytes
      .bss  size:   48064 bytes
Used Flash size : 1718569 bytes
           .text: 1113923 bytes
         .rodata:  604390 bytes
Total image size: 1814520 bytes (.bin may be padded larger)
```
Same core.py:
```
    RE_TARGET_PIO = re.compile(r'^LOAD .*?/framework-arduinoespressif32/tools/sdk/([^-]+)/lib/')
```

soosp, 2023-08-18:
Sorry, my mistake. I have esp32c3 not s3. The title of this ticket has been updated.

barbudor, 2023-08-18:
ok, I can reproduce with the standard `tasmota32c3` build on Win11

I understand why it fails on ESP32C3 and not on S3 (and may be others)

The code to identify the target is looking through the whole file until one of those regex have a match:
```
    RE_TARGET = re.compile(r'IDF_TARGET_(\S*) =')
    RE_TARGET_CMAKEv4x = re.compile(r'project_elf_src_(\S*)\.c.obj')
    RE_TARGET_MAKE = re.compile(r'^LOAD .*?/xtensa-(esp[^-]+)-elf/')
    RE_TARGET_PIO = re.compile(r'^LOAD .*?/framework-arduinoespressif32/tools/sdk/([^-]+)/lib/')
```

With the S3 build, the 3rd one gets a match with:
```
LOAD c:/users/jean-michel/.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/lib/no-rtti/crt0.o
```
before it reaches the
```
LOAD C:/Users/Jean-Michel/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32s3/lib\libesp_ringbuf.a
```
So the 4th regex is not used

But on C3, none of the 3 first regex get a match
And the 4th one doesn't neither on Windows

So your proposed fix seems to be correct

@Jason2866 Do you want to make the change directly or do you want me to send you a PR ?

279:
-     RE_TARGET_PIO = re.compile(r'^LOAD .*?/framework-arduinoespressif32/tools/sdk/([^-]+)/lib/')
+    RE_TARGET_PIO = re.compile(r'^LOAD .*?/framework-arduinoespressif32/tools/sdk/([^-]+)/lib')

barbudor, 2023-08-18:
https://github.com/Jason2866/tasmota-metrics/pull/4

Now needs @Jason2866 to merge and update PyPi
In the meantime applying the change directly seems to work fine

closed 2023-08-19 by Jason2866

Jason2866, 2023-08-19:
Thx @barbudor new version with the change is online on pypi. 
The newest version will not installed automatically. Needs manually update.
On a fresh install the latest will be installed

soosp, 2023-08-21:
Thank you guys, you are very quick!