arendst/Tasmota #19359: ESP32 using ShutterStopClose while shutter is opening generates random switch activities

sternegugger, 2023-08-20:
### PROBLEM DESCRIPTION

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): 3x MCP23017
- [X] Tasmota binary firmware version number used: 13.1.0.1 from Development (git pull: 20. Aug. 2023 20:02)
  - [ ] Pre-compiled
  - [x] Self-compiled (on full cleaned environment)
- [x] Flashing tools used: WiFi (Upload with Web Interface)
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
21:04:04.081 CMD: Backlog Template; Module; GPIO 255
21:04:04.119 MQT: Rolladen/stat/RESULT = {"NAME":"ESP32-DevKit","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,0,1,1,1,0,1,1,1,0,0,0,0,1,1,1,1,1,0,0,1],"FLAG":0,"BASE":1}
21:04:04.339 MQT: Rolladen/stat/RESULT = {"Module":{"1":"ESP32-DevKit"}}
21:04:04.549 MQT: Rolladen/stat/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"0":"None"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"640":"I2C SDA1"},"GPIO22":{"608":"I2C SCL1"},"GPIO23":{"0":"None"},"GPIO24":{"0":"None"},"GPIO25":{"0":"None"},"GPIO26":{"0":"None"},"GPIO27":{"0":"None"},"GPIO32":{"0":"None"},"GPIO33":{"0":"None"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}

```
- [x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
21:05:20.060 CMD: backlog rule1;rule2;rule3
21:05:20.096 MQT: Rolladen/stat/RESULT = {"Rule1":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":261,"Free":250,"Rules":"on file#mcp23x.dat do {\"NAME\":\"MCP23017 shutter\",\"BASE\":0,\"GPIO\":[160,161,162,163,164,165,166,167,256,257,258,259,260,261,262,263,168,169,170,171,172,173,174,175,264,265,266,267,0,0,270,271,176,177,178,179,180,181,182,183,272,273,274,275,276,277,268,269]} endon"}}
21:05:20.307 RUL: Rule2 on Switch1#state=2 do shutterStopOpen1 endon  on Switch1#state=3 do backlog shutterStopOpen1; shutterStopOpen2; shutterStopOpen3; shutterStopOpen4 endon  on Switch2#state=2 do shutterStopClose1 endon on Switch3#state=1 do shutterStopOpen2 endon  on Switch4#state=1 do shutterStopClose2 endon on Switch5#state=1 do backlog shutterStopOpen3; shutterStopOpen4; endon  on Switch6#state=1 do backlog shutterStopClose3; shutterStopClose4; endon on Switch9#state=1 do shutterStopOpen5 endon  on Switch10#state=1 do
21:05:20.308 RUL: Rule2 +shutterStopClose5 endon on Switch11#state=1 do shutterStopOpen6 endon  on Switch12#state=1 do shutterStopClose6 endon on Switch13#state=1 do shutterStopOpen7 endon  on Switch14#state=1 do shutterStopClose7 endon
21:05:20.319 MQT: Rolladen/stat/RESULT = {"Rule2":{"State":"ON","Once":"OFF","StopOnError":"OFF","Length":719,"Free":308,"Rules":"on Switch1#state=2 do shutterStopOpen1 endon  on Switch1#state=3 do backlog shutterStopOpen1; shutterStopOpen2; shutterStopOpen3; shutterStopOpen4 endon  on Switch2#state=2 do shutterStopClose1 endon on Switch3#state=1 do shutterStopOpen2 endon  on Switch4#state=1 do shutterStopClose2 endon on Switch5#state=1 do backlog shutterStopOpen3; shutterStopOpen4; endon  on Switch6#state=1 do backlog shutterStopClose3; shutterStopClose4; endon on Switch9#state=1 do shutterStopOpen5 endon  on Switch10#state=1 do shutterStopClose5 endon on Switch11#state=1 do shutterStopOpen6 endon  on Switch12#state=1 do shutterStopClose6 endon on Switch13#state=1 do shutterStopOpen7 endon  on Switch14#state=1 do shutterStopClose7 endon"}}
21:05:20.540 MQT: Rolladen/stat/RESULT = {"Rule3":{"State":"ON","Once":"OFF","StopOnError":"OFF","Length":379,"Free":132,"Rules":"on Switch15#state=1 do shutterStopOpen8 endon  on Switch16#state=1 do shutterStopClose8 endon on Switch17#state=1 do shutterStopOpen9 endon  on Switch18#state=1 do shutterStopClose9 endon on Switch19#state=1 do shutterStopOpen10 endon  on Switch20#state=1 do shutterStopClose10 endon on Switch21#state=1 do shutterStopOpen11 endon  on Switch22#state=1 do shutterStopClose11 endon"}}

```
- [x] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
21:07:09.362 MQT: Rolladen/stat/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":2,"LogHost":"vera-smart.fritz.box","LogPort":514,"SSId":["MB Fritz Keller","5G_Test_Tower"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["0000800B","0A05C80001000600003C5A0A192800000000","40800080","00006000","00004001","00000000"]}}
21:07:09.417 MQT: Rolladen/stat/STATUS4 = {"StatusMEM":{"ProgramSize":1835,"Free":1044,"Heap":110,"StackLowMark":3,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"1640D8","FlashFrequency":40,"FlashMode":"DIO","Features":["00000407","9F9AD7DF","0015A001","B7F7BFCF","05DA9BC4","E0360DC7","480840D2","20200000","D4BC482D","010A80B1"],"Drivers":"1,2,3,4,5,7,8,9,10,11,12,14,16,17,20,21,24,26,27,29,34,35,38,50,52,59,60,62,63,66,67,68,82,86,87,88","Sensors":"1,2,3,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,31,34,37,39,40,42,43,45,51,52,55,56,58,59,64,66,67,74,85,92,95,98,103,105,109,127","I2CDriver":"7,8,9,10,11,12,13,14,15,17,18,20,24,29,31,36,41,42,44,46,48,58,62,65,69,76,77,82"}}
21:07:09.474 MQT: Rolladen/stat/STATUS5 = {"StatusNET":{"Hostname":"ESP32-Rolladen","IPAddress":"192.168.2.17","Gateway":"192.168.2.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.2.1","DNSServer2":"0.0.0.0","Mac":"C8:C9:A3:D1:71:D0","IP6Global":"","IP6Local":"fe80::cac9:a3ff:fed1:71d0","Ethernet":{"Hostname":"","IPAddress":"0.0.0.0","Gateway":"0.0.0.0","Subnetmask":"0.0.0.0","DNSServer1":"192.168.2.1","DNSServer2":"0.0.0.0","Mac":"00:00:00:00:00:00","IP6Global":"","IP6Local":""},"Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
21:07:09.523 MQT: Rolladen/stat/STATUS6 = {"StatusMQT":{"MqttHost":"vera-smart","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_D171D0","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
21:07:09.566 MQT: Rolladen/stat/STATUS7 = {"StatusTIM":{"UTC":"2023-08-20T19:07:09","Local":"2023-08-20T21:07:09","StartDST":"2023-03-26T02:00:00","EndDST":"2023-10-29T03:00:00","Timezone":99,"Sunrise":"06:49","Sunset":"20:57"}}
21:07:09.619 MQT: Rolladen/stat/STATUS10 = {"StatusSNS":{"Time":"2023-08-20T21:07:09","Switch1":"OFF","Switch2":"OFF","Switch3":"ON","Switch4":"ON","Switch5":"ON","Switch6":"ON","Switch7":"ON","Switch8":"ON","Switch9":"OFF","Switch10":"OFF","Switch11":"OFF","Switch12":"OFF","Switch13":"OFF","Switch14":"OFF","Switch15":"OFF","Switch16":"OFF","Switch17":"OFF","Switch18":"OFF","Switch19":"OFF","Switch20":"OFF","Switch21":"OFF","Switch22":"OFF","Switch23":"OFF","Switch24":"OFF","Shutter1":{"Position":100,"Direction":0,"Target":100,"Tilt":0},"Shutter2":{"Position":100,"Direction":0,"Target":100,"Tilt":0},"Shutter3":{"Position":100,"Direction":0,"Target":100,"Tilt":0},"Shutter4":{"Position":100,"Direction":0,"Target":100,"Tilt":0},"Shutter5":{"Position":100,"Direction":0,"Target":100,"Tilt":0},"Shutter6":{"Position":100,"Direction":0,"Target":100,"Tilt":0},"Shutter7":{"Position":100,"Direction":0,"Target":100,"Tilt":0},"Shutter8":{"Position":100,"Direction":0,"Target":100,"Tilt":0},"Shutter9":{"Position":0,"Direction":0,"Target":0,"Tilt":0},"Shutter10":{"Position":100,"Direction":0,"Target":100,"Tilt":0},"Shutter11":{"Position":100,"Direction":0,"Target":100,"Tilt":0},"Shutter12":{"Position":0,"Direction":0,"Target":0,"Tilt":0}}}
[... cut ...]
### ADDITIONAL CONTEXT
This is what I have already checked and should not be the reason for this behavior:

- supply voltage: 5V 3A, Relays are directly connected to the power adapter. Voltage is stable all the time. (actually I'm using 5.2V)
- MCP23017 have stable 3.3V supply
- additionally to the build in pullup resistors I'm using 6.8 kOhm pullups, to be sure that the long lines to the switches do not produce interferrences.

**(Please, remember to close the issue when the problem has been addressed)**

stefanbode, 2023-08-22:
I fixed the other Issue. Now lets work on this one. is it only shutter7 or do you see the same behavior also on other shutters. I have no testboard MCP. Therefore a lack of testing this super high loaded configuration. But happy and willing to assist to fix it

stefanbode, 2023-08-22:
I tried to reproduce the issue with NO Switches but the commands and cannot reproduce.

```
Aug 20 20:32:18 ESP32-Rolladen ESP-MQT: Rolladen/stat/RESULT = {"Shutter7":{"Position":88,"Direction":0,"Target":88,"Tilt":0}}
Aug 20 20:32:18 ESP32-Rolladen ESP-MQT: Rolladen/stat/RESULT = {"Switch13":{"Action":"ON"}}
// I did not touch this switch
```

That a SWITCH is activated or not is NOT under the control of the shutter. I can do nothing here to avoid the SWITCH change. You have to look if this is part of one of the rules or maybe an electrical problem. Anyhow I do not see any relation to TASMOTA code or shutter code. Nothing we can debug or fix as far as I can see. I do see you're german. You can ping me on discord and we can do a screen sharing session. Account: stefanbode

sternegugger, 2023-08-23:
> That a SWITCH is activated or not is NOT under the control of the shutter.

I tried to exclude electrical problems by controlling the supply voltage and using lower pullup resistors for the inputs of the MCP23017.
You my be right that is has nothing to do with the shutter control, but whith the MCP23017 control or whith the persistance of its configuration during operation.

Is it possible to trace the I2C bus content?

sternegugger, 2023-08-23:
> You have to look if this is part of one of the rules or maybe an electrical problem.

It was definitely an electric problem (not the switch input, as I supposed before).
The interference came from the relays, switching the 250V of the running shutter motors.
Opening and closing the shutters until the end, causes an internal stop of the shutter motors (far away from the ESP32) with no interference.
But stopping the shutter by command while it moves, causes a stop of the motor by the tasmota controlled relay. The inductivity of the running motor causes an electric spark at the relais contacts. The electromagnetic field influenced the running tasmota software on the ESP32.

I tested the same scenario multiple times without connecting 250V to the shutter motors.
Now the up, down and stop switching of the relays works as designed, without random side effects. 

**Fix:** 
Shield the relays with a metal box and increase the distance of them to the ESP32.
Do not use the cheapest (chinese) Relais. I killed already two of my 24 Relays by testing stop the shutters while moving.
**==> It was not a software problem.**

Jason2866, 2023-08-23:
Closing since solved

closed 2023-08-23 by Jason2866