arendst/Tasmota #19367: Scripting: mean filter M: gives a value of -0.00

jkwim, 2023-08-23:
I am using Tasmota 12.3.1.3 custom compiled version with SCRIPTING on an ESP8266 Device.

I wanted to use mean filter `M: `for filtering out spurious `'0' `values.

The approach that I use is as follows:

```
D>
; 60 second moving average
M:mgrid=0 60

S>
mgrid=grid
```

The value of `grid` usually is above `230.00`. 
I want to use `mgrid` to detect the situation where `grid==0` continously (say for more than 60seconds) and filter out any spurios `grid==0` situations which lasts 10/20 seconds (these spurious values originate due to the sensor setup).

Everything works fine when moving from `grid=0` to say `grid=230.00`

When `grid=230` becomes `grid=0` in the other direction, after converging, the final value of the mean `mgrid` ends up with a value of `-0.00`

Jason2866, 2023-08-24:
Closing since answered.

closed 2023-08-24 by Jason2866

jkwim, 2023-08-25:
Understood the explanation.
However I am still confused as how the mean value of positive numbers can become a negative value.

Will do more investigation and post my findings here.