arendst/Tasmota #19387: tasmota-ir reboots at every command, after 13.1.0

fmarzocca, 2023-08-25:
### PROBLEM DESCRIPTION
My NodeMCU device with Tasmota-IR is rebooting after every command, since I installed v. 13.1.0

### REQUESTED INFORMATION

- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
16:24:35.505 CMD: status 0
16:24:35.513 MQT: stat/ac-salone/STATUS = {"Status":{"Module":18,"DeviceName":"AC Salone","FriendlyName":["ac-salone"],"Topic":"ac-salone","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
16:24:35.522 MQT: stat/ac-salone/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota-ir.bin.gz","RestartReason":"Exception","Uptime":"0T00:57:11","StartupUTC":"2023-08-25T13:27:24","Sleep":50,"CfgHolder":4617,"BootCount":76,"BCResetTime":"2021-08-18T16:09:15","SaveCount":352,"SaveAddress":"F7000"}}
16:24:35.529 MQT: stat/ac-salone/STATUS2 = {"StatusFWR":{"Version":"13.1.0(ir)","BuildDateTime":"2023-08-15T07:59:23","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"391/699"}}
16:24:35.536 MQT: stat/ac-salone/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Saltydog_1ext","Vodafone-saltydog"],"TelePeriod":3600,"Resolution":"558180C0","SetOption":["00008019","2805C80001000600003C5A00002800000000","00000081","00006000","00004000","00000000"]}}
16:24:35.548 MQT: stat/ac-salone/STATUS4 = {"StatusMEM":{"ProgramSize":648,"Free":352,"Heap":24,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640EF","FlashFrequency":40,"FlashMode":"DOUT","Features":["00000809","0F1A0582","04000001","00000003","00000000","00000000","00000000","00000000","04000000","00000080"],"Drivers":"1,2,4,5,7,9,10,12,62,68","Sensors":"1"}}
16:24:35.559 MQT: stat/ac-salone/STATUS5 = {"StatusNET":{"Hostname":"ac-salone","IPAddress":"192.168.1.57","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.1.1","DNSServer2":"0.0.0.0","Mac":"A4:CF:12:DF:66:DA","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
16:24:35.568 MQT: stat/ac-salone/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.90","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_DF66DA","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
16:24:35.577 MQT: stat/ac-salone/STATUS7 = {"StatusTIM":{"UTC":"2023-08-25T14:24:35","Local":"2023-08-25T16:24:35","StartDST":"2023-03-26T02:00:00","EndDST":"2023-10-29T03:00:00","Timezone":99,"Sunrise":"06:56","Sunset":"20:47"}}
16:24:35.584 MQT: stat/ac-salone/STATUS10 = {"StatusSNS":{"Time":"2023-08-25T16:24:35"}}
16:24:35.594 MQT: stat/ac-salone/STATUS11 = {"StatusSTS":{"Time":"2023-08-25T16:24:35","Uptime":"0T00:57:11","UptimeSec":3431,"Vcc":2.953,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"Saltydog_1ext","BSSId":"C8:9E:43:8B:D9:32","Channel":9,"Mode":"11n","RSSI":100,"Signal":-25,"LinkCount":1,"Downtime":"0T00:00:03"}}}
16:24:35.612 MQT: stat/ac-salone/STATUS12 = {"StatusSTK":{"Exception":3,"Reason":"Exception","EPC":["4000bf64","00000000","00000000"],"EXCVADDR":"40295cbf","DEPC":"00000000","CallChain":["40295cbf","40254110","4029006c","402012fe","4025016c","4022fd29","4020e763","40250aa0","4020ee96","40253cfc","4028d878","4020f724","4021c620","4028d870","4028d868","40290808","40100a5e","4021c89c","40212718","4021fab5","4020ddf8","40224f32","40295a18","40104685","4020f929","40100558","40102381","40101f50","401051eb","401043c6","40104837"]}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

16:25:13.071 MQT: tele/ac-salone/INFO3 = {"Info3":{"RestartReason":{"Exception":3,"Reason":"Exception","EPC":["4000bf64","00000000","00000000"],"EXCVADDR":"40295cbf","DEPC":"00000000","CallChain":["40295cbf","40254110","4029006c","402012fe","4025016c","4022fd29","4020e763","40250aa0","4020ee96","40253cfc","4028d878","4020f724","4021c620","4028d870","4028d868","40290808","40100a5e","4021c89c","40212718","4021fab5","4020b5c8","402254ae","40295a18","4020f929","40102eac","40227a63","40100558","40102381","40101f50","40100558","401043c6"]},"BootCount":77}}
16:25:16.435 QPC: Reset
```

### TO REPRODUCE
Just send any IRHVAC command

### EXPECTED BEHAVIOUR
Should not reboot

arendst, 2023-08-25:
What's the previous version where it worked fine?

fmarzocca, 2023-08-25:
last one: 13.0.0

fmarzocca, 2023-08-25:
I have just verified the other 2 IR-enabled devices I have (all NodeMCU), and all of them are rebooting after every command I send. This is since 13.1.0.

s-hadinger, 2023-08-26:
I confirm I can reproduce. I will investigate

arendst, 2023-08-26:
Between 3.0 and 3.1 only the library changed so I suggest to look there first.

s-hadinger, 2023-08-26:
Actually an old hidden bug that never surfaced.

closed 2023-08-26 by s-hadinger

fmarzocca, 2023-08-26:
> Actually an old hidden bug that never surfaced.

Thanks! Will it be in the next release?

s-hadinger, 2023-08-26:
Yes. But it's already available in the development version

fmarzocca, 2023-08-26:
Sorry, just updated to 13.1.0.1, but the device still reboots after any command...

fmarzocca, 2023-08-26:
This is the message I am sending to it (via MQTT):

`{"econ":"ON","vendor":"HAIER_AC_YRW02","fanspeed":"Auto","swingv":"ON","swingh":"ON","temp":24,"mode":"Cool","power":"OFF"}`

PR #19389 (2023-08-26): Fix crash in IRHVAC
## Description:

Fix crash in IRHVAC.

The bug was far fetched. It is actually an old bug when trying to report a NaN value in a JSON. But this never happened before #18310. Now NaN are reported as `null` since `NaN` is not a valid JSON value.

```
IRHVAC {"Vendor":"LG", "Power":"On","Mode":"Hot","FanSpeed":3,"Temp":22.5}
09:01:48.655 CMD: IRHVAC {"Vendor":"LG", "Power":"On","Mode":"Hot","FanSpeed":3,"Temp":22.5}
09:01:48.775 RSL: RESULT = {"IRHVAC":{"Vendor":"LG","Model":-1,"Command":"Control","Mode":"Auto","Power":"On","Celsius":"On","Temp":22.5,"FanSpeed":"Medium","Swing [... cut]