arendst/Tasmota #19401: Error when subscribing to DS18B20 sensor from another Tasmota device using MQTT

farcouet, 2023-08-27:
### PROBLEM DESCRIPTION
tasmota-54627B-0635 device publishes temperature from one DS18B20 sensor every 10 seconds to topic tele/tasmota_54627B/SENSOR.
Data sent: "tele/tasmota_54627B/SENSOR {"Time":"2023-08-27T19:20:20","DS18B20":{"Id":"0301A279BECE","Temperature":21.0},"Thermostat0":{"ThermostatModeSet":1,"TempTargetSet":18.00,"CtrDutyCycleRead":0,"ControlMethod":2,"HybridControllerPhase":1,"EmergencyState":0},"Thermostat1":{"ThermostatModeSet":0,"TempTargetSet":18.00,"CtrDutyCycleRead":0,"ControlMethod":0,"HybridControllerPhase":1,"EmergencyState":0},"Thermostat2":{"ThermostatModeSet":0,"TempTargetSet":18.00,"CtrDutyCycleRead":0,"ControlMethod":0,"HybridControllerPhase":1,"EmergencyState":0},"TempUnit":"C"}
"

I created this rule on the other Tasmota to receive the data "19:22:04.604 MQT: stat/tasmota_ACE5A5/RESULT = {"Rule2":{"State":"ON","Once":"OFF","StopOnError":"OFF","Length":142,"Free":369,"Rules":"ON mqtt#connected DO Subscribe DnTemp, tele/tasmota_54627B/SENSOR, DS18B20.Temperature ENDON ON Event#DnTemp DO TempMeasuredSet2 %value% ENDON"}}"

Each time the message is published I get a command error : "19:22:50.543 MQT: stat/tasmota_ACE5A5/RESULT = {"Command":"Error"}"

I have looked at the examples and started from those. I read that there was an issue previously but I have the firwmare (compiled) that is supposed to be fixed.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X ] Device used (e.g., Sonoff Basic): Sonoff 4CH Pro (receiver) & Sonoff TH publisher
- [ X] Tasmota binary firmware version number used: 13.0.0.3
  - [ ] Pre-compiled
  - [X ] Self-compiled
- [ X] Flashing tools used: serial
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
19:28:53.943 CMD: Backlog Template; Module; GPIO 255
19:28:53.945 SRC: WebConsole from 192.168.12.191
19:28:53.948 CMD: Grp 0, Cmd 'BACKLOG', Idx 1, Len 26, Pld -99, Data 'Template; Module; GPIO 255'
19:28:54.014 SRC: Backlog
19:28:54.017 CMD: Grp 0, Cmd 'TEMPLATE', Idx 1, Len 0, Pld -99, Data ''
19:28:54.024 MQT: stat/tasmota_ACE5A5/RESULT = {"NAME":"Generic","GPIO":[32,1,1,1,226,225,33,34,224,320,35,227,0,0],"FLAG":0,"BASE":23}
19:28:54.220 SRC: Backlog
19:28:54.223 CMD: Grp 0, Cmd 'MODULE', Idx 1, Len 0, Pld -99, Data ''
19:28:54.229 MQT: stat/tasmota_ACE5A5/RESULT = {"Module":{"23":"Sonoff 4CH Pro"}}
19:28:54.471 SRC: Backlog
19:28:54.473 CMD: Grp 0, Cmd 'GPIO', Idx 1, Len 3, Pld 255, Data '255'
19:28:54.482 MQT: stat/tasmota_ACE5A5/RESULT = {"GPIO0":{"32":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"1312":"DS18x201"},"GPIO3":{"0":"None"},"GPIO4":{"226":"Relay3"},"GPIO5":{"225":"Relay2"},"GPIO9":{"33":"Button2"},"GPIO10":{"34":"Button3"},"GPIO12":{"224":"Relay1"},"GPIO13":{"320":"Led_i1"},"GPIO14":{"35":"Button4"},"GPIO15":{"227":"Relay4"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
19:24:28.843 CMD: Backlog Rule1; Rule2; Rule3
19:24:28.845 SRC: WebConsole from 192.168.12.191
19:24:28.847 CMD: Grp 0, Cmd 'BACKLOG', Idx 1, Len 19, Pld -99, Data 'Rule1; Rule2; Rule3'
19:24:28.912 SRC: Backlog
19:24:28.915 CMD: Grp 0, Cmd 'RULE', Idx 1, Len 0, Pld -99, Data ''
19:24:28.923 MQT: stat/tasmota_ACE5A5/RESULT = {"Rule1":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":404,"Free":107,"Rules":"ON Power1#boot DO Backlog THERMOSTATMODESET1 1; TEMPHYSTSET1 0.2; SENSORINPUTSET1 1; TIMEPICYCLESET1 1; TempFrostProtectSet1 -10; CONTROLLERMODESET1 2;CLIMATEMODESET1 1; TIMEMINACTIONSET1 5;temptargetset1 %mem1%; THERMOSTATMODESET2 1; TEMPHYSTSET2 0.1; SENSORINPUTSET2 0;TIMEPICYCLESET2 1;  TempFrostProtectSet2 -10; CONTROLLERMODESET2 2;CLIMATEMODESET2 1; TIMEMINACTIONSET2 1;temptargetset2 %mem2% ENDON"}}
19:24:29.137 SRC: Backlog
19:24:29.140 CMD: Grp 0, Cmd 'RULE', Idx 2, Len 0, Pld -99, Data ''
19:24:29.146 MQT: stat/tasmota_ACE5A5/RESULT = {"Rule2":{"State":"ON","Once":"OFF","StopOnError":"OFF","Length":142,"Free":369,"Rules":"ON mqtt#connected DO Subscribe DnTemp, tele/tasmota_54627B/SENSOR, DS18B20.Temperature ENDON ON Event#DnTemp DO TempMeasuredSet2 %value% ENDON"}}
19:24:29.342 SRC: Backlog
19:24:29.345 CMD: Grp 0, Cmd 'RULE', Idx 3, Len 0, Pld -99, Data ''
19:24:29.355 MQT: stat/tasmota_ACE5A5/RESULT = {"Rule3":{"State":"ON","Once":"OFF","StopOnError":"OFF","Length":238,"Free":273,"Rules":"ON mem1#state DO Backlog TEMPTARGETSET1 %value%; Publish2 stat/tasmota_ACE5A5/targetTempGlycol {\"Temp\":%value%} ENDON ON mem2#state DO Backlog TEMPTARGETSET2 %value%; Publish2 stat/tasmota_ACE5A5/targetTempFermenter {\"Temp\":%value%} ENDON"}}
19:24:29.805 SRC: MQTT
19:24:29.809 CMD: Grp 0, Cmd 'SENSOR', Idx 1, Len 528, Pld -99, Data 

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
19:25:18.847 CMD: STATUS 0
19:25:18.849 SRC: WebConsole from 192.168.12.191
19:25:18.851 CMD: Grp 0, Cmd 'STATUS', Idx 1, Len 1, Pld 0, Data '0'
19:25:18.860 MQT: stat/tasmota_ACE5A5/STATUS = {"Status":{"Module":23,"DeviceName":"tasmota-glycol","FriendlyName":["tasmota-glycol-compressor","tasmota-glycol-pump1","tasmota-glycol-pump2","tasmota-glycol-relay"],"Topic":"tasmota_ACE5A5","ButtonTopic":"0","Power":2,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
19:25:18.875 MQT: stat/tasmota_ACE5A5/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:31:30","StartupUTC":"2023-08-27T17:53:48","Sleep":50,"CfgHolder":4617,"BootCount":73,"BCResetTime":"2023-07-24T18:06:08","SaveCount":307,"SaveAddress":"F6000"}}
19:25:18.888 MQT: stat/tasmota_ACE5A5/STATUS2 = {"StatusFWR":{"Version":"13.0.0.3(tasmota)","BuildDateTime":"2023-08-26T13:09:42","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8285N08","CR":"448/699"}}
19:25:18.899 MQT: stat/tasmota_ACE5A5/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":4,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["fred-iot",""],"TelePeriod":15,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A192800000000","00000080","00006000","00004000","00000000"]}}
[... 9 lines cut ...]
```lua
  Console output here:
19:25:50.270 SRC: MQTT
19:25:50.273 CMD: Grp 0, Cmd 'SENSOR', Idx 1, Len 528, Pld -99, Data '{"Time":"2023-08-27T19:25:50","DS18B20":{"Id":"0301A279BECE","Temperature":21.0},"Thermostat0":{"ThermostatModeSet":1,"TempTargetSet":18.00,"CtrDutyCycleRead":0,"ControlMethod":2,"HybridControllerPhase":1,"EmergencyState":0},"Thermostat1":{"ThermostatModeSet":0,"TempTargetSet":18.00,"CtrDutyCycleRead":0,"ControlMethod":0,"HybridControllerPhase":1,"EmergencyState":0},"Thermostat2":{"ThermostatModeSet":0,"TempTargetSet":18.00,"CtrDutyCycleRead":0,"ControlMethod":0,"HybridControllerPhase":1,"EmergencyState":0},"TempUnit":"C"}'
19:25:50.279 MQT: stat/tasmota_ACE5A5/RESULT = {"Command":"Error"}
19:25:57.035 WIF: Checking connection...
19:25:58.400 MQT: tele/tasmota_ACE5A5/STATE = {"Time":"2023-08-27T19:25:58","Uptime":"0T00:32:10","UptimeSec":1930,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":36,"MqttCount":1,"POWER1":"OFF","POWER2":"ON","POWER3":"OFF","POWER4":"OFF","Wifi":{"AP":1,"SSId":"fred-iot","BSSId":"6A:D7:9A:1F:25:FD","Channel":11,"Mode":"11n","RSSI":70,"Signal":-65,"LinkCount":1,"Downtime":"0T00:00:05"}}
19:25:58.415 MQT: tele/tasmota_ACE5A5/SENSOR = {"Time":"2023-08-27T19:25:58","DS18B20":{"Id":"030297791003","Temperature":-0.8},"Thermostat0":{"ThermostatModeSet":1,"TempTargetSet":0.00,"CtrDutyCycleRead" [... cut ...] et":18.00,"CtrDutyCycleRead":0,"ControlMethod":0,"HybridControllerPhase":1,"EmergencyState":0},"TempUnit":"C"}'
19:26:00.315 MQT: stat/tasmota_ACE5A5/RESULT = {"Command":"Error"}

```

### TO REPRODUCE
See description

### EXPECTED BEHAVIOUR
The temperature value should be read and used to do TempMeasuredSet2  (compiled with 4 thermostats) this should change the measured value.

### SCREENSHOTS

### ADDITIONAL CONTEXT

**(Please, remember to close the issue when the problem has been addressed)**

barbudor, 2023-08-27:
Can you please try with `Subscribe2` ?

sfromis, 2023-08-27:
I'm not seeing this issue when testing (using Subscribe, not Subscribe2), but IIRC it may be due to the input message being pretty long, payload 528 bytes. My testing was with a much shorter payload.

barbudor, 2023-08-27:
Ah yes, this has been identified in a previous discussion
Payload is limited to 256 bytes if I remember

farcouet, 2023-08-27:
> Can you please try with `Subscribe2` ?
I tried with subcribe2 and getting the same issue.

> Ah yes, this has been identified in a previous discussion Payload is limited to 256 bytes if I remember
I wanted also to try sending the updates on a separate topic but I get an error too when trying to setup this from the examples I found.

farcouet, 2023-08-27:
I tried this way to publish the temperature on a sparate topic:
on DS18B20#Temperature do publish tele/test {"Temp":"%value%"} endon

21:18:21.490 CMD: on DS18B20#Temperature do publish tele/test {"Temp":"%value%"} endon
21:18:21.498 MQT: stat/tasmota_54627B/RESULT = {"Command":"Unknown"}

sfromis, 2023-08-27:
You only have 3 rule sets, but can make each pretty long.

farcouet, 2023-08-27:
> You only have 3 rule sets, but can make each pretty long.

Ok thanks I see now ! So I have added it to my previous rules and it works perfectly. Thanks!

farcouet, 2023-08-27:
> Ah yes, this has been identified in a previous discussion Payload is limited to 256 bytes if I remember
Seems to be related to this. Used a workaround to do it.

closed 2023-08-27 by farcouet