arendst/Tasmota #19416: tasmota restarts frequently

xsp1989, 2023-08-30:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
tasmota 12.5.0 and 13.0.0 restart frequently on ZB-GW03, the worst case is restarting every 3-5 minutes, but there is no useful log in the web, usually there is no output, the device restarts itself, the reason for restarting Reboot for the software. But only frequent restarts on specific gateways have ruled out hardware problems and power problems.
The clues that have been found so far are as follows:
1. If the reset command is executed, the abnormal restart of the device will temporarily disappear
2. Delete the MQTT server from the WEB page, the device will no longer restart, reconfigure the mqtt server, and the device will no longer restart
3. Delete all zigbee devices, the gateway still restarts

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): ZB-GW03
- [ ] Tasmota binary firmware version number used: 12.5.0&13.0.0
  - [ ] Pre-compiled
  - [x] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Is it possible that this problem has something to do with the configuration file? How should I analyze this problem, or what doubts do you have, let me verify together

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

s-hadinger, 2023-08-30:
There are possibilities that the Tasmota device restarts by itself:
- losing mqtt connection and not being able to reconnect can trigger a restart 
- If the communication protocol between esp and zigbee MCU enters a bad state or if the zigbee MCU reports an unrecoverable error, Tasmota restarts

xsp1989, 2023-08-30:
I re-added some logs, but they don’t seem to be of much use, but it should be possible to rule out MQTT connection loss. I closed the MQTT server, and tasmota will reconnect multiple times, but what I see is that the gateway is sending the mqtt data after Thereafter, a reboot was performed immediately. And it seems that the data packet sent by EZSP is not received

arendst, 2023-08-30:
@duczz you have a different issue already fixed in latest dev branch.

CuriousJames, 2023-09-18:
Hello, I believe I have a related issue; when I turn on my Shelly 1PM it occasionally (not always) triggers a software OR hardware watchdog OR 'Exception' restart.

I have seen this issue only on this one shelly 1pm, I have at least two others I use regularly also running tasmota 13.1 but these haven't had any issues.  

I've found this is trigged by a turn on command from home assistant (via MQTT to the device) and by the 'toggle' on the device's tasmota webgui.

Weblog 4 our attached.
[shelly_log_hw.txt](https://github.com/arendst/Tasmota/files/12654035/shelly_log_hw.txt)
[shelly_log_hw2.txt](https://github.com/arendst/Tasmota/files/12654073/shelly_log_hw2.txt)
[shelly_log_exception.txt](https://github.com/arendst/Tasmota/files/12654072/shelly_log_exception.txt)

P.S.
I'm unsure if it's related but I have seen this line come up in the log quite frequently: "23:54:37.404 NRG: Energy reset by invalid data"

Thanks!

Jason2866, 2023-09-22:
@xsp1989 Please try latest development version. There was a crash when establishing mqtt connection took too long.

xsp1989, 2023-09-23:
Thank you very much for your support to the open source community, I will rebuild the firmware to verify this issue, thank you again @Jason2866

Jason2866, 2023-09-28:
Closing since not reproducible with current development version.
Please open a new issue when the problem is not solved for you.

closed 2023-09-28 by Jason2866