arendst/Tasmota #19425: ESP32 ethernet +tasmotaclient hangs after a while

Gifford47, 2023-08-30:
### PROBLEM DESCRIPTION
A have build an own firmware for an esp32 with SH1106 I2C display, LAN7820 module and an arduino pro mini @3v3, 8Mhz. After a certain amount of time (2-5 days), the ESP hangs on and it isnt active on the ethernet anymore. The TasmotaClient sketch for the arduino only sends a feedback to the esp after toggling any gpio. The toggling command comes from the esp by MQTT. Currently the esp only switches a relay twice a day (via command to the arduino-slave). After the gpio is toggled everything is fine. I don´t know what exactly causes the exception.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): ESP32 DevKit
- [X] Tasmota binary firmware version number used: 13.1.0
  - [ ] Pre-compiled
  - [X] Self-compiled
- [X] Flashing tools used: ESPFlasher
- [X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua

18:56:39.119 MQT: stat/tasmota_D286E8/RESULT = {"NAME":"ESP32-RAIL","GPIO":[1,1,608,1,4160,4192,0,0,5536,1,1,640,4224,0,5600,0,0,0,0,5568,0,0,0,0,0,0,0,0,1,1,1,1,1,0,0,1],"FLAG":0,"BASE":1}
18:56:39.352 MQT: stat/tasmota_D286E8/RESULT = {"Module":{"0":"ESP32-RAIL"}}
18:56:39.610 MQT: stat/tasmota_D286E8/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"608":"I2C SCL1"},"GPIO3":{"0":"None"},"GPIO4":{"4160":"Client TX"},"GPIO5":{"4192":"Client RX"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"5536":"ETH POWER"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"640":"I2C SDA1"},"GPIO16":{"4224":"Client RST"},"GPIO17":{"0":"None"},"GPIO18":{"5600":"ETH MDIO"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"0":"None"},"GPIO22":{"0":"None"},"GPIO23":{"5568":"ETH MDC"},"GPIO24":{"0":"None"},"GPIO25":{"0":"None"},"GPIO26":{"0":"None"},"GPIO27":{"0":"None"},"GPIO32":{"0":"None"},"GPIO33":{"0":"None"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}
```
- [X] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
18:57:37.447 MQT: stat/tasmota_D286E8/RESULT = {"Rule1":{"State":"ON","Once":"OFF","StopOnError":"OFF","Length":120,"Free":391,"Rules":"on var1#state do ClientSend S %value% endon on system#boot do DisplayDimmer 0 endon "}}
18:57:37.683 MQT: stat/tasmota_D286E8/RESULT = {"Rule2":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
18:57:37.933 MQT: stat/tasmota_D286E8/RESULT = {"Rule3":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
```
- [X] Provide the output of this command: `Status 0`:
```lua
18:58:01.564 MQT: stat/tasmota_D286E8/STATUS = {"Status":{"Module":0,"DeviceName":"esp32rail","FriendlyName":["esp32rail"],"Topic":"tasmota_D286E8","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
18:58:01.575 MQT: stat/tasmota_D286E8/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"0","RestartReason":"Software reset CPU","Uptime":"0T00:17:27","StartupUTC":"2023-08-30T17:40:34","Sleep":50,"CfgHolder":4617,"BootCount":234,"BCResetTime":"2022-06-02T20:06:36","SaveCount":378}}
18:58:01.584 MQT: stat/tasmota_D286E8/STATUS2 = {"StatusFWR":{"Version":"13.1.0(MyOwn)","BuildDateTime":"2023-08-24T20:25:45","Core":"2_0_11","SDK":"4.4.5.230722","CpuFrequency":160,"Hardware":"ESP32-D0WDQ6 rev.1","CR":"337/699"}}
18:58:01.592 MQT: stat/tasmota_D286E8/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["nodes_land",""],"TelePeriod":0,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A0A192800000000","00000080","00004000","00004000","00000000"]}}
18:58:01.606 MQT: stat/tasmota_D286E8/STATUS4 = {"StatusMEM":{"ProgramSize":1411,"Free":1856,"Heap":114,"StackLowMark":4,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"164020","FlashFrequency":40,"FlashMode":"DIO","Features":["00000809","8FBAC7CF","001493A1","000000CF","010013C0","C0010981","400040A0","00200000","5400002C","00000080"],"Drivers":"1,2,3,4,5,7,8,9,10,12,13,16,20,21,24,26,27,29,31,35,38,50,52,62,68,82","Sensors":"1,2,3,5,6,127","I2CDriver":"3,4,5,6,7,47"}}
18:58:01.619 MQT: stat/tasmota_D286E8/STATUS5 = {"StatusNET":{"Hostname":"tasmota-D286E8-1768","IPAddress":"0.0.0.0","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.1.1","DNSServer2":"0.0.0.0","Mac":"C8:C9:A3:D2:86:E8","IP6Global":"","IP6Local":"","Ethernet":{"Hostname":"tasmota-D286E8-1768-eth","IPAddress":"192.168.1.120","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.1.1","DNSServer2":"0.0.0.0","Mac":"C8:C9:A3:D2:86:EB","IP6Global":"","IP6Local":"fe80::cac9:a3ff:fed2:86eb"},"Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
18:58:01.631 MQT: stat/tasmota_D286E8/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.100","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_D286E8","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
18:58:01.642 MQT: stat/tasmota_D286E8/STATUS7 = {"StatusTIM":{"UTC":"2023-08-30T17:58:01","Local":"2023-08-30T18:58:01","StartDST":"2023-03-26T02:00:00","EndDST":"2023-10-29T03:00:00","Timezone":"+01:00","Sunrise":"06:04","Sunset":"19:37"}}
18:58:01.686 MQT: stat/tasmota_D286E8/STATUS10 = {"StatusSNS":{"Time":"2023-08-30T18:58:01","TasmotaClient":{"0":0,"1":0,"2":0,"3":0,"4":0,"5":0,"6":0,"7":0,"8":0,"9":0,"10":0,"11":0,"12":0,"13":0,"14":0,"15":0,"16":0,"17":0,"18":0,"19":0}}}
18:58:01.701 MQT: stat/tasmota_D286E8/STATUS11 = {"StatusSTS":{"Time":"2023-08-30T18:58:01","Uptime":"0T00:17:27","UptimeSec":1047,"Heap":116,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Berry":{"HeapUsed":3,"Objects":37},"POWER":"OFF"}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
- connect an ESP32 with an LAN7820 module like [here](https://sautter.com/blog/ethernet-on-esp32-using-lan8720/)
- build a firmware with the following:
```
#define USE_ETHERNET

#define USE_DISPLAY
#define USE_DISPLAY_SH1106

#ifndef USE_TASMOTA_CLIENT
#define USE_TASMOTA_CLIENT    // Enable the driver
#endif

#ifndef USE_TASMOTA_CLIENT_FLASH_SPEED
#define USE_TASMOTA_CLIENT_FLASH_SPEED 57600    // Configure the baud rate of the bootloader
#endif

#ifndef USE_TASMOTA_CLIENT_SERIAL_SPEED  
#define USE_TASMOTA_CLIENT_SERIAL_SPEED 57600    // Configure the baud rate at which the client microcontroller will be interfacing to Tasmota
#endif
```
- upload a sketch for the atmega
- run it at least for three days

### EXPECTED BEHAVIOUR
After a certain amount of time (2-5 days) the esp32 should be unreachable anymore. no ethernet connection alive, no response from mqtt. I don´t know if the arduino still works.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2023-08-30:
Can you try a ESP32 board with a LAN chip onboard? Connecting the LAN8720 PHY via wires is not an reliable solution.

Gifford47, 2023-09-01:
All modules are linked via a PCB. Only for first test purposes i had i flying wire solution. This worked also fine in the past.
At the moment i restart the ESP at midnight and it seems to be a workaround thus i have no exceptions anymore.
The ESP gets a relative long (about 240 chars) feedback from the tasmotaclient after each gpio action. Could this cause an exception? i can try to disable this behaviour temporary...

Jason2866, 2023-09-01:
Try. All I can say my Esp32 module with an LAN8720 on board does run many days without an issue.

Gifford47, 2023-09-08:
I still have exceptions... I investigated that the esp reboots several times. And sometimes it hangs on boot. The reset reason after a self-reboot is "software reset CPU" and it has a boot count of 238 after two weeks.
How can debug the root cause of the software reset?