arendst/Tasmota #19426: Upgrade fails with Shelly 2PM Plus

fir3drag0n, 2023-08-30:
Hi,
I wanted to upgrade my Shelly Plus 2PM from 12.5.0 to 13 via the Partition manager approach.
The first steps went fine, it seems.
But now I'm stuck in Safeboot mode.
I tried to switch witch Restart 3, but still end up in Safeboot mode.
Restart 1 same thing -> safeboot mode.
I also tried to upgrade FW, same result.
Information page shows at the end:

```
Partition safeboot*	832 KB (used 94%)
Partition app0	2688 KB (used 69%)
Partition fs	448 KB

```

I'm not sure why the device doesn't want to switch to the app0 partition. Any way to force it?

_Originally posted by @foobar26 in https://github.com/arendst/Tasmota/discussions/18983#discussioncomment-6620950_

Jason2866, 2023-08-31:
It is hard or nearly impossible to help here, since we have no logs or other Infos what happens or goes wrong. Only the result.

fir3drag0n, 2023-08-31:
`10:28:03.562 HTP: Firmware Upgrade
10:28:05.496 HTP: Upgrade started
10:28:05.498 SRC: WebGui from 192.168.1.205
10:28:05.499 CMD: Grp 0, Cmd 'OTAURL', Idx 1, Len 56, Pld -99, Data 'http://ota.tasmota.com/tasmota32/tasmota32.bin Upgrade 1'
10:28:05.500 CFG: CR 442/699, Busy 0
10:28:05.502 RSL: RESULT = {"OtaUrl":"http://ota.tasmota.com/tasmota32/tasmota32.bin Upgrade 1"}
10:28:05.634 SRC: WebGui from 192.168.1.205
10:28:05.636 CMD: Grp 0, Cmd 'UPGRADE', Idx 1, Len 1, Pld 1, Data '1'
10:28:05.638 RSL: RESULT = {"Upgrade":"Version 13.1.0 from http://ota.tasmota.com/tasmota32/tasmota32.bin Upgrade 1"}
10:28:08.053 UPL: http://ota.tasmota.com/tasmota32/tasmota32.bin Upgrade 1 13.1.0(safeboot)
10:28:08.819 OTA: Connection error 404 after 116 ms
10:28:09.128 WIF: Checking connection...
10:28:10.588 UPL: http://ota.tasmota.com/tasmota32/tasmota32.bin Upgrade 1 13.1.0(safeboot)
10:28:10.965 OTA: Connection error 404 after 71 ms
10:28:12.587 UPL: http://ota.tasmota.com/tasmota32/tasmota32.bin Upgrade 1 13.1.0(safeboot)
10:28:12.983 OTA: Connection error 404 after 79 ms
10:28:14.606 UPL: http://ota.tasmota.com/tasmota32/tasmota32.bin Upgrade 1 13.1.0(safeboot)
10:28:15.002 OTA: Connection error 404 after 87 ms
10:28:15.735 HTP: Main Menu
10:28:18.605 RSL: UPGRADE = {"Upgrade":"Failed "}
10:28:18.907 HTP: Console
10:28:29.083 WIF: Checking connection...
10:28:38.029 RSL: STATE = {"Time":"2023-08-31T10:28:38","Uptime":"0T13:30:13","UptimeSec":48613,"Heap":165,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"la familia 2.4","BSSId":"1C:ED:6F:4A:7B:5C","Channel":11,"Mode":"11n","RSSI":100,"Signal":-50,"LinkCount":4,"Downtime":"0T00:00:54"}}
10:28:38.031 RSL: SENSOR = {"Time":"2023-08-31T10:28:38","Switch1":"OFF","Switch2":"OFF","ANALOG":{"Temperature1":8.7},"TempUnit":"C"}
10:28:49.114 WIF: Checking connection...
10:29:09.092 WIF: Checking connection...
10:29:29.114 WIF: Checking connection...
10:29:49.106 WIF: Checking connection...`

arendst, 2023-09-03:
First:
```
10:28:05.502 RSL: RESULT = {"OtaUrl":"http://ota.tasmota.com/tasmota32/tasmota32.bin Upgrade 1"}
```
Wrong otaurl. There should be no ``Upgrade 1`` at the end.

Second:
You have a single core processor and therefor need a single-core binary. The correct otaurl should be:
```
http://ota.tasmota.com/tasmota32/release/tasmota32solo1.bin
```

Jason2866, 2023-09-08:
Closing since no feedback.

closed 2023-09-08 by Jason2866