arendst/Tasmota #19430: Tasmota LoadProhibited & watchdog reset when using HRG15 sensor

blacknell, 2023-08-31:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

I've compiled an ESP32 build using the define `USE_HRG15` to define pins `HRG15_Tx` & `HRG15_Tx`. As soon as I add the pins to the template the system reboots with no sensors and a blank template config.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): Firebeetle ESP32
- [x] Tasmota binary firmware version number used: 13.1.0.1
  - [ ] Pre-compiled
  - [x] Self-compiled
- [x] Flashing tools used: Visual Studio
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
12:43:33.800 MQT: stat/tasmota_CBF048/RESULT = {"NAME":"ESP32-WeatherStation","GPIO":[0,0,416,0,0,608,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,640,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"FLAG":0,"BASE":1}
12:43:34.032 MQT: stat/tasmota_CBF048/RESULT = {"Module":{"0":"ESP32-WeatherStation"}}
12:43:34.288 MQT: stat/tasmota_CBF048/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"416":"PWM1"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"608":"I2C SCL1"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"0":"None"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"0":"None"},"GPIO22":{"0":"None"},"GPIO23":{"0":"None"},"GPIO24":{"0":"None"},"GPIO25":{"640":"I2C SDA1"},"GPIO26":{"0":"None"},"GPIO27":{"0":"None"},"GPIO32":{"0":"None"},"GPIO33":{"0":"None"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}
```

But note that the device crashes and resets the template to empty. This is the template I am using
```lua
{"NAME":"ESP32-WeatherStation","GPIO":[0,0,416,0,0,608,0,0,0,224,0,0,0,0,0,0,0,0,0,0,0,640,8032,8064,0,0,0,0,0,0,0,1,0,0,0,0],"FLAG":0,"BASE":1}
```

- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
12:41:47.164 CMD: status 0
12:41:47.176 MQT: stat/tasmota_CBF048/STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_CBF048","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
12:41:47.217 MQT: stat/tasmota_CBF048/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"","RestartReason":"Deep Sleep reset digital core","Uptime":"0T00:01:46","StartupUTC":"2023-08-31T11:40:01","Sleep":50,"CfgHolder":4617,"BootCount":26,"BCResetTime":"2023-08-31T09:54:48","SaveCount":80}}
12:41:47.246 MQT: stat/tasmota_CBF048/STATUS2 = {"StatusFWR":{"Version":"13.1.0.1(weatherstation)","BuildDateTime":"2023-08-31T10:53:25","Core":"2_0_11","SDK":"4.4.5.230820","CpuFrequency":160,"Hardware":"ESP32-D0WD v1.0","CR":"363/699"}}
12:41:47.265 MQT: stat/tasmota_CBF048/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":4,"LogHost":"jumphost.local","LogPort":514,"SSId":["sandalwood",""],"TelePeriod":10,"Resolution":"558180C0","SetOption":["00088009","2805C80001000600003C5A0A192800000000","00000088","00006000","00004000","00000001"]}}
12:41:47.301 MQT: stat/tasmota_CBF048/STATUS4 = {"StatusMEM":{"ProgramSize":1348,"Free":1531,"Heap":161,"StackLowMark":4,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":16384,"FlashSize":16384,"FlashChipId":"1840C8","FlashFrequency":40,"FlashMode":"DIO","Features":["00000809","8F9AC7CF","00148001","000214CF","010013C0","C0000981","00004080","00200000","5400102C","02000080"],"Drivers":"1,2,3,4,5,7,8,9,10,12,16,20,21,24,26,27,29,35,38,50,52,62,68","Sensors":"1,2,3,5,6,9,10,14,90,110,127","I2CDriver":"7,10,11,15,83"}}
12:41:47.353 MQT: stat/tasmota_CBF048/STATUS5 = {"StatusNET":{"Hostname":"tasmota-CBF048-4168","IPAddress":"192.168.1.101","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.1.1","DNSServer2":"0.0.0.0","Mac":"EC:62:60:CB:F0:48","IP6Global":"fd99:6781:c0f5::ee62:60ff:fecb:f048","IP6Local":"fe80::ee62:60ff:fecb:f048","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":16.0}}
12:41:47.392 MQT: stat/tasmota_CBF048/STATUS6 = {"StatusMQT":{"MqttHost":"hatest.local","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_CBF048","MqttUser":"mqtt","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
12:41:47.421 MQT: stat/tasmota_CBF048/STATUS7 = {"StatusTIM":{"UTC":"2023-08-31T11:41:47","Local":"2023-08-31T12:41:47","StartDST":"2023-03-26T01:00:00","EndDST":"2023-10-29T02:00:00","Timezone":99,"Sunrise":"06:05","Sunset":"19:35"}}
12:41:47.479 MQT: stat/tasmota_CBF048/STATUS10 = {"StatusSNS":{"Time":"2023-08-31T12:41:47","BME280":{"Temperature":19.6,"Humidity":60.2,"DewPoint":11.7,"Pressure":1002.3},"BH1750":{"Illuminance":4876},"SHT3X":{"Temperature":15.4,"Humidity":87.2,"DewPoint":13.3},"MAX17043":{"Voltage":4.056,"BatteryPercentage":84.17},"ESP32":{"Temperature":52.8},"PressureUnit":"hPa","TempUnit":"C"}}
12:41:47.523 MQT: stat/tasmota_CBF048/STATUS11 = {"StatusSTS":{"Time":"2023-08-31T12:41:47","Uptime":"0T00:01:46","UptimeSec":106,"Heap":159,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":21,"MqttCount":1,"Berry":{"HeapUsed":3,"Objects":37},"POWER":"OFF","Dimmer":10,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"sandalwood","BSSId":"90:84:0D:E0:E6:9D","Channel":1,"Mode":"11n","RSSI":56,"Signal":-72,"LinkCount":1,"Downtime":"0T00:00:03"}}}
```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
Aug 31 12:58:19 tasmota-CBF048-4168 ESP-MQT: tele/tasmota_CBF048/INFO3 = {"Info3":{"RestartReason":{"Exception":28,"Reason":"LoadProhibited","EPC":"4008c020","EXCVADDR":"00000000","CallChain":["4008c01d","401b2e2a","401b74d9","401b5086","401b50c2","400e3b36","4013ac96","40103165","4010768e","4013496c","40142ec6","400dff0d"]},"BootCount":27}}

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Compile ESP with USE_HRG15
No need to connect device
Use this template
```
{"NAME":"ESP32-WeatherStation","GPIO":[0,0,416,0,0,608,0,0,0,224,0,0,0,0,0,0,0,0,0,0,0,640,8032,8064,0,0,0,0,0,0,0,1,0,0,0,0],"FLAG":0,"BASE":1}
```
### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
There should be some debug out from the HRG sensors 

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
I have seen some similar crashes in [another issue](https://github.com/espressif/arduino-esp32/discussions/6356) where a pin reserved for use by the ESP flash was being used by the user for serial. This resulted in a memory violation and crash. Turned out that pin was undefined.
I can't see the same problem here. GPIO26 & GPIO27 are available for use and the tasmota serial code calls
`TasmotaSerial(Pin(GPIO_HRG15_RX), Pin(GPIO_HRG15_TX), 2, 0, 150);`

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2023-08-31:
Just tried your template. No problem observed:
```
00:00:00.002-227/53 HDW: ESP32-D0WDQ6 
00:00:00.052-226/53 UFS: FlashFS mounted with 304 kB free
00:00:00.081-225/53 CFG: Loaded from File, Count 4307
00:00:00.092-225/53 QPC: Count 1
00:00:00.097-224/53 I2C: Bus1 using GPIO05(SCL) and GPIO25(SDA)
00:00:00.107-224/53 TFS: File 'mcp23x.dat' not found
00:00:00.111-224/52 MCP: Invalid template
00:00:00.116-224/52 ROT: Mode 1
00:00:00.201-218/51 CFG: No '*.autoconf' file found
00:00:00.206-219/51 BRY: Berry initialized, RAM used=3711 bytes
00:00:00.221-219/51 BRY: No 'preinit.be'
00:00:00.270-218/51 CFG: Energy loaded from file
00:00:00.273-218/51 SRC: Restart
00:00:00.279-217/51 Project tasmota - esp32d Version 13.1.0.2(theo)-2_0_11(2023-08-31T15:52:42)
00:00:00.280-218/51 ETH: No ETH MDC and/or ETH MDIO GPIO defined
00:00:00.346-215/51 BRY: No 'autoexec.be'
00:00:00.495-215/51 WIF: Attempting connection...
00:00:00.941-177/40 WIF: Dynamic Tx power enabled
00:00:00.951-177/40 WIF: Connecting to AP1 indebuurt_IoT Channel 11 BSSId 18:E8:29:CA:17:C1 in mode 11n as esp32d...
00:00:01.084-176/39 WIF: IPv4 192.168.2.115, mask 255.255.255.0, gateway 192.168.2.254
00:00:02.660-176/39 WIF: Connected
00:00:02.915-171/38 HTP: Web server active on esp32d with IP address 192.168.2.115
00:00:03.009-169/37 RTC: UTC 2023-08-31T13:59:14, DST 2023-03-26T02:00:00, STD 2023-10-29T03:00:00
15:59:14.000-169/37 RTC: Synced by NTP
15:59:14.695-167/36 WIF: IPv6 Local fe80::9af4:abff:fe6d:2db4
15:59:14.911-167/36 MQT: Attempting connection...
15:59:14.935-165/35 MQT: Connected
15:59:14.945-163/35 MQT: tele/esp32d/LWT = Online (retained)
15:59:14.950-161/34 MQT: cmnd/esp32d/POWER = 
15:59:14.952-165/35 MQT: Subscribe to cmnd/esp32d/#
15:59:14.956-160/33 MQT: Subscribe to cmnd/tasmotas/#
15:59:14.961-160/33 MQT: Subscribe to cmnd/DVES_6D2DB4_fb/#
15:59:14.967-164/35 MQT: Subscribe to domoticz/out/#
15:59:14.978-161/34 MQT: tele/esp32d/INFO1 = {"Info1":{"Module":"ESP32-WeatherStation","Version":"13.1.0.2(theo)","FallbackTopic":"cmnd/DVES_6D2DB4_fb/","GroupTopic":"cmnd/tasmotas/"}}
15:59:15.006-161/34 MQT: tele/esp32d/INFO2 = {"Info2":{"WebServerMode":"Admin","Hostname":"esp32d","IPAddress":"192.168.2.115","IP6Global":"","IP6Local":"fe80::9af4:abff:fe6d:2db4"}}
15:59:15.024-161/34 MQT: tele/esp32d/INFO3 = {"Info3":{"RestartReason":"Software reset CPU","BootCount":1395}}
15:59:15.034-161/34 MQT: stat/esp32d/RESULT = {"POWER1":"OFF"}
15:59:15.042-161/34 MQT: stat/esp32d/POWER1 = OFF
15:59:15.050-161/34 MQT: stat/esp32d/RESULT = {"POWER2":"OFF"}
15:59:15.057-161/34 MQT: stat/esp32d/POWER2 = OFF
15:59:15.066-164/35 TFS: File 'autoexec.bat' not found
15:59:15.892-161/34 MQT: domoticz/in = {"idx":403,"nvalue":0,"svalue":"37","Battery":100,"RSSI":10}
15:59:18.704-164/35 WIF: IPv6 Global fd79:962c:bf13:49d0:9af4:abff:fe6d:2db4
15:59:18.893-164/35 QPC: Reset
15:59:18.911-162/34 MQT: tele/esp32d/STATE = {"Time":"2023-08-31T15:59:18","Uptime":"0T00:00:07","UptimeSec":7,"Heap":164,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Berry":{"HeapUsed":3,"Objects":34},"POWER1":"OFF","POWER2":"OFF","Dimmer":37,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"indebuurt_IoT","BSSId":"18:E8:29:CA:17:C1","Channel":11,"Mode":"11n","RSSI":100,"Signal":-39,"LinkCount":1,"Downtime":"0T00:00:02"}}
15:59:18.954-162/35 MQT: tele/esp32d/SENSOR = {"Time":"2023-08-31T15:59:18","RG-15":{"Active":0.00,"Event":0.00,"Total":0.00,"FlowRate":0.00}}
15:59:20.546-161/34 HTP: Main Menu
15:59:20.883-163/35 APP: Boot Count 1395
15:59:21.621-161/34 CFG: Saved, Count 4308, Bytes 4096
```

arendst, 2023-08-31:
For further investigation I need your console log (log level 3 or up) to see where the issue occurs.

blacknell, 2023-08-31:
```lua
Aug 31 12:58:11 tasmota-CBF048-4168 ESP-SRC: WebGui from 192.168.1.13
Aug 31 12:58:11 tasmota-CBF048-4168 ESP-CMD: Grp 0, Cmd 'TEMPLATE', Idx 1, Len 142, Pld -99, Data '{"NAME":"ESP32-WeatherStation","GPIO":[0,0,416,0,0,608,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,640,8032,8064,0,0,0,0,0,0,0,0,0,0,0,0],"FLAG":0,"BASE":1}'
Aug 31 12:58:11 tasmota-CBF048-4168 ESP-CFG: CR 363/699, Busy 0
Aug 31 12:58:11 tasmota-CBF048-4168 ESP-MQT: stat/tasmota_CBF048/RESULT = {"NAME":"ESP32-WeatherStation","GPIO":[0,0,416,0,0,608,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,640,8032,8064,0,0,0,0,0,0,0,0,0,0,0,0],"FLAG":0,"BASE":1}
Aug 31 12:58:11 tasmota-CBF048-4168 ESP-HTP: Restart
Aug 31 12:58:11 tasmota-CBF048-4168 ESP-SRC: WebGui from 192.168.1.13
Aug 31 12:58:11 tasmota-CBF048-4168 ESP-CFG: Saved, Count 81, Bytes 4096
Aug 31 12:58:17 tasmota-CBF048-4168 ESP-HDW: ESP32-D0WD 
Aug 31 12:58:17 tasmota-CBF048-4168 ESP-UFS: FlashFS mounted with 12592 kB free
Aug 31 12:58:17 tasmota-CBF048-4168 ESP-CFG: Loaded from File, Count 81
Aug 31 12:58:17 tasmota-CBF048-4168 ESP-FRC: Some settings have been reset (6)
Aug 31 12:58:17 tasmota-CBF048-4168 ESP-CFG: CR 363/699, Busy 0
Aug 31 12:58:17 tasmota-CBF048-4168 ESP-ROT: Mode 1
Aug 31 12:58:17 tasmota-CBF048-4168 ESP-TFS: File '.drvset003' not found
Aug 31 12:58:17 tasmota-CBF048-4168 ESP-CFG: Energy use defaults as file system not ready or file not found
Aug 31 12:58:17 tasmota-CBF048-4168 ESP-SRC: Restart
Aug 31 12:58:17 tasmota-CBF048-4168 ESP-Project tasmota - Tasmota Version 13.1.0.1(weatherstation)-2_0_11(2023-08-31T10:53:25)
Aug 31 12:58:17 tasmota-CBF048-4168 ESP-CFG: Saved, Count 82, Bytes 4096
Aug 31 12:58:17 tasmota-CBF048-4168 ESP-WIF: Checking connection...
Aug 31 12:58:17 tasmota-CBF048-4168 ESP-WIF: Attempting connection...
Aug 31 12:58:17 tasmota-CBF048-4168 ESP-WIF: Connecting to AP1 sandalwood Channel 1 BSSId 90:84:0D:E0:E6:9D in mode 11n as tasmota-CBF048-4168...
Aug 31 12:58:17 tasmota-CBF048-4168 ESP-WIF: Checking connection...
Aug 31 12:58:17 tasmota-CBF048-4168 ESP-WIF: Attempting connection...
Aug 31 12:58:17 tasmota-CBF048-4168 ESP-WIF: IPv4 192.168.1.101, mask 255.255.255.0, gateway 192.168.1.1
Aug 31 12:58:17 tasmota-CBF048-4168 ESP-WIF: Checking connection...
Aug 31 12:58:17 tasmota-CBF048-4168 ESP-WIF: Connected
Aug 31 12:58:17 tasmota-CBF048-4168 ESP-HTP: Web server active on tasmota-CBF048-4168 with IP address 192.168.1.101
Aug 31 12:58:17 tasmota-CBF048-4168 ESP-WIF: DNS resolved 'jumphost.local' (192.168.1.4) in 11 ms
Aug 31 12:58:18 tasmota-CBF048-4168 ESP-NTP: Sync time...
Aug 31 12:58:18 tasmota-CBF048-4168 ESP-WIF: DNS resolved '2.pool.ntp.org' (85.199.214.102) in 15 ms
Aug 31 12:58:18 tasmota-CBF048-4168 ESP-RTC: UTC 2023-08-31T11:58:17, DST 2023-03-26T01:00:00, STD 2023-10-29T02:00:00
Aug 31 12:58:18 tasmota-CBF048-4168 ESP-RTC: Synced by NTP
Aug 31 12:58:19 tasmota-CBF048-4168 ESP-MQT: Attempting connection...
Aug 31 12:58:19 tasmota-CBF048-4168 ESP-WIF: DNS resolved 'hatest.local' (192.168.1.180) in 5 ms
Aug 31 12:58:19 tasmota-CBF048-4168 ESP-MQT: Connected
Aug 31 12:58:19 tasmota-CBF048-4168 ESP-MQT: tele/tasmota_CBF048/LWT = Online (retained)
Aug 31 12:58:19 tasmota-CBF048-4168 ESP-MQT: cmnd/tasmota_CBF048/POWER = 
Aug 31 12:58:19 tasmota-CBF048-4168 ESP-MQT: Subscribe to cmnd/tasmota_CBF048/#
Aug 31 12:58:19 tasmota-CBF048-4168 ESP-MQT: Subscribe to cmnd/tasmotas/#
Aug 31 12:58:19 tasmota-CBF048-4168 ESP-MQT: Subscribe to cmnd/DVES_CBF048_fb/#
Aug 31 12:58:19 tasmota-CBF048-4168 ESP-MQT: tele/tasmota_CBF048/INFO1 = {"Info1":{"Module":"ESP32-DevKit","Version":"13.1.0.1(weatherstation)","FallbackTopic":"cmnd/DVES_CBF048_fb/","GroupTopic":"cmnd/tasmotas/"}}
Aug 31 12:58:19 tasmota-CBF048-4168 ESP-MQT: tele/tasmota_CBF048/INFO2 = {"Info2":{"WebServerMode":"Admin","Hostname":"tasmota-CBF048-4168","IPAddress":"192.168.1.101","IP6Global":"","IP6Local":"fe80::ee62:60ff:fecb:f048"}}
Aug 31 12:58:19 tasmota-CBF048-4168 ESP-MQT: tele/tasmota_CBF048/INFO3 = {"Info3":{"RestartReason":{"Exception":28,"Reason":"LoadProhibited","EPC":"4008c020","EXCVADDR":"00000000","CallChain":["4008c01d","401b2e2a","401b74d9","401b5086","401b50c2","400e3b36","4013ac96","40103165","4010768e","4013496c","40142ec6","400dff0d"]},"BootCount":27}}
Aug 31 12:58:19 tasmota-CBF048-4168 ESP-SRC: MQTT
Aug 31 12:58:19 tasmota-CBF048-4168 ESP-CMD: Grp 0, Cmd 'DEEPSLEEPTIME', Idx 1, Len 1, Pld 0, Data '0'
Aug 31 12:58:19 tasmota-CBF048-4168 ESP-MQT: stat/tasmota_CBF048/RESULT = {"DeepSleepTime":0}
Aug 31 12:58:19 tasmota-CBF048-4168 ESP-SRC: MQTT
Aug 31 12:58:19 tasmota-CBF048-4168 ESP-CMD: Grp 0, Cmd 'STATUS', Idx 1, Len 1, Pld 0, Data '0'
Aug 31 12:58:19 tasmota-CBF048-4168 ESP-MQT: stat/tasmota_CBF048/STATUS = {"Status":{"Module":1,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_CBF048","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
Aug 31 12:58:19 tasmota-CBF048-4168 ESP-MQT: stat/tasmota_CBF048/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"","RestartReason":"Software reset CPU","Uptime":"0T00:00:05","StartupUTC":"2023-08-31T11:58:13","Sleep":50,"CfgHolder":4617,"BootCount":26,"BCResetTime":"2023-08-31T09:54:48","SaveCount":82}}
[... cut ...]
Aug 31 12:58:51 tasmota-CBF048-4168 ESP-MQT: tele/tasmota_CBF048/STATE = {"Time":"2023-08-31T12:58:49","Uptime":"0T00:00:36","UptimeSec":36,"Heap":167,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"sandalwood","BSSId":"90:84:0D:E0:E6:9D","Channel":1,"Mode":"11n","RSSI":62,"Signal":-69,"LinkCount":1,"Downtime":"0T00:00:03"}}
Aug 31 12:58:51 tasmota-CBF048-4168 ESP-MQT: tele/tasmota_CBF048/SENSOR = {"Time":"2023-08-31T12:58:49","ESP32":{"Temperature":52.2},"TempUnit":"C"}
```

arendst, 2023-08-31:
Yep I did. See ``15:59:18.954-162/35 MQT: tele/esp32d/SENSOR = {"Time":"2023-08-31T15:59:18","RG-15":{"Active":0.00,"Event":0.00,"Total":0.00,"FlowRate":0.00}}``

What you read is ``00:00:00.111-224/52 MCP: Invalid template``. That's about compiled in MCP2300 support irrelevant for your use-case.

arendst, 2023-08-31:
Thx. That's the end of a six cycle restart and doesn't show the initial cause of the exception. It's also a syslog logging which most likely doesn't show the initial cause either as the system is already restarting before sending the syslog log.

If it's a developement ESP32 you must be able to seriallog what happens from the start, so before the six cycle restart (which resets your config).

blacknell, 2023-08-31:
That was easier because it's remote from me. I'll go get the box...

blacknell, 2023-08-31:
Here we are. A lot more info as you said...

```
16:02:45.117 SRC: Backlog
16:02:45.119 CMD: Grp 0, Cmd 'TEMPLATE', Idx 1, Len 144, Pld -99, Data '{"NAME":"ESP32-WeatherStation","GPIO":[0,0,416,0,0,608,0,0,0,224,0,0,0,0,0,0,0,0,0,0,0,640,8032,8064,0,0,0,0,0,0,0,1,0,0,0,0],"FLAG":0,"BASE":1}'
16:02:45.133 CFG: CR 363/699, Busy 0
16:02:45.142 MQT: stat/tasmota_CBF048/RESULT = {"NAME":"ESP32-WeatherStation","GPIO":[0,0,416,0,0,608,0,0,0,224,0,0,0,0,0,0,0,0,0,0,0,640,8032,8064,0,0,0,0,0,0,0,1,0,0,0,0],"FLAG":0,"BASE":1}
16:02:45.525 CFG: Saved, Count 122, Bytes 4096
16:02:46.242 APP: Restarting
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:184
load:0x40078000,len:12732
ho 0 tail 12 room 4
load:0x40080400,len:2908
entry 0x400805c4

00:00:00.002 HDW: ESP32-D0WD 
00:00:00.064 UFS: FlashFS mounted with 12584 kB free
00:00:00.096 CFG: Loaded from File, Count 122
00:00:00.106 QPC: Count 1
00:00:00.107 CFG: CR 363/699, Busy 0
00:00:00.112 I2C: Bus1 using GPIO05(SCL) and GPIO25(SDA)
00:00:00.117 ROT: Mode 1
00:00:00.128 BRY: GC from 3899 to 3118 bytes, objects freed 5/16 (in 0 ms) - slots from 34/61 to 21/61
00:00:00.206 CFG: No '*.autoconf' file found
00:00:00.211 BRY: GC from 4599 to 3902 bytes, objects freed 5/35 (in 1 ms) - slots from 55/76 to 37/76
00:00:00.212 BRY: Berry initialized, RAM used=3902 bytes
00:00:00.237 BRY: No 'preinit.be'
00:00:00.243 TFS: File '.drvset003' not found
00:00:00.244 CFG: Energy use defaults as file system not ready or file not found
00:00:00.246 SRC: Restart
00:00:00.252 Project tasmota - Tasmota Version 13.1.0.1(weatherstation)-2_0_11(2023-08-31T15:57:54)
00:00:00.277 I2C: BME280 found at 0x77
00:00:00.279 I2C: BH1750 found at 0x23
00:00:00.311 I2C: SHT3X found at 0x44
00:00:00.313 HRG: Rg15Init 1073431492
00:00:00.315 HRG: Rg15Init Perform RG-15 init
00:00:00.338 I2C: MAX17043 found at 0x36
00:00:00.396 BRY: No 'autoexec.be'
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x4008c020  PS      : 0x00060e30  A0      : 0x801b2e2d  A1      : 0x3ffb1e70  
A2      : 0x00000000  A3      : 0x00000000  A4      : 0xffffffff  A5      : 0x00000000  
A6      : 0x00000000  A7      : 0x3ffb2064  A8      : 0x00000000  A9      : 0x3ffb1f90  
A10     : 0x3ffb53b4  A11     : 0x00000014  A12     : 0x3ffb0000  A13     : 0x3ffbe395  
A14     : 0x3ffc6f04  A15     : 0x0000016f  SAR     : 0x00000011  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x4008ce69  LEND    : 0x4008ce76  LCOUNT  : 0xffffffee  

Backtrace: 0x4008c01d:0x3ffb1e70 0x401b2e2a:0x3ffb1e90 0x401b74d9:0x3ffb1ed0 0x401b5086:0x3ffb1f80 0x401b50c2:0x3ffb2010 0x400e3b36:0x3ffb2050 0x4013ac96:0x3ffb20c0 0x40103165:0x3ffb2120 0x4010768e:0x3ffb21e0 0x4013496c:0x3ffb2200 0x40142ec6:0x3ffb2220 0x400dff0d:0x3ffb2290

ELF file SHA256: 0000000000000000

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:184
load:0x40078000,len:12732
ho 0 tail 12 room 4
load:0x40080400,len:2908
entry 0x400805c4
[... 271 lines cut ...]
16:02:56.005 MQT: tasmota/discovery/EC6260CBF048/config =  (retained)
16:02:56.008 MQT: tasmota/discovery/EC6260CBF048/sensors =  (retained)
16:02:56.072 SRC: Backlog
16:02:56.074 CMD: Grp 0, Cmd 'TIMEDST', Idx 1, Len 12, Pld 0, Data '0,0,3,1,1,60'
16:02:56.084 MQT: stat/tasmota_CBF048/RESULT = {"TimeDst":{"Hemisphere":0,"Week":0,"Month":3,"Day":1,"Hour":1,"Offset":60}}
16:02:56.773 SRC: MQTT
16:02:56.775 CMD: Grp 0, Cmd 'STATUS', Idx 1, Len 1, Pld 1, Data '1'
16:02:56.782 MQT: stat/tasmota_CBF048/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"","RestartReason":"Software reset CPU","Uptime":"0T00:00:08","StartupUTC":"2023-08-31T15:02:48","Sleep":50,"CfgHolder":4617,"BootCount":40,"BCResetTime":"2023-08-31T09:54:48","SaveCount":123}}
16:02:56.823 SRC: MQTT
16:02:56.826 CMD: Grp 0, Cmd 'STATUS', Idx 1, Len 2, Pld 11, Data '11'
16:02:56.836 MQT: stat/tasmota_CBF048/STATUS11 = {"StatusSTS":{"Time":"2023-08-31T16:02:56","Uptime":"0T00:00:08","UptimeSec":8,"Heap":170,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":50,"MqttCount":1,"Wifi":{"AP":1,"SSId":"sandalwood","BSSId":"9C:C9:EB:DE:2E:60","Channel":1,"Mode":"11n","RSSI":62,"Signal":-69,"LinkCount":1,"Downtime":"0T00:00:03"}}}
16:02:56.868 QPC: Reset
16:02:56.924 NTP: Sync time...
16:02:56.927 WIF: DNS resolved '2.pool.ntp.org' (195.171.43.12) in 0 ms
16:02:56.009 RTC: UTC 2023-08-31T15:02:56, DST 2023-03-26T01:00:00, STD 2023-10-29T02:00:00
16:02:56.000 RTC: Synced by NTP
16:02:56.005 SRC: MQTT
16:02:56.007 CMD: Grp 0, Cmd 'STATUS', Idx 1, Len 2, Pld 10, Data '10'
16:02:56.016 MQT: stat/tasmota_CBF048/STATUS10 = {"StatusSNS":{"Time":"2023-08-31T16:02:56","ESP32":{"Temperature":55.6},"TempUnit":"C"}}
16:02:56.767 WIF: IPv6 Global fd99:6781:c0f5::ee62:60ff:fecb:f048
16:02:57.914 APP: Boot Count 41
16:02:58.389 CFG: Saved, Count 124, Bytes 4096
16:03:04.903 MQT: tele/tasmota_CBF048/STATE = {"Time":"2023-08-31T16:03:04","Uptime":"0T00:00:16","UptimeSec":16,"Heap":169,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"sandalwood","BSSId":"9C:C9:EB:DE:2E:60","Channel":1,"Mode":"11n","RSSI":62,"Signal":-69,"LinkCount":1,"Downtime":"0T00:00:03"}}
16:03:04.931 MQT: tele/tasmota_CBF048/SENSOR = {"Time":"2023-08-31T16:03:04","ESP32":{"Temperature":55.0},"TempUnit":"C"}
```

blacknell, 2023-08-31:
I noticed that too. The same with I2C taken out of template completely

```
16:24:49.096 MQT: stat/tasmota_CBF048/RESULT = {"NAME":"ESP32-WeatherStation","GPIO":[0,0,416,0,0,0,0,0,0,224,0,0,0,0,0,0,0,0,0,0,0,0,8032,8064,0,0,0,0,0,0,0,1,0,0,0,0],"FLAG":0,"BASE":1}
16:24:49.384 CFG: Saved, Count 125, Bytes 4096
16:24:50.293 APP: Restarting
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:184
load:0x40078000,len:12732
ho 0 tail 12 room 4
load:0x40080400,len:2908
entry 0x400805c4

00:00:00.003 HDW: ESP32-D0WD 
00:00:00.032 UFS: FlashFS mounted with 12584 kB free
00:00:00.046 CFG: Loaded from File, Count 125
00:00:00.057 QPC: Count 1
00:00:00.058 CFG: CR 363/699, Busy 0
00:00:00.066 ROT: Mode 1
00:00:00.077 BRY: GC from 3899 to 3118 bytes, objects freed 5/16 (in 0 ms) - slots from 34/61 to 21/61
00:00:00.117 CFG: No '*.autoconf' file found
00:00:00.122 BRY: GC from 4599 to 3902 bytes, objects freed 5/35 (in 1 ms) - slots from 55/76 to 37/76
00:00:00.122 BRY: Berry initialized, RAM used=3902 bytes
00:00:00.143 BRY: No 'preinit.be'
00:00:00.147 TFS: File '.drvset003' not found
00:00:00.147 CFG: Energy use defaults as file system not ready or file not found
00:00:00.149 SRC: Restart
00:00:00.155 Project tasmota - Tasmota Version 13.1.0.1(weatherstation)-2_0_11(2023-08-31T15:57:54)
00:00:00.167 HRG: Rg15Init 1073430136
00:00:00.169 HRG: Rg15Init Perform RG-15 init
00:00:00.197 BRY: No 'autoexec.be'
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x4008c020  PS      : 0x00060e30  A0      : 0x801b2e2d  A1      : 0x3ffb1e70  
A2      : 0x00000000  A3      : 0x00000000  A4      : 0xffffffff  A5      : 0x00000000  
A6      : 0x00000000  A7      : 0x3ffb2064  A8      : 0x00000000  A9      : 0x3ffb1f90  
A10     : 0x3ffb4fc0  A11     : 0x00000014  A12     : 0x3ffb0000  A13     : 0x00006000  
A14     : 0x00000000  A15     : 0x3ffb2190  SAR     : 0x0000001d  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x4008ce69  LEND    : 0x4008ce76  LCOUNT  : 0xffffffee  

Backtrace: 0x4008c01d:0x3ffb1e70 0x401b2e2a:0x3ffb1e90 0x401b74d9:0x3ffb1ed0 0x401b5086:0x3ffb1f80 0x401b50c2:0x3ffb2010 0x400e3b36:0x3ffb2050 0x4013ac96:0x3ffb20c0 0x40103165:0x3ffb2120 0x4010768e:0x3ffb21e0 0x4013496c:0x3ffb2200 0x40142ec6:0x3ffb2220 0x400dff0d:0x3ffb2290

ELF file SHA256: 0000000000000000

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:184
load:0x40078000,len:12732
ho 0 tail 12 room 4
load:0x40080400,len:2908
entry 0x400805c4

00:00:00.002 HDW: ESP32-D0WD 
00:00:00.031 UFS: FlashFS mounted with 12584 kB free
00:00:00.045 CFG: Loaded from File, Count 125
00:00:00.048 FRC: Some settings have been reset (2)
00:00:00.049 CFG: CR 363/699, Busy 0
00:00:00.056 ROT: Mode 1
00:00:00.067 BRY: GC from 3899 to 3118 bytes, objects freed 5/16 (in 1 ms) - slots from 34/61 to 21/61
00:00:00.107 CFG: No '*.autoconf' file found
00:00:00.112 BRY: GC from 4599 to 3902 bytes, objects freed 5/35 (in 1 ms) - slots from 55/76 to 37/76
00:00:00.113 BRY: Berry initialized, RAM used=3902 bytes
00:00:00.133 BRY: No 'preinit.be'
00:00:00.137 TFS: File '.drvset003' not found
00:00:00.138 CFG: Energy use defaults as file system not ready or file not found
00:00:00.140 SRC: Restart
00:00:00.146 Project tasmota - Tasmota Version 13.1.0.1(weatherstation)-2_0_11(2023-08-31T15:57:54)
00:00:00.157 HRG: Rg15Init 1073430136
00:00:00.159 HRG: Rg15Init Perform RG-15 init
00:00:00.188 BRY: No 'autoexec.be'
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x4008c020  PS      : 0x00060e30  A0      : 0x801b2e2d  A1      : 0x3ffb1e70  
A2      : 0x00000000  A3      : 0x00000000  A4      : 0xffffffff  A5      : 0x00000000  
A6      : 0x00000000  A7      : 0x00060023  A8      : 0x00000000  A9      : 0x3ffb1f90  
A10     : 0x3ffb4fc0  A11     : 0x00000014  A12     : 0x3ffb0000  A13     : 0x00006000  
A14     : 0x00000000  A15     : 0x3ffb2190  SAR     : 0x0000001d  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x4008ce69  LEND    : 0x4008ce76  LCOUNT  : 0xffffffee  

Backtrace: 0x4008c01d:0x3ffb1e70 0x401b2e2a:0x3ffb1e90 0x401b74d9:0x3ffb1ed0 0x401b5086:0x3ffb1f80 0x401b50c2:0x3ffb2010 0x400e3b36:0x3ffb2050 0x4013ac96:0x3ffb20c0 0x40103165:0x3ffb2120 0x4010768e:0x3ffb21e0 0x4013496c:0x3ffb2200 0x40142ec6:0x3ffb2220 0x400dff0d:0x3ffb2290

ELF file SHA256: 0000000000000000

Rebooting...
```

I've pasted in the first boot log

blacknell, 2023-08-31:
I found the problem. An error in one of my debug lines.

**I apologise profoundly.** 🤭

I was confused because in trying to fix some bugs in the HRG driver (for which I will submit a PR) I switched to the Firebeetle board in platformio (since this is my board) and then the crashes started.

I wonder if defining the board makes a material difference??

closed 2023-08-31 by blacknell