arendst/Tasmota #19441: (Shutter) configuration lost on OTA update

JosefRypacek, 2023-09-03:
### PROBLEM DESCRIPTION
This issue is continuation of discussion here: https://github.com/arendst/Tasmota/pull/18295#issuecomment-1629469375
Today, I've updated my devices and I lost configuration again.

The same update procedure was done on 9 devices (same type, revision, firmware version and configuration) from/to following versions.
```
Program Version	13.0.0.2(tasmota32)
Build Date & Time	2023-07-12T10:49:37

Program Version	13.1.0(tasmota32)
Build Date & Time	2023-08-15T08:00:54
```

After update I lost all settings (including Wi-Fi) on 1 device.
The remaining 8 devices lost "only" shutter configuration. Please see following output for status0 before update, console output during/after OTA update and status0 after update.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): Sonoff Dual R3
- [X] Tasmota binary firmware version number used: 13.0.0.2 -> 13.1.0
  - [X] Pre-compiled
  - [ ] Self-compiled
- [X] Flashing tools used: OTA
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
08:42:36.635 CMD: Backlog Template; Module; GPIO 255
08:42:36.707 MQT: stat/tasmota_48C4D8/RESULT = {"NAME":"Sonoff Dual R3","GPIO":[34,0,0,0,0,0,0,0,0,576,225,0,0,0,0,0,0,0,0,0,0,7296,7328,224,0,0,0,0,32,33,0,0,0,0,0,0],"FLAG":0,"BASE":1}
08:42:36.907 MQT: stat/tasmota_48C4D8/RESULT = {"Module":{"0":"Sonoff Dual R3"}}
08:42:37.108 MQT: stat/tasmota_48C4D8/RESULT = {"GPIO0":{"34":"Button3"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"576":"LedLink_i"},"GPIO14":{"225":"Relay2"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"0":"None"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"0":"None"},"GPIO22":{"0":"None"},"GPIO23":{"0":"None"},"GPIO24":{"0":"None"},"GPIO25":{"7296":"CSE7761 Tx"},"GPIO26":{"7328":"CSE7761 Rx"},"GPIO27":{"224":"Relay1"},"GPIO32":{"32":"Button1"},"GPIO33":{"33":"Button2"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0` BEFORE UPDATE:
```lua
  STATUS 0 output here:
07:31:56.100 CMD: status0
07:31:56.153 MQT: stat/tasmota_48C4D8/STATUS0 = {"Status":{"Module":0,"DeviceName":"zaluzie-pokoj-levy","FriendlyName":["Tasmota",""],"Topic":"tasmota_48C4D8","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0},"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/tasmota32.bin","RestartReason":"Vbat power on reset","Uptime":"17T12:52:10","StartupUTC":"2023-08-16T17:39:46","Sleep":50,"CfgHolder":4617,"BootCount":10,"BCResetTime":"2023-07-09T22:57:25","SaveCount":32},"StatusFWR":{"Version":"13.0.0.2(tasmota32)","BuildDateTime":"2023-07-12T10:49:37","Core":"2_0_10","SDK":"4.4.5.230614","CpuFrequency":160,"Hardware":"ESP32-D0WD-V3 rev.3","CR":"399/699"},"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["RtIOT",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["0000C009","2805C80001000600003C5A0A192800000000","40800080","00006000","00004000","00000000"]},"StatusMEM":{"ProgramSize":1884,"Free":995,"Heap":106,"StackLowMark":4,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"1640EF","FlashFrequency":40,"FlashMode":"DIO","Features":["00000809","9F9AD7DF","0015A001","B7F7BFCF","05DA9BC0","E0360DC7","480840D2","20200000","D4BC482D","010A80B1"],"Drivers":"1,2,3,4,5,7,8,9,10,11,12,14,16,17,20,21,24,26,27,29,34,35,38,50,52,59,60,62,63,66,67,68,82,86,87,88","Sensors":"1,2,3,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,31,34,37,39,40,42,43,45,51,52,55,56,58,59,64,66,67,74,85,92,95,98,103,105,109,127","I2CDriver":"7,8,9,10,11,12,13,14,15,17,18,20,24,29,31,36,41,42,44,46,48,58,62,65,69,76,77,82"},"StatusNET":{"Hostname":"tasmota-48C4D8-1240","IPAddress":"xxx.yyy.zzz.40","Gateway":"xxx.yyy.zzz.1","Subnetmask":"255.255.255.0","DNSServer1":"xxx.yyy.zzz.1","DNSServer2":"0.0.0.0","Mac":"3C:61:05:48:C4:D8","IP6Global":"","IP6Local":"fe80::3e61:5ff:fe48:c4d8","Ethernet":{"Hostname":"","IPAddress":"0.0.0.0","Gateway":"0.0.0.0","Subnetmask":"0.0.0.0","DNSServer1":"xxx.yyy.zzz.1","DNSServer2":"0.0.0.0","Mac":"00:00:00:00:00:00","IP6Global":"","IP6Local":""},"Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":16.0},"StatusMQT":{"MqttHost":"mqtt.XXXX","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_48C4D8","MqttUser":"DVES_USER","MqttCount":13,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4},"StatusTIM":{"UTC":"2023-09-03T06:31:56","Local":"2023-09-03T07:31:56","StartDST":"2023-03-26T02:00:00","EndDST":"2023-10-29T03:00:00","Timezone":"+01:00","Sunrise":"06:09","Sunset":"19:29"},"StatusPTH":{"PowerDelta":[0,0,0,0,0,0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0},"StatusSNS":{"Time":"2023-09-03T07:31:56","ENERGY":{"TotalStartTime":"2023-07-09T22:01:18","Total":0.004,"Yesterday":0.000,"Today":0.000,"Power":[0,0],"ApparentPower":[0,0],"ReactivePower":[0,0],"Factor":[0.00,0.00],"Voltage":0,"Current":[0.000,0.000]},"Shutter1":{"Position":100,"Direction":0,"Target":100,"Tilt":0}},"StatusSTS":{"Time":"2023-09-03T07:31:56","Uptime":"17T12:52:10","UptimeSec":1515130,"Heap":104,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":20,"MqttCount":13,"Berry":{"HeapUsed":4,"Objects":47},"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"RtIOT","BSSId":"3A:60:F9:5D:26:62","Channel":9,"Mode":"11n","RSSI":96,"Signal":-52,"LinkCount":7,"Downtime":"0T00:01:27"}},"StatusSHT":{"SHT0":{"Relay1":1,"Relay2":2,"Open":340,"Close":340,"50perc":50,"Delay":2,"Opt":"0100","Calib":[300,500,700,900,1000],"Mode":"1","TiltConfig":[-90,0,19,0,-90]}}}
```

- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
DURING/AFTER UPDATE, `weblog` was not set
```lua
  Console output here:
00:00:00.003 HDW: ESP32-D0WD-V3 
00:00:00.024 UFS: FlashFS mounted with 300 kB free
00:00:00.040 CFG: Loaded from File, Count 33
00:00:00.044 SHT: About to save settings to file /.drvset027
00:00:00.331 QPC: Count 1
00:00:00.424 BRY: Berry initialized, RAM used=4470 bytes
00:00:00.468 CFG: Energy loaded from file
00:00:00.469 NRG: Init driver 19
00:00:00.470 Shutter: Use defaults
00:00:00.471 SHT: About to load settings from file /.drvset027
00:00:00.506 SHT: About to save settings to file /.drvset027
00:00:00.607 Project tasmota - zaluzie-pokoj-levy Version 13.1.0(tasmota32)-2_0_11(2023-08-15T08:00:54)
00:00:01.040 WIF: Connecting to AP1 RtIOT Channel 9 BSSId 3A:60:F9:5D:26:62 in mode 11n as tasmota-48C4D8-1240...
00:00:04.107 WIF: Connected
00:00:04.371 HTP: Web server active on tasmota-48C4D8-1240 with IP address xxx.yyy.zzz.40
07:33:49.395 MQT: Attempting connection...
[... 13 lines cut ...]
07:33:50.739 MQT: stat/tasmota_48C4D8/STATUS11 = {"StatusSTS":{"Time":"2023-09-03T07:33:50","Uptime":"0T00:00:07","UptimeSec":7,"Heap":113,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":41,"MqttCount":1,"Berry":{"HeapUsed":4,"Objects":47},"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"RtIOT","BSSId":"3A:60:F9:5D:26:62","Channel":9,"Mode":"11n","RSSI":94,"Signal":-53,"LinkCount":1,"Downtime":"0T00:00:05" [... cut ...] meSec":94,"Heap":103,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":20,"MqttCount":1,"Berry":{"HeapUsed":4,"Objects":47},"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"RtIOT","BSSId":"3A:60:F9:5D:26:62","Channel":9,"Mode":"11n","RSSI":92,"Signal":-54,"LinkCount":1,"Downtime":"0T00:00:05"}},"StatusSHT":{}}
```

### TO REPRODUCE
Update device with shutter config over the OTA.

### EXPECTED BEHAVIOUR
Retain shutter config.

### SCREENSHOTS
Not applicable.

### ADDITIONAL CONTEXT
No other context.

stefanbode, 2023-09-03:
Thanks for bringing it up. Will take a closer look on it on Monday. Gets priority

stefanbode, 2023-09-03:
@arendst do you think we have to switch back to the old driver until the fix on Monday or does this even create more confusion. Not sure where the upgrade lost the config. Quite sure it will be fixed in less than 24h

arendst, 2023-09-03:
No need to switch back. We'll just wait for your solution.

stefanbode, 2023-09-04:
With the submitted fix the problem should be away for new migrations. There is no easy way to fix this automated. Deleting the drvset0027 file is the most easiest way.

JosefRypacek, 2023-09-04:
Thanks, I'm glad I was helpful to identify the issue. I've actually recovered my devices manually, so don't need the workaround as the fix should be in new release. Thanks!

closed 2023-09-04 by JosefRypacek

Ansgar66, 2023-09-11:
Dear all, 
I do have the same issue and tried to follow this thread:
- updated from 9.4 to 13.1 (using web-installer since before got the "size error"
- restored the configuration file backuped before
- settings were accepted, but no shutter support is available

"SetOption80" leads to "ON"
Even after setting SetOption80 "OFF" and "OFF" again, the system doesn't know shuttercommands:
```
12:51:41.318 CMD: shutterposition
12:51:41.326 MQT: stat/Roll_K/RESULT = {"Command":"Error"}
```
- deleted "drvset0027" following this thread
- restarted device
--> no change ;-(

PR #19454 (2023-09-04): Fix migration error to filesystem settings
## Description:
File was saved before initialized. This causes migration error.
**Related issue (if applicable):** fixes #<Tasmota issue number goes here>
fix #19441

## Checklist:
  - [x] The pull request is done against the latest development branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR and the code change compiles without warnings
  - [x] The code change is tested and works with Tasmota core ESP8266 V.2.7.4.9
  - [x] The code change is tested and works with Tasmota core ESP32 V.2.0.11
  - [x] I accep [... cut]