arendst/Tasmota #19443: ESP32 does not receive serial data after "Serial Send 1"

Sbellins, 2023-09-03:
### PROBLEM DESCRIPTION
Hello, i have a KG battery meter that send data to a ESP32 and then i send it over mqtt on home assistant. Until a couple of days ago it worked all fine but now i cannot send the serial data anymore. 
If i set SerialSend 1, the ESP no longher read the data provvided by the battery monitor.
I've set:
  Teleperiod to 10
  Setoption19 1
  Baudrate 115200

This are the strings the battery monito sends to the ESP, the data i need are just in the R50 string.
[image]

And after "SerialSend 1" it shows nothing anymore (when it was working it showed something like in the console and it was sending it to home assistent)

``` lua
tele/BatteryMonitor/RESULT = {"SerialReceived":":R50=1,\n:r50=1,243,1242,0,11000,70,63,55062,125,0,0,0,0,0,\r\n"} "
```

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): ESP-32 Board
- [x] Tasmota binary firmware version number used:  Tasmota 13.1.0 by Theo Arends
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: Web Installer
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
12:38:47.915 CMD: Backlog Template; Module; GPIO 255
12:38:47.948 MQT: stat/BatteryMonitor/RESULT = {"NAME":"ESP32-DevKit","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,0,1,1,1,0,1,1,1,0,0,0,0,1,1,1,1,1,0,0,1],"FLAG":0,"BASE":1}
12:38:47.963 MQT: tele/BatteryMonitor/STATE = {"Time":"2023-09-03T12:38:47","Uptime":"0T00:05:18","UptimeSec":318,"Heap":111,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Berry":{"HeapUsed":4,"Objects":42},"Wifi":{"AP":1,"SSId":"IoT","BSSId":"1C:ED:6F:B3:69:75","Channel":11,"Mode":"11n","RSSI":46,"Signal":-77,"LinkCount":1,"Downtime":"0T00:00:03"}}
12:38:48.174 MQT: stat/BatteryMonitor/RESULT = {"Module":{"1":"ESP32-DevKit"}}
12:38:48.419 MQT: stat/BatteryMonitor/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"0":"None"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"0":"None"},"GPIO22":{"0":"None"},"GPIO23":{"0":"None"},"GPIO24":{"0":"None"},"GPIO25":{"0":"None"},"GPIO26":{"0":"None"},"GPIO27":{"0":"None"},"GPIO32":{"0":"None"},"GPIO33":{"0":"None"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
12:42:01.972 CMD: Status 0
12:42:01.977 MQT: stat/BatteryMonitor/STATUS = {"Status":{"Module":1,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"BatteryMonitor","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
12:42:01.983 MQT: stat/BatteryMonitor/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32.bin","RestartReason":"Software reset CPU","Uptime":"0T00:08:32","StartupUTC":"2023-09-03T11:33:29","Sleep":50,"CfgHolder":4617,"BootCount":5,"BCResetTime":"2023-09-03T12:21:32","SaveCount":17}}
12:42:01.988 MQT: stat/BatteryMonitor/STATUS2 = {"StatusFWR":{"Version":"13.1.0(tasmota32)","BuildDateTime":"2023-08-15T08:00:54","Core":"2_0_11","SDK":"4.4.5.230722","CpuFrequency":160,"Hardware":"ESP32-D0WD-V3 rev.3","CR":"426/699"}}
12:42:01.995 MQT: stat/BatteryMonitor/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["IoT",""],"TelePeriod":10,"Resolution":"558180C0","SetOption":["00408009","2805C80001000600003C5A0A192800000000","00000080","00006000","00004000","00000000"]}}
12:42:02.681 MQT: stat/BatteryMonitor/STATUS4 = {"StatusMEM":{"ProgramSize":0,"Free":2880,"Heap":109,"StackLowMark":4,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"1640D8","FlashFrequency":40,"FlashMode":"DIO","Features":["00000809","9F9AD7DF","0015A001","B7F7BFCF","05DA9BC4","E0360DC7","480840D2","20200000","D4BC482D","010A80B1"],"Drivers":"1,2,3,4,5,7,8,9,10,11,12,14,16,17,20,21,24,26,27,29,34,35,38,50,52,59,60,62,63,66,67,68,82,86,87,88","Sensors":"1,2,3,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,31,34,37,39,40,42,43,45,51,52,55,56,58,59,64,66,67,74,85,92,95,98,103,105,109,127","I2CDriver":"7,8,9,10,11,12,13,14,15,17,18,20,24,29,31,36,41,42,44,46,48,58,62,65,69,76,77,82"}}
12:42:02.693 MQT: stat/BatteryMonitor/STATUS5 = {"StatusNET":{"Hostname":"BatteryMonitor-7036","IPAddress":"192.168.1.151","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.1.1","DNSServer2":"0.0.0.0","Mac":"40:22:D8:E6:1B:7C","IP6Global":"","IP6Local":"fe80::4222:d8ff:fee6:1b7c","Ethernet":{"Hostname":"","IPAddress":"0.0.0.0","Gateway":"0.0.0.0","Subnetmask":"0.0.0.0","DNSServer1":"192.168.1.1","DNSServer2":"0.0.0.0","Mac":"00:00:00:00:00:00","IP6Global":"","IP6Local":""},"Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":16.0}}
12:42:02.700 MQT: stat/BatteryMonitor/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.26","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_E61B7C","MqttUser":"mqtt-sbellins","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
12:42:02.706 MQT: stat/BatteryMonitor/STATUS7 = {"StatusTIM":{"UTC":"2023-09-03T11:42:02","Local":"2023-09-03T12:42:02","StartDST":"2023-03-26T02:00:00","EndDST":"2023-10-29T03:00:00","Timezone":"+01:00","Sunrise":"06:09","Sunset":"19:29"}}
12:42:02.715 MQT: stat/BatteryMonitor/STATUS10 = {"StatusSNS":{"Time":"2023-09-03T12:42:02"}}
12:42:02.723 MQT: stat/BatteryMonitor/STATUS11 = {"StatusSTS":{"Time":"2023-09-03T12:42:02","Uptime":"0T00:08:33","UptimeSec":513,"Heap":103,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":20,"MqttCount":1,"Berry":{"HeapUsed":4,"Objects":42},"Wifi":{"AP":1,"SSId":"IoT","BSSId":"1C:ED:6F:B3:69:75","Channel":11,"Mode":"11n","RSSI":46,"Signal":-77,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
the console shouled say something like this and send it to home assistant
```lua
12:38:48.432 MQT: tele/BatteryMonitor/RESULT = {"SerialReceived":":R51=1,\n:r51=1,27,3000,2000,0,0,0,100,0,0,3040,100,100,100,0,0,1,\r\n"}
12:38:48.924 MQT: tele/BatteryMonitor/RESULT = {"SerialReceived":":R50=1,\n:r50=1,211,2685,460,271237,9,31833,3586,130,0,0,1,167,2222,\r\n:R51=1,J:r51=1,27,3000,2000,0,0,0,100,0,0,3040,100,100,100,0,0,1,\r\n"}

```

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

barbudor, 2023-09-03:
What did you changed?
Obviously Tasmota is receiving data but tries to understand this as commands from the serial port
Check your template, you may have lost the GPIO for the serial bridge

sfromis, 2023-09-03:
The data you post indicate that the device was "factory reset" 4 hours ago, without any gpio config (like for serial bridge) done since then. Clearly not the state in which it used to work for you, you likely need to config the same way as you did earlier.

Sbellins, 2023-09-03:
I also tried to restore the backup af the configuration i had before, it does not work. No gpio was configured before, i follower the same guide i followed last time to configure tasmota

barbudor, 2023-09-03:
> i follower the same guide i followed last time to configure tasmota

Which is where ?

Actually you are right, you can use the serial bridge without defining GPIO
I just made some tests and as soon as you issue `SerialSend 1`, the serial port enters the bridge mode. This even survice a restart.
However it has to be either
`SerialSend 1`
or
`SerialSend1 1`
which are esquivalent as in the 1st case the index 1 is implicit.

But `SerialSend1` would give a command error and does not enter the serial bridge mode.

=> No problem on my side. You probably made a mistake with the `SerialSend`syntax and did not noticed the command error.

Beside, on ESP32 it is a pity not to take advantage of the additional UARTs using the so called Software-Serial-Bridge which is actually hardware on ESP32 (software on ESP8266).
You can pickup any other GPIOS and set them as `SerBr TX` and `SerBr RX` and use them with the `S` prefixed commands such as `SBaudrate` and `SSerialSend1`.
The advantages is that
- Your Tasmota serial console remains available if you need to hook a PC for debug
- The text that the ESP is sending at boot time as well as the first 3 lines of tasmota logs won't be sent to your device

Sbellins, 2023-09-05:
> > i follower the same guide i followed last time to configure tasmota
> 
> Which is where ?
> 
> Actually you are right, you can use the serial bridge without defining GPIO I just made some tests and as soon as you issue `SerialSend 1`, the serial port enters the bridge mode. This even survice a restart. However it has to be either `SerialSend 1` or `SerialSend1 1` which are equivalent as in the 1st case the index 1 is implicit.
> 
> But `SerialSend1` alone would give a command error and does not enter the serial bridge mode.
> 
> => No problem on my side. You probably made a mistake with the `SerialSend`syntax and did not noticed the command error.
> 
> Beside, on ESP32 it is a pity not to take advantage of the additional UARTs using the so called Software-Serial-Bridge which is actually hardware on ESP32 (software on ESP8266). You can pickup any other GPIOS and set them as `SerBr TX` and `SerBr RX` and use them with the `S` prefixed commands such as `SBaudrate` and `SSerialSend1`. The advantages is that
> 
> * Your Tasmota serial console remains available if you need to hook a PC for debug
> * The text that the ESP is sending at boot time as well as the first 3 lines of tasmota logs won't be sent to your device

Tank you, SSerial send and the configuration you suggested is now working.
If you can give me a hand i'm experiencing another issue. Tasmota is receiveing R50 and R51 strings in 2 separate messages and this cause errors in home assistant. 
```lua
10:08:40.337 MQT: tele/BatteryMonitor/RESULT = {"SSerialReceived":":R51=1,\n:r51=1,27,3000,2000,0,0,0,100,0,0,3040,100,100,100,0,0,1,\r\n"}
10:08:40.716 MQT: tele/BatteryMonitor/RESULT = {"SSerialReceived":":R50=1,\n:r50=1,243,2532,1450,77914,272348,302933,166440,129,0,99,0,334,2518,\r\n"}
```
Do you know to to send only the R50 string?
i've tried with this rule i've found online but it does not work
```lua
Rule1 on System#Boot do RuleTimer1 10 endon on Rules#Timer=1 do backlog SSerialSend :R50=1,2,1,; RuleTimer1 10 endon

Rule1 1
```
Before this series of error, tasmota was sending the r51 string and i worked, now it's not working as intended (i see the sensors in home assistant switch from the value to 0 and then back to the value)

barbudor, 2023-09-05:
No need to quote the full message that is just above.

There is no way of filtering the SSerialReceived message. Every string received will be sent to MQTT.
Can't you instruct your device to send only 1 type?
Or can't you filter that in your HA integration (no need to ask, I have no idea how HA is working)

Alternatively, as you have an ESP32, you could use a Berry rule to capture the data received and send only the messages matching your criteria on another topic. HA would have to listen to that other topic