arendst/Tasmota #19482: When updating Sonoff Basic from Tasmota 7.2 to 8.5.1 physical button stops working

kmgalanakis, 2023-09-09:
### PROBLEM DESCRIPTION
Yesterday I updated my Sonoff Basic from 7.2 to 8.5.1 (and then to 9.1 and eventually 13.1 to get auto-discovery on HA) and I realized that even though everything was working properly, the physical button was no longer working. 

In order to get the button back to a working state I have to downgrade again to 7.2. Any version after 7.2 disables the button.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Sonoff Basic
- [x] Tasmota binary firmware version number used: 8.5.1, 9.1, 13.1
  - [x] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2023-09-09:
No support or bug fixing for such very old versions. Update to llatest release Version.
If the issue is still there, you can reopen.

closed 2023-09-09 by Jason2866

kmgalanakis, 2023-09-09:
@Jason2866 as I mentioned in the description, the issue is still there even for 13.1.0. I only mentioned 8.5.1 because it was the first version that broke the physical button functionality.

I'm reopening the issue.

kmgalanakis, 2023-09-09:
@Jason2866 apparently I cannot reopen the issue, could you please do it for me?

Jason2866, 2023-09-09:
Moving to discussions, since it is not an issue with current version