arendst/Tasmota #19489: Tasmota crashes with a display.ini for GC9503V displays

dgwynne, 2023-09-10:
### PROBLEM DESCRIPTION

The udisplay driver crashes when I'm trying to get a display.ini working for GC9503V displays. I think the config is a lot bigger than what the parser is expecting, which leads to a bunch of buffer overflows and smashes.

I've got a ZX3D95CE01S-TR-4848 device that contains a GC9503V display in it. I've previously got it working with OpenHASP, but would like to get Tasmota working on it so I can play around with actual LVGL.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): ZX3D95CE01S-TR-4848
- [x] Tasmota binary firmware version number used: tasmota32s3-lvgl
  - [ ] Pre-compiled
  - [x] Self-compiled
- [x] Flashing tools used: webinstall with tasmota32s3, then upgrade to my own fw, and serial flash of my own firmware.
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
14:09:56.381 MQT: stat/display_E74CDC/RESULT = {"NAME":"ESP32S3","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,0,0,0,0,0,768,6210,1,1,1,1,1,736,1,1,704],"FLAG":0,"BASE":1}
14:09:56.612 MQT: stat/display_E74CDC/RESULT = {"Module":{"0":"ESP32S3"}}
14:09:56.857 MQT: stat/display_E74CDC/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"0":"None"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"0":"None"},"GPIO33":{"0":"None"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"768":"SPI CS1"},"GPIO39":{"6210":"Option A3"},"GPIO40":{"0":"None"},"GPIO41":{"0":"None"},"GPIO42":{"0":"None"},"GPIO43":{"0":"None"},"GPIO44":{"0":"None"},"GPIO45":{"736":"SPI CLK1"},"GPIO46":{"0":"None"},"GPIO47":{"0":"None"},"GPIO48":{"704":"SPI MOSI1"}}
14:09:57.899 MQT: tele/display_E74CDC/STATE = {"Time":"2023-09-10T14:09:57","Uptime":"0T03:10:09","UptimeSec":11409,"Heap":211,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"things79","BSSId":"1A:FD:74:4A:2F:88","Channel":1,"Mode":"11n","RSSI":100,"Signal":-50,"LinkCount":1,"Downtime":"0T00:00:04"}}
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```
14:10:58.507 MQT: stat/display_E74CDC/STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"display_E74CDC","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
14:10:58.553 MQT: stat/display_E74CDC/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32s3.bin","RestartReason":"Software reset CPU","Uptime":"0T03:11:10","StartupUTC":"2023-09-10T00:59:48","Sleep":50,"CfgHolder":4617,"BootCount":18,"BCResetTime":"2023-09-09T11:15:57","SaveCount":36}}
14:10:58.588 MQT: stat/display_E74CDC/STATUS2 = {"StatusFWR":{"Version":"13.1.0(lvgl-haspmota)","BuildDateTime":"2023-09-09T12:48:34","Core":"2_0_11","SDK":"4.4.5.230722","CpuFrequency":240,"Hardware":"ESP32-S3","CR":"413/699"}}
14:10:58.602 MQT: stat/display_E74CDC/STATUS3 = {"StatusLOG":{"SerialLog":4,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["things79",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A192800000000","00000080","00006000","00004000","00000000"]}}
14:10:58.645 MQT: stat/display_E74CDC/STATUS4 = {"StatusMEM":{"ProgramSize":2356,"Free":523,"Heap":206,"StackLowMark":19,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":16384,"FlashSize":16384,"FlashChipId":"18405E","FlashFrequency":80,"FlashMode":"QIO","Features":["00000809","9FAAD7DF","0005A021","B7F7BFCF","05DA9BC4","60360DC7","080840D2","20200000","D4BC483D","010A80B1"],"Drivers":"1,2,3,4,5,8,9,10,11,12,13,14,16,17,20,24,26,27,29,34,38,50,52,54,55,56,59,60,62,63,66,67,68,86,87,88","Sensors":"1,2,3,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,31,34,37,39,40,42,43,45,51,52,55,56,58,59,64,66,67,74,85,92,95,98,103,105,109,127","I2CDriver":"7,8,9,10,11,12,13,14,15,17,18,20,24,29,31,36,41,42,44,46,48,58,59,62,65,69,76,77,82"}}
14:10:58.713 MQT: stat/display_E74CDC/STATUS5 = {"StatusNET":{"Hostname":"display-E74CDC-3292","IPAddress":"10.0.128.174","Gateway":"10.0.128.1","Subnetmask":"255.255.255.0","DNSServer1":"10.0.127.1","DNSServer2":"0.0.0.0","Mac":"48:27:E2:E7:4C:DC","IP6Global":"","IP6Local":"fe80::4a27:e2ff:fee7:4cdc","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":19.0}}
14:10:58.748 MQT: stat/display_E74CDC/STATUS6 = {"StatusMQT":{"MqttHost":"mqtt.lan.animata.net","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_E74CDC","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
14:10:58.773 MQT: stat/display_E74CDC/STATUS7 = {"StatusTIM":{"UTC":"2023-09-10T04:10:58","Local":"2023-09-10T14:10:58","StartDST":"2023-03-26T02:00:00","EndDST":"2023-10-29T03:00:00","Timezone":"+10:00","Sunrise":"15:19","Sunset":"04:14"}}
14:10:58.789 MQT: stat/display_E74CDC/STATUS10 = {"StatusSNS":{"Time":"2023-09-10T14:10:58"}}
14:10:58.805 MQT: stat/display_E74CDC/STATUS11 = {"StatusSTS":{"Time":"2023-09-10T14:10:58","Uptime":"0T03:11:10","UptimeSec":11470,"Heap":206,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"things79","BSSId":"1A:FD:74:4A:2F:88","Channel":1,"Mode":"11n","RSSI":98,"Signal":-51,"LinkCount":1,"Downtime":"0T00:00:04"}}}
14:10:58.843 MQT: stat/display_E74CDC/STATUS12 = {"StatusSTK":{"Exception":28,"Reason":"LoadProhibited","EPC":"4209a7ad","EXCVADDR":"0000000c","CallChain":["4209a7aa","4209ae4f","4207da19","420840a9","420b8a94","420840c7","420c99d2","420dc3de","4202a1d2"]}}
```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
With the OptionA3 pin set up and the display.ini below, and `SerialLog` set to 4, the following was caught from the Serial console:

?ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0xa (SPI_FAST_FLASH_BOOT)
Saved PC:0x4215e74e
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x128
load:0x403c9700,len:0xb90
load:0x403cc700,len:0x28cc
SHA-256 comparison failed:
Calculated: b7185e43e2ed3b52bbf7578551293b4d4392a1c655a1eae819286c9eb2a7ca6d
Expected: 9b01bd686907a92c7287b9b74d6f0a736ec17a1ca0d18863e9d9e5ffcb3c7d78
Attempting to boot anyway...
entry 0x403c98b8

00:00:00.002 HDW: ESP32-S3 
00:00:00.043 UFS: FlashFS mounted with 12588 kB free
00:00:00.061 CFG: Loaded from File, Count 36
00:00:00.063 FRC: Some settings have been reset (3)
00:00:00.064 CFG: CR 413/699, Busy 0
00:00:00.069 SPI: Hardware using GPIO45(CLK) and GPIO48(MOSI)
[... 74 lines cut ...]
### EXPECTED BEHAVIOUR

The display should work, but I'd be happy for the udisplay library to not crash as a first step.

### SCREENSHOTS

N/A

### ADDITIONAL CONTEXT

The following links about the hardware are useful for understanding the hardware:

- https://vi.aliexpress.com/item/1005005188547984.html
- https://openhasp.haswitchplate.com/0.7.0/hardware/wireless-tag/wt-86-32-3zw1/
- https://github.com/HASwitchPlate/openHASP/discussions/435-
- https://github.com/wireless-tag-com were useful to me

s-hadinger helped me convert the following SPI init code into a display.ini:

- https://github.com/bekencorp/armino/blob/6fd9f73a06b098e5a562b95875aac1b96b879c70/components/demos/media/lcd/common/lcd_spi_config.c#L141
- https://github.com/SuGlider/Arduino_GFX/blob/c309dfe08493a5f97a465b0ebca766f7ecaebcff/src/display/Arduino_RGB_Display.h#L11

I'm worried that the SPI pins being shared with the display pins might be a problem too.

Lastly, I've got a mofo sized headache otherwise I'd be trying to push through this myself. Sorry.

dgwynne, 2023-09-11:
It should be possible to reproduce the parser bugs without the hardware just by trying the display.ini on any esp32s3.

gemu2015, 2023-09-12:
very unusual device descriptor!

related buffers are:
in uDisplay.cpp
uDisplay::uDisplay(char *lp) : Renderer(800, 600) {

 char linebuff[128];

change this to 256 to fit the long lines

then eventually enlarge also the cmds buffer

in uDisplay.h
class uDisplay : public Renderer {
   
uint8_t dsp_cmds[256];

probably to 512

gemu2015, 2023-09-12:
however at a closer look this setup can not work at all!
RGB displays do not need any SPI setup at all
section I:  sends SPI cmds which will not work since no SPI has been defined.
this requires more research!

dgwynne, 2023-09-12:
> however at a closer look this setup can not work at all! RGB displays do not need any SPI setup at all section I: sends SPI cmds which will not work since no SPI has been defined. this requires more research!

https://github.com/wireless-tag-com/ZX3D95CE01S-TR-4848/blob/master/main/board.h sure makes it look like there's some SPI in use on the board, and worse, some of the pin numbers overlap with RGB signal pins.

dgwynne, 2023-09-12:
With this diff:

```diff
diff --git a/lib/lib_display/UDisplay/uDisplay.cpp b/lib/lib_display/UDisplay/uDisplay.cpp
index 66ab35bac..5583800f8 100755
--- a/lib/lib_display/UDisplay/uDisplay.cpp
+++ b/lib/lib_display/UDisplay/uDisplay.cpp
@@ -133,11 +133,16 @@ uDisplay::uDisplay(char *lp) : Renderer(800, 600) {
   }
   lut_partial = 0;
   lut_full = 0;
-  char linebuff[128];
+  char linebuff[256];
   while (*lp) {
 
     uint16_t llen = strlen_ln(lp);
-    strncpy(linebuff, lp, llen);
+    if (llen >= sizeof(linebuff)) {
+      /* XXX what do you do here? throw an exception? return NULL? */
+      Serial.printf("display configuration line is too long");
+      return;
+    }
+    memcpy(linebuff, lp, llen);
     linebuff[llen] = 0;
     lp += llen;
     char *lp1 = linebuff;
@@ -948,16 +953,21 @@ Renderer *uDisplay::Init(void) {
     _panel_config->flags.relax_on_idle = 0;
     _panel_config->flags.fb_in_psram = 1;             // allocate frame buffer in PSRAM
 
+  Serial.printf("l %u\n",__LINE__);
     ESP_ERROR_CHECK(esp_lcd_new_rgb_panel(_panel_config, &_panel_handle));
     ESP_ERROR_CHECK(esp_lcd_panel_reset(_panel_handle));
     ESP_ERROR_CHECK(esp_lcd_panel_init(_panel_handle));
+  Serial.printf("l %u\n",__LINE__);
 
     uint16_t color = random(0xffff);
+  Serial.printf("l %u\n",__LINE__);
     ESP_ERROR_CHECK(_panel_handle->draw_bitmap(_panel_handle, 0, 0, 1, 1, &color));
 
     _rgb_panel = __containerof(_panel_handle, esp_rgb_panel_t, base);
+  Serial.printf("l %u\n",__LINE__);
 
     rgb_fb = (uint16_t *)_rgb_panel->fb;
+  Serial.printf("l %u\n",__LINE__);
 
 #endif // USE_ESP32_S3
   }
@@ -1027,6 +1037,7 @@ Renderer *uDisplay::Init(void) {
     // to disable SPI TRANSACTION
     spi_nr = 3;
     spi_cs = par_cs;
+  Serial.printf("l %u\n",__LINE__);
 
     _i80_bus = nullptr;
 
@@ -1052,6 +1063,7 @@ Renderer *uDisplay::Init(void) {
 
     _dev = &LCD_CAM;
 
+  Serial.printf("l %u\n",__LINE__);
     pb_beginTransaction();
     uint16_t index = 0;
     while (1) {
@@ -1095,6 +1107,7 @@ Renderer *uDisplay::Init(void) {
     }
 
     pb_endTransaction();
+  Serial.printf("l %u\n",__LINE__);
 
[... 261 lines cut ...]
19:42:10.709 MQT: Subscribe to cmnd/display_E74CDC/#
19:42:10.710 MQT: Subscribe to cmnd/tasmotas/#
19:42:10.722 MQT: Subscribe to cmnd/DVES_E74CDC_fb/#
19:42:10.726 MQT: tele/display_E74CDC/INFO1 = {"Info1":{"Module":"ESP32S3","Version":"13.1.0(lvgl-haspmota)","FallbackTopic":"cmnd/DVES_E74CDC_fb/","GroupTopic":"cmnd/tasmotas/"}}
19:42:10.740 MQT: tele/display_E74CDC/INFO2 = {"Info2":{"WebServerMode":"Admin","Hostname":"display-E74CDC-3292","IPAddress":"10.0.128.174","IP6Global":"","IP6Local":"fe80::4a27:e2ff:fee7:4cdc"}}
19:42:10.755 MQT: tele/display_E74CDC/INFO3 = {"Info3":{"RestartReason":{"Exception":28,"Reason":"LoadProhibited","EPC":"4209a47f","EXCVADDR":"0000000c","CallChain":["4209a47c","4209aafb","4207d84d","42083edd","420b873c","42083efb","420c967a","420dc086","4202a00a"]},"BootCount":27}}
19:42:10.778 MQT: stat/display_E74CDC/RESULT = {"POWER":"OFF"}
19:42:10.790 MQT: stat/display_E74CDC/POWER = OFF
19:42:12.211 SRC: MQTT
19:42:12.212 CMD: Grp 0, Cmd 'STATUS', Idx 1, Len 1, Pld 1, Data '1'
19:42:12.217 MQT: stat/display_E74CDC/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32s3.bin","RestartReason":"Software reset CPU","Uptime":"0T00:00:06","StartupUTC":"2023-09-12T09:42:06","Sleep":50,"CfgHolder":4617,"BootCount":26,"BCResetTime":"2023-09-09T11:15:57","SaveCount":44}}
19:42:12.262 SRC: MQTT
19:42:12.263 CMD: Grp 0, Cmd 'STATUS', Idx 1, Len 2, Pld 11, Data '11'
19:42:12.269 MQT: stat/display_E74CDC/STATUS11 = {"StatusSTS":{"Time":"2023-09-12T19:42:12","Uptime":"0T00:00:06","UptimeSec":6,"Heap":217,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":23,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"things79","BSSId":"1A:FD:74:4A:2F:88","Channel":1,"Mode":"11n","RSSI":100,"Signal":-44,"LinkCount":1,"Downtime":"0T00:00:03"}}}
19:42:12.312 SRC: MQTT
19:42:12.313 CMD: Grp 0, Cmd 'STATE', Idx 1, Len 0, Pld -99, Data ''
19:42:12.317 MQT: stat/display_E74CDC/RESULT = {"Time":"2023-09-12T19:42:12","Uptime":"0T00:00:06","UptimeSec":6,"Heap":214,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":26,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"things79","BSSId":"1A:FD:74:4A:2F:88","Channel":1,"Mode":"11n","RSSI":100,"Signal":-44,"LinkCount":1,"Downtime":"0T00:00:03"}}
19:42:13.925 HTP: Main Menu
19:42:14.042 QPC: Reset
19:42:14.916 MQT: tele/display_E74CDC/STATE = {"Time":"2023-09-12T19:42:14","Uptime":"0T00:00:08","UptimeSec":8,"Heap":215,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"things79","BSSId":"1A:FD:74:4A:2F:88","Channel":1,"Mode":"11n","RSSI":100,"Signal":-43,"LinkCount":1,"Downtime":"0T00:00:03"}}
19:42:15.922 APP: Boot Count 27
19:42:16.312 CFG: Saved, Count 45, Bytes 4096
19:42:19.918 MQT: tasmota/discovery/4827E2E74CDC/config = {"ip":"10.0.128.174","dn":"Tasmota","fn":["Tasmota",null,null,null,null,null,null,null],"hn":"display-E74CDC-3292","mac":"4827E2E74CDC","md":"ESP32S3","ty":0,"if":0,"ofln":"Offline","onln":"Online","state":["OFF","ON","TOGGLE","HOLD"],"sw":"13.1.0","t":"display_E74CDC","ft":"%prefix%/%topic%/","tp":["cmnd","stat","tele"],"rl":[1,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"swc":[-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1],"swn":[null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null],"btn":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"so":{"4":0,"11":0,"13":0,"17":0,"20":0,"30":0,"68":0,"73":0,"82":0,"114":0,"117":0},"lk":0,"lt_st":0,"bat":0,"dslp":0,"sho":[],"sht":[],"ver":1} (retained)
19:42:19.995 MQT: tasmota/discovery/4827E2E74CDC/sensors = {"sn":{"Time":"2023-09-12T19:42:19"},"ver":1} (retained)
```

dgwynne, 2023-09-12:
The serial print thing from line 963 is around here:

```c
    uint16_t color = random(0xffff);
  Serial.printf("l %u\n",__LINE__);
    ESP_ERROR_CHECK(_panel_handle->draw_bitmap(_panel_handle, 0, 0, 1, 1, &color));
```

As an aside, I'd recommend using `strlcpy()` instead of `strncpy()`, and you should always pass the size of the destination buffer as the length argument to guard against overflow. This also means that you should only use a nul terminated buffer as the source. If you are copying a chunk of a source string, ie, characters from a buffer that is not nul terminated, into the destination buffer then the length check, `memcpy()`, and nul termination I used above is a good option.

smcgann99, 2023-09-14:
Looks like an awesome bit of kit for the price, I couldn't build one for less. I will have a play with openhasp, but would really like to make some mods to it and so Tasmota would be the best option. fingers crossed you guys can figure it out ;-)

dgwynne, 2023-09-25:
> @dgwynne Does the board have PSRAM? Without no joy with Tasmota and LVGL. If there is PSRAM please post screen snapshot of Tasmota Information Page. From the specs i have found it seems a ESP32-S3N16R8 is used. This is QIO Flash and OPI PSRAM. The build you probably use is an QIO/QSPI build. So no PSRAM support. Use build `tasmota32s3-qio_opi`

Is `Status 4` kind of the same thing as a screenshot?

```
{
   "StatusMEM" : {
      "Drivers" : "1,2,3,4,5,8,9,10,11,12,13,14,16,17,20,24,26,27,29,34,38,50,52,54,55,56,59,60,62,63,66,67,68,86,87,88",
      "Features" : [
         "00000809",
         "9FAAD7DF",
         "0005A021",
         "B7F7BFCF",
         "05DA9BC4",
         "60360DC7",
         "080840D2",
         "20200000",
         "D4BC483D",
         "010A80B1"
      ],
      "FlashChipId" : "18405E",
      "FlashFrequency" : 80,
      "FlashMode" : "QIO",
      "FlashSize" : 16384,
      "Free" : 524,
      "Heap" : 210,
      "I2CDriver" : "7,8,9,10,11,12,13,14,15,17,18,20,24,29,31,36,41,42,44,46,48,58,59,62,65,69,76,77,82",
      "ProgramFlashSize" : 16384,
      "ProgramSize" : 2355,
      "PsrFree" : 0,
      "PsrMax" : 0,
      "Sensors" : "1,2,3,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,31,34,37,39,40,42,43,45,51,52,55,56,58,59,64,66,67,74,85,92,95,98,103,105,109,127",
      "StackLowMark" : 19
   }
}
```

I'll try tweaking my build when I find some time.

dgwynne, 2023-09-26:
I'm using this now:

```
[env:tasmota32s3-lvgl]^M
extends                 = env:tasmota32s3^M
board                   = esp32s3-qio_opi^M
board_build.f_cpu       = 240000000L^M
board_build.f_flash     = 80000000L^M
build_flags             = ${env:tasmota32s3.build_flags}^M
                          -DUSE_BERRY_ULP^M
                          -DFIRMWARE_LVGL^M
                          -DUSE_LVGL_OPENHASP^M
                          -DUDSP_DEBUG^M
lib_extra_dirs          = lib/libesp32, lib/libesp32_lvgl, lib/lib_basic, lib/lib_i2c, lib/lib_rf, lib/lib_div, lib/lib_ssl, lib/lib_display^M
```

Which results PSRAM showing up in `Status 4`:

```
{
   "StatusMEM" : {
      "Drivers" : "1,2,3,4,5,8,9,10,11,12,13,14,16,17,20,24,26,27,29,34,38,50,52,54,55,56,59,60,62,63,66,67,68,86,87,88",
      "Features" : [
         "00000809",
         "9FAAD7DF",
         "0005A021",
         "B7F7BFCF",
         "05DA9BC4",
         "60360DC7",
         "080840D2",
         "20200000",
         "D4BC483D",
         "010A80B1"
      ],
      "FlashChipId" : "18405E",
      "FlashFrequency" : 80,
      "FlashMode" : "QIO",
      "FlashSize" : 16384,
      "Free" : 521,
      "Heap" : 216,
      "I2CDriver" : "7,8,9,10,11,12,13,14,15,17,18,20,24,29,31,36,41,42,44,46,48,58,59,62,65,69,76,77,82",
      "ProgramFlashSize" : 16384,
      "ProgramSize" : 2358,
      "PsrFree" : 7673,
      "PsrMax" : 8187,
      "Sensors" : "1,2,3,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,31,34,37,39,40,42,43,45,51,52,55,56,58,59,64,66,67,74,85,92,95,98,103,105,109,127",
      "StackLowMark" : 20
   }
}
```

`Display` now produces `{"Display":{"Model":17,"Type":0,"Width":480,"Height":480,"Mode":0,"Dimmer":50,"Size":1,"Font":1,"Rotate":0,"Invert":0,"Refresh":2,"Cols":[16,8],"Rows":2}}`, which is good.

LVGL in the Berry console appears to respond, but nothing appears on the screen. Maybe the SPI init is important?

Jason2866, 2023-09-26:
If the display.ini is correct and activated with assigning a GPIO the Option A3 the Tasmota logo will be shown when booting.
Not saying that there is verified no bug which such long init sequences. But a least it will not crash anymore because the needed PSRAM was not active.

dgwynne, 2023-09-26:
With the increased buffer sizes and the right psram it no longer crashes, but it's still not working. Progress at least

gemu2015, 2023-09-26:
there is absolutely no chance to get this display to work with the current driver.
this display needs an SPI initialization before finally working in RGB mode
the current driver supports only one interface for a given display.

i would suggest a different approach.
use the unmodified udisplay driver in RGB mode without SPI initialization in section I: (use no section :I at all)
define the SPI interface pins in Tasmota GPIO UserIO  and send the needed SPI initialisation sequence with berry

gemu2015, 2023-09-27:
i created a version that should probably work with your display.ini above.

it is a quick hack, it sends each section I: line by line to Tasmota defined SPI before initializing RGB mode.

this display uses 3 wire SPI, so do not define SPI_DC

at least on my ESP32S3 it shows the appropriate debug infos without crashing

[uDisplay.cpp.zip](https://github.com/arendst/Tasmota/files/12735367/uDisplay.cpp.zip)

dgwynne, 2023-09-27:
with that uDisplay.cpp I see:

00:00:00.001 HDW: ESP32-S3 (PSRAM)
00:00:00.024 UFS: FlashFS mounted with 12588 kB free
00:00:00.037 CFG: Loaded from File, Count 64
00:00:00.043 QPC: Count 1
00:00:00.043 CFG: CR 413/699, Busy 0
00:00:00.045 SPI: Hardware using GPIO45(CLK) and GPIO48(MOSI)
00:00:00.050 ROT: Mode 1
00:00:00.053 BRY: GC from 3832 to 3632 bytes, objects freed 4/21 (in 0 ms) - slots from 37/61 to 29/61
00:00:00.093 CFG: No '*.autoconf' file found
00:00:00.096 BRY: GC from 7082 to 5676 bytes, objects freed 6/49 (in 0 ms) - slots from 72/91 to 49/91
00:00:00.097 BRY: Berry initialized, RAM used=5676 bytes
00:00:00.112 BRY: No 'preinit.be'
00:00:00.115 TFS: File '.drvset003' not found
00:00:00.115 CFG: Energy use defaults as file system not ready or file not found
00:00:00.126 DSP: init?
00:00:00.126 DSP: XDSP_17
00:00:00.145 DSP: File descriptor used
00:00:00.146 DSP: rule?
00:00:00.146 DSP: udisp?
start send cmd table
cmd, args 00, 0 
end send cmd table
start send cmd table
cmd, args f0, 5 55 aa 52 08 00 
end send cmd table
start send cmd table
cmd, args f6, 2 5a 87 
end send cmd table
start send cmd table
cmd, args c1, 1 3f 
end send cmd table
start send cmd table
cmd, args c2, 1 0e 
end send cmd table
start send cmd table
cmd, args c6, 1 f8 
end send cmd table
start send cmd table
cmd, args c9, 1 10 
end send cmd table
start send cmd table
cmd, args cd, 1 25 
end send cmd table
start send cmd table
cmd, args f8, 1 8a 
end send cmd table
start send cmd table
cmd, args ac, 1 45 
end send cmd table
start send cmd table
cmd, args a0, 1 dd 
end send cmd table
start send cmd table
cmd, args a7, 1 47 
end send cmd table
start send cmd table
cmd, args fa, 4 00 00 00 04 
end send cmd table
start send cmd table
cmd, args 86, 4 99 a3 a3 51 
end send cmd table
start send cmd table
cmd, args a3, 1 ee 
end send cmd table
start send cmd table
cmd, args fd, 3 3c 3c 00 
end send cmd table
start send cmd table
[... 217 lines cut ...]
18:39:49.694 MQT: Subscribe to cmnd/tasmotas/#
18:39:49.695 MQT: Subscribe to cmnd/DVES_E74CDC_fb/#
18:39:49.697 MQT: tele/display_E74CDC/INFO1 = {"Info1":{"Module":"ESP32S3","Version":"13.1.0(lvgl-haspmota)","FallbackTopic":"cmnd/DVES_E74CDC_fb/","GroupTopic":"cmnd/tasmotas/"}}
18:39:49.720 MQT: tele/display_E74CDC/INFO2 = {"Info2":{"WebServerMode":"Admin","Hostname":"display-E74CDC-3292","IPAddress":"10.0.128.174","IP6Global":"","IP6Local":"fe80::4a27:e2ff:fee7:4cdc"}}
18:39:49.732 MQT: tele/display_E74CDC/INFO3 = {"Info3":{"RestartReason":"Software reset CPU","BootCount":32}}
18:39:49.745 MQT: stat/display_E74CDC/RESULT = {"POWER":"ON"}
18:39:49.746 MQT: stat/display_E74CDC/POWER = ON
18:39:49.749 TFS: File 'autoexec.bat' not found
18:39:50.789 LVG: Refreshed 19200 pixels in 36 ms (533 pix/ms)
18:39:50.826 LVG: Refreshed 19200 pixels in 36 ms (533 pix/ms)
18:39:50.863 LVG: Refreshed 19200 pixels in 35 ms (548 pix/ms)
18:39:50.900 LVG: Refreshed 19200 pixels in 35 ms (548 pix/ms)
18:39:50.938 LVG: Refreshed 19200 pixels in 35 ms (548 pix/ms)
18:39:50.975 LVG: Refreshed 19200 pixels in 36 ms (533 pix/ms)
18:39:51.013 LVG: Refreshed 19200 pixels in 36 ms (533 pix/ms)
18:39:51.049 LVG: Refreshed 19200 pixels in 35 ms (548 pix/ms)
18:39:51.087 LVG: Refreshed 19200 pixels in 36 ms (533 pix/ms)
18:39:51.124 LVG: Refreshed 19200 pixels in 35 ms (548 pix/ms)
18:39:51.162 LVG: Refreshed 19200 pixels in 35 ms (548 pix/ms)
18:39:51.200 LVG: Refreshed 19200 pixels in 37 ms (518 pix/ms)
18:39:51.202 SRC: MQTT

but i dont see any splash drawn on the display :(

is udisplay allowed to take over the pins?

dgwynne, 2023-09-27:
is there a reset pin somewhere that needs to flip it from spi to display mode maybe?

gemu2015, 2023-09-27:
do you have a list of used pins of this board ?
if it has a display reset pin it must be implemented which i will update in short

dgwynne, 2023-09-27:
> do you have a list of used pins of this board ? if it has a display reset pin it must be implemented which i will update in short

https://github.com/wireless-tag-com/ZX3D95CE01S-TR-4848/blob/master/main/board.h is the pins i think.

dgwynne, 2023-09-27:
> > do you have a list of used pins of this board ? if it has a display reset pin it must be implemented which i will update in short
> 
> https://github.com/wireless-tag-com/ZX3D95CE01S-TR-4848/blob/master/main/board.h is the pins i think.

I can config 41 as "Display Rst" in the Tasmota template for uDisplay to use like the SPI pins.

gemu2015, 2023-09-27:
ok, now with support of reset and back panel 

[uDisplay.cpp.zip](https://github.com/arendst/Tasmota/files/12736863/uDisplay.cpp.zip)

gemu2015, 2023-09-27:
remark:
do not forget to set displaydimmer 100, sometime display is extremely dark in the beginning

dgwynne, 2023-09-27:
Still no splash screen, and using lvgl from berry doesnt seem to produce any output. I can't see anything obviously different between what you're doing and arduinogfx/openhasp either. I haven't looked super closely though.

gemu2015, 2023-09-27:
there is something wrong with spi when inspecting with oscilloscope.
will report when i found the reason

Jason2866, 2023-09-27:
Note. There is no device specific code in Tasmota accepted anymore. Every device special init / config needed has to be provided with a Berry driver or init sequence loaded from a config file placed in Filesystem.

gemu2015, 2023-09-27:
no device specific code!  all derived from display.ini

gemu2015, 2023-09-27:
ok there were issues with ESP32-S3  3 wire SPI,  so we switch to software SPI until solved.

select SSPI_CS, SSPI_SCLK, SSPI_MOSI and SPI init sequence should work

[uDisplay.cpp.zip](https://github.com/arendst/Tasmota/files/12739405/uDisplay.cpp.zip)

dgwynne, 2023-09-27:
OK, using SSPI works much better, the display wakes up now. I can see stuff on the screen now, but it's messed up. Hopefully that's just a matter of tweaking timing params in the ini file now though.

gemu2015, 2023-09-28:
ok there was also a limit in number of data points per line not only line length.
this is fixed now.
[Archiv.zip](https://github.com/arendst/Tasmota/files/12745822/Archiv.zip)

gemu2015, 2023-09-28:
this is a new version with adapted syntax, everything defined via display.ini
use this init line for section I: for your display
:IS,45,48,38,41
=> init software SPI with sclk, mosi, cs, reset

do not define anything for display except OPTION 3 in Tasmota GUI.

[Archiv.zip](https://github.com/arendst/Tasmota/files/12749558/Archiv.zip)

dgwynne, 2023-09-28:
The latest uDisplay files with the :IS line in display.ini work as well as the pin config and the previous uDisplay files, ie, the display is reset, initialised via (software) SPI, and the backlight works, however, whatever is drawn on the screen is still garbled. If I draw stuff via the berry console, I can see the display change, but it's still garbled.

I do like having the board/display config in once place though, so :IS makes sense to me.

I don't have any experience getting displays working though, so I'm feeling a bit stuck and don't know what to do next.

dgwynne, 2023-09-28:
fwiw, this is my current display.ini. The values are based on what was working for me in openHASP, including the SPI init sequence they're using.

```
:H,GC9503V,480,480,16,RGB,40,41,42,39,5,45,48,47,0,21,14,13,12,11,16,17,18,8,3,46,10,15
:V,1,10,8,50,1,10,8,20,1
:S,2,1,1,0,42,20
:IS,45,48,38,41
:I
F0,05,55,AA,52,08,00
F6,02,5A,87
C1,01,3F
C2,01,0E
C6,01,F8
C9,01,10
CD,01,25
F8,01,8A
AC,01,45
A0,01,DD
A7,01,47
FA,04,00,00,00,04
A3,01,EE
FD,03,28,28,00
71,01,48
72,01,48
73,02,00,48
97,01,EE
83,01,93
9A,01,72
9B,01,5a
82,02,2c,2c
B1,01,10
6D,20,00,1F,19,1A,10,0e,0c,0a,02,07,1E,1E,1E,1E,1E,1E,1E,1E,1E,1E,1E,1E,08,01,09,0b,0D,0F,1a,19,1f,00
64,10,38,05,01,db,03,03,38,04,01,dc,03,03,7A,7A,7A,7A
65,10,38,03,01,dd,03,03,38,02,01,de,03,03,7A,7A,7A,7A
66,10,38,01,01,df,03,03,38,00,01,e0,03,03,7A,7A,7A,7A
67,10,30,01,01,e1,03,03,30,02,01,e2,03,03,7A,7A,7A,7A
68,0d,00,08,15,08,15,7A,7A,08,15,08,15,7A,7A
60,08,38,08,7A,7A,38,09,7A,7A
63,08,31,e4,7A,7A,31,E5,7A,7A
68,01,07
7A,02,08,13
7B,02,08,13
D1,34,00,00,00,04,00,12,00,18,00,21,00,2a,00,35,00,47,00,56,00,90,00,e5,01,68,01,d5,01,d7,02,36,02,a6,02,ee,03,48,03,a0,03,ba,03,c5,03,d0,03,E0,03,ea,03,Fa,03,FF
D2,34,00,00,00,04,00,12,00,18,00,21,00,2a,00,35,00,47,00,56,00,90,00,e5,01,68,01,d5,01,d7,02,36,02,a6,02,ee,03,48,03,a0,03,ba,03,c5,03,d0,03,E0,03,ea,03,Fa,03,FF
D3,34,00,00,00,04,00,12,00,18,00,21,00,2a,00,35,00,47,00,56,00,90,00,e5,01,68,01,d5,01,d7,02,36,02,a6,02,ee,03,48,03,a0,03,ba,03,c5,03,d0,03,E0,03,ea,03,Fa,03,FF
D4,34,00,00,00,04,00,12,00,18,00,21,00,2a,00,35,00,47,00,56,00,90,00,e5,01,68,01,d5,01,d7,02,36,02,a6,02,ee,03,48,03,a0,03,ba,03,c5,03,d0,03,E0,03,ea,03,Fa,03,FF
D5,34,00,00,00,04,00,12,00,18,00,21,00,2a,00,35,00,47,00,56,00,90,00,e5,01,68,01,d5,01,d7,02,36,02,a6,02,ee,03,48,03,a0,03,ba,03,c5,03,d0,03,E0,03,ea,03,Fa,03,FF
D6,34,00,00,00,04,00,12,00,18,00,21,00,2a,00,35,00,47,00,56,00,90,00,e5,01,68,01,d5,01,d7,02,36,02,a6,02,ee,03,48,03,a0,03,ba,03,c5,03,d0,03,E0,03,ea,03,Fa,03,FF
3a,01,66
11,80
29,80
#
```

gemu2015, 2023-09-29:
:IS replaces :I, so remove single :I line.
however this should not cause the problem.

carefully check the gpio definitions of :H and :V line 

check the init sequence by comparing with the source code of the working driver
btw, the above sample source code is not exactly for your board, since the gpio pins do not match.

do you have a working code exactly for your board ?

dgwynne, 2023-09-29:
> :IS replaces :I, so remove single :I line. however this should not cause the problem.
>
> carefully check the gpio definitions of :H and :V line
>
> check the init sequence by comparing with the source [...]

They look right though...

> do you have a working code exactly for your board ?

looks like openHASP has committed the code I was running for this board:

https://github.com/HASwitchPlate/openHASP/blob/master/user_setups/esp32s3/zw3d95ce01s_ar.ini#L135-L176

gemu2015, 2023-09-29:
but what about the SPI init sequence? it should also be given. can't find it in hasp code.

dgwynne, 2023-09-29:
openHASP gets this from https://github.com/moononournation/Arduino_GFX/blob/9e12ba7f984348d87c698721aa6bbe7074b35f8a/src/display/Arduino_RGB_Display.h#L11 as a dependency.

gemu2015, 2023-09-29:
so carefully recheck this init sequence with display.ini

dgwynne, 2023-10-02:
I've double checked and rewritten the display.ini and file and had no joy. Still feels like a sync issue, but none of the V: params or clock speed things I've tweaked have made any difference. Could the SPI setup used to configure the display still be fiddling with the bus after the display is supposed to have taken over?

dgwynne, 2023-10-02:
> reading the chips data sheet i realized that it has an inverted Reset line. try this new driver version
> 

That didnt work :(

It seems to want to pin left high after reset. This works:

```c
              if (reset >= 0) {
                pinMode(reset, OUTPUT);
                digitalWrite(reset, LOW);
                delay(5);
                digitalWrite(reset, HIGH);
                delay(5);
              }
```

dgwynne, 2023-10-02:
By "didn't work" I mean that I got no display output with the new code. With the previous code and the chunk in the last comment the display seems to start up, but is still garbled.

dgwynne, 2023-10-06:
OK. I got it working with the following config:

```
:H,GC9503V,480,480,16,RGB,40,41,42,39,5,11,16,17,18,8,3,46,10,45,48,47,0,21,14,13,12,15
:V,1,8,10,50,1,8,10,20,0
:S,2,1,1,0,42,20
:IS,45,48,38,41
F0,05,55,AA,52,08,00
F6,02,5A,87
C1,01,3F
C2,01,0E
C6,01,F8
C9,01,10
CD,01,25
F8,01,8A
AC,01,45
A0,01,DD
A7,01,47
FA,04,00,00,00,04
A3,01,EE
FD,3,28,28,00
71,01,48
72,01,48
73,02,00,44
97,01,EE
83,01,93
9A,01,72
9B,01,5a
82,02,2c,2c
B1,01,10
6D,20,00,1F,19,1A,10,0e,0c,0a,02,07,1E,1E,1E,1E,1E,1E,1E,1E,1E,1E,1E,1E,08,01,09,0b,0D,0F,1a,19,1f,00
64,10,38,05,01,db,03,03,38,04,01,dc,03,03,7A,7A,7A,7A
65,10,38,03,01,dd,03,03,38,02,01,de,03,03,7A,7A,7A,7A
66,10,38,01,01,df,03,03,38,00,01,e0,03,03,7A,7A,7A,7A
67,10,30,01,01,e1,03,03,30,02,01,e2,03,03,7A,7A,7A,7A
68,0d,00,08,15,08,15,7A,7A,08,15,08,15,7A,7A
60,08,38,08,7A,7A,38,09,7A,7A
63,08,31,e4,7A,7A,31,e5,7A,7A
6B,01,07
7A,02,08,13
7B,02,08,13
D1,34,00,00,00,04,00,12,00,18,00,21,00,2a,00,35,00,47,00,56,00,90,00,e5,01,68,01,d5,01,d7,02,36,02,a6,02,ee,03,48,03,a0,03,ba,03,c5,03,d0,03,E0,03,ea,03,Fa,03,FF
D2,34,00,00,00,04,00,12,00,18,00,21,00,2a,00,35,00,47,00,56,00,90,00,e5,01,68,01,d5,01,d7,02,36,02,a6,02,ee,03,48,03,a0,03,ba,03,c5,03,d0,03,E0,03,ea,03,Fa,03,FF
D3,34,00,00,00,04,00,12,00,18,00,21,00,2a,00,35,00,47,00,56,00,90,00,e5,01,68,01,d5,01,d7,02,36,02,a6,02,ee,03,48,03,a0,03,ba,03,c5,03,d0,03,E0,03,ea,03,Fa,03,FF
D4,34,00,00,00,04,00,12,00,18,00,21,00,2a,00,35,00,47,00,56,00,90,00,e5,01,68,01,d5,01,d7,02,36,02,a6,02,ee,03,48,03,a0,03,ba,03,c5,03,d0,03,E0,03,ea,03,Fa,03,FF
D5,34,00,00,00,04,00,12,00,18,00,21,00,2a,00,35,00,47,00,56,00,90,00,e5,01,68,01,d5,01,d7,02,36,02,a6,02,ee,03,48,03,a0,03,ba,03,c5,03,d0,03,E0,03,ea,03,Fa,03,FF
D6,34,00,00,00,04,00,12,00,18,00,21,00,2a,00,35,00,47,00,56,00,90,00,e5,01,68,01,d5,01,d7,02,36,02,a6,02,ee,03,48,03,a0,03,ba,03,c5,03,d0,03,E0,03,ea,03,Fa,03,FF
3a,01,66
11,80
29,80
#
```

The main fix was changing `pclk_active_neg` to 0 on the `V:` line. That fixed the noise/sync issues.

The other change was moving the pin assignments around to swap the "endianness" of the pixel format.

Now to get touch working...

dgwynne, 2023-10-06:
This uses `:B` to fix the endianness, and adds support for the touch controller too:

```
:H,GC9503V,480,480,16,RGB,40,41,42,39,5,45,48,47,0,21,14,13,12,11,16,17,18,8,3,46,10,15
:B,2
:V,1,8,10,50,1,8,10,20,0
:S,2,1,1,0,42,20
:TI1,38,*,*,4,-1
:IS,45,48,38,41
F0,05,55,AA,52,08,00
F6,02,5A,87
C1,01,3F
C2,01,0E
C6,01,F8
C9,01,10
CD,01,25
F8,01,8A
AC,01,45
A0,01,DD
A7,01,47
FA,04,00,00,00,04
A3,01,EE
FD,3,28,28,00
71,01,48
72,01,48
73,02,00,44
97,01,EE
83,01,93
9A,01,72
9B,01,5a
82,02,2c,2c
B1,01,10
6D,20,00,1F,19,1A,10,0e,0c,0a,02,07,1E,1E,1E,1E,1E,1E,1E,1E,1E,1E,1E,1E,08,01,09,0b,0D,0F,1a,19,1f,00
64,10,38,05,01,db,03,03,38,04,01,dc,03,03,7A,7A,7A,7A
65,10,38,03,01,dd,03,03,38,02,01,de,03,03,7A,7A,7A,7A
66,10,38,01,01,df,03,03,38,00,01,e0,03,03,7A,7A,7A,7A
67,10,30,01,01,e1,03,03,30,02,01,e2,03,03,7A,7A,7A,7A
68,0d,00,08,15,08,15,7A,7A,08,15,08,15,7A,7A
60,08,38,08,7A,7A,38,09,7A,7A
63,08,31,e4,7A,7A,31,e5,7A,7A
6B,01,07
7A,02,08,13
7B,02,08,13
D1,34,00,00,00,04,00,12,00,18,00,21,00,2a,00,35,00,47,00,56,00,90,00,e5,01,68,01,d5,01,d7,02,36,02,a6,02,ee,03,48,03,a0,03,ba,03,c5,03,d0,03,E0,03,ea,03,Fa,03,FF
D2,34,00,00,00,04,00,12,00,18,00,21,00,2a,00,35,00,47,00,56,00,90,00,e5,01,68,01,d5,01,d7,02,36,02,a6,02,ee,03,48,03,a0,03,ba,03,c5,03,d0,03,E0,03,ea,03,Fa,03,FF
D3,34,00,00,00,04,00,12,00,18,00,21,00,2a,00,35,00,47,00,56,00,90,00,e5,01,68,01,d5,01,d7,02,36,02,a6,02,ee,03,48,03,a0,03,ba,03,c5,03,d0,03,E0,03,ea,03,Fa,03,FF
D4,34,00,00,00,04,00,12,00,18,00,21,00,2a,00,35,00,47,00,56,00,90,00,e5,01,68,01,d5,01,d7,02,36,02,a6,02,ee,03,48,03,a0,03,ba,03,c5,03,d0,03,E0,03,ea,03,Fa,03,FF
D5,34,00,00,00,04,00,12,00,18,00,21,00,2a,00,35,00,47,00,56,00,90,00,e5,01,68,01,d5,01,d7,02,36,02,a6,02,ee,03,48,03,a0,03,ba,03,c5,03,d0,03,E0,03,ea,03,Fa,03,FF
D6,34,00,00,00,04,00,12,00,18,00,21,00,2a,00,35,00,47,00,56,00,90,00,e5,01,68,01,d5,01,d7,02,36,02,a6,02,ee,03,48,03,a0,03,ba,03,c5,03,d0,03,E0,03,ea,03,Fa,03,FF
3a,01,66
11,80
29,80
#
```

template is:

```
{"NAME":"ESP32S3","GPIO":[1,1,1,1,1,0,608,1,1,1,1,1,1,1,1,640,1,1,1,1,1,1,0,0,0,0,0,0,6210,0,0,0,0,0,0,0,0,0],"FLAG":0,"BASE":1}
```

PR #19682 (2023-10-07): udisplay suport for GC3503V
## Description:

**Related issue (if applicable):** fixes #19489

## Checklist:
  - [x] The pull request is done against the latest development branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR and the code change compiles without warnings
  - [x] The code change is tested and works with Tasmota core ESP8266 V.2.7.4.9
  - [x] The code change is tested and works with Tasmota core ESP32 V.2.0.14
  - [x] I accept the [CLA](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md#contributor-license-agreement-cla).
 [... cut]
closed 2023-10-07 by arendst