arendst/Tasmota #19496: Shelly Dimmer 2 does not work after flashing Tasmota

DerDom, 2023-09-11:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
I did flash Tasmota to my Shelly Dimmer 2. Before, I could dimm the light, switch it on and off. After flashing Tasmota it does not work anymore.

- I am not sure my Shelly Dimmer 2 works 100% correct. However, it was able to dimm and swith the light before with the original Firmware
- Flashing via http://shellyip/ota?url=http://ota.tasmota.com/tasmota/shelly/mg2tasmota-ShellyDimmer2.zip worked without any issue
- I flashed Tasmota 13.1.0 
- connecting to my wifi worked without any issue
- I did update the template: {"NAME":"Shelly Dimmer 2","GPIO":[0,3200,0,3232,5568,5600,0,0,193,0,192,0,320,4736],"FLAG":0,"BASE":18}

Could you please let me know what additional information you need to determine whether it is a code issue or a setup issue?

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Shelly Dimmer 2
- [x] Tasmota binary firmware version number used: Tasmota 13.1.0 
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: https://github.com/arendst/mgos-to-tasmota
- [x] Provide the output of command: `power on`:
```lua
  Configuration output here:
11:19:11.729 CMD: power on
11:19:11.739 RSL: RESULT = {"POWER":"ON"}
11:19:11.742 RSL: POWER = ON
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
11:20:57.632 RSL: STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_858984","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
11:20:57.639 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:16:41","StartupUTC":"2023-09-11T10:04:16","Sleep":50,"CfgHolder":4617,"BootCount":5,"BCResetTime":"2023-09-11T10:57:48","SaveCount":54,"SaveAddress":"F6000"}}
11:20:57.643 RSL: STATUS2 = {"StatusFWR":{"Version":"13.1.0(tasmota)","BuildDateTime":"2023-08-15T07:58:36","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"393/699"}}
11:20:57.647 RSL: STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Vodafone-64DF",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A192800000000","00000080","00006000","00004000","00000000"]}}
11:20:57.659 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":634,"Free":368,"Heap":24,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"15400B","FlashFrequency":40,"FlashMode":"DOUT","Features":["00000809","8F9AC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","54000020","00000080"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45,62,68","Sensors":"1,2,3,4,5,6","I2CDriver":"7"}}
11:20:57.665 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota-858984-2436","IPAddress":"192.168.178.30","Gateway":"192.168.178.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.178.1","DNSServer2":"0.0.0.0","Mac":"E8:9F:6D:85:89:84","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
11:20:57.670 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_858984","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
11:20:57.676 RSL: STATUS7 = {"StatusTIM":{"UTC":"2023-09-11T10:20:57","Local":"2023-09-11T11:20:57","StartDST":"2023-03-26T02:00:00","EndDST":"2023-10-29T03:00:00","Timezone":"+01:00","Sunrise":"06:21","Sunset":"19:12"}}
11:20:57.679 RSL: STATUS9 = {"StatusPTH":{"PowerDelta":[0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
11:20:57.687 RSL: STATUS10 = {"StatusSNS":{"Time":"2023-09-11T11:20:57","Switch1":"ON","Switch2":"OFF","ANALOG":{"Temperature":41.8},"ENERGY":{"TotalStartTime":"2023-09-11T10:58:41","Total":0.002,"Yesterday":0.000,"Today":0.002,"Power":4},"TempUnit":"C"}}
11:20:57.693 RSL: STATUS11 = {"StatusSTS":{"Time":"2023-09-11T11:20:57","Uptime":"0T00:16:41","UptimeSec":1001,"Heap":24,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":0,"POWER":"ON","Dimmer":100,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"Vodafone-64DF","BSSId":"3C:37:12:DB:3B:56","Channel":5,"Mode":"11n","RSSI":100,"Signal":-46,"LinkCount":1,"Downtime":"0T00:00:04"}}}
```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
11:22:25.491 CMD: power off
11:22:25.493 SRC: WebConsole from 192.168.178.27
11:22:25.496 CMD: Grp 0, Cmd 'POWER', Idx 1, Len 3, Pld 0, Data 'off'
11:22:25.501 RSL: RESULT = {"POWER":"OFF"}
11:22:25.503 RSL: POWER = OFF
11:22:26.096 CFG: Saved to flash at F4, Count 56, Bytes 4096
11:22:30.511 CMD: power on
11:22:30.513 SRC: WebConsole from 192.168.178.27
11:22:30.515 CMD: Grp 0, Cmd 'POWER', Idx 1, Len 2, Pld 1, Data 'on'
11:22:30.526 RSL: RESULT = {"POWER":"ON"}
11:22:30.529 RSL: POWER = ON
11:22:31.068 CFG: Saved to flash at FB, Count 57, Bytes 4096
11:22:41.254 WIF: Checking connection...

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
I switched with the Shelly Firmware which worked. Then flashed. Now I cannot switch anymore.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
I expect the switch to work.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2023-09-11:
Have you flashed the Coprocessor with the needed custom firmware?
As described here https://templates.blakadder.com/shelly_Dimmer_2.html

DerDom, 2023-09-11:
Thanks @Jason2866. This is the info I was missing. Sorry.

closed 2023-09-11 by DerDom