arendst/Tasmota #19501: Sonoff DUAL R3 v2 shows wrong results on both chanels if the load is inductive

vanLierBernd, 2023-09-12:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
18:44:19.874 CMD: Backlog Rule1; Rule2; Rule3
18:44:19.911 MQT: stat/tasmota_7C9AAC/RESULT = {"Rule1":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
18:44:20.120 MQT: stat/tasmota_7C9AAC/RESULT = {"Rule2":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
18:44:20.352 MQT: stat/tasmota_7C9AAC/RESULT = {"Rule3":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
18:43:12.393 CMD: Status 0
18:43:12.401 MQT: stat/tasmota_7C9AAC/STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["Tasmota",""],"Topic":"tasmota_7C9AAC","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
18:43:12.437 MQT: stat/tasmota_7C9AAC/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32.bin","RestartReason":"Vbat power on reset","Uptime":"0T00:05:01","StartupUTC":"2023-09-12T15:38:11","Sleep":50,"CfgHolder":4617,"BootCount":18,"BCResetTime":"2023-09-11T16:12:22","SaveCount":75}}
18:43:12.476 MQT: stat/tasmota_7C9AAC/STATUS2 = {"StatusFWR":{"Version":"13.1.0(tasmota32)","BuildDateTime":"2023-08-15T08:00:54","Core":"2_0_11","SDK":"4.4.5.230722","CpuFrequency":160,"Hardware":"ESP32-D0WD-V3 rev.3","CR":"450/699"}}
18:43:12.493 MQT: stat/tasmota_7C9AAC/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Villa Hahn IoT","Villa Hahn IoT_EXT2G"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000680003C5A0A192800000000","00000081","00006000","00004000","00000000"]}}
18:43:12.541 MQT: stat/tasmota_7C9AAC/STATUS4 = {"StatusMEM":{"ProgramSize":1870,"Free":1009,"Heap":107,"StackLowMark":4,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"16405E","FlashFrequency":40,"FlashMode":"DIO","Features":["00000809","9F9AD7DF","0015A001","B7F7BFCF","05DA9BC4","E0360DC7","480840D2","20200000","D4BC482D","010A80B1"],"Drivers":"1,2,3,4,5,7,8,9,10,11,12,14,16,17,20,21,24,26,27,29,34,35,38,50,52,59,60,62,63,66,67,68,82,86,87,88","Sensors":"1,2,3,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,31,34,37,39,40,42,43,45,51,52,55,56,58,59,64,66,67,74,85,92,95,98,103,105,109,127","I2CDriver":"7,8,9,10,11,12,13,14,15,17,18,20,24,29,31,36,41,42,44,46,48,58,62,65,69,76,77,82"}}
18:43:12.616 MQT: stat/tasmota_7C9AAC/STATUS5 = {"StatusNET":{"Hostname":"tasmota-7C9AAC-6828","IPAddress":"192.168.188.130","Gateway":"192.168.188.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.188.1","DNSServer2":"2a02:587:312a:9800:2e91:abff:fef7:d4ae","Mac":"24:D7:EB:7C:9A:AC","IP6Global":"2a02:587:312a:9800:26d7:ebff:fe7c:9aac","IP6Local":"fe80::26d7:ebff:fe7c:9aac","Ethernet":{"Hostname":"","IPAddress":"0.0.0.0","Gateway":"0.0.0.0","Subnetmask":"0.0.0.0","DNSServer1":"192.168.188.1","DNSServer2":"2a02:587:312a:9800:2e91:abff:fef7:d4ae","Mac":"00:00:00:00:00:00","IP6Global":"","IP6Local":""},"Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":16.0}}
18:43:12.683 MQT: stat/tasmota_7C9AAC/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.188.20","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_7C9AAC","MqttUser":"MQTT","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
18:43:12.713 MQT: stat/tasmota_7C9AAC/STATUS7 = {"StatusTIM":{"UTC":"2023-09-12T15:43:12","Local":"2023-09-12T18:43:12","StartDST":"2023-03-26T03:00:00","EndDST":"2023-10-29T04:00:00","Timezone":99,"Sunrise":"08:22","Sunset":"21:10"}}
18:43:12.733 MQT: stat/tasmota_7C9AAC/STATUS9 = {"StatusPTH":{"PowerDelta":[0,0,0,0,0,0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
18:43:12.760 MQT: stat/tasmota_7C9AAC/STATUS10 = {"StatusSNS":{"Time":"2023-09-12T18:43:12","Switch1":"ON","Switch2":"ON","ENERGY":{"TotalStartTime":"2023-09-11T00:00:00","Total":0.037,"Yesterday":0.008,"Today":0.029,"Power":[2,0],"ApparentPower":[2,0],"ReactivePower":[2,0],"Factor":[0.81,0.00],"Voltage":232,"Current":[0.011,0.000],"BL09XX":{"Temperature":49.1}},"TempUnit":"C"}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

Current 0.011 A
Active Power 2W
Apparent Power 2 VA
Reactive Power 1VAr
Power Factor 0,81

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
I did the calibration on both channels with a 75W lamp. Everything looking fine. When I connect the pool pump (235V, 1100 Watts), I get the following results:

**(Please, remember to close the issue when the problem has been addressed)**

Braehead, 2023-09-15:
Power can only be caclulated if you measure both voltage and current.  Current only is only valid for power of resistive loads.  BUT, you also need to measure the phase angle between the voltage and current waveforms for reactive loads.  Otherwise it means nothing, but it does also allow you to measure generation and grid feeds (direction) too.  If you want accurate power figures for reactive loads (capcitance or inductance) you would be better using a dedicated energy measuring chip.  Tasmota does not sample fast enough to calculate true power measurements.

vanLierBernd, 2023-09-18:
Thank you Adam for the explanation.