arendst/Tasmota #19507: Typo in xdsp_19_max7219_matrix.ino

SN-8023, 2023-09-13:
in the file tasmota/tasmota_xdsp_display/xdsp_19_max7219_matrix.ino for the MAX7219 there are several typos (max2719, max7291 etc). Is this supposed to be like this?

btw. great project. Hats off.

Jason2866, 2023-09-22:
The driver supports the max2719 and max7219

closed 2023-09-22 by Jason2866

sfromis, 2023-09-22:
As those names are from internal labels without material impact, I'd expect the differences being to separate the later matrix support from the original 7-segment support, as the Tasmota build structure inherited from Arduino needs labels to be unique, also outside of each `.ino` file.

barbudor, 2023-09-22:
> The driver supports the max2719 and max7219

There is no such chip as the MAX2719