arendst/Tasmota #19521: tasmota32-scripting build fails - needs adopting to the new email implementation

hpuac, 2023-09-14:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

I wanted to install the unofficial "Tasmota Scripting (english)" via the Web-UI and got a 404 error.
The error message was: `Downlading firmware https://tasmota.github.io/install/firmware/unofficial//tasmota32-scripting.factory.bin failed: 404`
After posting this in the "flashing-issues" discord it turns out that a failed build is the cause.
See: https://github.com/Jason2866/Tasmota-build/actions/runs/6186683718/job/16794919497
Apparently the code needs adoption to the new email implementation.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Not relevant, skipping this point
- [x] Tasmota binary firmware version number used: latest version by Web-UI
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: Web-UI
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

Not relevant, skipping this point
```
- [x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

Not relevant, skipping this point
```
- [x] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

Not relevant, skipping this point
```
- [x] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

Not relevant, skipping this point
```

### TO REPRODUCE
_Steps to reproduce the behavior:_

1. Go to the [Tasmota install page](https://tasmota.github.io/install/)
2. Select Tasmota Scripting (english)
3. Try to install

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

I'm expecting the firmware to download and to flash

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

[image]
[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

Fyi @gemu2015 as maintainer of scripter

**(Please, remember to close the issue when the problem has been addressed)**

gemu2015, 2023-09-15:
file system access has been completely changed in new email library.
the library uses a different approach than tasmota.

it has to be evaluated how to adapt the code.

i am short in time and it may take some time to solve this.

currently it will compile by commenting this line in xdrv_01_2_webserver_esp32_mail.ino
 
//att.file.storage_type = esp_mail_file_storage_type_univ;

 any email attachments are no longer possible, but simple mail should work

hagenbuch, 2023-09-27:
Not wanting to add pressure, I just had the same problem :) Could upload and flash the traditional way, so no worries :)

PR #19608 (2023-09-27): fix compile error with new email lib
## Description:

**Related issue (if applicable):** fixes #19521

fixes compile error until file system problems with email lib are solved

however no more file attachments in mail

## Checklist:
  - [x] The pull request is done against the latest development branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR and the code change compiles without warnings
  - [x] The code change is tested and works with Tasmota core ESP8266 V.2.7.4.9
  - [x] The code change is tested and works with Tasmota core ESP32 V.2.0.13
  - [x [... cut]
closed 2023-09-27 by Jason2866