arendst/Tasmota #19527: Webquery not working in pre-built bins

pmknowles, 2023-09-15:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
WebQuery does not work unless #define USE_WEBSEND_RESPONSE is compiled but is not mentioned in the docs.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ X] Device used (e.g., Sonoff Basic): various
- [ X] Tasmota binary firmware version number used: 13.1.0
  - [ X] Pre-compiled
  - [ X] Self-compiled
- [ X] Flashing tools used: Web installer
- [ X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
- 
```lua
  Configuration output here:
13:41:23.661 MQT: stat/Computer/RESULT = {"NAME":"sonoff-dual-r3","GPIO":[0,0,1,0,0,0,3232,3200,0,0,225,0,0,0,0,0,0,0,0,0,1,7296,7328,224,0,0,0,0,160,161,0,0,0,0,0,0],"FLAG":0,"BASE":1}
13:41:23.883 MQT: stat/Computer/RESULT = {"Module":{"0":"sonoff-dual-r3"}}
13:41:24.171 MQT: stat/Computer/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"3232":"Serial Rx"},"GPIO10":{"3200":"Serial Tx"},"GPIO11":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"225":"Relay2"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"0":"None"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"0":"None"},"GPIO22":{"0":"None"},"GPIO23":{"0":"None"},"GPIO24":{"0":"None"},"GPIO25":{"7296":"CSE7761 Tx"},"GPIO26":{"7328":"CSE7761 Rx"},"GPIO27":{"224":"Relay1"},"GPIO32":{"160":"Switch1"},"GPIO33":{"161":"Switch2"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}
```
- [X ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [X] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
13:42:06.183 MQT: stat/Computer/STATUS = {"Status":{"Module":0,"DeviceName":"Computer","FriendlyName":["PC",""],"Topic":"Computer","ButtonTopic":"0","Power":3,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
13:42:06.220 MQT: stat/Computer/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release-12.5.0/tasmota32-bluetooth.bin","RestartReason":"Vbat power on reset","Uptime":"27T23:20:48","StartupUTC":"2023-08-18T13:21:18","Sleep":50,"CfgHolder":4617,"BootCount":111,"BCResetTime":"2022-02-23T08:39:43","SaveCount":6152}}
13:42:06.258 MQT: stat/Computer/STATUS2 = {"StatusFWR":{"Version":"12.5.0(bluetooth)","BuildDateTime":"2023-04-17T08:07:04","Core":"2_0_7","SDK":"4.4.4.20230310","CpuFrequency":160,"Hardware":"ESP32-D0WD-V3 rev.3","CR":"440/699"}}
13:42:06.276 MQT: stat/Computer/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["pkm-24m",""],"TelePeriod":60,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A192800000000","00000080","00006000","0000C002","00000000"]}}
13:42:06.309 MQT: stat/Computer/STATUS4 = {"StatusMEM":{"ProgramSize":1644,"Free":1856,"Heap":72,"StackLowMark":0,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"16405E","FlashFrequency":40,"FlashMode":"DOUT","Features":["00000809","8F9AC7CF","00148001","000000CF","010013C0","C0000981","40004080","00200000","5400082C","00000080"],"Drivers":"1,2,3,4,5,7,8,9,10,12,16,20,21,24,26,27,29,35,38,50,52,59,62,79,82,85","Sensors":"1,2,3,5,6,52,62,127","I2CDriver":"7"}}
13:42:06.373 MQT: stat/Computer/STATUS5 = {"StatusNET":{"Hostname":"Computer-4376","IPAddress":"192.168.2.111","Gateway":"192.168.0.1","Subnetmask":"255.255.0.0","DNSServer1":"192.168.0.1","DNSServer2":"0.0.0.0","Mac":"08:3A:F2:2B:B1:18","IP6Global":"fd00::a3a:f2ff:fe2b:b118","IP6Local":"fe80::a3a:f2ff:fe2b:b118","Ethernet":{"Hostname":"","IPAddress":"0.0.0.0","Gateway":"0.0.0.0","Subnetmask":"0.0.0.0","DNSServer1":"192.168.0.1","DNSServer2":"0.0.0.0","Mac":"00:00:00:00:00:00","IP6Global":"","IP6Local":""},"Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
13:42:06.422 MQT: stat/Computer/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.5.109","MqttPort":1883,"MqttClientMask":"Sonoff-Computer","MqttClient":"Sonoff-Computer","MqttUser":"openHAB","MqttCount":10,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
13:42:06.453 MQT: stat/Computer/STATUS7 = {"StatusTIM":{"UTC":"2023-09-15T12:42:06","Local":"2023-09-15T13:42:06","StartDST":"2023-03-26T01:00:00","EndDST":"2023-10-29T02:00:00","Timezone":99,"Sunrise":"06:26","Sunset":"19:03"}}
13:42:06.469 MQT: stat/Computer/STATUS9 = {"StatusPTH":{"PowerDelta":[0,0,0,0,0,0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
13:42:06.497 MQT: stat/Computer/STATUS10 = {"StatusSNS":{"Time":"2023-09-15T13:42:06","Switch1":"ON","Switch2":"ON","ENERGY":{"TotalStartTime":"1970-01-01T00:00:00","Total":[32.659,35.659],"Yesterday":[0.140,0.123],"Today":[0.098,0.079],"Power":[17,17],"ApparentPower":[31,25],"ReactivePower":[25,19],"Factor":[0.57,0.66],"Frequency":50,"Voltage":245,"Current":[0.125,0.103]}}}
13:42:06.541 MQT: stat/Computer/STATUS11 = {"StatusSTS":{"Time":"2023-09-15T13:42:06","Uptime":"27T23:20:48","UptimeSec":2416848,"Heap":76,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":20,"MqttCount":10,"Berry":{"HeapUsed":3,"Objects":44},"POWER1":"ON","POWER2":"ON","Wifi":{"AP":1,"SSId":"pkm-24m","BSSId":"3C:A6:2F:6F:F1:65","Channel":11,"Mode":"11n","RSSI":62,"Signal":-69,"LinkCount":4,"Downtime":"0T00:00:16"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
13:45:49.590 MQT: stat/Computer/RESULT = {"WebQuery":"Done"}

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Webquery https://apigenerator.dronahq.com/api/vuDg_Sdh/path GET

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

13:47:22.673 CMD: Webquery https://apigenerator.dronahq.com/api/vuDg_Sdh/path GET
13:47:24.239 RSL: RESULT = {"WebQuery":[  {    "MMM": "DONE",    "id": 1  },  {    "Are you there": "YES",    "id": 2  },  {    "Are you there 1": "YES",    "id": 3  },  {    "id": 4  },  {    "id": 5  },  {    "id": 6  },  {    "id": 7  },  {    "id": 8  },  {    "id": 9  },  {    "query": "\n    query IntrospectionQuery {\n      __schema {\n        \n        queryType { name }\n        mutationType { name }\n        subscriptionType { name }\n        types {\n          ...FullType\n        }\n        directives {\n          name\n          description\n          \n          locations\n          args {\n            ...InputValue\n          }\n        }\n      }\n    }\n\n    fragment FullType on __Type {\n      kind\n      name\n      description\n      \n      fields(includeDeprecated: true) {\n        name\n        description\n        args {\n          ...InputValue\n        }\n        type {\n          ...TypeRef\n        }\n        isDeprecated\n        deprecationReason\n      }\n      inputFields {\n        ...InputValue\n      }\n      interfaces {\n        ...TypeRef\n      }\n      enumValues(includeDeprecated: true) {\n        name\n        description\n        isDeprecated\n        deprecationReason\n      }\n      possibleTypes {\n        ...TypeRef\n      }\n    }\n\n    fragment InputValue on __InputValue {\n      name\n      description\n      type { ...TypeRef }\n      defaultValue\n      \n      \n    }\n\n    fragment TypeRef on __Type {\n      kind\n      name\n      ofType {\n        kind\n        name\n        ofType {\n        [... cut ...]
_Add any other context about the problem here._
There was a post in the old Sonoff forum that they couldn't get webquery ... GET to work
The docs do not state (nor does the original PR) that #define USE_WEBSEND_RESPONSE needs to be compiled for this command to work. I would insert it into the docs but I need confirmation that this is the case. I also read somewhere that ESP8266 does not work with https for webqueries so that needs confirming too.

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2023-09-15:
Yes, `USE_WEBSEND_RESPONSE` is needed to have the response, and this option has not been configured in precompiled binaries. With a custom build including this, you do get  the message on both ESP8266 and ESP32. Of course, your test case gives a redirect response instead of the JSON, if executed via http instead of https.

Indeed, https is not supported for ESP8266 at all, also for other things than WebQuery/WebSend.

barbudor, 2023-09-15:
Your description is not correct
WebQuery and WebSend works perfectly as the expected GET/POST request is effectly sent.

It is only the response part that is an option as it is not necesseraly needed and use quite some buffer memory

Feel free to update the docs by using the pencil icon on the top right of the page.

Thanks

closed 2023-09-15 by pmknowles