arendst/Tasmota #19538: Prometheus Units for Sonoff S31 are Incorrect for Daily and Total Power

subtlepseudonym, 2023-09-17:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
The prometheus metrics `energy_power_kilowatts_daily` and `energy_power_kilowatts_total` report the number of kilowatt _hours_ on the Sonoff S31 rather than kilowatts, as implied by the metric name.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Sonoff S31
- [x] Tasmota binary firmware version number used: 12.1.1
  - [ ] Pre-compiled
  - [x] Self-compiled
- [x] Flashing tools used: Tasmotizer
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
19:57:02.435 RSL: RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
19:57:02.659 RSL: RESULT = {"Module":{"41":"Sonoff S31"}}
19:57:02.875 RSL: RESULT = {"GPIO0":{"32":"Button1"},"GPIO1":{"3072":"CSE7766 Tx"},"GPIO2":{"0":"None"},"GPIO3":{"3104":"CSE7766 Rx"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"224":"Relay1"},"GPIO13":{"320":"Led_i1"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
 
```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Enable prometheus metrics during compilation and visit the /metrics endpoint.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
The metric names should match the units of the reported values. These metrics, on the S31, should be named `energy_power_kilowatt_hours_daily` and `energy_power_kilowatt_hours_total`.

I haven't looked into any other power monitoring devices that report [these metrics](https://github.com/arendst/Tasmota/blob/ed33d61ab73efa1719567a85ac317730ffd6f191/tasmota/tasmota_xsns_sensor/xsns_75_prometheus.ino#L270). Assuming that other devices are reporting kilowatts rather than kilowatt hours, the best approach is likely to add two new metrics with the proposed names.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

subtlepseudonym, 2023-09-17:
This issue can be fixed on the prometheus side with [metric relabeling](https://prometheus.io/docs/prometheus/latest/configuration/configuration/#metric_relabel_configs), so this issue is a relatively high effort change for low reward, but I figured it was worth documenting.

Jason2866, 2023-09-24:
@subtlepseudonym Would you add to the Documentation? It is Open Source too. Just klick the sheet with the pencil in the upper right corner.

subtlepseudonym, 2023-09-24:
Can do -- [tasmota/docs#1283](https://github.com/tasmota/docs/pull/1283)

closed 2023-09-24 by subtlepseudonym