arendst/Tasmota #19556: PzemAc 1 error 9 - reading pzem values intermittently

IVCrossway, 2023-09-20:
### PROBLEM DESCRIPTION
Tasmota with a PZEM016 Sensor is not reporting data correctly since yesterday. Rebooting the tasmota+pzem didn't change anything. 

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): ESP12F
- [X] Tasmota binary firmware version number used: Tasmota 13.1.0
  - [X] Pre-compiled
  - [ ] Self-compiled
- [X] Flashing tools used: WebInstaller
- [X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
19:57:02.540 CMD: Backlog Template; Module; GPIO 255
19:57:02.542 SRC: WebConsole from 192.168.188.27
19:57:02.545 CMD: Grp 0, Cmd 'BACKLOG', Idx 1, Len 26, Pld -99, Data 'Template; Module; GPIO 255'
19:57:02.615 SRC: Backlog
19:57:02.617 CMD: Grp 0, Cmd 'TEMPLATE', Idx 1, Len 0, Pld -99, Data ''
19:57:02.624 MQT: stat/tasmota_8C5B15/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
19:57:02.830 SRC: Backlog
19:57:02.833 CMD: Grp 0, Cmd 'MODULE', Idx 1, Len 0, Pld -99, Data ''
19:57:02.839 MQT: stat/tasmota_8C5B15/RESULT = {"Module":{"0":"Generic"}}
19:57:03.080 SRC: Backlog
19:57:03.083 CMD: Grp 0, Cmd 'GPIO', Idx 1, Len 3, Pld 255, Data '255'
19:57:03.093 MQT: stat/tasmota_8C5B15/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"1536":"PZEM016 Rx"},"GPIO2":{"0":"None"},"GPIO3":{"1472":"PZEM0XX Tx"},"GPIO4":{"480":"Buzzer"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [X] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
20:02:49.547 CMD: Grp 0, Cmd 'BACKLOG', Idx 1, Len 19, Pld -99, Data 'Rule1; Rule2; Rule3'
20:02:49.570 SRC: Backlog
20:02:49.574 CMD: Grp 0, Cmd 'RULE', Idx 1, Len 0, Pld -99, Data ''
20:02:49.580 MQT: stat/tasmota_8C5B15/RESULT = {"Rule1":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
20:02:49.790 SRC: Backlog
20:02:49.793 CMD: Grp 0, Cmd 'RULE', Idx 2, Len 0, Pld -99, Data ''
20:02:49.800 MQT: stat/tasmota_8C5B15/RESULT = {"Rule2":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
20:02:50.041 SRC: Backlog
20:02:50.044 CMD: Grp 0, Cmd 'RULE', Idx 3, Len 0, Pld -99, Data ''
20:02:50.053 MQT: stat/tasmota_8C5B15/RESULT = {"Rule3":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}

```
- [X] Provide the output of this command: `Status 0`:
```lua
19:57:39.213 CMD: Grp 0, Cmd 'STATUS', Idx 1, Len 1, Pld 0, Data '0'
19:57:39.220 MQT: stat/tasmota_8C5B15/STATUS = {"Status":{"Module":0,"DeviceName":"powermeter","FriendlyName":["powermeter"],"Topic":"tasmota_8C5B15","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
19:57:39.229 MQT: stat/tasmota_8C5B15/STATUS1 = {"StatusPRM":{"Baudrate":9600,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:21:40","StartupUTC":"2023-09-20T18:35:59","Sleep":50,"CfgHolder":4617,"BootCount":54,"BCResetTime":"2023-09-04T15:07:54","SaveCount":117,"SaveAddress":"F6000"}}
19:57:39.235 MQT: stat/tasmota_8C5B15/STATUS2 = {"StatusFWR":{"Version":"13.1.0(tasmota)","BuildDateTime":"2023-08-15T07:58:36","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"380/699"}}
19:57:39.241 MQT: stat/tasmota_8C5B15/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":4,"MqttLog":4,"SysLog":4,"LogHost":"","LogPort":514,"SSId":["FRITZ!Box 7590",""],"TelePeriod":10,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A192800000000","00000080","00006000","00004000","00000000"]}}
19:57:39.254 MQT: stat/tasmota_8C5B15/STATUS4 = {"StatusMEM":{"ProgramSize":634,"Free":368,"Heap":21,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640C8","FlashFrequency":40,"FlashMode":"DOUT","Features":["00000809","8F9AC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","54000020","00000080"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45,62,68","Sensors":"1,2,3,4,5,6","I2CDriver":"7"}}
19:57:39.262 MQT: stat/tasmota_8C5B15/STATUS5 = {"StatusNET":{"Hostname":"tasmota-8C5B15-6933","IPAddress":"192.168.188.184","Gateway":"192.168.188.2","Subnetmask":"255.255.255.0","DNSServer1":"192.168.188.2","DNSServer2":"0.0.0.0","Mac":"refacted","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
19:57:39.271 MQT: stat/tasmota_8C5B15/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.188.73","MqttPort":1883,"MqttClientMask":"pzempowermeter","MqttClient":"pzempowermeter","MqttUser":"redacted","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
19:57:39.281 MQT: stat/tasmota_8C5B15/STATUS7 = {"StatusTIM":{"UTC":"2023-09-20T18:57:39","Local":"2023-09-20T19:57:39","StartDST":"2023-03-26T02:00:00","EndDST":"2023-10-29T03:00:00","Timezone":"+01:00","Sunrise":"06:33","Sunset":"18:53"}}
19:57:39.286 MQT: stat/tasmota_8C5B15/STATUS9 = {"StatusPTH":{"PowerDelta":[0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
19:57:39.298 MQT: stat/tasmota_8C5B15/STATUS10 = {"StatusSNS":{"Time":"2023-09-20T19:57:39","ENERGY":{"TotalStartTime":"2023-09-04T15:12:16","Total":85.670,"Yesterday":4.438,"Today":0.320,"Period":0,"Power":482,"ApparentPower":534,"ReactivePower":231,"Factor":0.90,"Frequency":50,"Voltage":234,"Current":2.284}}}
19:57:39.305 MQT: stat/tasmota_8C5B15/STATUS11 = {"StatusSTS":{"Time":"2023-09-20T19:57:39","Uptime":"0T00:21:40","UptimeSec":1300,"Heap":19,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"FRITZ!Box 7590","BSSId":"redacted","Channel":1,"Mode":"11n","RSSI":100,"Signal":-46,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [X] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
19:57:59.403 PAC: PzemAc 1 error 9
19:57:59.476 WIF: Sending Gratuitous ARP
19:58:00.401 DMP: 01 04 14 08 32 08 E5 00 00 12 EC 00 00 52 08 00 01 00 F4 00 5A 00 00 92 D3
19:58:00.403 PAC: PzemAc 1 error 9
19:58:01.401 DMP: 01 04 14 08 32 08 E5 00 00 12 EC 00 00 52 09 00 01 01 F4 00 5A 00 00 92 D3
19:58:01.403 PAC: PzemAc 1 error 9
19:58:01.768 WIF: Checking connection...
19:58:02.401 DMP: 01 04 14 09 32 08 E9 00 00 12 DE 00 00 52 0A 00 01 00 F4 00 5A 00 00 04 92
19:58:02.403 PAC: PzemAc 1 error 9
19:58:03.401 DMP: 01 04 14 09 30 08 D4 00 00 12 C6 00 00 52 0A 00 00 01 F4 00 5A 00 00 8C FE
19:58:03.403 PAC: PzemAc 1 error 9
19:58:04.401 DMP: 01 04 14 09 30 08 BD 00 00 12 85 00 00 52 0A 00 01 00 F4 00 5A 00 00 D8 21
19:58:04.403 PAC: PzemAc 1 error 9
19:58:05.401 DMP: 01 04 14 08 30 08 BD 00 00 12 85 00 00 52 0A 00 01 00 F4 00 5A 00 00 D8 21
19:58:05.403 PAC: PzemAc 1 error 9
19:58:06.401 DMP: 01 04 14 09 3A 08 D5 00 00 12 BA 00 00 52 0A 00 01 00 F4 00 5A 00 00 BB 5A
19:58:06.403 PAC: PzemAc 1 error 9
19:58:07.401 DMP: 01 04 14 09 3A 08 C4 00 00 12 9C 00 00 52 0A 00 01 01 F4 00 5A 00 00 91 A9
19:58:07.403 PAC: PzemAc 1 error 9

```

### TO REPRODUCE
Not really a way to reproduce. It just happend randomly.

### EXPECTED BEHAVIOUR
Tasmota should read pzem values as always. Insted it randomly switches between reading values and reading 0 reporting "pzemac 1 error 9".

### SCREENSHOTS
Here you can see the sensor reporting correct values until today. Today it started to measure uncorrectly.
[image]
[image]
[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Black6spdZ, 2023-09-21:
do you ever see "pzemac 1 error 7" in your log? I'm trying to get a new setup working and this is all it spits out

barbudor, 2023-09-21:
If you haven't changed anything in Tasmota (firmware, version, setting) it could be a hardware problem
Check anything about wiring, connections... Something may have become loose or damaged over time

IVCrossway, 2023-09-21:
Well basically the problem solved itself today. Tasmota is reading correctly everytime. No more error 9 in the logs. Just something strange happened with Home Assistant that now is reporting two entity for each value (like pzem.energy0 / pzem.energy1).