arendst/Tasmota #19562: Shelly Plus 2PM and Shuttersetup: crashes with exception

Marosso, 2023-09-21:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
I try to use two new Shelly Plus 2PM for shutter control. But i. e. when I use AutoSetup (https://tasmota.github.io/docs/Blinds-and-Shutters/), the Shutters interrupt for a second and in the logs I see that Tasmota has crashed ("RestartReason":{"Exception":1,"Reason":"Syscall"[...]). The AutoSetup is not successfull.

First I made an OTA update to Tasmota according to https://github.com/tasmota/mgos32-to-tasmota32. I did find out, that this installes the solo1-Version of Tasmota although the newer Shelly Plus 2PMs are dual core ESP32.
What I did try:
- install tasmota32.bin via web interface
- flash with web flasher over serial (this installs 13.1.0(tasmota32))
- I tried everything with two brand new Shellys and they both crash most of the time

What I did notice (no idea if relevant):
- Analog1 temperatrue readings are funny: sometimes jumps, is in the range of 4-15°, although it should be ~20.
- after I flashed via serial, I did AutoConfig to Shelly Plus 2PM, than the Shelly did not come up again (invalid header error) in serial console
- the shutter AutoSetup seemingly worked once. But shortly after, the Shelly factory resetted itself?

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Shelly Plus 2PM PCB v0.1.9 with ESP32-U4WDH with Dual Core
- [x] Tasmota binary firmware version number used: 13.1.0(tasmota32)-2_0_11(2023-08-15T08:00:54)
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: Web Flasher
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
08:57:17.557 CMD: Backlog Template; Module; GPIO 255
08:57:17.609 RSL: RESULT = {"NAME":"Shelly Plus 2PM PCB v0.1.9","GPIO":[320,0,0,0,34,192,0,0,225,224,0,0,0,0,193,0,0,0,0,0,0,608,640,3458,0,0,0,0,0,9472,0,4736,0,0,0,0],"FLAG":0,"BASE":1}
08:57:17.819 RSL: RESULT = {"Module":{"0":"Shelly Plus 2PM PCB v0.1.9"}}
08:57:18.030 RSL: RESULT = {"GPIO0":{"320":"Led_i1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"34":"Button3"},"GPIO5":{"192":"Switch_n1"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"225":"Relay2"},"GPIO13":{"224":"Relay1"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"193":"Switch_n2"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"0":"None"},"GPIO22":{"0":"None"},"GPIO23":{"0":"None"},"GPIO24":{"0":"None"},"GPIO25":{"608":"I2C SCL1"},"GPIO26":{"640":"I2C SDA1"},"GPIO27":{"3458":"ADE7953 IRQ3"},"GPIO32":{"0":"None"},"GPIO33":{"9472":"ADE7953 RST"},"GPIO34":{"0":"None"},"GPIO35":{"4736":"ADC Temp1"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}

```
- [x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
n/a

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
08:58:15.198 CMD: Status 0
08:58:15.203 RSL: STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["Tasmota",""],"Topic":"tasmota_D06540","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
08:58:15.244 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32.bin","RestartReason":"Software reset CPU","Uptime":"0T00:06:18","StartupUTC":"2023-09-21T07:51:57","Sleep":50,"CfgHolder":4617,"BootCount":7,"BCResetTime":"2023-09-21T08:33:34","SaveCount":21}}
08:58:15.272 RSL: STATUS2 = {"StatusFWR":{"Version":"13.1.0(tasmota32)","BuildDateTime":"2023-08-15T08:00:54","Core":"2_0_11","SDK":"4.4.5.230722","CpuFrequency":160,"Hardware":"ESP32-U4WDH-D rev.3","CR":"428/699"}}
08:58:15.299 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Wolke",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["0000C009","2805C80001000600003C5A0A192800000000","40000080","00006000","00004000","00000000"]}}
08:58:15.343 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":1870,"Free":1009,"Heap":110,"StackLowMark":3,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"164020","FlashFrequency":40,"FlashMode":"DIO","Features":["00000809","9F9AD7DF","0015A001","B7F7BFCF","05DA9BC4","E0360DC7","480840D2","20200000","D4BC482D","010A80B1"],"Drivers":"1,2,3,4,5,7,8,9,10,11,12,14,16,17,20,21,24,26,27,29,34,35,38,50,52,59,60,62,63,66,67,68,82,86,87,88","Sensors":"1,2,3,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,31,34,37,39,40,42,43,45,51,52,55,56,58,59,64,66,67,74,85,92,95,98,103,105,109,127","I2CDriver":"7,8,9,10,11,12,13,14,15,17,18,20,24,29,31,36,41,42,44,46,48,58,62,65,69,76,77,82"}}
08:58:15.408 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota-D06540-1344","IPAddress":"192.168.178.62","Gateway":"192.168.178.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.178.1","DNSServer2":"2a02:810b:5c0:3c57:e228:6dff:fe4f:c56","Mac":"CC:DB:A7:D0:65:40","IP6Global":"2a02:810b:5c0:3c57:cedb:a7ff:fed0:6540","IP6Local":"fe80::cedb:a7ff:fed0:6540","Ethernet":{"Hostname":"","IPAddress":"0.0.0.0","Gateway":"0.0.0.0","Subnetmask":"0.0.0.0","DNSServer1":"192.168.178.1","DNSServer2":"2a02:810b:5c0:3c57:e228:6dff:fe4f:c56","Mac":"00:00:00:00:00:00","IP6Global":"","IP6Local":""},"Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":16.0}}
08:58:15.465 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_D06540","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
08:58:15.479 RSL: STATUS7 = {"StatusTIM":{"UTC":"2023-09-21T07:58:15","Local":"2023-09-21T08:58:15","StartDST":"2023-03-26T02:00:00","EndDST":"2023-10-29T03:00:00","Timezone":"+01:00","Sunrise":"06:35","Sunset":"18:51"}}
08:58:15.505 RSL: STATUS9 = {"StatusPTH":{"PowerDelta":[0,0,0,0,0,0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
08:58:15.532 RSL: STATUS10 = {"StatusSNS":{"Time":"2023-09-21T08:58:15","Switch1":"OFF","Switch2":"OFF","ANALOG":{"Temperature1":13.1},"ENERGY":{"TotalStartTime":"2023-09-21T08:39:38","Total":0.001,"Yesterday":0.000,"Today":0.001,"Power":[0,0],"ApparentPower":[0,0],"ReactivePower":[0,0],"Factor":[0.00,0.00],"Voltage":0,"Current":[0.000,0.000]},"Shutter1":{"Position":0,"Direction":0,"Target":0,"Tilt":0},"TempUnit":"C"}}
08:58:15.572 RSL: STATUS11 = {"StatusSTS":{"Time":"2023-09-21T08:58:15","Uptime":"0T00:06:18","UptimeSec":378,"Heap":113,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Berry":{"HeapUsed":4,"Objects":47},"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"Wolke","BSSId":"E0:28:6D:4F:0C:58","Channel":1,"Mode":"11n","RSSI":100,"Signal":-43,"LinkCount":1,"Downtime":"0T00:00:04"}}}
08:58:15.613 RSL: STATUS12 = {"StatusSTK":{"Exception":1,"Reason":"Syscall","EPC":"4009167b","EXCVADDR":"00000000","CallChain":["40091678","40092a40","40091d03","40190a46","40190c61","4017ea9c","40180129","401910f9","400ef3e9","40165451","40165476","400f1cf8","4014ef25","4014ef42","4014f732","400d2f0e","4014ece1","4014b80e","4014cf29","40107092","40155e56","40152797","401049c9","400f1c66","400f2aa1","400f4235","401507fd","401071ff","40155f7a","4016241d","400de243","401362f1","4016056a","40136aff","401071ff","4016204d","40162171","40152e03","401049c9","400f1c66","400f2aa1","400f4235","401507fd","401071ff","40155f7a","4016241d","400de243","401362f1"]}}
[... cut ...]
After I run "shuttersetup", the shutters go up a little bit. Than they usually stop for a second, go up further, sometimes stop for one more time. The log shows that the shelly crashed by then.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
I expect the shelly not to crash and the shuttersetup to complete.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

stefanbode, 2023-09-21:
Thanks for giving detailed log info. Indeed during the movement the shutter resets and the. Just continues to move up. Anyhow at this point everything is screwed up already. Normally the power is measured every second and this is fine. During calibration we need 20 readings and here it can happen that the device crashes. Or let me say the driver to read the energy data. There is no easy fix for this because the sensors ore not build for high frequency readouts. I know Shelly is doing it but not how. They must have a trick. The autosetup is early stage and we still improve. Any feedback welcome.

Jason2866, 2023-09-21:
@Marosso Please follow the description exactly for converting Shelly devices to Tasmota. EVERY step is needed.
Doing different can and will most of the time make problems and can end in a bricked device. Yes there is the solo1 variant used since there are different Shelly versions sold. Solo1 version runs on every ESP32. You can easily change to the tasmota32 version by just adopting the OTA URL and hit upgrade.

Marosso, 2023-09-21:
@stefanbode 
Thank you for your swift reply. Okay, I see that. Any idea how I could help? Can I debug somehow where it crashes? I usually do not work with IoT devices, remote debugging is probably not possible ;-) can I introduce some global perisistant variable and fill set in in different places in the code to find out where it crashed?

@Jason2866 I did try to follow as closely as I could (for point 12 I did not know which filesize to set, I just chose the pre selected 512). Do you think something went wrong here? I did flash one Shelly over serial afterwards just to make sure that the OTA update wasn't the issue.

Jason2866, 2023-09-21:
No, the convert worked. The steps you additionaly did are not needed.

stefanbode, 2023-09-21:
I optimized some code that may help. stop calculation is now 10x a sec and not 20x like before this should remove stress from the sensor without impacting the accuracy to much. To compensate the additional error I implemented a method to better capture the time where the 0Watt really occurs. On top of this I now check for 0.6 seconds instead 0.1 sec if the motor is really stopped and do not consume any energy anymore. All in all maybe much better than before. PR is on the way into the development built. Debugged and checked with Shelly Plus 2PM PCB v0.1.9 and the tasmota32.bin.  Be careful some shellys require the solo1 bin.

stefanbode, 2023-09-21:
One additional comment regarding the "energy consumption" is 150kWh or something super strange. This is a known issue by calling the sensor 10x/20x a second and not only once a second as it is the default. Maybe fix in a later version

stefanbode, 2023-09-21:
ok also fixed the energy thing

Marosso, 2023-09-22:
thank you very much @stefanbode . I will try it out as soon as the development release is online.

closed 2023-09-22 by arendst

Marosso, 2023-09-22:
@stefanbode I tried the new version. The shutters stop two times when going up, but the shutters go down completely afterwards (which shows that it was still in shuttersetup, right?)
Unfortunately, after it fully closes, it crashes again:
`11:26:42.096 RSL: INFO3 = {"Info3":{"RestartReason":{"Exception":28,"Reason":"LoadProhibited","EPC":"400f33a3","EXCVADDR":"00000010","CallChain":["400f33a0","400f3473","40150045","4015352b","40103895","400f0f2a","400f1d91","400f27f5","40150f05","40106b2f","401565b2","40162b49","400de073","40136098","40160cee","40136a7d","40106b2f","40162779","4016289d","40153503","40103895","400f0f2a","400f1d91","400f27f5","40150f05","40106b2f","401565b2","40162b49","400e512b"]},"BootCount":27}}`

stefanbode, 2023-09-22:
What template do you use? There are two different ones. And the testing was done also with a Shelly plus 2pm. Therefore quite strange the crashes. Which board is your template.

Marosso, 2023-09-22:
I have two Shelly Plus 2PM version according to board v0.1.9 and dual core ESP32. I used tasmota32.bin (13.1.0.3(tasmota32))
Template:
11:51:38.207 CMD: Backlog Template; Module; GPIO 255
11:51:38.211 SRC: WebConsole from 192.168.178.20
11:51:38.214 CMD: Grp 0, Cmd 'BACKLOG', Idx 1, Len 26, Pld -99, Data 'Template; Module; GPIO 255'
11:51:38.249 SRC: Backlog
11:51:38.251 ADE: Channel1 ACCMODE 0x2D3C00, VRMS 6309892, Period 4473, IRMS 1773, WATT 0, VA 29, VAR 0
11:51:38.253 CMD: Grp 0, Cmd 'TEMPLATE', Idx 1, Len 0, Pld -99, Data ''
11:51:38.255 ADE: Channel2 ACCMODE 0x2D3C00, VRMS 6311255, Period 4473, IRMS 1773, WATT 0, VA 29, VAR 0
11:51:38.259 RSL: RESULT = {"NAME":"Shelly Plus 2PM PCB v0.1.9","GPIO":[320,0,0,0,34,192,0,0,225,224,0,0,0,0,193,0,0,0,0,0,0,608,640,3458,0,0,0,0,0,9472,0,4736,0,0,0,0],"FLAG":0,"BASE":1}
11:51:38.492 SRC: Backlog
11:51:38.495 CMD: Grp 0, Cmd 'MODULE', Idx 1, Len 0, Pld -99, Data ''
11:51:38.497 RSL: RESULT = {"Module":{"0":"Shelly Plus 2PM PCB v0.1.9"}}
11:51:38.744 SRC: Backlog
11:51:38.746 CMD: Grp 0, Cmd 'GPIO', Idx 1, Len 3, Pld 255, Data '255'
11:51:38.756 RSL: RESULT = {"GPIO0":{"320":"Led_i1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"34":"Button3"},"GPIO5":{"192":"Switch_n1"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"225":"Relay2"},"GPIO13":{"224":"Relay1"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"193":"Switch_n2"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"0":"None"},"GPIO22":{"0":"None"},"GPIO23":{"0":"None"},"GPIO24":{"0":"None"},"GPIO25":{"608":"I2C SCL1"},"GPIO26":{"640":"I2C SDA1"},"GPIO27":{"3458":"ADE7953 IRQ3"},"GPIO32":{"0":"None"},"GPIO33":{"9472":"ADE7953 RST"},"GPIO34":{"0":"None"},"GPIO35":{"4736":"ADC Temp1"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}

stefanbode, 2023-09-22:
Now it gets interesting. Same hard and software and different behavior. The new version does not make any power measure during the first 5 seconds. Therefore the reboot must be later. Can you confirm? You can set Shuttersethalfway 50 before calibration. If it is 70 afterwards the calibration finished at least. Anyhow I do not see any chance that the calibration run continues after a crash.

Marosso, 2023-09-22:
I checked afterwards if shutteropenduration is plausible, bit it was 120 - so it did not work.

I tried to get complete logging. I attached two logs. During the first one, it crashes while going up (and it stopped there). In the second one it went down again and crashed after reaching the end point. Maybe it helps.
[log_crash_during_open.txt](https://github.com/arendst/Tasmota/files/12700267/log_crash_during_open.txt)
[log_crash_after_closed.txt](https://github.com/arendst/Tasmota/files/12700268/log_crash_after_closed.txt)

stefanbode, 2023-09-23:
120 is the initial value for the autosetup. This is max run time. If it is 120 then for sure there was a crash. I will take a look into the logs later.

stefxx, 2023-10-19:
I am having the same issue with my Shelly Plus 2PM and shutter. Running shuttersetup moves the shutter up briefly twice, and then continues all the way up while the Shelly crashes. Shutteroperationduration was set to 120 afterwards.

stefanbode, 2023-10-21:
Hi. Ideas are welcome. Currently the high frequency we call the measure cause the reboot some times. You may try it a second time. Open and closeduration are set to high values before starting. 120 looks like the opening was fine and the crash happens on closing. Keep in mind there is always a manual way.

stefxx, 2023-10-21:
It never closed. I think the crash happened after a few seconds going up already. Tries 3 times, same result.

I have a few more shutters to install, so I will have more chances to try. If I find anything interesting I will let you know. For this shutter, I did calibrate manually which is just fine. I was just looking forward to see the magic happen automatically ;)

stefanbode, 2023-10-21:
Thanks. Currently my best guess is that the used method works for some batches and not for others. I have a good one. Maybe I have to deep dive into the driver for the measurement system. If we just catch the crash and continue this is either fine. Shelly seems to have a smart method to do it. Anyhow debugging this is quite time consuming.

stefxx, 2023-10-21:
Great. Let me know if I can help. I might not be an expert on those devices or Tasmota, but I am technically skilled enough that if you explain me what and how, I can do it. Keep up the good work!

PR #19565 (2023-09-22): Improvement in shuttersetup
- reduce call of energy to 10x a sec.
- stop detection more rigid to avoid wrong messages
- more accurate calculation of duration

## Description:

**Related issue (if applicable):** fixes #<Tasmota issue number goes here>
try to fix: #19562

## Checklist:
  - [x] The pull request is done against the latest development branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR and the code change compiles without warnings
  - [x] The code change is tested and works with Tasmota core ESP8266 V.2.7.4.9
  - [x] The code change is test [... cut]