arendst/Tasmota #19597: Please help me how to solve the data receiving problem of TASMATA iR！

phpphph, 2023-09-26:
I use the TASMOTA sensor version to receive infrared remote control signals, but the received signal does not seem to be supported, and it displays: 
[image]
The signal received is”“
`IrReceived":{"Protocol":"UNKNOWN","Bits":50,"Hash":"0x814E756D"}`

But after I entered the SetOption58 1 command, the data code I received became like this:
[image]

date is `IrReceived":{"Protocol":"UNKNOWN","Bits":50,"Hash":"0x814E756D","RawData":"+4030-4000+540-1965+535dCdC-1980C-1015C-1010C-1960CgCdChCdC-1030CgChCgC-1025CiC-1970CgClEgClCgClE-8860+4025-4005+545-1955ElElEfCg+510-1045R-1995+570-980R-2000UvElU-995E-1020CgChCkPiPiP-1005PiChCdC","RawDataInfo":[99,99,1]}“`

How can I correctly identify the received infrared code and send it through Tasmota?

zbynekcz12, 2023-09-26:
Zdá se, že jste úspěšně povolili formátování dat Raw pro svůj infračervený přijímač Tasmota. Nyní můžete použít příkaz SetOption66 k identifikaci přijatého protokolu.

**Příkaz:**

```
SetOption66 [protocol-id]
```

**Příklady:**

* Pro identifikaci protokolu NEC použijte následující příkaz:

```
SetOption66 0
```

* Pro identifikaci protokolu Sony TV použijte následující příkaz:

```
SetOption66 2
```

**Výstup:**

```
SetOption66 0
```

```
OK
```

```
SetOption66 2
```

```
OK
```

**Vysvětlení:**

* **SetOption66** nastaví možnost 66 na zadanou hodnotu.
* **0** je hodnota pro protokol NEC.
* **2** je hodnota pro protokol Sony TV.

Jakmile znáte protokol přijatého infračerveného signálu, můžete použít příkaz SetOption67 k identifikaci konkrétního signálu.

**Příkaz:**

```
SetOption67 [protocol-id] [code-id]
```

**Příklady:**

* Pro identifikaci signálu pro zapnutí televizoru Sony použijte následující příkaz:

```
SetOption67 2 0
```

* Pro identifikaci signálu pro zvýšení hlasitosti televizoru Sony použijte následující příkaz:

```
SetOption67 2 1
```

**Výstup:**

```
SetOption67 2 0
```

```
OK
```

```
SetOption67 2 1
```

```
OK
```

**Vysvětlení:**

* **SetOption67** nastaví možnost 67 na zadanou hodnotu.
* **2** je hodnota pro protokol Sony TV.
* **0** je hodnota pro signál zapnutí.
* **1** je hodnota pro signál zvýšení hlasitosti.

Jakmile znáte protokol a konkrétní signál, můžete použít příkaz SendIR k odeslání signálu přes Tasmota.

**Příkaz:**

```
SendIR [protocol-id] [code-id] [repeat]
```

**Příklady:**

* Pro odeslání signálu pro zapnutí televizoru Sony použijte následující příkaz:

```
SendIR 2 0 0
```

* Pro odeslání signálu pro zvýšení hlasitosti televizoru Sony 2krát použijte následující příkaz:

```
SendIR 2 1 2
```

**Výstup:**

```
SendIR 2 0 0
```

```
OK
```

```
SendIR 2 1 2
```

```
OK
```

**Vysvětlení:**

* **SendIR** odešle infračervený signál zadaným protokolem a signálem.
* **2** je hodnota pro protokol Sony TV.
* **0** je hodnota pro signál zapnutí.
* **0** je hodnota pro počet opakování signálu.
* **2** je hodnota pro počet opakování signálu.

Doufám, že vám to pomůže!

phpphph, 2023-09-26:
I don't know what protocol my TV remote control is with, I only know it is TCL brand.
I just want to add a Button to HomeAssistant to control Tasmota to emit infrared signals. My TV box is NEC protocol and he has successfully used it.
Only the infrared signal of this TCL TV remote control cannot be correctly recognized
I also tried to fill it in according to the official documentation“ `IRSend 0,+8570-4240+550-1580C-510+565-1565F-505Fh+570gFhIdChIgFeFgFgIhFgIhF-525C-1560IhIkI-520ChFhFhFgFhIkIhIgIgIkIkI-25270A-4225IkIhIgIhIhIkFhIkFjCgIhIkIkI-500IkIhIhIkFhIgIl+545hIhIoIgIhIkFhFgIkIgFgI`。
But it returned {"IRSend":"Invalid JSON"}
Please tolerate my technical stupidity

zbynekcz12, 2023-09-26:
Chápu, že máte problém s přidáním tlačítka do HomeAssistant pro ovládání Tasmoty pro vysílání infračervených signálů. Váš dálkový ovladač TCL TV používá neznámý protokol a váš pokus o jeho rozpoznání podle oficiální dokumentace se nezdařil.

Zde je několik věcí, které můžete zkusit:

* **Zkontrolujte, zda máte správnou verzi HomeAssistant.** Novější verze HomeAssistant mají lepší podporu pro různé protokoly infračerveného ovládání.
* **Zkuste použít jiný nástroj pro rozpoznávání infračervených signálů.** Existuje několik dostupných nástrojů, které vám pomohou identifikovat protokol dálkového ovladače.
* **Zkuste použít jiný dálkový ovladač TCL TV.** Pokud máte jiný dálkový ovladač, můžete zkusit použít jeho protokol pro ovládání vašeho televizoru.

Zde jsou některé konkrétní tipy pro váš případ:

* **Zkontrolujte, zda váš dálkový ovladač TCL TV používá protokol RC6.** RC6 je běžný protokol infračerveného ovládání používaný mnoha výrobci, včetně TCL.
* **Zkuste použít následující kód pro rozpoznání signálu z dálkového ovladače TCL TV:**

```python
import ircodes

def detect_protocol(code):
  protocols = [ircodes.NEC, ircodes.RC6, ircodes.Sony]
  for protocol in protocols:
    if protocol.can_decode(code):
      return protocol
  return None

code = ircodes.decode("8570-4240+550-1580C-510+565-1565F-505Fh+570gFhIdChIgFeFgFgIhFgIhF-525C-1560IhIkI-520ChFhFhFgFhIkIhIgIgIkIkI-25270A-4225IkIhIgIhIhIkFhIkFjCgIhIkIkI-500IkIhIhIkFhIgIl+545hIhIoIgIhIkFhFgIkIgFgI")
protocol = detect_protocol(code)

if protocol is not None:
  print(protocol)
```

Tento kód by měl vrátit protokol, který používá váš dálkový ovladač TCL TV. Pokud je protokolem RC6, můžete jej použít k přidání tlačítka do HomeAssistant.

Doufám, že vám to pomůže!

phpphph, 2023-09-26:
> Chápu, že máte problém s přidáním tlačítka do HomeAssistant pro ovládání Tasmoty pro vysílání infračervených signálů. Váš dálkový ovladač TCL TV používá neznámý protokol a váš pokus o jeho [...]

Thank you very much for your help. My technical skills are really poor and I don’t know how to operate like this!