arendst/Tasmota #19602: shutter result MQTT: invalid JSON (comma missing)

sternegugger, 2023-09-26:
### PROBLEM DESCRIPTION
Using more than one shutter in the ESP32 configuration.
When moving a shutter (up or down), an MQTT result will be generated each second.
The contained JSON should contain a set of shutters with their position parameters.
Receiving MQTT apps could update the position for each shutter, while one is moving.

Error:
Between each shutter in the JSON list should be a comma (","). But it is missing.
The MQTT receiving app (in my case OpenHab) can not interprate the JSON and updates only the position of the first shutter.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [  ] Device used (e.g., Sonoff Basic): _____
- [x] Tasmota binary firmware version number used: 13.1.0.3 (development 2023-09-26 20:00)
  - [ ] Pre-compiled
  - [x] Self-compiled
- [  ] Flashing tools used: _____
- [  ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [  ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [  ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [x] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
21:31:30.321 MQT: Rolladen/stat/RESULT = {"Shutter1":{"Position":100,"Direction":0,"Target":100,"Tilt":0}"Shutter2":{"Position":100,"Direction":0,"Target":100,"Tilt":0}"Shutter3":{"Position":100,"Direction":0,"Target":100,"Tilt":0}"Shutter4":{"Position":100,"Direction":0,"Target":100,"Tilt":0}"Shutter5":{"Position":100,"Direction":0,"Target":100,"Tilt":0}"Shutter6":{"Position":100,"Direction":0,"Target":100,"Tilt":0}"Shutter7":{"Position":100,"Direction":0,"Target":100,"Tilt":0}"Shutter8":{"Position":0,"Direction":0,"Target":0,"Tilt":0}"Shutter9":{"Position":1,"Direction":-1,"Target":100,"Tilt":0}"Shutter10":{"Position":0,"Direction":0,"Target":0,"Tilt":0}"Shutter11":{"Position":0,"Direction":0,"Target":0,"Tilt":0}"Shutter12":{"Position":0,"Direction":0,"Target":0,"Tilt":0}}

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Configurate mor than one shutter.
Let one shutter move.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
The MQTT JSON restult should have a comma seperated list of shutters. Like this:
```lua
21:31:30.321 MQT: Rolladen/stat/RESULT = {"Shutter1":{"Position":100,"Direction":0,"Target":100,"Tilt":0},"Shutter2":{"Position":100,"Direction":0,"Target":100,"Tilt":0},"Shutter3":{"Position":100,"Direction":0,"Target":100,"Tilt":0},"Shutter4":{"Position":100,"Direction":0,"Target":100,"Tilt":0},"Shutter5":{"Position":100,"Direction":0,"Target":100,"Tilt":0},"Shutter6":{"Position":100,"Direction":0,"Target":100,"Tilt":0},"Shutter7":{"Position":100,"Direction":0,"Target":100,"Tilt":0},"Shutter8":{"Position":0,"Direction":0,"Target":0,"Tilt":0},"Shutter9":{"Position":1,"Direction":-1,"Target":100,"Tilt":0},"Shutter10":{"Position":0,"Direction":0,"Target":0,"Tilt":0},"Shutter11":{"Position":0,"Direction":0,"Target":0,"Tilt":0},"Shutter12":{"Position":0,"Direction":0,"Target":0,"Tilt":0}}

```
Correct, with comma:
  "Tilt":0}**,**"Shutter2"

Wrong, comma is missing:
  "Tilt":0}"Shutter2"

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2023-09-27:
@stefanbode Can you have a look?

stefanbode, 2023-09-27:
Yes. One second.

PR #19607 (2023-09-27): fix JSON Bug
## Description:

**Related issue (if applicable):** fixes #<Tasmota issue number goes here>
fixed #19602

## Checklist:
  - [x] The pull request is done against the latest development branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR and the code change compiles without warnings
  - [x] The code change is tested and works with Tasmota core ESP8266 V.2.7.4.9
  - [x] The code change is tested and works with Tasmota core ESP32 V.2.0.13
  - [x] I accept the [CLA](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md#contributor [... cut]
closed 2023-09-27 by Jason2866