arendst/Tasmota #19616: Orno WE517 script scaling problem

HaraldNoack, 2023-09-28:
So script is now also supported. You should add a line like this:
```
>y{"Name":"WE517","Baud":9600,"Config":"8E1","Address":1,"Function":3,"Voltage":[0xE,0x10,0x12],"Current":[0x16,0x18,0x1A],"Power":[0x1E,0x20,0x22],"ReactivePower":[0x26,0x28,0x2A],"Factor":[0x36,0x38,0x3A],"Frequency":0x14,"Total":0x100,"ExportActive":0x110,"User":[{"J":"ImportActive","G":"Import Active","R":0x108,"U":"kWh","D":24},{"J":"TotalPower","G":"Active Power Total","R":0x1C,"U":"W","D":23}]}#
```

_Originally posted by @arendst in https://github.com/arendst/Tasmota/discussions/17545#discussioncomment-4583744_

THANKS for the help but I think there is a scaling problem of some registers:
https://old-b2b.orno.pl/download-resource/26065/
Register 0x1C,0x1E,0x20,0x22 is in "kW" and not in "W" as in the script above.

Please can someone help me and correct the script above.
THANKS!!!!

gemu2015, 2023-10-01:
not script related but related to generic modbus driver "xnrg_29_modbus.ino" which only fetches its definition data from script

arendst, 2023-10-01:
you might want to try this:
```
>y{"Name":"WE517","Baud":9600,"Config":"8E1","Address":1,"Function":3,"Voltage":[0xE,0x10,0x12],"Current":[0x16,0x18,0x1A],"Power":[{"R":0x1E,"F":-3},{"R":0x20,"F":-3},{"R":0x22,"F":-3}],"ReactivePower":[0x26,0x28,0x2A],"Factor":[0x36,0x38,0x3A],"Frequency":0x14,"Total":0x100,"ExportActive":0x110,"User":[{"J":"ImportActive","G":"Import Active","R":0x108,"U":"kWh","D":24},{"J":"TotalPower","G":"Active Power Total","R":0x1C,"U":"W","D":23,"F":-3}]}#
```

HaraldNoack, 2023-10-02:
Thanks for the try but I think it doesn't work 

[image]

Active Power and Active Power Total are too low.

THANKS for your help!!!

HaraldNoack, 2023-10-02:
`>y{"Name":"WE517","Baud":9600,"Config":"8E1","Address":1,"Function":3,"Voltage":[0xE,0x10,0x12],"Current":[0x16,0x18,0x1A],"Power":[0x1E,0x20,0x22],"ReactivePower":[0x26,0x28,0x2A],"Factor":[0x36,0x38,0x3A],"Frequency":0x14,"Total":0x100,"ExportActive":0x110,"User":[{"J":"ImportActive","G":"Import Active","R":0x108,"U":"kWh","D":24},{"J":"TotalPower","G":"Active Power Total","R":0x1C,"U":"W","D":23}]}#
`
results in:
[image]

`>y{"Name":"WE517","Baud":9600,"Config":"8E1","Address":1,"Function":3,"Voltage":[0xE,0x10,0x12],"Current":[0x16,0x18,0x1A],"Power":[{"R":0x1E,"F":-3},{"R":0x20,"F":-3},{"R":0x22,"F":-3}],"ReactivePower":[0x26,0x28,0x2A],"Factor":[0x36,0x38,0x3A],"Frequency":0x14,"Total":0x100,"ExportActive":0x110,"User":[{"J":"ImportActive","G":"Import Active","R":0x108,"U":"kWh","D":24},{"J":"TotalPower","G":"Active Power Total","R":0x1C,"U":"W","D":23,"F":-3}]}#
`
results in:
[image]

Thanks!!!!

arendst, 2023-10-02:
Try this:
```
>y{"Name":"WE517","Baud":9600,"Config":"8E1","Address":1,"Function":3,"Voltage":[0xE,0x10,0x12],"Current":[0x16,0x18,0x1A],"Power":[{"R":0x1E,"F":3},{"R":0x20,"F":3},{"R":0x22,"F":3}],"ReactivePower":[0x26,0x28,0x2A],"Factor":[0x36,0x38,0x3A],"Frequency":0x14,"Total":0x100,"ExportActive":0x110,"User":[{"J":"ImportActive","G":"Import Active","R":0x108,"U":"kWh","D":24},{"J":"TotalPower","G":"Active Power Total","R":0x1C,"U":"W","D":23,"F":3}]}#
```

HaraldNoack, 2023-10-02:
Hello,

`>y{"Name":"WE517","Baud":9600,"Config":"8E1","Address":1,"Function":3,"Voltage":[0xE,0x10,0x12],"Current":[0x16,0x18,0x1A],"Power":[{"R":0x1E,"F":3},{"R":0x20,"F":3},{"R":0x22,"F":3}],"ReactivePower":[0x26,0x28,0x2A],"Factor":[0x36,0x38,0x3A],"Frequency":0x14,"Total":0x100,"ExportActive":0x110,"User":[{"J":"ImportActive","G":"Import Active","R":0x108,"U":"kWh","D":24},{"J":"TotalPower","G":"Active Power Total","R":0x1C,"U":"W","D":23,"F":3}]}#`

results in:

[image]

Active Power Total is displayed correct.
Active Power and Reactive Power are still too low.

THANKS!!!!

arendst, 2023-10-03:
Ok. Reset to the original config.

According to your docs Active Power is in kW, Apparent Power is in kVA and Reactive Power is in kVAr. That's fine and that's what Tasmota expects. 

For further debugging enable ``weblog 4`` in the console for half a minute and provide the log here. Expect something like this:
```
10:41:08.559-026 NRG: Modbus send Device 1, Function 4, Register 0000 (0/0), Size 2
10:41:08.757-026 NRG: Modbus register 0, phase 0, rcvd 01040443675DF6E709, T 0, F 0, value 231.3670
10:41:08.758-026 NRG: Modbus send Device 1, Function 4, Register 0006 (1/0), Size 2
10:41:08.956-026 NRG: Modbus register 1, phase 0, rcvd 0104043E8A311E421E, T 0, F 0, value 0.2699
10:41:08.958-026 NRG: Modbus send Device 1, Function 4, Register 000C (2/0), Size 2
10:41:09.156-026 NRG: Modbus register 2, phase 0, rcvd 010404427978229C3C, T 0, F 3, value 62367.3164
10:41:09.157-026 NRG: Modbus send Device 1, Function 4, Register 0012 (3/0), Size 2
10:41:09.356-026 NRG: Modbus register 3, phase 0, rcvd 01040441BF7ADE7D64, T 0, F 0, value 23.9350
10:41:09.358-026 NRG: Modbus send Device 1, Function 4, Register 0018 (4/0), Size 2
10:41:09.556-026 NRG: Modbus register 4, phase 0, rcvd 01040400000000FB84, T 0, F 0, value 0.0000
10:41:09.557-026 NRG: Modbus send Device 1, Function 4, Register 001E (5/0), Size 2
10:41:09.756-026 NRG: Modbus register 5, phase 0, rcvd 0104043F7FFD914774, T 0, F 0, value 1.0000
10:41:09.757-026 NRG: Modbus send Device 1, Function 4, Register 0046 (6/0), Size 2
10:41:09.956-026 NRG: Modbus register 6, phase 0, rcvd 0104044248320DBB4F, T 0, F 0, value 50.0489
10:41:09.957-026 NRG: Modbus send Device 1, Function 4, Register 0156 (7/0), Size 2
10:41:10.156-026 NRG: Modbus register 7, phase 0, rcvd 01040442D2F7CF49A1, T 0, F 0, value 105.4840
10:41:10.157-026 NRG: Modbus send Device 1, Function 4, Register 004A (8/0), Size 2
10:41:10.357-026 NRG: Modbus register 8, phase 0, rcvd 0104043A83126F4A38, T 0, F 0, value 0.0010
10:41:10.358-026 NRG: EnergyTotal[0] 105.4840 kWh
```

HaraldNoack, 2023-10-03:
Hello,
Thanks!
The load is my heat pump and I added additional impulse counter to separate in heating energy and warm water energy but there are also some problems. Maybe you have a solution.
[(https://github.com/arendst/Tasmota/discussions/19638)]

```
18:39:29.282 NRG: Modbus register 0, phase 0, rcvd 010304436F0000DFAA, T 0, F 0, value 239.0000
18:39:29.482 NRG: Modbus register 0, phase 1, rcvd 010304436DE666B420, T 0, F 0, value 237.9000
18:39:29.682 NRG: Modbus register 0, phase 2, rcvd 010304436E3333DA8F, T 0, F 0, value 238.2000
18:39:29.882 NRG: Modbus register 1, phase 0, rcvd 0103043FF0A3D7CEBA, T 0, F 0, value 1.8800
18:39:29.909 BRY: GC from 6338 to 4868 bytes, objects freed 2/49 (in 1 ms) - slots from 51/122 to 51/122
18:39:29.920 MQT: tele/Tasmota_Heatpump/STATE = {"Time":"2023-10-03T18:39:29","Uptime":"0T00:00:15","UptimeSec":15,"Heap":143,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":47,"MqttCount":1,"Berry":{"HeapUsed":4,"Objects":49},"Wifi":{"AP":1,"SSId":"Tier4","BSSId":"E4:C3:2A:47:2B:4F","Channel":1,"Mode":"11n","RSSI":100,"Signal":-47,"LinkCount":1,"Downtime":"0T00:00:04"}}
18:39:30.123 NRG: Modbus register 1, phase 1, rcvd 0103043FF5C28FF6D1, T 0, F 0, value 1.9200
18:39:30.070 MQT: tele/Tasmota_Heatpump/SENSOR = {"Time":"2023-10-03T18:39:29","ANALOG":{"A1":1053,"A2":990},"ENERGY":{"TotalStartTime":"2023-10-03T18:36:37","Total":94.090,"Yesterday":3.720,"Today":2.140,"TodaySumImport":0.000,"TodaySumExport":0.000,"ExportActive":0.010,"Period":[0.0,0.0,0.0],"Power":[0.0,0.0,0.0],"ApparentPower":[449.3,456.8,438.3],"ReactivePower":[0.0,0.0,0.0],"Factor":[1.00,1.00,1.00],"Frequency":50.000,"Voltage":[239.00,237.90,238.20],"Current":[1.880,1.920,1.840],"ImportActive":94.100,"TotalPower":1039.0},"DS18B20_1":{"Id":"0621611B4003","Temperature":42.625},"DS18B20_2":{"Id":"07216016D18D","Temperature":40.750},"WPumpe ":{"Total_in":13.019,"Power_actual":1040,"total_day":0.000,"total_week":0.000,"total_month":0.000,"total_year":0.000},"Heizung":{"Total_in":3.023,"Power_actual":0,"total_day":0.000,"total_week":0.000,"total_month":0.000,"total_year":0.000},"WWasser":{"Total_in":15.512,"Power_actual":1040,"total_day":0.000,"total_week":0.000,"total_month":0.000,"total_year":0.000},"Heatpump":{"Druck_Vorlauf":1.203,"Druck_Ruecklauf":1.147,"Differenzdruck":0.056,"Temperatur_Vorlauf":42.625,"Temperatur_Ruecklauf":40.750},"TempUnit":"C"}
18:39:30.208 HTP: Consoles
18:39:30.283 NRG: Modbus register 1, phase 2, rcvd 0103043FECCCCDA347, T 0, F 0, value 1.8500
18:39:30.483 NRG: Modbus register 2, phase 0, rcvd 01030400000000FA33, T 0, F 0, value 0.0000
18:39:30.683 NRG: Modbus register 2, phase 1, rcvd 01030400000000FA33, T 0, F 0, value 0.0000
18:39:30.883 NRG: Modbus register 2, phase 2, rcvd 01030400000000FA33, T 0, F 0, value 0.0000
18:39:31.084 NRG: Modbus register 4, phase 0, rcvd 0103043BA3D70AD8C2, T 0, F 0, value 0.0050
18:39:31.284 NRG: Modbus register 4, phase 1, rcvd 0103043B03126F4B9B, T 0, F 0, value 0.0020
18:39:31.483 NRG: Modbus register 4, phase 2, rcvd 01030400000000FA33, T 0, F 0, value 0.0000
18:39:31.683 NRG: Modbus register 5, phase 0, rcvd 0103043F800000F7CF, T 0, F 0, value 1.0000
18:39:31.883 NRG: Modbus register 5, phase 1, rcvd 0103043F800000F7CF, T 0, F 0, value 1.0000
18:39:31.913 BRY: GC from 6338 to 4868 bytes, objects freed 2/49 (in 1 ms) - slots from 51/122 to 51/122
18:39:31.924 MQT: tele/Tasmota_Heatpump/STATE = {"Time":"2023-10-03T18:39:31","Uptime":"0T00:00:17","UptimeSec":17,"Heap":142,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":25,"MqttCount":1,"Berry":{"HeapUsed":4,"Objects":49},"Wifi":{"AP":1,"SSId":"Tier4","BSSId":"E4:C3:2A:47:2B:4F","Channel":1,"Mode":"11n","RSSI":100,"Signal":-46,"LinkCount":1,"Downtime":"0T00:00:04"}}
18:39:32.123 NRG: Modbus register 5, phase 2, rcvd 0103043F800000F7CF, T 0, F 0, value 1.0000
18:39:32.074 MQT: tele/Tasmota_Heatpump/SENSOR = {"Time":"2023-10-03T18:39:31","ANALOG":{"A1":1053,"A2":996},"ENERGY":{"TotalStartTime":"2023-10-03T18:36:37","Total":94.090,"Yesterday":3.720,"Today":2.140,"TodaySumImport":0.000,"TodaySumExport":0.000,"ExportActive":0.010,"Period":[0.0,0.0,0.0],"Power":[0.0,0.0,0.0],"ApparentPower":[449.3,456.8,440.7],"ReactivePower":[0.0,0.0,0.0],"Factor":[1.00,1.00,1.00],"Frequency":50.000,"Voltage":[239.00,237.90,238.20],"Current":[1.880,1.920,1.850],"ImportActive":94.100,"TotalPower":1039.0},"DS18B20_1":{"Id":"0621611B4003","Temperature":42.625},"DS18B20_2":{"Id":"07216016D18D","Temperature":40.813},"WPumpe ":{"Total_in":13.019,"Power_actual":1040,"total_day":0.000,"total_week":0.000,"total_month":0.000,"total_year":0.000},"Heizung":{"Total_in":3.023,"Power_actual":0,"total_day":0.000,"total_week":0.000,"total_month":0.000,"total_year":0.000},"WWasser":{"Total_in":15.513,"Power_actual":1039,"total_day":0.000,"total_week":0.000,"total_month":0.000,"total_year":0.000},"Heatpump":{"Druck_Vorlauf":1.201,"Druck_Ruecklauf":1.144,"Differenzdruck":0.057,"Temperatur_Vorlauf":42.625,"Temperatur_Ruecklauf":40.750},"TempUnit":"C"}
18:39:32.283 NRG: Modbus register 6, phase 0, rcvd 0103044247F5C3595F, T 0, F 0, value 49.9900
18:39:32.483 NRG: Modbus register 7, phase 0, rcvd 01030442BC2E143200, T 0, F 0, value 94.0900
18:39:32.683 NRG: Modbus register 8, phase 0, rcvd 0103043C23D70AD85E, T 0, F 0, value 0.0100
18:39:32.787 HTP: Console
18:39:32.884 NRG: Modbus register 9, phase 0, rcvd 01030442BC33337B4A, T 0, F 0, value 94.1000
18:39:33.083 NRG: Modbus register 10, phase 0, rcvd 0103043F851EB8EE1C, T 0, F 3, value 1040.0000
[... cut ...]
18:40:18.483 NRG: Modbus register 1, phase 1, rcvd 0103043FD70A3D816E, T 0, F 0, value 1.6800
18:40:18.684 NRG: Modbus register 1, phase 2, rcvd 0103043FD33333533B, T 0, F 0, value 1.6500
18:40:19.083 NRG: Modbus register 2, phase 0, rcvd 01030400000000FA33, T 0, F 0, value 0.0000
18:40:19.283 NRG: Modbus register 2, phase 1, rcvd 01030400000000FA33, T 0, F 0, value 0.0000
18:40:19.483 NRG: Modbus register 2, phase 2, rcvd 01030400000000FA33, T 0, F 0, value 0.0000
18:40:19.683 NRG: Modbus register 4, phase 0, rcvd 0103043C23D70AD85E, T 0, F 0, value 0.0100 
```

arendst, 2023-10-04:
Using the default config and knowing the relation between the config and the Modbus registers from the log above ...
```
>y{"Name":"WE517","Baud":9600,"Config":"8E1","Address":1,"Function":3,"Voltage":[0xE,0x10,0x12],"Current":[0x16,0x18,0x1A],"Power":[0x1E,0x20,0x22],"ReactivePower":[0x26,0x28,0x2A],"Factor":[0x36,0x38,0x3A],"Frequency":0x14,"Total":0x100,"ExportActive":0x110,"User":[{"J":"ImportActive","G":"Import Active","R":0x108,"U":"kWh","D":24},{"J":"TotalPower","G":"Active Power Total","R":0x1C,"U":"W","D":23}]}#
Modbus register:                                                                 0                         1                        2                                4                         5                           6            7                    8                                                         9                                                                      10
```
... you could conclude that the Orno WE517 does not provide Active Power values. So no or low values is perfectly reasonable.

Jason2866, 2023-10-16:
@arendst For me it looks more like a question than an issue. If so i would move to discussions to keep issues for issues :-)

arendst, 2023-10-17:
It is now indeed.