arendst/Tasmota #19628: TuyaMCU dimmer ignores setoption20 1 and turns power on after a dimmer value is set

Kobbe1, 2023-09-30:
### PROBLEM DESCRIPTION

TuyaMCU dimmer seems to ignore setoption20 1 and turns power on after a dimmer value is set.
Interestingly downgrading to tasmota <=8.2 makes it work but I would rather avoid that.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): wall mounted TuyaMCU dimmer
- [x] Tasmota binary firmware version number used: Tasmota 13.1.0
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: Do not quite remember as it was long ago - but I think it was flashed through hardware serial
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
18:17:05.043 MQT: stat/dimmer/RESULT = {"NAME":"Tuya Dimmer","GPIO":[1,2272,1,2304,1,1,0,0,1,0,1,0,1,0],"FLAG":0,"BASE":54}
18:17:05.257 MQT: stat/dimmer/RESULT = {"Module":{"0":"Tuya Dimmer"}}
18:17:05.512 MQT: stat/dimmer/RESULT = {"GPIO0":{"32":"Button1"},"GPIO1":{"2272":"Tuya Tx"},"GPIO2":{"0":"None"},"GPIO3":{"2304":"Tuya Rx"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"288":"Led1"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
Rules are off

```
- [x] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
18:18:39.598 CMD: Status 0
18:18:39.607 MQT: stat/dimmer/STATUS = {"Status":{"Module":0,"DeviceName":"Dimmer","FriendlyName":["Dimmer"],"Topic":"dimmer","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":0,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
18:18:39.616 MQT: stat/dimmer/STATUS1 = {"StatusPRM":{"Baudrate":9600,"SerialConfig":"8N1","GroupTopic":"sonoffs","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota-DE.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:12:13","StartupUTC":"2023-09-30T17:06:26","Sleep":50,"CfgHolder":4617,"BootCount":18,"BCResetTime":"2023-09-30T17:54:20","SaveCount":127,"SaveAddress":"FA000"}}
18:18:39.625 MQT: stat/dimmer/STATUS2 = {"StatusFWR":{"Version":"13.1.0(tasmota)","BuildDateTime":"2023.08.15 08:03:05","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"385/699"}}
18:18:39.634 MQT: stat/dimmer/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["hzgtred453",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00108009","2805C80001000600003C5AFF002800000000","00000080","00006000","00004000","00000000"]}}
18:18:39.650 MQT: stat/dimmer/STATUS4 = {"StatusMEM":{"ProgramSize":635,"Free":368,"Heap":22,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1440C8","FlashFrequency":40,"FlashMode":"DOUT","Features":["00000407","8F9AC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","54000020","00000080"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45,62,68","Sensors":"1,2,3,4,5,6","I2CDriver":"7"}}
18:18:39.658 MQT: stat/dimmer/STATUS5 = {"StatusNET":{"Hostname":"dimmer-7320","IPAddress":"xxx","Gateway":"xxx","Subnetmask":"xxx","DNSServer1":"xxx","DNSServer2":"0.0.0.0","Mac":"EC:FA:BC:97:5C:98","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
18:18:39.663 MQT: stat/dimmer/STATUS6 = {"StatusMQT":{"MqttHost":"xxx","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_975C98","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
18:18:39.672 MQT: stat/dimmer/STATUS7 = {"StatusTIM":{"UTC":"2023-09-30T17:18:39","Local":"2023-09-30T18:18:39","StartDST":"2023-03-26T02:00:00","EndDST":"2023-10-29T03:00:00","Timezone":"+01:00","Sunrise":"06:48","Sunset":"18:32"}}
18:18:39.680 MQT: stat/dimmer/STATUS10 = {"StatusSNS":{"Time":"2023-09-30T18:18:39"}}
18:18:39.689 MQT: stat/dimmer/STATUS11 = {"StatusSTS":{"Time":"2023-09-30T18:18:39","Uptime":"0T00:12:13","UptimeSec":733,"Heap":22,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Dimmer":64,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"xxx","BSSId":"xxx","Channel":9,"Mode":"11n","RSSI":46,"Signal":-77,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [x] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
18:22:59.882 MQT: stat/dimmer/RESULT = {"WebLog":4}
18:23:00.381 CFG: in Flash gespeichert an F7, Anzahl 130, Bytes 4096
18:23:07.568 TYA: Send "55aa00000000ff"
18:23:07.571 {"TuyaReceived":{"Data":"55AA000000010101","Cmnd":0,"CmndData":"01"}}
18:23:07.573 TYA: Heartbeat
18:23:08.522 WIF: Prüfe Verbindung...
18:23:10.029 CMD: dimmer 44
18:23:10.031 SRC: WebConsole from xxx
18:23:10.034 CMD: Grp 0, Cmd 'DIMMER', Idx 1, Len 2, Pld 44, Data '44'
18:23:10.040 MQT: stat/dimmer/RESULT = {"POWER":"OFF","Dimmer":44}
18:23:10.118 TYA: Send "55aa00060008020200040000002c41"
18:23:10.122 {"TuyaReceived":{"Data":"55AA000000010101","Cmnd":0,"CmndData":"01"}}
18:23:10.124 TYA: Heartbeat
18:23:10.346 CFG: in Flash gespeichert an F6, Anzahl 131, Bytes 4096
18:23:18.520 TYA: Send "55aa00000000ff"
18:23:18.532 {"TuyaReceived":{"Data":"55AA00070008020200040000002C42","Cmnd":7,"CmndData":"020200040000002C","DpType2Id2":44,"2":{"DpId":2,"DpIdType":2,"DpIdData":"0000002C"}}}
18:23:18.535 TYA: fnId=21 is set for dpId=2
18:23:18.536 TYA: RX value 44 from dpId 2 
18:23:18.544 {"TuyaReceived":{"Data":"55AA0007000501010001010F","Cmnd":7,"CmndData":"0101000101","DpType1Id1":1,"1":{"DpId":1,"DpIdType":1,"DpIdData":"01"}}}
18:23:18.547 TYA: fnId=11 is set for dpId=1
18:23:18.548 TYA: RX Relay-1 --> MCU State: On Current State:Off
18:23:18.550 SRC: Switch
18:23:18.559 MQT: stat/dimmer/RESULT = {"POWER":"ON"}
18:23:18.563 MQT: stat/dimmer/POWER = ON
18:23:18.569 {"TuyaReceived":{"Data":"55AA00070008020200040000002A40","Cmnd":7,"CmndData":"020200040000002A","DpType2Id2":42,"2":{"DpId":2,"DpIdType":2,"DpIdData":"0000002A"}}}
18:23:18.571 TYA: fnId=21 is set for dpId=2
18:23:18.572 TYA: RX value 42 from dpId 2 
18:23:18.574 SRC: Switch
18:23:18.577 CMD: Grp 0, Cmd 'DIMMER', Idx 3, Len 2, Pld 42, Data '42'
18:23:18.585 MQT: stat/dimmer/RESULT = {"POWER":"ON","Dimmer":42}
18:23:19.323 CFG: in Flash gespeichert an F5, Anzahl 132, Bytes 4096
18:23:25.501 TYA: Send "55aa001c00080117091e12171906aa"
18:23:25.504 {"TuyaReceived":{"Data":"55AA000000010101","Cmnd":0,"CmndData":"01"}}
18:23:25.506 TYA: Heartbeat
18:23:26.479 WIF: Sending Gratuitous ARP
18:23:28.508 WIF: Prüfe Verbindung...
18:23:29.492 TYA: Send "55aa00000000ff"
18:23:40.486 TYA: Send "55aa00000000ff"
18:23:40.489 {"TuyaReceived":{"Data":"55AA000000010101","Cmnd":0,"CmndData":"01"}}
18:23:40.491 TYA: Heartbeat
18:23:48.503 WIF: Prüfe Verbindung...
18:23:51.483 TYA: Send "55aa00000000ff"
18:23:51.486 {"TuyaReceived":{"Data":"55AA000000010101","Cmnd":0,"CmndData":"01"}}
18:23:51.488 TYA: Heartbeat
```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Use firmware > 8.2
setoption20 1
Switch device off with 'power 0'
Set dimmer level with 'dimmer xy'
Device turns on

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
with setoption20 1 device should support setting brightness without turning on as it does <=8.2

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

barbudor, 2023-09-30:
Check [SetOption54](https://tasmota.github.io/docs/Commands/#setoption54)
By default Tuya do not respect Setoption20, you need to enforce it with SetOption54

barbudor, 2023-09-30:
Actually, SetOption54 was introduced in v6.3.0.4, but it was in v8.2.0 that default values for SetOptions was introduce enforcing a default value of "false" for SetOption54.

Kobbe1, 2023-09-30:
Thank you, but already had set that before, forgot to say that. It gives the following weird behaviour that whenever i try to switch off it switches on and also changes the dimmer value:

```
19:12:29.674 CMD: power off
19:12:29.715 MQT: stat/dimmer/RESULT = {"POWER":"OFF"}
19:12:29.720 MQT: stat/dimmer/POWER = OFF
19:12:37.689 MQT: stat/dimmer/RESULT = {"POWsetoption54 ER":"ON","Dimmer":53}
19:12:37.778 MQT: stat/dimmer/RESULT = {"POWER":"OFF"}
19:12:37.782 MQT: stat/dimmer/POWER = OFF
19:12:38.086 MQT: stat/dimmer/RESULT = {"POWER":"ON","Dimmer":39}
19:12:38.161 MQT: stat/dimmer/RESULT = {"POWER":"ON","Dimmer":42}
19:12:38.240 MQT: stat/dimmer/RESULT = {"POWER":"ON","Dimmer":44}
19:12:38.340 MQT: stat/dimmer/RESULT = {"POWER":"ON","Dimmer":47}
19:12:38.401 MQT: stat/dimmer/RESULT = {"POWER":"ON","Dimmer":50}
19:12:38.481 MQT: stat/dimmer/RESULT = {"POWER":"ON","Dimmer":53}
19:12:38.561 MQT: stat/dimmer/RESULT = {"POWER":"ON","Dimmer":56}
```

Kobbe1, 2023-09-30:
Here with weblog 4 and power off. It turns on when I set dimmer

```
19:21:48.816 CMD: setoption54 1
19:21:48.818 SRC: WebConsole from xxx
19:21:48.821 CMD: Grp 0, Cmd 'SETOPTION', Idx 54, Len 1, Pld 1, Data '1'
19:21:48.827 MQT: stat/dimmer/RESULT = {"SetOption54":"ON"}
19:21:49.367 CFG: in Flash gespeichert an F5, Anzahl 196, Bytes 4096
19:21:58.501 TYA: Send "55aa00000000ff"
19:21:58.505 {"TuyaReceived":{"Data":"55AA000000010101","Cmnd":0,"CmndData":"01"}}
19:21:58.507 TYA: Heartbeat
19:22:00.384 CMD: dimmer 88
19:22:00.386 SRC: WebConsole from xxx
19:22:00.389 CMD: Grp 0, Cmd 'DIMMER', Idx 1, Len 2, Pld 88, Data '88'
19:22:00.393 SRC: Light
19:22:00.411 TYA: Send "55aa0006000802020004000000586d"
19:22:00.426 TYA: Send "55aa0006000501010001010e"
19:22:00.451 MQT: stat/dimmer/RESULT = {"POWER":"ON","Dimmer":88}
19:22:00.520 {"TuyaReceived":{"Data":"55AA000000010101","Cmnd":0,"CmndData":"01"}}
19:22:00.523 TYA: Heartbeat
19:22:00.526 {"TuyaReceived":{"Data":"55AA0007000802020004000000586E","Cmnd":7,"CmndData":"0202000400000058","DpType2Id2":88,"2":{"DpId":2,"DpIdType":2,"DpIdData":"00000058"}}}
19:22:00.529 TYA: fnId=21 is set for dpId=2
19:22:00.531 TYA: RX value 88 from dpId 2 
19:22:00.534 {"TuyaReceived":{"Data":"55AA0007000501010001010F","Cmnd":7,"CmndData":"0101000101","DpType1Id1":1,"1":{"DpId":1,"DpIdType":1,"DpIdData":"01"}}}
19:22:00.537 TYA: fnId=11 is set for dpId=1
19:22:00.538 TYA: RX Relay-1 --> MCU State: On Current State:On
19:22:00.542 {"TuyaReceived":{"Data":"55AA0007000802020004000000566C","Cmnd":7,"CmndData":"0202000400000056","DpType2Id2":86,"2":{"DpId":2,"DpIdType":2,"DpIdData":"00000056"}}}
19:22:00.545 TYA: fnId=21 is set for dpId=2
19:22:00.547 TYA: RX value 86 from dpId 2 
19:22:00.566 TYA: Send "55aa0006000802020004000000586d"
19:22:00.572 {"TuyaReceived":{"Data":"55AA0007000501010001010F","Cmnd":7,"CmndData":"0101000101","DpType1Id1":1,"1":{"DpId":1,"DpIdType":1,"DpIdData":"01"}}}
19:22:00.574 TYA: fnId=11 is set for dpId=1
19:22:00.576 TYA: RX Relay-1 --> MCU State: On Current State:On
19:22:01.338 CFG: in Flash gespeichert an F4, Anzahl 197, Bytes 4096
19:22:08.504 WIF: Prüfe Verbindung...
19:22:10.484 TYA: Send "55aa00000000ff"
19:22:10.495 {"TuyaReceived":{"Data":"55AA0007000802020004000000586E","Cmnd":7,"CmndData":"0202000400000058","DpType2Id2":88,"2":{"DpId":2,"DpIdType":2,"DpIdData":"00000058"}}}
19:22:10.498 TYA: fnId=21 is set for dpId=2
19:22:10.500 TYA: RX value 88 from dpId 2 
19:22:21.487 TYA: Send "55aa00000000ff"
19:22:21.491 {"TuyaReceived":{"Data":"55AA000000010101","Cmnd":0,"CmndData":"01"}}
19:22:21.493 TYA: Heartbeat
19:22:25.492 TYA: Send "55aa001c00080117091e13161906aa"
19:22:25.496 {"TuyaReceived":{"Data":"55AA000000010101","Cmnd":0,"CmndData":"01"}}
19:22:25.498 TYA: Heartbeat
19:22:26.529 WIF: Sending Gratuitous ARP
19:22:28.505 WIF: Prüfe Verbindung...
19:22:32.485 TYA: Send "55aa00000000ff"
```
After I try to turn off but it will tun back on and als change dimmer value:

```

19:24:10.033 CMD: power off
19:24:10.036 SRC: WebConsole from xxx
19:24:10.039 CMD: Grp 0, Cmd 'POWER', Idx 1, Len 3, Pld 0, Data 'off'
19:24:10.055 TYA: Send "55aa0006000501010001000d"
19:24:10.080 MQT: stat/dimmer/RESULT = {"POWER":"OFF"}
19:24:10.084 MQT: stat/dimmer/POWER = OFF
19:24:12.088 TYA: fnId=21 is set for dpId=2
19:24:12.090 TYA: RX value 70 from dpId 2 
19:24:12.107 {"TuyaReceived":{"Data":"55AA0007000501010001010F","Cmnd":7,"CmndData":"0101000101","DpType1Id1":1,"1":{"DpId":1,"DpIdType":1,"DpIdData":"01"}}}
19:24:12.110 TYA: fnId=11 is set for dpId=1
19:24:12.111 TYA: RX Relay-1 --> MCU State: On Current State:On
19:24:12.122 {"TuyaReceived":{"Data":"55AA0007000802020004000000485E","Cmnd":7,"CmndData":"0202000400000048","DpType2Id2":72,"2":{"DpId":2,"DpIdType":2,"DpIdData":"00000048"}}}
19:24:12.125 TYA: fnId=21 is set for dpId=2
19:24:12.127 TYA: RX value 72 from dpId 2 
19:24:12.129 SRC: Switch
19:24:12.132 CMD: Grp 0, Cmd 'DIMMER', Idx 3, Len 2, Pld 72, Data '72'
19:24:12.139 MQT: stat/dimmer/RESULT = {"POWER":"ON","Dimmer":72}
[... 24 lines cut ...]
19:24:12.390 TYA: RX value 81 from dpId 2 
19:24:12.391 SRC: Switch
19:24:12.394 CMD: Grp 0, Cmd 'DIMMER', Idx 3, Len 2, Pld 81, Data '81'
19:24:12.402 MQT: stat/dimmer/RESULT = {"POWER":"ON","Dimmer":81}
19:24:12.426 {"TuyaReceived":{"Data":"55AA0007000501010001010F","Cmnd":7,"CmndData":"0101000101","DpType1Id1":1,"1":{"DpId":1,"DpIdType":1,"DpIdData":"01"}}}
19:24:12.430 TYA: fnId=11 is set for dpId=1
19:24:12.431 TYA: RX Relay-1 --> MCU State: On Current State:On
19:24:12.441 {"TuyaReceived":{"Data":"55AA0007000802020004000000546A","Cmnd":7,"CmndData":"0202000400000054","DpType2Id2":84,"2":{"DpId":2,"DpIdType":2,"DpIdData":"00000054"}}}
19:24:12.444 TYA: fnId=21 is set for dpId=2
19:24:12.445 TYA: RX value 84 from dpId 2 
19:24:12.447 SRC: Switch
19:24:12.450 CMD: Grp 0, Cmd 'DIMMER', Idx 3, Len 2, Pld 84, Data '84'
19:24:12.458 MQT: stat/dimmer/RESULT = {"POWER":"ON","Dimmer":84}
19:24:12.506 {"TuyaReceived":{"Data":"55AA0007000501010001010F","Cmnd":7,"CmndData":"0101000101","DpType1Id1":1,"1":{"DpId":1,"DpIdType":1,"DpIdData":"01"}}}
19:24:12.509 TYA: fnId=11 is set for dpId=1
19:24:12.511 TYA: RX Relay-1 --> MCU State: On Current State:On
19:24:12.639 {"TuyaReceived":{"Data":"55AA0007000802020004000000566C","Cmnd":7,"CmndData":"0202000400000056","DpType2Id2":86,"2":{"DpId":2,"DpIdType":2,"DpIdData":"00000056"}}}
19:24:12.641 TYA: fnId=21 is set for dpId=2
19:24:12.643 TYA: RX value 86 from dpId 2 
19:24:12.645 SRC: Switch
19:24:12.648 CMD: Grp 0, Cmd 'DIMMER', Idx 3, Len 2, Pld 86, Data '86'
19:24:12.655 MQT: stat/dimmer/RESULT = {"POWER":"ON","Dimmer":86}
19:24:13.332 CFG: in Flash gespeichert an F6, Anzahl 203, Bytes 4096

```

barbudor, 2023-09-30:
From what I see, Tasmora receive commands from the TuyaMCU which updates the status of POWER and DIMMER
It looks like the TuyaMCU is considering that a switch is pressed and keep on increasing the dimmer

What is your device ?
Do you have buttons/switchs that are connected to the TuyaMCU and not under control of Tasmota ?

Kobbe1, 2023-09-30:
It is the device depicted here: https://tasmota.github.io/docs/TuyaMCU-Devices/#touch-eu-and-us-multiple-manufacturers
It only has 3 touch buttons - on/off, + and -

Interesting that it works in 8.2 and lower ...

barbudor, 2023-10-01:
@Kobbe1 Not having the device in hand makes it difficult
So you confirm that with v8.2.0 it works as expected but not anymore with v8.3.0 ?

Kobbe1, 2023-10-02:
I rechecked again, it works with 8.2. I could not find 8.3 so tried 8.4. That also works. I had never tried with 8.4.
Then with 8.5.1 it fails.
I can do some more tests but guess I should stay with 8.4 then, after all everything is working as far as I can tell.

Here w/ weblog 4 @ 8.5.1:
```
23:11:57 CMD: weblog 4
23:11:57 SRC: WebConsole from xxx
23:11:57 CMD: Gruppe 0, Index 1, Befehl "WEBLOG", Daten "4"
23:11:57 MQT: stat/dimmer/RESULT = {"WebLog":4}
23:12:06 TYA: Send "55aa00000000ff"
23:12:06 {"TuyaReceived":{"Data":"55AA000000010101","Cmnd":0,"CmndData":"01"}}
23:12:06 TYA: Heartbeat
23:12:08 CMD: dimmer 88
23:12:08 SRC: WebConsole from xxx
23:12:08 CMD: Gruppe 0, Index 1, Befehl "DIMMER", Daten "88"
23:12:08 MQT: stat/dimmer/RESULT = {"POWER":"OFF","Dimmer":88}
23:12:08 CFG: in Flash gespeichert am F5, zählen 288, Bytes 4096
23:12:08 TYA: Send "55aa0006000802020004000000586d"
23:12:08 {"TuyaReceived":{"Data":"55AA000000010101","Cmnd":0,"CmndData":"01"}}
23:12:08 TYA: Heartbeat
23:12:08 WIF: Prüfe Verbindung...
23:12:16 WIF: Sending Gratuitous ARP
23:12:17 TYA: Send "55aa00000000ff"
23:12:17 {"TuyaReceived":{"Data":"55AA0007000802020004000000586E","Cmnd":7,"CmndData":"0202000400000058","DpType2Id2":88,"2":{"DpId":2,"DpIdType":2,"DpIdData":"00000058"}}}
23:12:17 TYA: fnId=21 is set for dpId=2
23:12:17 TYA: RX value 88 from dpId 2 
23:12:17 {"TuyaReceived":{"Data":"55AA0007000501010001010F","Cmnd":7,"CmndData":"0101000101","DpType1Id1":1,"1":{"DpId":1,"DpIdType":1,"DpIdData":"01"}}}
23:12:17 TYA: fnId=11 is set for dpId=1
23:12:17 TYA: RX Relay-1 --> MCU State: On Current State:Off
23:12:17 SRC: Switch
23:12:17 MQT: stat/dimmer/RESULT = {"POWER":"ON"}
23:12:17 MQT: stat/dimmer/POWER = ON
23:12:17 {"TuyaReceived":{"Data":"55AA0007000802020004000000566C","Cmnd":7,"CmndData":"0202000400000056","DpType2Id2":86,"2":{"DpId":2,"DpIdType":2,"DpIdData":"00000056"}}}
23:12:17 TYA: fnId=21 is set for dpId=2
23:12:17 TYA: RX value 86 from dpId 2 
23:12:18 CFG: in Flash gespeichert am F4, zählen 289, Bytes 4096
```

And the same with setoption54 1 where it then again does strange things after 'power off':

```

23:15:07 CMD: dimmer 44
23:15:07 SRC: WebConsole from xxx
23:15:07 CMD: Gruppe 0, Index 1, Befehl "DIMMER", Daten "44"
23:15:07 MQT: stat/dimmer/RESULT = {"POWER":"OFF","Dimmer":44}
23:15:07 CFG: in Flash gespeichert am F7, zählen 294, Bytes 4096
23:15:07 TYA: Send "55aa00060008020200040000002c41"
23:15:07 {"TuyaReceived":{"Data":"55AA000000010101","Cmnd":0,"CmndData":"01"}}
23:15:07 TYA: Heartbeat
23:15:08 WIF: Prüfe Verbindung...
23:15:13 TYA: Send "55aa00000000ff"
23:15:13 {"TuyaReceived":{"Data":"55AA00070008020200040000002C42","Cmnd":7,"CmndData":"020200040000002C","DpType2Id2":44,"2":{"DpId":2,"DpIdType":2,"DpIdData":"0000002C"}}}
23:15:13 TYA: fnId=21 is set for dpId=2
23:15:13 TYA: RX value 44 from dpId 2 
23:15:13 {"TuyaReceived":{"Data":"55AA0007000501010001010F","Cmnd":7,"CmndData":"0101000101","DpType1Id1":1,"1":{"DpId":1,"DpIdType":1,"DpIdData":"01"}}}
23:15:13 TYA: fnId=11 is set for dpId=1
23:15:13 TYA: RX Relay-1 --> MCU State: On Current State:Off
23:15:13 SRC: Switch
23:15:13 MQT: stat/dimmer/RESULT = {"POWER":"ON"}
23:15:13 MQT: stat/dimmer/POWER = ON
23:15:13 {"TuyaReceived":{"Data":"55AA00070008020200040000002A40","Cmnd":7,"CmndData":"020200040000002A","DpType2Id2":42,"2":{"DpId":2,"DpIdType":2,"DpIdData":"0000002A"}}}
23:15:13 TYA: fnId=21 is set for dpId=2
23:15:13 TYA: RX value 42 from dpId 2 
23:15:14 CFG: in Flash gespeichert am F6, zählen 295, Bytes 4096
23:15:16 WIF: Sending Gratuitous ARP
23:15:24 TYA: Send "55aa00000000ff"
23:15:24 {"TuyaReceived":{"Data":"55AA000000010101","Cmnd":0,"CmndData":"01"}}
23:15:24 TYA: Heartbeat
[... 34 lines cut ...]
23:15:36 {"TuyaReceived":{"Data":"55AA0007000501010001000E","Cmnd":7,"CmndData":"0101000100","DpType1Id1":0,"1":{"DpId":1,"DpIdType":1,"DpIdData":"00"}}}
23:15:36 TYA: fnId=11 is set for dpId=1
23:15:36 TYA: RX Relay-1 --> MCU State: Off Current State:Off
23:15:36 {"TuyaReceived":{"Data":"55AA0007000802020004000000081E","Cmnd":7,"CmndData":"0202000400000008","DpType2Id2":8,"2":{"DpId":2,"DpIdType":2,"DpIdData":"00000008"}}}
23:15:36 TYA: fnId=21 is set for dpId=2
23:15:36 TYA: RX value 8 from dpId 2 
23:15:36 {"TuyaReceived":{"Data":"55AA0007000501010001000E","Cmnd":7,"CmndData":"0101000100","DpType1Id1":0,"1":{"DpId":1,"DpIdType":1,"DpIdData":"00"}}}
23:15:36 TYA: fnId=11 is set for dpId=1
23:15:36 TYA: RX Relay-1 --> MCU State: Off Current State:Off
23:15:36 {"TuyaReceived":{"Data":"55AA0007000802020004000000051B","Cmnd":7,"CmndData":"0202000400000005","DpType2Id2":5,"2":{"DpId":2,"DpIdType":2,"DpIdData":"00000005"}}}
23:15:36 TYA: fnId=21 is set for dpId=2
23:15:36 TYA: RX value 5 from dpId 2 
23:15:36 {"TuyaReceived":{"Data":"55AA0007000501010001000E","Cmnd":7,"CmndData":"0101000100","DpType1Id1":0,"1":{"DpId":1,"DpIdType":1,"DpIdData":"00"}}}
23:15:36 TYA: fnId=11 is set for dpId=1
23:15:36 TYA: RX Relay-1 --> MCU State: Off Current State:Off
23:15:36 {"TuyaReceived":{"Data":"55AA00070008020200040000000218","Cmnd":7,"CmndData":"0202000400000002","DpType2Id2":2,"2":{"DpId":2,"DpIdType":2,"DpIdData":"00000002"}}}
23:15:36 TYA: fnId=21 is set for dpId=2
23:15:36 TYA: RX value 2 from dpId 2 
23:15:36 {"TuyaReceived":{"Data":"55AA0007000501010001000E","Cmnd":7,"CmndData":"0101000100","DpType1Id1":0,"1":{"DpId":1,"DpIdType":1,"DpIdData":"00"}}}
23:15:36 TYA: fnId=11 is set for dpId=1
23:15:36 TYA: RX Relay-1 --> MCU State: Off Current State:Off
23:15:36 {"TuyaReceived":{"Data":"55AA00070008020200040000000016","Cmnd":7,"CmndData":"0202000400000000","DpType2Id2":0,"2":{"DpId":2,"DpIdType":2,"DpIdData":"00000000"}}}
23:15:36 TYA: fnId=21 is set for dpId=2
23:15:36 TYA: RX value 0 from dpId 2 
```

barbudor, 2023-10-03:
If you can help narrowing down exactly between which versions it changes from OK to KO it will help reducing the guessing.
Because without the device in hand, I can't debug in real.

Kobbe1, 2023-10-03:
Thanks.
v8.4.0 does work, v8.5.1 does not. I just tried to upgrade to v8.5.0 but this fails with some "buffer does not match", tried different files inclusing a lite image, all from https://github.com/arendst/Tasmota/releases/tag/v8.5.0

barbudor, 2023-10-03:
Thanks
I will see what I can find by comparing those 2 versions

closed 2023-11-02 by github-actions[bot]