arendst/Tasmota #19670: Wifi range_extender: compile error for ESP8266

sternegugger, 2023-10-04:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): ESP8266 (NodeMCU)
- [x] Tasmota binary firmware version number used: 13.1.0.3
  - [ ] Pre-compiled
  - [x] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
I followed the instructions described in [documentation "Range Extender"](https://tasmota.github.io/docs/Range-Extender/)
- git clone https://github.com/arendst/Tasmota.git
- open new tasmota directory in Visual Code
- copied `platformio_tasmota_cenv_sample.ini` to `platformio_tasmota_cenv.ini`
- added the following to `tasmota/user_config_override.ini`
<code>
#define USE_WIFI_RANGE_EXTENDER
#define USE_WIFI_RANGE_EXTENDER_NAPT
</code>

- Start Build under PlatformIO - tasmota - General 

Here the build output with errors:
<code>
 Executing task: C:\Users\Micha\.platformio\penv\Scripts\platformio.exe run --environment tasmota 

Processing tasmota (platform: https://github.com/tasmota/platform-espressif8266/releases/download/2023.04.00/platform-espressif8266.zip; framework: arduino; board: esp8266_1M)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
-DCONFIG_TASMOTA_FLASHMODE_DOUT
*** use provided user_config_override.h as planned ***
*** use provided platformio_override.ini as planned ***
*** use provided platformio_tasmota_cenv.ini as planned ***
*** use provided platformio_tasmota_core3_env.ini as planned ***
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/esp8266_1M.html
PLATFORM: Espressif 8266 (2023.4.0) > Espressif Generic ESP8266 Tasmota 1M sketch NO FS
HARDWARE: ESP8266 80MHz, 80KB RAM, 972.00KB Flash
PACKAGES:
 - framework-arduinoespressif8266 @ 2.7.4+9 
 - tool-esptoolpy @ 1.40501.0 (4.5.1)
 - toolchain-xtensa @ 2.40802.200502 (4.8.2)
Converting tasmota.ino
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ strict
Found 102 compatible libraries
Scanning dependencies...
...
Indexing .pio\build\tasmota\lib2d2\libAdafruit_SGP40-1.1.0.a
D:/Michael/tasmota_code/Tasmota_range_extender/tasmota/tasmota_xdrv_driver/xdrv_58_range_extender.ino:65:2: warning: #warning **** USE_WIFI_RANGE_EXTENDER is enabled **** [-Wcpp]
 #warning **** USE_WIFI_RANGE_EXTENDER is enabled ****
  ^
D:/Michael/tasmota_code/Tasmota_range_extender/tasmota/tasmota_xdrv_driver/xdrv_58_range_extender.ino:71:2: error: #error LWIP_FEATURES required, add "-D PIO_FRAMEWORK_ARDUINO_LWIP2_HIGHER_BANDWIDTH" to build_flags #error LWIP_FEATURES required, add "-D PIO_FRAMEWORK_ARDUINO_LWIP2_HIGHER_BANDWIDTH" to build_flags
  ^
Archiving .pio\build\tasmota\libee6\libAdafruit_VEML7700.a
...
Compiling .pio\build\tasmota\lib766\ESP8266WiFi\ESP8266WiFiAP.cpp.o
D:/Michael/tasmota_code/Tasmota_range_extender/tasmota/tasmota_xdrv_driver/xdrv_58_range_extender.ino: In function 'void CmndRgxPortMap(uint8_t, uint16_t, uint32_t, uint16_t)':
D:/Michael/tasmota_code/Tasmota_range_extender/tasmota/tasmota_xdrv_driver/xdrv_58_range_extender.ino:296:61: error: 'ip_portmap_add' was not declared in this scope
   if (ip_portmap_add(proto, gw_ip, gw_port, dst_ip, dst_port))
                                                             ^
D:/Michael/tasmota_code/Tasmota_range_extender/tasmota/tasmota_xdrv_driver/xdrv_58_range_extender.ino: In function 'void rngxSetupNAPT()':
D:/Michael/tasmota_code/Tasmota_range_extender/tasmota/tasmota_xdrv_driver/xdrv_58_range_extender.ino:429:45: error: 'ip_napt_init' was not declared in this scope
     err_t ret = ip_napt_init(NAPT, NAPT_PORT);
                                             ^
D:/Michael/tasmota_code/Tasmota_range_extender/tasmota/tasmota_xdrv_driver/xdrv_58_range_extender.ino:433:49: error: 'ip_napt_enable_no' was not declared in this scope
       err_t ret = ip_napt_enable_no(SOFTAP_IF, 1);
                                                 ^
Compiling .pio\build\tasmota\lib766\ESP8266WiFi\ESP8266WiFiGeneric.cpp.o
Compiling .pio\build\tasmota\lib766\ESP8266WiFi\ESP8266WiFiGratuitous.cpp.o
*** [.pio\build\tasmota\src\tasmota.ino.cpp.o] Error 1
============================================================================================ [FAILED] Took 91.43 seconds ============================================================================================

Environment    Status    Duration
-------------  --------  ------------
tasmota        FAILED    00:01:31.429
======================================================================================= 1 failed, 0 succeeded in 00:01:31.429 ======================================================================================= 

 *  The terminal process "C:\Users\Micha\.platformio\penv\Scripts\platformio.exe 'run', '--environment', 'tasmota'" terminated with exit code: 1. 
 * </code>

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Compile and build without errors and warnings.

Please fix the code to be compialable with the documented options or adapt the documentation.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
Adding `#define PIO_FRAMEWORK_ARDUINO_LWIP2_HIGHER_BANDWIDTH` to `tasmota/user_config_override.ini` did not help.

The documentation seems to be outdated. 

Copying the environment mentionend in the documentation to the corresponding section in `platformio_tasmota_cenv.ini` did not help.

<code>
[env:tasmota-rangeextender]
monitor_port = /dev/cu.SLAB*
upload_port  = /dev/cu.SLAB*
upload_speed = 921600
build_unflags               = ${esp_defaults.build_unflags}
build_flags                 = ${esp82xx_defaults.build_flags}
                              -D FIRMWARE_RANGE_EXTENDER
                              -D USE_WIFI_RANGE_EXTENDER                ; adds about 11k to flash size
                              -D USE_WIFI_RANGE_EXTENDER_NAPT           ; Optional, adds about 1k to flash size
                              -D PIO_FRAMEWORK_ARDUINO_LWIP2_HIGHER_BANDWIDTH   ; required option to use this feature
</code>

**(Please, remember to close the issue when the problem has been addressed)**

barbudor, 2023-10-04:
Strictly following the first 10 lines of hte documentation lead me to 
```
============================================================================================= [SUCCESS] Took 115.61 seconds =============================================================================================

Environment            Status    Duration
---------------------  --------  ------------
tasmota-rangeextender  SUCCESS   00:01:55.610
============================================================================================== 1 succeeded in 00:01:55.610 ============================================================================================== 
```

barbudor, 2023-10-04:
Just to be clear, I just did that:
[image]

I didn't even tried what is described as an alternative

Before retrying, don't forget to clear the folders `.pio/` and `.cache/` to insure a full recompile of the framework with the proper defines

closed 2023-10-05 by Jason2866

sternegugger, 2023-10-06:
The altenative, using Visual Source Code (VSC), is documented here: https://tasmota.github.io/docs/Visual-Studio-Code/
Under PlatFormIO I had to scroll much more down to find the section "tasmota-rangeexternder". Clicking on General -> Build works now perfectly.

In the meanwhile I found also the precompiled bin file which I could load directly with te OTA wen interface, using this URL: https://tasmota.github.io/install/firmware/unofficial/tasmota-rangeextender.bin

And additionally the [WebInstaller](https://tasmota.github.io/install/) also supports "rangeextender".  Scroll down to “unofficial - Tasmota Regexextender (english)”

**So everything is fine**. 
Only in the documentation of [Range Extender](url) the following line could be removed:
<code>
This feature is not included in precompiled binaries
</code>
and replace it with a hint, where to get the precompiled bin file.

By the way you could correct these lines, which make no sense:
<code>
// Optionally, to enable NAT/NAPT
#ifndef USE_WIFI_RANGE_EXTENDER_NAPT
#undef USE_WIFI_RANGE_EXTENDER_NAPT
#endif
</code>
Replace `#undef` by `#define`