arendst/Tasmota #19686: shutterfrequency not changeable

DaBaBW, 2023-10-06:
### PROBLEM DESCRIPTION
I´m not able to change shutterfrequency in shuttermode 4 with stepper motors on ESP32.
I need 500Hz because 1000Hz is too fast for my 28BYJ-48 steppers.
On ESP8266 it works fine with same Tasmota version.

### REQUESTED INFORMATION

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): ESP32 NodeMCU
- [x] Tasmota binary firmware version number used: 13.1.0.4
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: ESPFlasher
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
18:23:29.677 MQT: stat/Rollos_Wozi_S/RESULT = {"NAME":"ESP32-DevKit","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,0,1,1,1,0,1,1,1,0,0,0,0,1,1,1,1,1,0,0,1],"FLAG":0,"BASE":1}
18:23:29.907 MQT: stat/Rollos_Wozi_S/RESULT = {"Module":{"1":"ESP32-DevKit"}}
18:23:30.115 MQT: stat/Rollos_Wozi_S/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"355":"Counter4"},"GPIO3":{"0":"None"},"GPIO4":{"225":"Relay2"},"GPIO5":{"263":"Relay_i8"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"258":"Relay_i3"},"GPIO15":{"0":"None"},"GPIO16":{"416":"PWM1"},"GPIO17":{"352":"Counter1"},"GPIO18":{"256":"Relay_i1"},"GPIO19":{"419":"PWM4"},"GPIO20":{"0":"None"},"GPIO21":{"640":"I2C SDA1"},"GPIO22":{"608":"I2C SCL1"},"GPIO23":{"262":"Relay_i7"},"GPIO24":{"0":"None"},"GPIO25":{"260":"Relay_i5"},"GPIO26":{"259":"Relay_i4"},"GPIO27":{"417":"PWM2"},"GPIO32":{"229":"Relay6"},"GPIO33":{"418":"PWM3"},"GPIO34":{"353":"Counter2"},"GPIO35":{"354":"Counter3"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
18:46:48.360 MQT: stat/Rollos_Wozi_S/STATUS = {"Status":{"Module":1,"DeviceName":"Rollos Wohnzimmer Süd","FriendlyName":["Tasmota","Tasmota2","Tasmota3","Tasmota4","","","",""],"Topic":"Rollos_Wozi_S","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
18:46:48.368 MQT: stat/Rollos_Wozi_S/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/tasmota32-DE.bin","RestartReason":"Software reset CPU","Uptime":"1T20:56:37","StartupUTC":"2023-10-04T19:50:11","Sleep":50,"CfgHolder":4617,"BootCount":8,"BCResetTime":"2023-07-21T17:10:17","SaveCount":29}}
18:46:48.373 MQT: stat/Rollos_Wozi_S/STATUS2 = {"StatusFWR":{"Version":"13.1.0.4(tasmota32)","BuildDateTime":"2023.10.04 17:23:51","Core":"2_0_14","SDK":"4.4.6","CpuFrequency":160,"Hardware":"ESP32-D0WD-V3 v3.0","CR":"492/699"}}
18:46:48.377 MQT: stat/Rollos_Wozi_S/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["W-PD19-2S",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["04000009","2805C80001000600003C5A0A192800000000","40800080","00006000","00004000","00000001"]}}
18:46:48.393 MQT: stat/Rollos_Wozi_S/STATUS4 = {"StatusMEM":{"ProgramSize":1870,"Free":1009,"Heap":129,"StackLowMark":3,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"1640EF","FlashFrequency":40,"FlashMode":"DIO","Features":["00000407","9F9AD7DF","0015A001","B7F7BFCF","05DA9BC4","E0360DC7","480840D2","20200000","D4BC482D","010A80B1"],"Drivers":"1,2,3,4,5,7,8,9,10,11,12,14,16,17,20,21,24,26,27,29,34,35,38,50,52,59,60,62,63,66,67,68,82,86,87,88","Sensors":"1,2,3,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,31,34,37,39,40,42,43,45,51,52,55,56,58,59,64,66,67,74,85,92,95,98,103,105,109,127","I2CDriver":"7,8,9,10,11,12,13,14,15,17,18,20,24,29,31,36,41,42,44,46,48,58,62,65,69,76,77,82"}}
18:46:48.402 MQT: stat/Rollos_Wozi_S/STATUS5 = {"StatusNET":{"Hostname":"Rollos-Wozi-S-0572","IPAddress":"192.168.110.108","Gateway":"192.168.110.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.110.1","DNSServer2":"0.0.0.0","Mac":"54:43:B2:46:42:3C","IP6Global":"","IP6Local":"fe80::5643:b2ff:fe46:423c","Ethernet":{"Hostname":"","IPAddress":"0.0.0.0","Gateway":"0.0.0.0","Subnetmask":"0.0.0.0","DNSServer1":"192.168.110.1","DNSServer2":"0.0.0.0","Mac":"00:00:00:00:00:00","IP6Global":"","IP6Local":""},"Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":16.0}}
18:46:48.408 MQT: stat/Rollos_Wozi_S/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.110.24","MqttPort":1883,"MqttClientMask":"Rollos_Wozi_S","MqttClient":"Rollos_Wozi_S","MqttUser":"Daniel","MqttCount":2,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
18:46:48.414 MQT: stat/Rollos_Wozi_S/STATUS7 = {"StatusTIM":{"UTC":"2023-10-06T16:46:48","Local":"2023-10-06T18:46:48","StartDST":"2023-03-26T02:00:00","EndDST":"2023-10-29T03:00:00","Timezone":99,"Sunrise":"07:23","Sunset":"18:46"}}
18:46:48.426 MQT: stat/Rollos_Wozi_S/STATUS10 = {"StatusSNS":{"Time":"2023-10-06T18:46:48","COUNTER":{"C1":3005,"C2":1290,"C3":43000,"C4":34000},"ESP32":{"Temperature":49.4},"Shutter1":{"Position":0,"Direction":0,"Target":0,"Tilt":0},"Shutter2":{"Position":0,"Direction":0,"Target":0,"Tilt":0},"Shutter3":{"Position":0,"Direction":0,"Target":0,"Tilt":0},"Shutter4":{"Position":0,"Direction":0,"Target":0,"Tilt":0},"TempUnit":"C"}}
18:46:48.439 MQT: stat/Rollos_Wozi_S/STATUS11 = {"StatusSTS":{"Time":"2023-10-06T18:46:48","Uptime":"1T20:56:37","UptimeSec":161797,"Heap":127,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":2,"Berry":{"HeapUsed":23,"Objects":306},"POWER1":"off","POWER2":"off","POWER3":"off","POWER4":"off","POWER5":"off","POWER6":"off","POWER7":"off","POWER8":"off","PWM":{"PWM1":0,"PWM2":0,"PWM3":0,"PWM4":0},"Wifi":{"AP":1,"SSId":"W-PD19-2S","BSSId":"E0:28:6D:4D:5E:79","Channel":1,"Mode":"11n","RSSI":100,"Signal":-46,"LinkCount":1,"Downtime":"0T00:00:06"}}}
18:46:48.447 MQT: stat/Rollos_Wozi_S/STATUS13 = {"StatusSHT":{"SHT0":{"Relay1":1,"Relay2":2,"Open":340,"Close":340,"50perc":50,"Delay":20,"Opt":"0001","Calib":[300,500,700,900,1000],"Mode":"4","TiltConfig":[0,0,0,0,0]},"SHT1":{"Relay1":3,"Relay2":4,"Open":430,"Close":430,"50perc":50,"Delay":20,"Opt":"0001","Calib":[300,500,700,900,1000],"Mode":"4","TiltConfig":[0,0,0,0,0]},"SHT2":{"Relay1":5,"Relay2":6,"Open":430,"Close":430,"50perc":50,"Delay":20,"Opt":"0001","Calib":[300,500,700,900,1000],"Mode":"4","TiltConfig":[0,0,0,0,0]},"SHT3":{"Relay1":7,"Relay2":8,"Open":340,"Close":340,"50perc":50,"Delay":20,"Opt":"0001","Calib":[300,500,700,900,1000],"Mode":"4","TiltConfig":[0,0,0,0,0]}}}

```
- [x] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
18:48:25.437 CMD: shutterfrequency 1100
18:48:25.439 SRC: WebConsole from 192.168.110.21
18:48:25.441 CMD: Grp 0, Cmd 'SHUTTERFREQUENCY', Idx 1, Len 4, Pld 1100, Data '1100'
18:48:25.444 SHT: ShutterMode: 4
18:48:25.445 SHT: Shtr1 min realpos_chg: 2000, min tilt_chg 2
18:48:25.446 SHT: Shtr1 Openvel 1000, Closevel: 1000
18:48:25.448 SHT: Shtr1 Init. Pos 680000, Inv 1, Locked 0, Endstop enab 0, webButt inv 0, Motordel: 20
18:48:25.449 SHT: ShutterMode: 4
18:48:25.451 SHT: Shtr2 min realpos_chg: 2000, min tilt_chg 2
18:48:25.452 SHT: Shtr2 Openvel 1000, Closevel: 1000
18:48:25.453 SHT: Shtr2 Init. Pos 860000, Inv 1, Locked 0, Endstop enab 0, webButt inv 0, Motordel: 20
18:48:25.454 SHT: ShutterMode: 4
18:48:25.456 SHT: Shtr3 min realpos_chg: 2000, min tilt_chg 2
18:48:25.457 SHT: Shtr3 Openvel 1000, Closevel: 1000
18:48:25.458 SHT: Shtr3 Init. Pos 860000, Inv 1, Locked 0, Endstop enab 0, webButt inv 0, Motordel: 20
18:48:25.459 SHT: ShutterMode: 4
[... cut ...]
18:48:25.463 SHT: Shtr4 Init. Pos 680000, Inv 1, Locked 0, Endstop enab 0, webButt inv 0, Motordel: 20
18:48:25.471 MQT: stat/Rollos_Wozi_S/RESULT = {"ShutterFrequency":1000}

```

### TO REPRODUCE
Everytime I want to change shutterfrequency

### EXPECTED BEHAVIOUR
Shutterfrequency should change (I want 500Hz)

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

stefanbode, 2023-10-11:
Hi @DaBaBW . The stepper motor should now work with the frequency. Can you give it try after merged into the development branch. I could only test with one shutter and not 4 simoultanously.

DaBaBW, 2023-10-11:
Hello @stefanbode, thank you very much! I will try it after merging.

closed 2023-10-11 by arendst

stefanbode, 2023-10-14:
I added an additional fix and checked with an oszilloscope channel 1 and 2. Looks like it is now idepedent working. please give it a try and let me know if it also works for channel 3+4.
@arendst : Hi Theo, is there a CONST already I can use to limit to the maximum amount of available channels for de ledfrequency. https://espressif-docs.readthedocs-hosted.com/projects/arduino-esp32/en/latest/api/ledc.html
C3,S2 and S3 all have different ones.

arendst, 2023-10-14:
Let me chk.....

arendst, 2023-10-14:
You might want to try define LEDC_CHANNELS.

Although the above link suggests ESP32 has 16 channels, arduino only supports 8. Same for ESP32S2 and S3. For ESP32C3 it's 6.

s-hadinger, 2023-10-14:
You should fine in `#include "soc/soc_caps.h"` all the per devices characteristics

stefanbode, 2023-10-15:
thanks @s-hadinger. I went this way and also do some management of the channels. Is there known if and which channels are used by the light driver? I haven't seen anything. Anyhow it is unusual if shutters with stepper motors are defined also a PWM Light is on the same device. From my side this is fine for a commit.

s-hadinger, 2023-10-15:
I'm out for now, but I remember that you are supposed to use `analogAttach()` to get a ledc channel and change the frequency. It's here : https://github.com/arendst/Tasmota/blob/715b64dc2821f4cb6e03404118ec49b8afa33cb0/lib/libesp32/ESP32-to-ESP8266-compat/src/esp8266toEsp32.h#L26

DaBaBW, 2023-10-15:
I just updated the last fix, but unfortunately it got worse.
Now all of the 4 shutters influences each other in frequency.

stefanbode, 2023-10-16:
Can you please send an image of the configuration page. Looks like I have to rebuild it the same here to move forward.

DaBaBW, 2023-10-16:
[Shutter Config.pdf](https://github.com/arendst/Tasmota/files/12913244/Shutter.Config.pdf)

backlog SetOption80 1; Shuttermode 4; SetOption15 OFF
backlog PulseTime1 0; PulseTime2 0; ShutterRelay1 1; ShutterButton1 1 up 1; ShutterButton1 2 down 1; ShutterInvert1 1
backlog PulseTime3 0; PulseTime4 0; ShutterRelay2 3; ShutterButton2 3 up 1; ShutterButton2 4 down 1; ShutterInvert2 1
backlog PulseTime5 0; PulseTime6 0; ShutterRelay3 5; ShutterButton3 5 up 1; ShutterButton3 6 down 1; ShutterInvert3 1
backlog PulseTime7 0; PulseTime8 0; ShutterRelay4 7; ShutterButton4 7 up 1; ShutterButton4 8 down 1; ShutterInvert4 1

stefanbode, 2023-10-18:
Ok, nice channel management implemented but did not persist the variable. Sorry. I submitted a new fix and this is tested on my 2 sliders door and with your config. I was able to move all 4 doors in parallel. There is ONE limit. If you start mooving a door while another door is still in movement the counter steps up 1. If all doors are stopped the counter get RESETTED to 0. This does mean if you start moving doors and take care that at least one is still on its way you can break it :-) after start number 9. There are only 8 channels. #19784

DaBaBW, 2023-10-18:
Unfortunately this still isn´t the final solution.
Now shutters 1+3 and 2+4 are "connected". When shutter 1 stops then shutter 3 is slowing down to nearly zero movement.
The same with shutter 2 and 4.

I don´t understand why the frequency is such a problem. The version before your first fix where shutterfrequency was unchangeable at 1000Hz worked fine in function, only was too fast for my steppers.

stefanbode, 2023-10-18:
This is more than strange. Because there is the first started shutter at channel 1 and the second at channel 2. there is no connect to any shutter number. It is no difference if you start 1+2 or 1+3 or 1+4. the second shutter is always on channel 2. can you share the log with loglevel 4. thx.

DaBaBW, 2023-10-18:
After more testing I found out that it isn´t everytime the same shutter that slows down when another stops.
At stop of shutter 4 it is sometimes shutter 2 and sometimes shutter 3 that slowes down.
You say the channels are changed in order which one starts moving first this could be an explanation for this.

At this log I started opening shutter 1 to 4, then stopped 1 to 4 while moving.
Shutter 3 slowed down when stopping shutter 2.

20:12:25.395 CMD: weblog 4
20:12:25.405 MQT: stat/Rollos_Wozi_S/RESULT = {"WebLog":4}
20:12:25.823 CFG: Saved, Anzahl 285, Bytes 4096
20:12:34.883 SHT: Shtr1 Time 22(0), cStop 10500, cVelo 500, mVelo 500, aVelo 0, mRun 1000, aPos 11480, nStop 21980, Trgt 1360000, mVelo 25, Dir 1, Tilt 0, TrgtTilt: 0, Tiltmove: 0
20:12:34.934 SHT: Shtr1 Time 23(0), cStop 10500, cVelo 500, mVelo 500, aVelo 0, mRun 1000, aPos 12480, nStop 22980, Trgt 1360000, mVelo 25, Dir 1, Tilt 0, TrgtTilt: 0, Tiltmove: 0
20:12:34.939 SHT: Shtr1 Real 12480, Start 0, Stop 1360000, Dir 1, Delay 20, Rtc 1.15 [s], Freq 500, PWM 0, Tilt 0
20:12:34.947 MQT: stat/Rollos_Wozi_S/RESULT = {"Shutter1":{"Position":99,"Direction":1,"Target":0,"Tilt":0},"Shutter2":{"Position":100,"Direction":0,"Target":100,"Tilt":0},"Shutter3":{"Position":100,"Direction":0,"Target":100,"Tilt":0},"Shutter4":{"Position":100,"Direction":0,"Target":100,"Tilt":0}}
20:12:34.983 SHT: Shtr1 Time 24(0), cStop 10500, cVelo 500, mVelo 500, aVelo 0, mRun 1000, aPos 13520, nStop 24020, Trgt 1360000, mVelo 25, Dir 1, Tilt 0, TrgtTilt: 0, Tiltmove: 0
20:12:35.036 SHT: Shtr1 Time 25(0), cStop 10500, cVelo 500, mVelo 500, aVelo 0, mRun 1000, aPos 14480, nStop 24980, Trgt 1360000, mVelo 25, Dir 1, Tilt 0, TrgtTilt: 0, Tiltmove: 0
20:12:35.083 SHT: Shtr1 Time 26(0), cStop 10500, cVelo 500, mVelo 500, aVelo 0, mRun 1000, aPos 15480, nStop 25980, Trgt 1360000, mVelo 25, Dir 1, Tilt 0, TrgtTilt: 0, Tiltmove: 0
20:12:35.135 SHT: Shtr1 Time 27(0), cStop 10500, cVelo 500, mVelo 500, aVelo 0, mRun 1000, aPos 16480, nStop 26980, Trgt 1360000, mVelo 25, Dir 1, Tilt 0, TrgtTilt: 0, Tiltmove: 0
20:12:35.183 SHT: Shtr1 Time 28(0), cStop 10500, cVelo 500, mVelo 500, aVelo 0, mRun 1000, aPos 17480, nStop 27980, Trgt 1360000, mVelo 25, Dir 1, Tilt 0, TrgtTilt: 0, Tiltmove: 0
20:12:35.235 SHT: Shtr1 Time 29(0), cStop 10500, cVelo 500, mVelo 500, aVelo 0, mRun 1000, aPos 18480, nStop 28980, Trgt 1360000, mVelo 25, Dir 1, Tilt 0, TrgtTilt: 0, Tiltmove: 0
20:12:35.283 SHT: Shtr1 Time 30(0), cStop 10500, cVelo 500, mVelo 500, aVelo 0, mRun 1000, aPos 19480, nStop 29980, Trgt 1360000, mVelo 25, Dir 1, Tilt 0, TrgtTilt: 0, Tiltmove: 0
20:12:35.336 SHT: Shtr1 Time 31(0), cStop 10500, cVelo 500, mVelo 500, aVelo 0, mRun 1000, aPos 20480, nStop 30980, Trgt 1360000, mVelo 25, Dir 1, Tilt 0, TrgtTilt: 0, Tiltmove: 0
20:12:35.383 SHT: Shtr1 Time 32(0), cStop 10500, cVelo 500, mVelo 500, aVelo 0, mRun 1000, aPos 21480, nStop 31980, Trgt 1360000, mVelo 25, Dir 1, Tilt 0, TrgtTilt: 0, Tiltmove: 0
20:12:35.435 SHT: Shtr1 Time 33(0), cStop 10500, cVelo 500, mVelo 500, aVelo 0, mRun 1000, aPos 22480, nStop 32980, Trgt 1360000, mVelo 25, Dir 1, Tilt 0, TrgtTilt: 0, Tiltmove: 0
20:12:35.483 SHT: Shtr1 Time 34(0), cStop 10500, cVelo 500, mVelo 500, aVelo 0, mRun 1000, aPos 23480, nStop 33980, Trgt 1360000, mVelo 25, Dir 1, Tilt 0, TrgtTilt: 0, Tiltmove: 0
20:12:35.534 SHT: Shtr1 Time 35(0), cStop 10500, cVelo 500, mVelo 500, aVelo 0, mRun 1000, aPos 24480, nStop 34980, Trgt 1360000, mVelo 25, Dir 1, Tilt 0, TrgtTilt: 0, Tiltmove: 0
20:12:35.575 SRC: WebGui from 192.168.110.21
20:12:35.576 SRC: WebGui from 192.168.110.21
20:12:35.579 CMD: Grp 0, Cmd 'SHUTTERPOSITION', Idx 2, Len 8, Pld -99, Data 'StopOpen'
20:12:35.581 SHT: Pos. payload <StopOpen> (8), payload -99, idx 2 (1), src 7
20:12:35.582 SHT: Pos. payload <StopOpen> (0), payload 100, idx 2 (1), src 7
20:12:35.584 SHT: lastsource 7:, real 0, target 1720000, tiltreal: 0, tilttarget: 0, payload 100
20:12:35.586 SHT: Use channel 6
20:12:35.588 SHT: Delay Start? var2 <99>=<>, max10s?
20:12:35.589 SRC: Shutter
20:12:35.592 SHT: Skipping switch off relay 0
20:12:35.597 MQT: stat/Rollos_Wozi_S/RESULT = {"POWER4":"off"}
20:12:35.600 MQT: stat/Rollos_Wozi_S/POWER4 = off
20:12:35.702 SRC: Shutter
20:12:35.712 MQT: stat/Rollos_Wozi_S/RESULT = {"POWER3":"on"}
20:12:35.716 MQT: stat/Rollos_Wozi_S/POWER3 = on
20:12:35.717 SHT: Shtr2 Real 0, Start 0, Stop 1720000, Dir 1, Delay 40, Rtc 0.00 [s], Freq 0, PWM 0, Tilt 0
20:12:35.723 MQT: stat/Rollos_Wozi_S/RESULT = {"Shutter2":{"Position":100,"Direction":1,"Target":0,"Tilt":0}}
20:12:35.792 SHT: Shtr1 Time 40(4), cStop 10500, cVelo 500, mVelo 500, aVelo 0, mRun 0, aPos 29480, nStop 39980, Trgt 1360000, mVelo 12, Dir 1, Tilt 0, TrgtTilt: 0, Tiltmove: 0
20:12:35.794 SHT: Shtr2 Time 1(1), cStop 0, cVelo 12, mVelo 500, aVelo 12, mRun 0, aPos 0, nStop 0, Trgt 1720000, mVelo 12, Dir 1, Tilt 0, TrgtTilt: 0, Tiltmove: 0
20:12:35.873 SHT: Shtr1 Time 42(1), cStop 10500, cVelo 500, mVelo 500, aVelo 0, mRun 100, aPos 31480, nStop 41980, Trgt 1360000, mVelo 12, Dir 1, Tilt 0, TrgtTilt: 0, Tiltmove: 0
20:12:35.875 SHT: Shtr2 Time 3(1), cStop 72, cVelo 36, mVelo 500, aVelo 12, mRun 100, aPos 0, nStop 72, Trgt 1720000, mVelo 12, Dir 1, Tilt 0, TrgtTilt: 0, Tiltmove: 0
20:12:35.921 SHT: Shtr1 Time 43(0), cStop 10500, cVelo 500, mVelo 500, aVelo 0, mRun 150, aPos 32480, nStop 42980, Trgt 1360000, mVelo 12, Dir 1, Tilt 0, TrgtTilt: 0, Tiltmove: 0
[... cut ...]
20:13:07.655 SHT: Velocity 41, Delta -27
20:13:07.657 SHT: Real 512680, Pulsecount 12817, tobe 12570, Start 0
20:13:07.659 SRC: Shutter
20:13:07.670 MQT: stat/Rollos_Wozi_S/RESULT = {"POWER7":"off"}
20:13:07.673 MQT: stat/Rollos_Wozi_S/POWER7 = off
20:13:07.675 SHT: Shtr4 Real 512680, Start 512680, Stop 502800, Dir 0, Delay 20, Rtc 26.60 [s], Freq 0, PWM 0, Tilt 0
20:13:07.681 MQT: stat/Rollos_Wozi_S/SHUTTER4 = 38
20:13:07.685 MQT: stat/Rollos_Wozi_S/RESULT = {"Shutter4":{"Position":62,"Direction":0,"Target":63,"Tilt":0}}

stefanbode, 2023-10-18:
You can ignore last message. Will roll back and fix it differently.

stefanbode, 2023-10-18:
committed a new release: #19791
as you mentioned the storage did not work and I now reverted back most of the functionality and just moved over the storage issue and some general fixes on disabling file access during shutter movement. #19791 is waiting

stefanbode, 2023-10-18:
`20:12:35.586 SHT: Use channel 6` This is already strange because on 4 shutters this number should never be >4. Anyhow removed all the function here. Therefore no error at this stage hopefully anymore

DaBaBW, 2023-10-19:
Seems to work fine, thank you!!

PR #19717 (2023-10-11): Fix shutterfrequency esp32 for stepper motors
## Description:

**Related issue (if applicable):** fixes #<Tasmota issue number goes here>
fixes #19686

## Checklist:
  - [x] The pull request is done against the latest development branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR and the code change compiles without warnings
  - [x] The code change is tested and works with Tasmota core ESP8266 V.2.7.4.9
  - [x] The code change is tested and works with Tasmota core ESP32 V.2.0.14
  - [x] I accept the [CLA](https://github.com/arendst/Tasmota/blob/develo [... cut]
PR #19791 (2023-10-18): reverted back some function and fix frequency storage
## Description:
reverted back some function and fix frequency storage
**Related issue (if applicable):** fixes #<Tasmota issue number goes here>
fix #19686
## Checklist:
  - [x] The pull request is done against the latest development branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR and the code change compiles without warnings
  - [x] The code change is tested and works with Tasmota core ESP8266 V.2.7.4.9
  - [x] The code change is tested and works with Tasmota core ESP32 V.2.0.14
  - [x] I accep [... cut]