arendst/Tasmota #19696: 上电之后，继电器跳动

maxadc, 2023-10-07:
使用esp8266 
IO配置如下
[image]
固件版本 | 13.1.0(tasmota)
ESP 芯片 ID | 591779 (ESP8266EX)
Flash 芯片 |ID 0x146085 (DOUT)

控制及联网正常。遇到一个问题，那就是，上电之后，继电器会跳动，应该是，固件初始化时输出了控制动作！这是怎么回事

closed 2023-10-07 by maxadc

hikevin117, 2023-10-29:
I have the same problem, and there is currently no solution. This is not suitable for use on gate, it will cause danger.

barbudor, 2023-10-29:
And what is actually the problem ?

hikevin117, 2023-10-29:
Thank you for your answer. 

I set commands 
poweronstate was  0 or 3  ==> Not resolved
savedata 1 ==> Not resolved

I also recorded a video of the problem. You can take a look at the voltage behavior when the problem occurred.

https://mega.nz/file/0kh3GDLJ#em_ee0aKOu3nV-4NdHrYSJsaZU3SBdJxL53k6JTQs9E