arendst/Tasmota #19733: Modbusbridge causes memory leak

jeroenst, 2023-10-13:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
When using modbusbridge Tasmota gets out of memory and becomes unreachable. 

It started after upgrading to 13.0 but back on 12.5 the issue remains. 

I'm already investigating it and trying to find the leak. 

PR will follow when fixed. 

[image]

I also have another device running with Modbus bridge on 12.5.1 without memory leak. 

I suspect writing registers is causing this but it is still under investigation. 

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic):  Astra Rocketcontroller 
- [x] Tasmota binary firmware version number used: 13.1
  - [x] Pre-compiled 13.1, 12.5
  - [x] Self-compiled 12.5
- [x] Flashing tools used: website
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Poll and write to Modbus device using mqtt commands

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
No decrease of free heap

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2023-10-13:
Discussed on Discord:
https://discord.com/channels/479389167382691863/490244847568158751/1162097276609626153

jeroenst, 2023-10-14:
The leak only happens when writing data to the Modbus device registers.

I searched tasmota/tasmota_xdrv_driver/xdrv_63_modbus_bridge.ino and lib/lib_basic/TasmotaModbus-3.6.0/src/TasmotaModbus.cpp but every malloc is followed by a free, I can't find an exception on that.

Could fragmentation cause such a big memory occupation?

jeroenst, 2023-10-14:
Also on the latest commit from this morning, the memory leak is still occurring.

jeroenst, 2023-10-15:
I did several changes and now don't have the memory leak anymore. I suspect that the bytes reserved in the malloc at https://github.com/arendst/Tasmota/blob/715b64dc2821f4cb6e03404118ec49b8afa33cb0/tasmota/tasmota_xdrv_driver/xdrv_63_modbus_bridge.ino#L903 has to be doubled. 

Also ModbusTCP suffers from this I suppose:
https://github.com/arendst/Tasmota/blob/715b64dc2821f4cb6e03404118ec49b8afa33cb0/tasmota/tasmota_xdrv_driver/xdrv_63_modbus_bridge.ino#L699

jeroenst, 2023-10-15:
That seems to work without any leak. I also added some other memory related fixes that can cause leaking which I'm testing this night. If results are good, I will send the PR tomorrow.

jeroenst, 2023-10-16:
Tested the fix this night and no memory leak occurs anymore.

[image]

Thanks to @Rolandwalraven for discussing about the wrong memory allocation and to the Tasmota team for their great work.

PR #19758 (2023-10-16): ModbusBridge write memleak fix (19733)
## Description:

**Related issue (if applicable):** fixes #19733 

## Checklist:
  - [x] The pull request is done against the latest development branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR and the code change compiles without warnings
  - [ ] The code change is tested and works with Tasmota core ESP8266 V.2.7.4.9
  - [x] The code change is tested and works with Tasmota core ESP32 V.2.0.14
  - [x] I accept the [CLA](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md#contributor-license-agre [... cut]
closed 2023-10-16 by arendst