arendst/Tasmota #19735: Sonoff ZbBridgePro: Zigbee net not starting

pottigotti, 2023-10-14:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
After flashing a Sonoff ZBBridgePro with Tasmota binary (tasmota32-zbbrdgpro.bin) inclusive firmware upgrade and auto-configuration with ZBBridge pro the zigbee net is not starting.
I tried this with two Sonoff devices. The issue is the same.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): Sonoff ZigBee Bridge Pro
- [ ] Tasmota binary firmware version number used: 13.1.0
  - [ X] Pre-compiled
  - [ ] Self-compiled
- [ X] Flashing tools used: esp_flasher on ubuntu
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
10:00:02.923 MQT: tele/tasmota_8997E4/INFO3 = {"Info3":{"RestartReason":"Software reset CPU","BootCount":7}} 

10:00:02.940 TFS: File 'autoexec.bat' not found 

10:00:04.002 SRC: MQTT 

10:00:04.003 CMD: Grp 0, Cmd 'STATUS', Idx 1, Len 1, Pld 1, Data '1' 

10:00:04.010 MQT: stat/tasmota_8997E4/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"https://ota.tasmota.com/tasmota32/tasmota32-zbbrdgpro.bin","RestartReason":"Software reset CPU","Uptime":"0T00:00:06","StartupUTC":"2023-10-14T08:59:58","Sleep":50,"CfgHolder":4617,"BootCount":6,"BCResetTime":"2023-10-14T08:41:07","SaveCount":19}} 

10:00:04.055 SRC: MQTT 

10:00:04.057 CMD: Grp 0, Cmd 'STATUS', Idx 1, Len 2, Pld 11, Data '11' 

10:00:04.061 BRY: GC from 6299 to 3717 bytes, objects freed 28/38 (in 1 ms) - slots from 81/122 to 41/91 

10:00:04.080 MQT: stat/tasmota_8997E4/STATUS11 = {"StatusSTS":{"Time":"2023-10-14T10:00:04","Uptime":"0T00:00:06","UptimeSec":6,"Heap":162,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":26,"MqttCount":1,"Berry":{"HeapUsed":3,"Objects":38},"Wifi":{"AP":1,"SSId":"FRITZ!Powerline 546E","BSSId":"E8:DF:70:1C:40:A0","Channel":11,"Mode":"11n","RSSI":100,"Signal":-43,"LinkCount":1,"Downtime":"0T00:00:03"}}} 

10:00:05.780 QPC: Reset 

10:00:06.623 WIF: IPv6 Global 2001:9e8:d80b:6b00:3ee9:eff:fe89:97e4 

10:00:06.783 BRY: GC from 4079 to 3717 bytes, objects freed 1/38 (in 1 ms) - slots from 41/91 to 41/91 

10:00:06.791 MQT: tele/tasmota_8997E4/STATE = {"Time":"2023-10-14T10:00:06","Uptime":"0T00:00:08","UptimeSec":8,"Heap":160,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Berry":{"HeapUsed":3,"Objects":38},"Wifi":{"AP":1,"SSId":"FRITZ!Powerline 546E","BSSId":"E8:DF:70:1C:40:A0","Channel":11,"Mode":"11n","RSSI":100,"Signal":-41,"LinkCount":1,"Downtime":"0T00:00:03"}} 

10:00:07.794 APP: Boot Count 7 

10:00:08.458 CFG: Saved, Count 20, Bytes 4096 

10:00:11.699 HTP: Consoles 

10:00:11.813 MQT: tasmota/discovery/3CE90E8997E4/config = {"ip":"192.168.178.79","dn":"Tasmota","fn":["Tasmota",null,null,null,null,null,null,null],"hn":"tasmota-8997E4-6116","mac":"3CE90E8997E4","md":"Sonoff Zigbee Pro","ty":0,"if":0,"ofln":"Offline","onln":"Online","state":["OFF","ON","TOGGLE","HOLD"],"sw":"13.1.0.4","t":"tasmota_8997E4","ft":"%prefix%/%topic%/","tp":["cmnd","stat","tele"],"rl":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"swc":[-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1],"swn":[null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null],"btn":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"so":{"4":0,"11":0,"13":0,"17":0,"20":0,"30":0,"68":0,"73":0,"82":0,"114":0,"117":0},"lk":0,"lt_st":0,"bat":0,"dslp":0,"sho":[],"sht":[],"ver":1} (retained) 

10:00:11.896 MQT: tasmota/discovery/3CE90E8997E4/sensors = {"sn":{"Time":"2023-10-14T10:00:11"},"ver":1} (retained) 

10:00:14.652 ZIG: rebooting ZNP device 

10:00:15.652 ZbInput discarding byte 00 

10:00:15.654 ZIG: {"ZbZNPReceived":"4180000200020712"} 

10:00:15.662 MQT: tele/tasmota_8997E4/RESULT = {"ZbState":{"Status":1,"Message":"CCxxxx ZNP booted","RestartReason":"Power-up","MajorRel":2,"MinorRel":7}} 

10:00:15.802 ZIG: checking device configuration 

10:00:15.804 ZIG: ZbZNPSent 2102 

10:00:15.853 ZIG: {"ZbZNPReceived":"610202000207120000000000"} 

10:00:15.860 MQT: tele/tasmota_8997E4/RESULT = {"ZbState":{"Status":50,"MajorRel":2,"MinorRel":7,"MaintRel":18,"Revision":0}} 

10:00:15.864 ZIG: ZbZNPSent 2108600000 

10:00:15.903 ZIG: {"ZbZNPReceived":"6108000155"} 

10:00:15.905 ZIG: ZbZNPSent 2108870000 

10:00:15.953 ZIG: {"ZbZNPReceived":"6108000100"} 

10:00:15.955 ZIG: ZbZNPSent 2108830000 

10:00:16.003 ZIG: {"ZbZNPReceived":"610800023C29"} 

10:00:16.005 ZIG: ZbZNPSent 21082D0000 

10:00:16.053 ZIG: {"ZbZNPReceived":"610800083CE90E89CCCCCCCC"} 

10:00:16.055 ZIG: ZbZNPSent 2108840000 

10:00:16.103 ZIG: {"ZbZNPReceived":"6108000400080000"} 

10:00:16.106 ZIG: ZbZNPSent 2108630000 

10:00:16.153 ZIG: {"ZbZNPReceived":"6108000100"} 

10:00:16.154 ZIG: ZbZNPSent 2108620000 

10:00:16.203 ZIG: {"ZbZNPReceived":"61080200"} 

10:00:16.212 MQT: tele/tasmota_8997E4/RESULT = {"ZbState":{"Status":2,"Message":"Resetting configuration"}} 

10:00:16.216 ZIG: ZbZNPSent 21090300000103 

10:00:16.253 ZIG: {"ZbZNPReceived":"610900"} 

10:00:16.255 ZIG: ZbZNPSent 410000 

10:00:16.706 HTP: Main Menu 

10:00:17.491 ZbInput discarding byte 00 

10:00:17.494 ZIG: {"ZbZNPReceived":"4180000200020712"} 

10:00:17.495 ZIG: ZbZNPSent 2109830000023C29 

10:00:17.541 ZIG: {"ZbZNPReceived":"610900"} 

10:00:17.544 ZIG: ZbZNPSent 21092D0000083CE90E89CCCCCCCC 

10:00:17.591 ZIG: {"ZbZNPReceived":"610900"} 

10:00:17.593 ZIG: ZbZNPSent 21098400000400080000 

10:00:17.641 ZIG: {"ZbZNPReceived":"610900"} 

10:00:17.644 ZIG: ZbZNPSent 21098700000100 

10:00:17.691 ZIG: {"ZbZNPReceived":"610900"} 

10:00:17.694 ZIG: ZbZNPSent 2109620000100000E4973CE90E890000E4973CE90E89 

10:00:17.741 ZIG: {"ZbZNPReceived":"610900"} 

10:00:17.744 ZIG: ZbZNPSent 21096300000100 

10:00:17.791 ZIG: {"ZbZNPReceived":"610900"} 

10:00:17.794 ZIG: ZbZNPSent 210901010020FFFFFFFFFFFFFFFF5A6967426565416C6C69616E636530390000000000000000 

10:00:17.873 ZIG: {"ZbZNPReceived":"610902"} 

10:00:17.875 ZIG: ZbZNPSent 21098F00000101 

10:00:17.922 ZIG: {"ZbZNPReceived":"610900"} 

10:00:17.925 ZIG: ZbZNPSent 2107600001000100 

10:00:17.972 ZIG: {"ZbZNPReceived":"610700"} 

10:00:17.975 ZIG: ZbZNPSent 21096000000155 

10:00:18.022 ZIG: {"ZbZNPReceived":"610900"} 

10:00:18.025 ZIG: ZbZNPSent 2108620000 

10:00:18.072 ZIG: {"ZbZNPReceived":"61080200"} 

10:00:18.081 MQT: tele/tasmota_8997E4/RESULT = {"ZbState":{"Status":99,"Message":"Abort"}} 

10:00:18.085 ZIG: Abort 

10:00:18.086 ZIG: Stopping (99) 

10:00:19.997 HTP: Consoles 

10:00:21.548 WIF: Checking connection... 

10:00:21.799 HTP: Console 

10:00:41.563 WIF: Checking connection... 

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Trying a second device yielded the same.
It's interesting that the behaviour and log output is in principle the same, but for the first device the zigbee LED was on all the time. For the second device it only flashed while trying to start.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
My expectation was a running ZigBee network and active buttons in the Tasmota Menu.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2023-10-14:
Have you flashed the changed Zigbee MCU firmware?

Jason2866, 2023-10-14:
We have considered, since it is non reversible we decided against. So it is never done by "accident"