arendst/Tasmota #19757: Flash of Shelly plus 2PM works, but after waiting some minutes, the ESP32 has the state after flashing (reverting all configuration steps)

gon0, 2023-10-15:
### PROBLEM DESCRIPTION
I flashed a shelly plus 2PM with the latest tasmota32solo1.bin 
I configure the shelly
After 30 seconds to 2 minutes, all the configuration is lost, the ESP32 is not connected to my WiFi anymore, the AP is open = state after flashing is reverted.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues) Closest I could find is: https://github.com/arendst/Tasmota/issues/19426
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): Shelly Plus 2PM:
[image]

- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [X] Flashing tools used: Tasmota Web Installer
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
Console just before self-erase:
00:00:00.002 HDW: ESP32-U4WDH-D 
00:00:00.011 UFS: FlashFS mounted with 304 kB free
00:00:00.020 CFG: Loaded from File, Count 6
00:00:00.024 QPC: Count 1
00:00:00.033 I2C: Bus1 using GPIO25(SCL) and GPIO26(SDA)
00:00:00.039 TFS: File 'mcp23x.dat' not found
00:00:00.088 BRY: Berry initialized, RAM used=4438 bytes
00:00:00.106 CFG: Energy loaded from file
00:00:00.209 I2C: ADE7953 found at 0x38
00:00:00.211 TFS: File 'calib.dat' not found
00:00:00.212 NRG: Init driver 7
00:00:00.220 Project tasmota - Tasmota_plus2PM_Steffi_PC Version 13.1.0(solo1)-2_0_11(2023-08-15T08:02:07)
00:00:00.947 WIF: Connecting to AP1 Heizung Channel 1 BSSId E0:28:6D:5E:D7:FE in mode 11n as tasmota-882F70-3952...
00:00:02.456 WIF: Connected
00:00:02.716 HTP: Web server active on tasmota-882F70-3952 with IP address 192.168.122.238
23:10:55.627 MQT: Attempting connection...
23:10:55.655 MQT: Connected
23:10:55.667 MQT: tele/tasmota_882F70/LWT = Online (retained)
23:10:55.676 MQT: cmnd/tasmota_882F70/POWER = 
23:10:55.703 MQT: tele/tasmota_882F70/INFO1 = {"Info1":{"Module":"Shelly Plus 2PM PCB v0.1.9","Version":"13.1.0(solo1)single-core","FallbackTopic":"cmnd/DVES_882F70_fb/","GroupTopic":"cmnd/tasmotas/"}}
23:10:55.727 MQT: tele/tasmota_882F70/INFO2 = {"Info2":{"WebServerMode":"Admin","Hostname":"tasmota-882F70-3952","IPAddress":"192.168.122.238","IP6Global":"","IP6Local":"fe80::c249:efff:fe88:2f70"}}
23:10:55.753 MQT: tele/tasmota_882F70/INFO3 = {"Info3":{"RestartReason":"Software reset CPU","BootCount":2}}
23:10:55.764 MQT: stat/tasmota_882F70/RESULT = {"POWER1":"OFF"}
23:10:55.769 MQT: stat/tasmota_882F70/POWER1 = OFF
23:10:55.780 MQT: stat/tasmota_882F70/RESULT = {"POWER2":"OFF"}
23:10:55.789 MQT: stat/tasmota_882F70/POWER2 = OFF
23:10:56.829 MQT: stat/tasmota_882F70/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32solo1.bin","RestartReason":"Software reset CPU","Uptime":"0T00:00:05","StartupUTC":"2023-10-15T22:10:51","Sleep":50,"CfgHolder":4617,"BootCount":1,"BCResetTime":"2023-10-15T22:35:40","SaveCount":6}}
23:10:56.988 MQT: stat/tasmota_882F70/STATUS11 = {"StatusSTS":{"Time":"2023-10-15T23:10:56","Uptime":"0T00:00:05","UptimeSec":5,"Heap":131,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":66,"MqttCount":1,"Berry":{"HeapUsed":4,"Objects":46},"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"Heizung","BSSId":"E0:28:6D:5E:D7:FE","Channel":1,"Mode":"11n","RSSI":82,"Signal":-59,"LinkCount":1,"Downtime":"0T00:00:03"}}}
23:10:57.066 MQT: stat/tasmota_882F70/RESULT = {"Time":"2023-10-15T23:10:57","Uptime":"0T00:00:06","UptimeSec":6,"Heap":132,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":78,"MqttCount":1,"Berry":{"HeapUsed":4,"Objects":46},"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"Heizung","BSSId":"E0:28:6D:5E:D7:FE","Channel":1,"Mode":"11n","RSSI":84,"Signal":-58,"LinkCount":1,"Downtime":"0T00:00:03"}}
23:10:57.130 MQT: stat/tasmota_882F70/STATUS10 = {"StatusSNS":{"Time":"2023-10-15T23:10:57","ANALOG":{"Temperature1":60.5},"ENERGY":{"TotalStartTime":"2023-10-15T14:16:48","Total":0.001,"Yesterday":0.000,"Today":0.001,"Power":[0,0],"ApparentPower":[0,0],"ReactivePower":[0,0],"Factor":[0.00,0.00],"Voltage":0,"Current":[0.000,0.000]},"TempUnit":"C"}}
23:10:58.996 QPC: Reset
23:11:00.020 MQT: tele/tasmota_882F70/STATE = {"Time":"2023-10-15T23:11:00","Uptime":"0T00:00:09","UptimeSec":9,"Heap":132,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":39,"MqttCount":1,"Berry":{"HeapUsed":4,"Objects":46},"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"Heizung","BSSId":"E0:28:6D:5E:D7:FE","Channel":1,"Mode":"11n","RSSI":86,"Signal":-57,"LinkCount":1,"Downtime":"0T00:00:03"}}
23:11:00.079 MQT: tele/tasmota_882F70/SENSOR = {"Time":"2023-10-15T23:11:00","ANALOG":{"Temperature1":60.1},"ENERGY":{"TotalStartTime":"2023-10-15T14:16:48","Total":0.001,"Yesterday":0.000,"Today":0.001,"Period":[0,0],"Power":[0,0],"ApparentPower":[0,0],"ReactivePower":[0,0],"Factor":[0.00,0.00],"Voltage":0,"Current":[0.000,0.000]},"TempUnit":"C"}
23:11:28.781 UFS: File '/' download
23:11:38.104 MQT: stat/tasmota_882F70/RESULT = {"Reset":"Reset and Restarting"}
23:11:38.128 MQT: stat/tasmota_882F70/RESULT = {"Reset":"Reset and Restarting"}
23:11:38.204 CFG: Use defaults

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
23:14:16.698 CMD: Status 0
23:14:16.705 MQT: stat/tasmota_882F70/STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota_plus2PM_Steffi_PC","FriendlyName":["Tasmota_plus2PM_Steffi_PC",""],"Topic":"tasmota_882F70","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[3,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
23:14:16.750 MQT: stat/tasmota_882F70/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32solo1.bin","RestartReason":"Software reset CPU","Uptime":"0T00:00:15","StartupUTC":"2023-10-15T22:14:01","Sleep":50,"CfgHolder":4617,"BootCount":2,"BCResetTime":"2023-10-15T22:35:40","SaveCount":7}}
23:14:16.794 MQT: stat/tasmota_882F70/STATUS2 = {"StatusFWR":{"Version":"13.1.0(solo1)single-core","BuildDateTime":"2023-08-15T08:02:07","Core":"2_0_11","SDK":"4.4.5.230722","CpuFrequency":240,"Hardware":"ESP32-U4WDH-D rev.3","CR":"484/699"}}
[... 10 lines cut ...]
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
Difficult to do, because after the issue, the blank tasmota32solo1 configuration is present

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
1. flash shelly plus 2PM via web installer with tasmota32solo1.bin
2. configure WiFi
3. configure MQTT server
4. wait 30 seconds to 2 minutes
5. the ESP32 is in the state of step one and accessible only via AP

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
The ESP32 should keep my configuration.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
Is it possible to upload the .dmp-file (Restore) without WiFi password?

sfromis, 2023-10-15:
Getting `{"Reset":"Reset and Restarting"}` 40 seconds after boot indicates that a Button pin was registering as constantly pushed.  However, as you did not supply the requested information about the template config, hard to see exactly what you did. It could be bad template, or something wrong about handling of the switching inputs or on-device button.

Did you take care to select the right one of the two templates on the page, depending on board revision?

gon0, 2023-10-16:
Thank you for investigation and you quick answer!

I used the template for v0.1.9 ({"NAME":"Shelly Plus 2PM PCB v0.1.9","GPIO":[320,0,0,0,34,192,0,0,225,224,0,0,0,0,193,0,0,0,0,0,0,608,640,3458,0,0,0,0,0,9472,0,4736,0,0,0,0],"FLAG":0,"BASE":1}) according to the board present.

Good point! regarding Buttons:
I have connected two N.O. buttons to the S1 and S2 pin of the shelly plus 2PM. In closed state (while being pressed) they connect L to the S1 or S2 input, respectively. In open state (not pushed), the voltage between N and S1 or S2 drops from 50V~ near 20V~ over a minute or two.

When GPIO5 and GPIO18 is defined as "Button 1" and "Button 2", as is per default, the buttons do not switch the relays, see here:
[image]

My aim is that each button press (rising edge; L connected to input) toggles the respective relay.

The "reverting to state after flash"-behavior occurs, when I define GPIO5 and GPIO18 as "Button_n 1" and "Button_n 2", Here, the relays work on every second release. 

I set both buttons to "Button_d" with the result, that each second button press toggles the relay:

```
06:05:49.999 QPC: Reset
06:05:51.009 MQT: tele/tasmota_882F70/STATE = {"Time":"2023-10-16T06:05:50","Uptime":"0T00:00:08","UptimeSec":8,"Heap":133,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":35,"MqttCount":1,"Berry":{"HeapUsed":4,"Objects":46},"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"Heizung","BSSId":"E0:28:6D:5E:D7:FE","Channel":1,"Mode":"11n","RSSI":82,"Signal":-59,"LinkCount":1,"Downtime":"0T00:00:03"}}
06:05:51.057 MQT: tele/tasmota_882F70/SENSOR = {"Time":"2023-10-16T06:05:51","ANALOG":{"Temperature1":41.2},"ENERGY":{"TotalStartTime":"2023-10-15T14:16:48","Total":0.001,"Yesterday":0.000,"Today":0.000,"Period":[0,0],"Power":[0,0],"ApparentPower":[0,0],"ReactivePower":[0,0],"Factor":[0.00,0.00],"Voltage":0,"Current":[0.000,0.000]},"TempUnit":"C"}
06:05:57.365 MQT: stat/tasmota_882F70/RESULT = {"POWER1":"ON"}
06:05:57.378 MQT: stat/tasmota_882F70/POWER1 = ON
06:06:01.644 MQT: stat/tasmota_882F70/RESULT = {"POWER2":"ON"}
06:06:01.658 MQT: stat/tasmota_882F70/POWER2 = ON
06:06:04.838 MQT: stat/tasmota_882F70/RESULT = {"POWER2":"OFF"}
06:06:04.844 MQT: stat/tasmota_882F70/POWER2 = OFF
06:06:09.230 MQT: stat/tasmota_882F70/RESULT = {"POWER1":"OFF"}
06:06:09.242 MQT: stat/tasmota_882F70/POWER1 = OFF
06:06:13.081 MQT: stat/tasmota_882F70/RESULT = {"POWER1":"ON"}
06:06:13.090 MQT: stat/tasmota_882F70/POWER1 = ON
06:06:15.719 MQT: stat/tasmota_882F70/RESULT = {"POWER1":"OFF"}
06:06:15.732 MQT: stat/tasmota_882F70/POWER1 = OFF
06:06:18.537 MQT: stat/tasmota_882F70/RESULT = {"POWER2":"ON"}
06:06:18.550 MQT: stat/tasmota_882F70/POWER2 = ON
```

Also here, Button_n, Button_d and Button_i cause the "reverting to state after flash"-behavior.

When I changed GPIO5 and GPIO18 to "Button_in", the relays are toggled on each button release. For me, this behavior is fine.

When I then tried to send an additional "HOLD" via MQTT, I get this behavior on long-pressing the button:
```

06:21:30.748 CMD: SetOption32
06:21:30.755 MQT: stat/tasmota_882F70/RESULT = {"SetOption32":40}
06:21:35.862 CMD: SetOption32 4
06:21:35.869 MQT: stat/tasmota_882F70/RESULT = {"SetOption32":4}
06:21:47.877 MQT: stat/tasmota_882F70/RESULT = {"Reset":"Reset and Restarting"}
06:21:48.217 CFG: Use defaults
```

I could not find a list of what "Button_d" means (here https://tasmota.github.io/docs/Components/#tasmota I only find information about Button, Button_i, Button_n and Button_in.

I'll close this issue, since, you helped me getting on the right path with the button-settings. Thank you very much! :)

closed 2023-10-16 by gon0

barbudor, 2023-10-16:
As per the template pages, gpio 5 and 18 should be defined as switches, not buttons
By changing that you put yourselves in those water
https://templates.blakadder.com/shelly_plus_2PM.html

Revert back to the official template and then read the Switchmode section of the doc to see how to get the proper behavior from your type of switch (momentary or rocker)