arendst/Tasmota #19760: Device has become unavailable, seems to think its something else

cmjennnings21, 2023-10-16:
### PROBLEM DESCRIPTION
One of my devices has become unresponsive, when I look into the dashboard it the device is red, as if offline. When navigating to the IP address of the device its asking to login to MyEnergi, I do have this integration in my HA setup, its for my EV charger

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x ] Device used (e.g., Sonoff Basic): Sonoff Mini (Early version)
- [x ] Tasmota binary firmware version number used: Can't see in the dashboard
  - [x ] Pre-compiled Yes
  - [ ] Self-compiled
- [x ] Flashing tools used: Can't recall, this has been working for years and updating the FW hasn't been an issue
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
[image]

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2023-10-16:
Looks like your device was "factory reset", can happen as a result of "pulsing" power outages, as this can trigger a recovery feature. Connect to the AP it is broadcasting, and reconfigure the device. You may want to consider [SetOption65](https://tasmota.github.io/docs/Commands/#setoption65).

barbudor, 2023-10-16:
The point is that we have NO clues of what you are doing with Tasmota in relation to you EV charger
There is no description of your Tasmota setup