arendst/Tasmota #19763: DS18alias wrong sensor match

spcqike, 2023-10-16:
### PROBLEM DESCRIPTION
ds18alias doesnt work if multiple sensors are connected, the sensors get the wrong name

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): generic esp8266
- [ ] Tasmota binary firmware version number used: 13.1.0
  - [ ] Pre-compiled
  - [ ] Self-compiled  - self compiled with sml, ds18alias and max6675 sensor
- [ ] Flashing tools used: tasmotiser
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
14:12:09.933 MQT: stat/tasmota_heizung/TEMPLATE = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
14:12:10.147 MQT: stat/tasmota_heizung/MODULE = {"Module":{"0":"Generic"}}
14:12:10.359 MQT: stat/tasmota_heizung/GPIO = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"1313":"DS18x202"},"GPIO5":{"1312":"DS18x201"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"2560":"MX31855 DO"},"GPIO13":{"0":"None"},"GPIO14":{"2528":"MX31855 CLK"},"GPIO15":{"2496":"MX31855 CS"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
no rules

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
14:13:10.357 CMD: status 0
14:13:10.365 MQT: stat/tasmota_heizung/STATUS = {"Status":{"Module":0,"DeviceName":"Heizung","FriendlyName":["Heizung"],"Topic":"tasmota_heizung","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
14:13:10.402 MQT: stat/tasmota_heizung/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"","RestartReason":"External System","Uptime":"0T00:08:28","StartupUTC":"2023-10-16T12:04:42","Sleep":50,"CfgHolder":4617,"BootCount":63,"BCResetTime":"2023-09-26T13:12:38","SaveCount":81,"SaveAddress":"3FB000"}}
14:13:10.431 MQT: stat/tasmota_heizung/STATUS2 = {"StatusFWR":{"Version":"13.1.0(TasmoCompiler-esp82664M)","BuildDateTime":"2023.09.26 12:22:32","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"302/699"}}
14:13:10.453 MQT: stat/tasmota_heizung/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Hier_WLAN",""],"TelePeriod":10,"Resolution":"558180C0","SetOption":["00008019","2805C80001000600003C5A0A192800000000","00000080","00007000","00004000","00000000"]}}
14:13:10.496 MQT: stat/tasmota_heizung/STATUS4 = {"StatusMEM":{"ProgramSize":713,"Free":1332,"Heap":9,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"1640D8","FlashFrequency":40,"FlashMode":"DOUT","Features":["00000407","0780458E","042E0001","61F607C3","1A4C0004","D036F889","0009209C","00201000","5400E0C0","00008080"],"Drivers":"1,2,4,6,8,9,10,12,18,19,22,24,26,29,30,35,37,38,45,50,62,68","Sensors":"1,2,4,5,6,7,8,9,14,15,17,18,19,20,21,26,27,31,38,39,42,44,46,47,53,55,56,58,59,63,65,66,72,73,74,88,89,91,92,93,103","I2CDriver":"8,9,10,15,17,18,20,21,24,29,30,32,36,41,42,43,45,46,51,52,60,61,62,63,76"}}
14:13:10.548 MQT: stat/tasmota_heizung/STATUS5 = {"StatusNET":{"Hostname":"tasmota-heizung-5930","IPAddress":"192.168.8.187","Gateway":"192.168.8.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.8.1","DNSServer2":"0.0.0.0","Mac":"98:CD:AC:26:B7:2A","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
14:13:10.573 MQT: stat/tasmota_heizung/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.8.1","MqttPort":1883,"MqttClientMask":"Heizung","MqttClient":"Heizung","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
14:13:10.595 MQT: stat/tasmota_heizung/STATUS7 = {"StatusTIM":{"UTC":"2023-10-16T12:13:10","Local":"2023-10-16T14:13:10","StartDST":"2023-03-26T02:00:00","EndDST":"2023-10-29T03:00:00","Timezone":99,"Sunrise":"08:11","Sunset":"18:59"}}
14:13:10.619 MQT: stat/tasmota_heizung/STATUS10 = {"StatusSNS":{"Time":"2023-10-16T14:13:10","-4":{"Id":"3CE1E38139D4","Temperature":26.1},"abc":{"Id":"3CE1E381479B","Temperature":24.6},"-2":{"Id":"3CE1E381D7F2","Temperature":23.9},"-1":{"Id":"3CE1E381F942","Temperature":24.6},"MAX6675":{"Temperature":26.3,"ReferenceTemperature":null,"Error":0},"Öl":{"Count":16023.20},"TempUnit":"C"}}
14:13:10.652 MQT: stat/tasmota_heizung/STATUS11 = {"StatusSTS":{"Time":"2023-10-16T14:13:10","Uptime":"0T00:08:28","UptimeSec":508,"Heap":9,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":22,"MqttCount":1,"Wifi":{"AP":1,"SSId":"Hier_WLAN","BSSId":"D8:3A:DD:54:35:30","Channel":1,"Mode":"11n","RSSI":68,"Signal":-66,"LinkCount":1,"Downtime":"0T00:00:05"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

boot with 3 sensors (without weblog 4)
00:00:00.001 HDW: ESP8266EX
00:00:00.035 UFS: FlashFS mounted with 1976 kB free
00:00:00.106 CFG: Loaded from File, Anzahl 79
00:00:00.111 QPC: Count 1
00:00:00.119 SCR: ufilesystem found
00:00:00.149 SCR: nv=0, tv=0, vns=0, vmem=16, smem=8192
00:00:00.164 Projekt tasmota - Heizung Version 13.1.0(TasmoCompiler-esp82664M)-2_7_4_9(2023.09.26 12:22:32)
00:00:00.215 meters: 1 , decode lines: 1, memory used: 1033 bytes
00:00:00.805 WIF: verbinden mit AP1 Hier_WLAN Channel 1 BSSId D8:3A:DD:54:35:30 in Modus 11n als tasmota-heizung-5930...
00:00:01.758 WIF: verbunden
00:00:02.016 HTP: Web-Server aktiv bei tasmota-heizung-5930 mit IP-Adresse 192.168.8.187
14:04:21.044 MQT: Verbindungsversuch...
14:04:21.059 MQT: verbunden
14:04:21.062 MQT: tele/tasmota_heizung/LWT = Online (beibehalten)
14:04:21.065 MQT: cmnd/tasmota_heizung/POWER = 
14:04:21.071 MQT: tele/tasmota_heizung/INFO1 = {"Info1":{"Module":"Generic","Version":"13.1.0(TasmoCompiler-esp82664M)","FallbackTopic":"cmnd/Heizung_fb/","GroupTopic":"cmnd/tasmotas/"}}
14:04:21.081 MQT: tele/tasmota_heizung/INFO2 = {"Info2":{"WebServerMode":"Admin","Hostname":"tasmota-heizung-5930","IPAddress":"192.168.8.187"}}
14:04:21.094 MQT: tele/tasmota_heizung/INFO3 = {"Info3":{"RestartReason":"External System","BootCount":62}}
14:04:21.135 MQT: stat/tasmota_heizung/DS18ALIAS = {"DS18B20-1":{"Id":"A83CE1E38139D428"},"DS18B20-2":{"Id":"D43CE1E381D7F228"},"-1":{"Id":"363CE1E381479B28"}}
[... 69 lines cut ...]
put ds18alias in autoexec.bat
```
DS18Alias 363CE1E381479B28,-1
DS18Alias D43CE1E381D7F228,-2
DS18Alias 893CE1E381F94228,abc
ds18alias A83CE1E38139D428,-4
```

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
match the right ID to the right sensor

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
- see log

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
the firmware is self compiled with tasmocompiler, i try to build a oil heating monitor and want to monitor oil consumption (oil sensor as SML counter), air temperature (max6675) and up to 8 water temperatures (8 DS18B20, 2 inputs with 4 sensors each)

as you can see, when the last sensor (the third in my list, named "abc") is attached, it gets the ID "3CE1E381479B", which is the ID of my first sensor namend "-1".

as i now see in the logs, even with 3 sensors, the ID "3CE1E381479B" is wrongly mapped to sensor "-2". therefore, sensor "-2" has its real ID, when 4 sensors are connected.

**(Please, remember to close the issue when the problem has been addressed)**

barbudor, 2023-10-16:
I see no problems in the above logs

At the beginning you have
``` 
"DS18B20-1":{"Id":"A83CE1E38139D428"},
"DS18B20-2":{"Id":"893CE1E381F94228"},
"DS18B20-3":{"Id":"D43CE1E381D7F228"},
"DS18B20-4":{"Id":"363CE1E381479B28"}
```
Then you apply those commands:
```
DS18Alias 363CE1E381479B28,-1
DS18Alias D43CE1E381D7F228,-2
DS18Alias 893CE1E381F94228,abc
ds18alias A83CE1E38139D428,-4
```
At the end you get:
```
"-4":{"Id":"A83CE1E38139D428"},
"abc":{"Id":"893CE1E381F94228"},
"-2":{"Id":"D43CE1E381D7F228"},
"-1":{"Id":"363CE1E381479B28"}
```
Which is exactly what you have asked for

And in the sensor message:
```
"-4":{"Id":"3CE1E38139D4","Temperature":23.4},
"abc":{"Id":"3CE1E381479B","Temperature":23.2},
"-2":{"Id":"3CE1E381D7F2","Temperature":23.4},
"-1":{"Id":"3CE1E381F942","Temperature":23.6}
```

The only thing that I would say to be incorrect is your idea to name devices with a starting minus sign.
Using a minus sign in a JSON key is a bad idea, starting a key with a minus sign is even worse.

barbudor, 2023-10-16:
By heating the either sensor "-1" or "abc", the temperature match the Alias or the ID ?

spcqike, 2023-10-16:
it matches the id, so if i heat physical "abc", the temperature of alias "-1" rises.

barbudor, 2023-10-16:
Thanks, I will investigate tonight

spcqike, 2023-10-16:
Thanks for your time and effort. 

I want to mention, it’s not only that „abc“ and „-1“ are switched. If I remove „abc“ it switches „-2“ and „-1“ 

I remember that I noticed the problem with sensor 4, named „-4“, in the first place. After removing and re attaching it (I’m not sure but I think in this step I also changed its input Pin, as I need to use 2 GPIOs with 4 sensors each when it’s finished, now with this logs I only use one pin for all 4 sensors), the problem occurred like shown in the logs. 

The names are just placeholder until I know where the sensors are placed at the end. But until than I need to use numbers. And I would like to keep the number in the name somehow, so that I can match the numbered cables. „1_abc“ gives strange results in the UI and a comment like „ ;1“ isn’t possible. 

    ds18alias ID,abc ;1

is later shown as „abc ;1 Temperatur“. So I thought I use „-1“ for the beginning and „abc-1“ later, when I know where the sensor will be placed. 

I thought, as everything seems to be interpreted as string, „-1“ won’t make problems. Also, because „abc“ was „-3“ in the first place. And changing it to „abc“ hasn’t changed anything.

barbudor, 2023-10-16:
Fixed with PR above
Just wait for the merge

arendst, 2023-10-16:
Merged just now.

closed 2023-10-16 by arendst

barbudor, 2023-10-16:
thanks @arendst

sfromis, 2023-10-17:
Not sure if this changed just now, or is an older issue. But with 6 sensors connected, after a build with `#define DS18x20_USE_ID_ALIAS`, without having set any aliases, I'm seeing this:
[image]

In the JSON, the corresponding keys are empty, like `"":{"Id":"xxxxxxxxxxxx","Temperature":20.37}` meaning that it is invalid JSON with duplicate keys.

Of course, it helps to set the `DS18Alias` for each (confusingly not just with the `Id` value provided by Tasmota), but I do think that just having the feature included in the binary should default to old naming, until actually setting new aliases.

barbudor, 2023-10-17:
Let me check.
Indeed I tested only with aliases provided but clearly it should not change the default behavior

barbudor, 2023-10-17:
@sfromis I don't see the same as you do
Here is for 4 chips:
```
21:03:46.499 MQT: tele/nodemcu/SENSOR = {"Time":"2023-10-17T21:03:46+02:00","DS18B20-3":{"Id":"0000001549DE","Temperature":18.8},"DS18B20-4":{"Id":"000004716165","Temperature":19.2},"DS18B20-2":{"Id":"00000471A6BC","Temperature":19.2},"DS18B20-1":{"Id":"00000472022C","Temperature":19.3},"TempUnit":"C"} (retained)
```
Are you on ESP8266 or ESP32 or C3 ?

barbudor, 2023-10-17:
I only have 4 DS18, so can't test with more for now

sfromis, 2023-10-17:
My setup is ESP32-C3, having worked with sensors for some time now, but only today flashed with a binary built with current dev source, and the feature enabled. It did return to the default naming with `DS18Alias xxxx,0`.

barbudor, 2023-10-17:
Interresting
I don't think that C3 has a impact on that but you never used that ESP with DS18 Alias before ?
I'll try with a blank flash to see if it makes a difference

sfromis, 2023-10-17:
Changed my `autoexec.bat` to only have 3 of the sensors. When booting, it reverted to the behavior of empty names for the sensors now again without DS18Alias.

barbudor, 2023-10-17:
Ah ah
I tested without a filesystem using an on system init rule, so later than autoexec.bat
Need to try that

sfromis, 2023-10-17:
Also seeing the lack of default names on a "classic" ESP32 board, after boot DS18Alias lists default name for the first of the 6 sensors, and empty names for 2-5. But not when hooking up to an ESP8266, where it seemed to work.

spcqike, 2023-10-18:
wow, that was fast :) thanks for your time and effort.

short update on my site:

autoexec.bat (only 3 defined sensors, with 4 sensors connected)
> DS18Alias 363CE1E381479B28,-1
> DS18Alias D43CE1E381D7F228,-2
> DS18Alias 893CE1E381F94228,abc

log
```
10:27:45.039 MQT: verbunden
10:27:45.042 MQT: tele/tasmota_heizung/LWT = Online (beibehalten)
10:27:45.045 MQT: cmnd/tasmota_heizung/POWER = 
10:27:45.050 MQT: tele/tasmota_heizung/INFO1 = {"Info1":{"Module":"Generic","Version":"13.1.0.4(tasmota-4M)","FallbackTopic":"cmnd/Heizung_fb/","GroupTopic":"cmnd/tasmotas/"}}
10:27:45.059 MQT: tele/tasmota_heizung/INFO2 = {"Info2":{"WebServerMode":"Admin","Hostname":"tasmota-heizung-5930","IPAddress":"192.168.8.187"}}
10:27:45.072 MQT: tele/tasmota_heizung/INFO3 = {"Info3":{"RestartReason":"Software/System restart","BootCount":75}}
10:27:45.109 MQT: stat/tasmota_heizung/DS18ALIAS = {"DS18B20-1":{"Id":"A83CE1E38139D428"},"-1":{"Id":"893CE1E381F94228"},"DS18B20-3":{"Id":"D43CE1E381D7F228"},"DS18B20-2":{"Id":"363CE1E381479B28"}}
10:27:45.158 MQT: stat/tasmota_heizung/DS18ALIAS = {"DS18B20-1":{"Id":"A83CE1E38139D428"},"-1":{"Id":"893CE1E381F94228"},"-2":{"Id":"D43CE1E381D7F228"},"DS18B20-2":{"Id":"363CE1E381479B28"}}
10:27:45.207 MQT: stat/tasmota_heizung/DS18ALIAS = {"DS18B20-1":{"Id":"A83CE1E38139D428"},"-1":{"Id":"893CE1E381F94228"},"-2":{"Id":"D43CE1E381D7F228"},"abc":{"Id":"363CE1E381479B28"}}
10:27:48.227 QPC: Reset
10:27:49.153 MQT: tele/tasmota_heizung/STATE = {"Time":"2023-10-18T10:27:49","Uptime":"0T00:00:09","UptimeSec":9,"Heap":16,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"Hier_WLAN","BSSId":"D8:3A:DD:54:35:65","Channel":1,"Mode":"11n","RSSI":70,"Signal":-65,"LinkCount":1,"Downtime":"0T00:00:03"}}
10:27:49.181 MQT: tele/tasmota_heizung/SENSOR = {"Time":"2023-10-18T10:27:49","DS18B20-1":{"Id":"3CE1E38139D4","Temperature":22.4},"-1":{"Id":"3CE1E381479B","Temperature":22.1},"-2":{"Id":"3CE1E381D7F2","Temperature":22.5},"abc":{"Id":"3CE1E381F942","Temperature":22.7},"MAX6675":{"Temperature":24.0,"ReferenceTemperature":null,"Error":0},"Öl":{"Count":27435.32},"TempUnit":"C"}
```

naming in UI
[image]

if i heat "abc", "abc" gets warm. same for the other, even if i place one sensor at the second pin

but:
i looks like there is still some confusion.
the alias process doesnt match the IDs correctly. "-1" gets the ID of "abc" (which was the problem in the first place), but now it later switches back and uses the real ID for the sensors. so it works, with my esp8266 i also see the original standard name for not "aliased" sensors. 
```
DS18Alias 363CE1E381479B28,-1
DS18Alias D43CE1E381D7F228,-2
DS18Alias 893CE1E381F94228,abc
```
before one sensor was moved to another input (full log see above)
```
10:27:45.109 MQT: stat/tasmota_heizung/DS18ALIAS = {"DS18B20-1":{"Id":"A83CE1E38139D428"},"-1":{"Id":"893CE1E381F94228"},"DS18B20-3":{"Id":"D43CE1E381D7F228"},"DS18B20-2":{"Id":"363CE1E381479B28"}}
10:27:45.158 MQT: stat/tasmota_heizung/DS18ALIAS = {"DS18B20-1":{"Id":"A83CE1E38139D428"},"-1":{"Id":"893CE1E381F94228"},"-2":{"Id":"D43CE1E381D7F228"},"DS18B20-2":{"Id":"363CE1E381479B28"}}
10:27:45.207 MQT: stat/tasmota_heizung/DS18ALIAS = {"DS18B20-1":{"Id":"A83CE1E38139D428"},"-1":{"Id":"893CE1E381F94228"},"-2":{"Id":"D43CE1E381D7F228"},"abc":{"Id":"363CE1E381479B28"}}
```

after one sensor was moved
```
10:43:19.096 MQT: stat/tasmota_heizung/DS18ALIAS = {"DS18B20-2":{"Id":"D43CE1E381D7F228"},"-1":{"Id":"A83CE1E38139D428"},"DS18B20-1":{"Id":"893CE1E381F94228"},"DS18B20-3":{"Id":"363CE1E381479B28"}}
10:43:19.146 MQT: stat/tasmota_heizung/DS18ALIAS = {"DS18B20-2":{"Id":"D43CE1E381D7F228"},"-1":{"Id":"A83CE1E38139D428"},"-2":{"Id":"893CE1E381F94228"},"DS18B20-3":{"Id":"363CE1E381479B28"}}
10:43:19.195 MQT: stat/tasmota_heizung/DS18ALIAS = {"DS18B20-2":{"Id":"D43CE1E381D7F228"},"-1":{"Id":"A83CE1E38139D428"},"-2":{"Id":"893CE1E381F94228"},"abc":{"Id":"363CE1E381479B28"}}
10:43:22.227 QPC: Reset
10:43:23.189 MQT: tele/tasmota_heizung/STATE = {"Time":"2023-10-18T10:43:23","Uptime":"0T00:00:09","UptimeSec":9,"Heap":15,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"Hier_WLAN","BSSId":"D8:3A:DD:54:35:65","Channel":1,"Mode":"11n","RSSI":70,"Signal":-65,"LinkCount":1,"Downtime":"0T00:00:03"}}
10:43:23.219 MQT: tele/tasmota_heizung/SENSOR = {"Time":"2023-10-18T10:43:23","DS18B20-2":{"Id":"3CE1E38139D4","Temperature":23.5},"-1":{"Id":"3CE1E381479B","Temperature":22.4},"-2":{"Id":"3CE1E381D7F2","Temperature":23.6},"abc":{"Id":"3CE1E381F942","Temperature":23.8},"MAX6675":{"Temperature":24.8,"ReferenceTemperature":null,"Error":0},"Öl":{"Count":27415.48},"TempUnit":"C"}
````

one thing to add:

> Indeed I tested only with aliases provided but clearly it should not change the default behavior

if it would be written in the documentation, it wouldn't be a bug, but a feature :) i guess, someone, who decides to use ``ds18alias``, knows what he does. you want to name your sensors in a certain way. so, maybe it is ok to have empty names in the first place, as you want to name your sensors anyway?

sfromis, 2023-10-18:
> maybe it is ok to have empty names in the first place

Not really, as it generates invalid JSON.

arendst, 2023-10-18:
Let me fix the empty names now.

arendst, 2023-10-18:
I think the OP's issue is caused by mixing the unsorted id with the sorted alias name.

I'll examine the new addition (Alias) with the original driver and add sorted indexes where needed.

arendst, 2023-10-18:
@spcqike try latest dev if it still messes up the alias relations.

spcqike, 2023-10-18:
update

```
00:00:00.001 HDW: ESP8266EX
00:00:00.032 UFS: FlashFS mounted with 1976 kB free
00:00:00.104 CFG: Loaded from File, Anzahl 23
00:00:00.110 QPC: Count 1
00:00:00.115 SCR: ufilesystem found
00:00:00.152 Projekt tasmota - Tasmota Version 13.1.0.4(tasmota-4M)-2_7_4_9(2023.10.18 17:30:25)
00:00:00.770 WIF: verbinden mit AP1 Hier_WLAN Channel 6 BSSId 62:9A:E1:FD:6A:2E in Modus 11n als tasmota-26B72A-5930...
00:00:01.796 WIF: verbunden
00:00:02.049 HTP: Web-Server aktiv bei tasmota-26B72A-5930 mit IP-Adresse 172.20.10.3
17:50:30.004 RSL: INFO1 = {"Info1":{"Module":"Generic","Version":"13.1.0.4(tasmota-4M)","FallbackTopic":"cmnd/DVES_26B72A_fb/","GroupTopic":"cmnd/tasmotas/"}}
17:50:30.009 RSL: INFO2 = {"Info2":{"WebServerMode":"Admin","Hostname":"tasmota-26B72A-5930","IPAddress":"172.20.10.3"}}
17:50:30.020 RSL: INFO3 = {"Info3":{"RestartReason":"External System","BootCount":14}}
17:50:30.066 RSL: RESULT = {"DS18B20-1":{"Id":"503CE1E3811BE628"},"DS18B20-2":{"Id":"A83CE1E38139D428"},"-1":{"Id":"363CE1E381479B28"},"DS18B20-4":{"Id":"D43CE1E381D7F228"},"DS18B20-5":{"Id":"893CE1E381F94228"}}
17:50:30.115 RSL: RESULT = {"DS18B20-1":{"Id":"503CE1E3811BE628"},"DS18B20-2":{"Id":"A83CE1E38139D428"},"-1":{"Id":"363CE1E381479B28"},"-2":{"Id":"D43CE1E381D7F228"},"DS18B20-5":{"Id":"893CE1E381F94228"}}
17:50:30.166 RSL: RESULT = {"DS18B20-1":{"Id":"503CE1E3811BE628"},"DS18B20-2":{"Id":"A83CE1E38139D428"},"-1":{"Id":"363CE1E381479B28"},"-2":{"Id":"D43CE1E381D7F228"},"-3":{"Id":"893CE1E381F94228"}}
17:50:30.215 RSL: RESULT = {"DS18B20-1":{"Id":"503CE1E3811BE628"},"-4":{"Id":"A83CE1E38139D428"},"-1":{"Id":"363CE1E381479B28"},"-2":{"Id":"D43CE1E381D7F228"},"-3":{"Id":"893CE1E381F94228"}}
17:50:33.237 QPC: Reset
17:50:34.208 RSL: STATE = {"Time":"2023-10-18T17:50:34","Uptime":"0T00:00:09","UptimeSec":9,"Heap":16,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Wifi":{"AP":1,"SSId":"Hier_WLAN","BSSId":"62:9A:E1:FD:6A:2E","Channel":6,"Mode":"11n","RSSI":44,"Signal":-78,"LinkCount":1,"Downtime":"0T00:00:03"}}
17:50:34.230 RSL: SENSOR = {"Time":"2023-10-18T17:50:34","DS18B20-1":{"Id":"3CE1E3811BE6","Temperature":14.9},"-4":{"Id":"3CE1E38139D4","Temperature":15.5},"-1":{"Id":"3CE1E381479B","Temperature":15.3},"-2":{"Id":"3CE1E381D7F2","Temperature":15.4},"-3":{"Id":"3CE1E381F942","Temperature":15.7},"TempUnit":"C"}
```

it works. doesnt matter if sensors are connected to input 1 or input 2.  and new sensors get the standard  naming.

one thing i saw: when the mqtt server isnt available, aliasing doesnt work.

mqtt logs

```

no alias without configured mqtt server 

00:00:00.001 HDW: ESP8266EX
00:00:00.013 UFS: FlashFS mounted with 1976 kB free
00:00:00.070 CFG: Loaded from File, Anzahl 122
00:00:00.076 QPC: Count 1
00:00:00.081 SCR: ufilesystem found
00:00:00.091 SCR: nv=0, tv=0, vns=0, vmem=16, smem=8192
00:00:00.102 Projekt tasmota - Heizung Version 13.1.0.4(tasmota-4M)-2_7_4_9(2023.10.18 10:13:49)
00:00:00.171 meters: 1 , decode lines: 1, memory used: 1033 bytes
00:00:00.560 WIF: verbinden mit AP1 Hier_WLAN Channel 6 BSSId 06:3D:2F:6F:E9:4F in Modus 11n als tasmota-heizung-5930...
00:00:01.781 WIF: verbunden
00:00:02.033 HTP: Web-Server aktiv bei tasmota-heizung-5930 mit IP-Adresse 172.20.10.3
17:29:28.034 MQT: Verbindungsversuch...
17:29:28.249 MQT: Verbindung fehlgeschlagen aufgrund von 192.168.8.1:1883, rc -2. Wiederversuch in 10 s
17:29:31.456 QPC: Reset
17:29:37.372 RSL: STATE = {"Time":"2023-10-18T17:29:37","Uptime":"0T00:00:13","UptimeSec":13,"Heap":15,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Wifi":{"AP":1,"SSId":"Hier_WLAN","BSSId":"06:3D:2F:6F:E9:4F","Channel":6,"Mode":"11n","RSSI":70,"Signal":-65,"LinkCount":1,"Downtime":"0T00:00:03"}}
17:29:37.396 RSL: SENSOR = {"Time":"2023-10-18T17:29:37","DS18B20-2":{"Id":"3CE1E38139D4","Temperature":16.6},"DS18B20-4":{"Id":"3CE1E381479B","Temperature":16.3},"DS18B20-1":{"Id":"3CE1E381D7F2","Temperature":16.1},"DS18B20-3":{"Id":"3CE1E381F942","Temperature":16.5},"MAX6675":{"Temperature":18.8,"ReferenceTemperature":null,"Error":0},"Öl":{"Count":27415.48},"TempUnit":"C"}
17:29:39.010 MQT: Verbindungsversuch...
17:29:39.227 MQT: Verbindung fehlgeschlagen aufgrund von 192.168.8.1:1883, rc -2. Wiederversuch in 20 s

mqtt removed

00:00:00.001 HDW: ESP8266EX
00:00:00.014 UFS: FlashFS mounted with 1976 kB free
00:00:00.072 CFG: Loaded from File, Anzahl 124
00:00:00.077 QPC: Count 1
00:00:00.082 SCR: ufilesystem found
00:00:00.094 SCR: nv=0, tv=0, vns=0, vmem=16, smem=8192
00:00:00.104 Projekt tasmota - Heizung Version 13.1.0.4(tasmota-4M)-2_7_4_9(2023.10.18 10:13:49)
00:00:00.174 meters: 1 , decode lines: 1, memory used: 1033 bytes
00:00:00.564 WIF: verbinden mit AP1 Hier_WLAN Channel 6 BSSId 06:3D:2F:6F:E9:4F in Modus 11n als tasmota-heizung-5930...
00:00:01.781 WIF: verbunden
00:00:02.033 HTP: Web-Server aktiv bei tasmota-heizung-5930 mit IP-Adresse 172.20.10.3
17:30:16.052 RSL: INFO1 = {"Info1":{"Module":"Generic","Version":"13.1.0.4(tasmota-4M)","FallbackTopic":"cmnd/Heizung_fb/","GroupTopic":"cmnd/tasmotas/"}}
17:30:16.056 RSL: INFO2 = {"Info2":{"WebServerMode":"Admin","Hostname":"tasmota-heizung-5930","IPAddress":"172.20.10.3"}}
17:30:16.067 RSL: INFO3 = {"Info3":{"RestartReason":"Software/System restart","BootCount":85}}
17:30:16.110 RSL: DS18ALIAS = {"DS18B20-2":{"Id":"D43CE1E381D7F228"},"-1":{"Id":"A83CE1E38139D428"},"DS18B20-1":{"Id":"893CE1E381F94228"},"DS18B20-3":{"Id":"363CE1E381479B28"}}
[... cut ...]
17:30:20.380 RSL: STATE = {"Time":"2023-10-18T17:30:20","Uptime":"0T00:00:09","UptimeSec":9,"Heap":15,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":23,"MqttCount":0,"Wifi":{"AP":1,"SSId":"Hier_WLAN","BSSId":"06:3D:2F:6F:E9:4F","Channel":6,"Mode":"11n","RSSI":68,"Signal":-66,"LinkCount":1,"Downtime":"0T00:00:03"}}
17:30:20.403 RSL: SENSOR = {"Time":"2023-10-18T17:30:20","DS18B20-2":{"Id":"3CE1E38139D4","Temperature":16.4},"-1":{"Id":"3CE1E381479B","Temperature":16.2},"-2":{"Id":"3CE1E381D7F2","Temperature":16.0},"abc":{"Id":"3CE1E381F942","Temperature":16.4},"MAX6675":{"Temperature":17.3,"ReferenceTemperature":null,"Error":0},"Öl":{"Count":27415.48},"TempUnit":"C"}
```

sfromis, 2023-10-18:
Looks like the way you execute those DS18Alias commands are only after Tasmota fully initialized, like `System#boot` - maybe use `System#init` instead for an earlier execution, not depending on network connections (including MQTT if configured).

spcqike, 2023-10-18:
I use autoexec.bat as it’s mentioned in the docs. 

Is there another script / file / way to execute this without rules?

sfromis, 2023-10-18:
`autoexec.bat` is at the same time as `System#boot`, meaning after connection to Wifi and (if configured) MQTT. Since Berry is not available on ESP8266, the way to have commands executed at an earlier point is rules. Why do you ask for "without rules"?

If you do not want to connect to a MQTT broker, you can disable MQTT, `SetOption3 0`. Unless you do that, the expectation is that MQTT is needed for the Tasmota device to be fully operational, and `autoexec.bat` will be executed when that's the case.

spcqike, 2023-10-18:
I use SML (counter input meter to keep track of oil consumption) which uses scripts which disallows rules. 

No. I want to use MQTT. But as the device is placed in your new house, where I don’t have remote access yet and I forgot to bring the temporary raspberry server, I haven’t had MQTT today :) I just noticed this while setting up the Tasmota device. 

For now I just cleared the MQTT IP, as the server isn’t there anyway. I will set it up in the next days. 

Isn’t there a way to call a file / script which sets the alias? Just in case the device crashes and restarts after a power outage and the server isn’t (yet or in general) up and running?

sfromis, 2023-10-18:
IDK what options there are with scripting, but I'd not be surprised if it is possible there.

PR #19769 (2023-10-16): fix mis-order
## Description:

**Related issue (if applicable):** fixes #19763

The 2 layer indexing was not taken into account in DS18Alias

## Checklist:
  - [X] The pull request is done against the latest development branch
  - [X] Only relevant files were touched
  - [X] Only one feature/fix was added per PR and the code change compiles without warnings
  - [X] The code change is tested and works with Tasmota core ESP8266 V.2.7.4.9
  - [X] The code change is tested and works with Tasmota core ESP32 V.2.0.14
  - [X] I accept the [CLA](https://github.com/arendst/Tasmota/blob/development/CONT [... cut]