arendst/Tasmota #19802: MQTT client ID is corrupted

ramanraja, 2023-10-20:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
 I need to set the MQTT client id programmatically for each of my Tasmota devices.  
I was trying to mimic this line in my_user_config.h:
#define MQTT_CLIENT_ID         "DVES_%06X"    
When I issued the command from the URL window of a browser, The client id was corrupted with strange looking characters inserted in place of %06.  My fallback topic was also corrupted because of this.
I tried to issue the following command through the browser:
http://192.168.0.125/cm?cmnd=mqttclient%20DVES_%06X
I got the response:
{"MqttClient":"DVES_X"}
The expected response was something like  {"MqttClient":"DVES_ECADF6"}
### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _13.2.0____
  - [x ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _Tasmotizer____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
06:41:30.909 CMD: Backlog Template; Module; GPIO 255
06:41:30.973 RSL: RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
06:41:31.187 RSL: RESULT = {"Module":{"1":"Sonoff Basic"}}
06:41:31.441 RSL: RESULT = {"GPIO0":{"32":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"224":"Relay1"},"GPIO13":{"320":"Led_i1"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
Tasmota is already restarted after command, so I am providing the relevant info-1
 INFO1 = {"Info1":{"Module":"Sonoff Basic","Version":"13.2.0(tasmota)","FallbackTopic":"cmnd/DVES_X_fb/","GroupTopic":"cmnd/IN-RAJA/tasmotas/"}}
```

### TO REPRODUCE
_Steps to reproduce the behavior:_
On a browser URL, issue this command to the Tasmota device:
http://192.168.0.125/cm?cmnd=mqttclient%20DVES_%06X

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
{"MqttClient":"DVES_859E6D"}

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
There are many commands with the % symbol in them, but the mqttclient command is behaving differently.
This corrupts the fallback topic also.

**(Please, remember to close the issue when the problem has been addressed)**

ramanraja, 2023-10-20:
The work around is to encode % as %25.

sfromis, 2023-10-20:
Making sure to do the correct URL encoding is no workaround, it is fixing your error. And when you use an URL in a browser, it is up to **you** to first to the URL encoding. You *did* URL encode the blank correctly with `%20`, meaning that `%06` has to become hex 06, without it being treated as 3 separate chars.