arendst/Tasmota #19826: Camera OV5640 / AI Thinker Cam failed with Tasmota 13.x

BmdOnline, 2023-10-23:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

Since Tasmota ugrade, I was unable to use my AITHINKER CAM with OV5640 camera.
Rolling back to Tasmota 12.5.0 solves the problem.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): AITHINKER CAM
- [x] Tasmota binary firmware version number used: 13.2.0.1
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: ESP-Flasher 1.4.1
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
19:33:33.556 RSL: RESULT = {"NAME":"AITHINKER CAM","GPIO":[4992,1,672,1,416,5088,1,1,1,6720,736,704,1,1,5089,5090,0,5091,5184,5152,0,5120,5024,5056,0,0,0,0,4928,576,5094,5095,5092,0,0,5093],"FLAG":0,"BASE":2}
19:33:33.779 RSL: RESULT = {"Module":{"0":"AITHINKER CAM"}}
19:33:33.988 RSL: RESULT = {"GPIO0":{"4992":"CAM_XCLK"},"GPIO1":{"0":"None"},"GPIO2":{"672":"SPI MISO1"},"GPIO3":{"0":"None"},"GPIO4":{"416":"PWM1"},"GPIO5":{"5088":"CAM_DATA1"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"6720":"SDCard CS"},"GPIO14":{"736":"SPI CLK1"},"GPIO15":{"704":"SPI MOSI1"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"5089":"CAM_DATA2"},"GPIO19":{"5090":"CAM_DATA3"},"GPIO20":{"0":"None"},"GPIO21":{"5091":"CAM_DATA4"},"GPIO22":{"5184":"CAM_PCLK"},"GPIO23":{"5152":"CAM_HREF"},"GPIO24":{"0":"None"},"GPIO25":{"5120":"CAM_VSYNC"},"GPIO26":{"5024":"CAM_SIOD"},"GPIO27":{"5056":"CAM_SIOC"},"GPIO32":{"4928":"CAM_PWDN"},"GPIO33":{"576":"LedLink_i"},"GPIO34":{"5094":"CAM_DATA7"},"GPIO35":{"5095":"CAM_DATA8"},"GPIO36":{"5092":"CAM_DATA5"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"5093":"CAM_DATA6"}}

```
- [x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
19:34:30.266 RSL: RESULT = {"Rule1":{"State":"ON","Once":"OFF","StopOnError":"OFF","Length":30,"Free":481,"Rules":"ON System#Boot DO WcInit ENDON"}}
19:34:30.489 RSL: RESULT = {"Rule2":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
19:34:30.739 RSL: RESULT = {"Rule3":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}

```
- [x] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
19:35:02.201 RSL: STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota 1956","FriendlyName":["AITHINKER CAM"],"Topic":"tasmota_78A7A4","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
19:35:02.237 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32.bin","RestartReason":"Software reset CPU","Uptime":"0T00:04:18","StartupUTC":"2023-10-23T17:30:44","Sleep":50,"CfgHolder":4617,"BootCount":7,"BCResetTime":"2022-01-17T10:44:49","SaveCount":20}}
19:35:02.278 RSL: STATUS2 = {"StatusFWR":{"Version":"13.2.0.1(webcam)","BuildDateTime":"2023-10-23T07:41:18","Core":"2_0_14","SDK":"4.4.6.231011","CpuFrequency":240,"Hardware":"ESP32-D0WDQ6 v1.0","CR":"403/699"}}
19:35:02.292 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["xxxxxx",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A000000000000","00000080","00006000","00004000","00000000"]}}
19:35:02.320 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":1227,"Free":1856,"Heap":166,"StackLowMark":4,"PsrMax":4093,"PsrFree":4088,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"16405E","FlashFrequency":80,"FlashMode":"DIO","Features":["00000809","0F0005CE","00040001","00000001","00000000","00000880","80000080","00200000","54000008","00000080"],"Drivers":"1,2,4,9,10,12,29,38,50,52,62,81","Sensors":"1,127","I2CDriver":""}}
19:35:02.359 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota-78A7A4-1956","IPAddress":"10.10.10.101","Gateway":"10.10.10.1","Subnetmask":"255.255.255.0","DNSServer1":"10.10.10.1","DNSServer2":"0.0.0.0","Mac":"78:21:84:78:A7:A4","IP6Global":"fd46:901c:be2a::7a21:84ff:fe78:a7a4","IP6Local":"fe80::7a21:84ff:fe78:a7a4","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
19:35:02.397 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"xxxxxx","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_78A7A4","MqttUser":"tasmota_78A7A4","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
19:35:02.415 RSL: STATUS7 = {"StatusTIM":{"UTC":"2023-10-23T17:35:02","Local":"2023-10-23T19:35:02","StartDST":"2023-03-26T02:00:00","EndDST":"2023-10-29T03:00:00","Timezone":99,"Sunrise":"08:22","Sunset":"18:45"}}
19:35:02.443 RSL: STATUS10 = {"StatusSNS":{"Time":"2023-10-23T19:35:02"}}
19:35:02.451 RSL: STATUS11 = {"StatusSTS":{"Time":"2023-10-23T19:35:02","Uptime":"0T00:04:18","UptimeSec":258,"BatteryPercentage":0,"Heap":166,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Berry":{"HeapUsed":3,"Objects":33},"POWER":"OFF","Dimmer":27,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"xxxxxx","BSSId":"30:46:9A:0B:9B:16","Channel":11,"Mode":"11n","RSSI":82,"Signal":-59,"LinkCount":1,"Downtime":"0T00:00:02"}}}

```
- [x] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
00:00:00.002 HDW: ESP32-D0WDQ6 v1.0 (PSRAM)
00:00:00.043 UFS: FlashFS mounted with 304 kB free
00:00:00.067 CFG: Loaded from File, Count 21
00:00:00.075 QPC: Count 1
00:00:00.077 CFG: CR 403/699, Busy 0
00:00:00.079 SPI: Hardware using GPIO14(CLK), GPIO15(MOSI) and GPIO02(MISO)
00:00:00.700 BRY: GC from 3752 to 2971 bytes, objects freed 5/15 (in 1 ms) - slots from 33/61 to 20/61
00:00:00.756 CFG: No '*.autoconf' file found
00:00:00.760 BRY: GC from 4296 to 3599 bytes, objects freed 5/31 (in 1 ms) - slots from 50/76 to 32/76
00:00:00.760 BRY: Berry initialized, RAM used=3599 bytes
00:00:00.771 BRY: No 'preinit.be'
00:00:00.772 SRC: Restart
00:00:00.777 Project tasmota - Tasmota 1956 Version 13.2.0.1(webcam)-2_0_14(2023-10-23T07:41:18)
00:00:00.778 CAM: WcSetup
00:00:00.778 CAM: i2c_enabled_2: 0
00:00:00.778 CAM: Template pin config
00:00:00.779 CAM: XCLK on GPIO 0 using ledc channel 1
[... 66 lines cut ...]

```

And using Tasmota 13.2.0.1
```lua
19:04:06.707 RUL: SYSTEM#BOOT performs "WcInit"
19:04:06.747 CAM: InitErr 0x106 try 1
19:04:06.780 CAM: InitErr 0x106 try 2
19:04:06.813 CAM: InitErr 0x106 try 3
19:04:06.814 CAM: InitErr 0x106
19:04:06.850 CAM: InitErr 0x106 try 1
19:04:06.883 CAM: InitErr 0x106 try 2
19:04:06.916 CAM: InitErr 0x106 try 3
19:04:06.917 CAM: InitErr 0x106
19:04:06.926 MQT: stat/tasmota_78A7A4/RESULT = {"WCInit":"Done"}

```

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2023-10-23:
Support for OV5640 was dropped in actual Tasmota Arduino framework.

closed 2023-10-23 by Jason2866

marcosterland, 2023-12-30:
> Support for OV5640 was dropped in actual Tasmota Arduino framework.

Is there a particular reason for that? What would be necessary to add support again? I would be willing to create a PR for that.

Jason2866, 2023-12-30:
@marcosterland It is done by design. Support is removed in IDF and in Arduino precompiled libs to save code space. No changes are done in Tasmota.

edwin-jones, 2024-01-24:
Hi @Jason2866, I'm having the exact same issue with the M5 Stack Timer Camera F with an ov3660 module and I can only get it working with 12.5 - has support for this camera also been dropped in v13 onwards? Thank you.

Jason2866, 2024-01-24:
@edwin-jones Yes.

edwin-jones, 2024-01-24:
Got it, so basically skip v13 for both these camera modules and don't upgrade until a version is released that moves to core 3.0.0 which is planned but not scheduled? Thanks.

Jason2866, 2024-01-24:
We will switch to Arduino core 3.0.0 right after it is released (probably a few days after).
So the switch is depending on Arduino espressif work ;-)