arendst/Tasmota #19863: Master and Development does not build anymore

deleonio, 2023-10-29:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Sonoff
- [x] Tasmota binary firmware version number used: 13.2
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: 
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:v
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
********************************************************************
* Looking for han_Parser.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:han_Parser.h"
* Web  > https://registry.platformio.org/search?q=header:han_Parser.h
*
********************************************************************

 #include "han_Parser.h"
                        ^
compilation terminated.
Retrieved `.pio/build/tasmota-ir/libbd8/ESP8266mDNS/ESP8266mDNS.cpp.o' from cache
*** [.pio/build/tasmota-ir/src/tasmota.ino.cpp.o] Error 1
====================================== [FAILED] Took 5.96 seconds ======================================

Environment    Status    Duration
-------------  --------  ------------
tasmota-ir     FAILED    00:00:05.958
================================ 1 failed, 0 succeeded in 00:00:05.958 ================================

```

### TO REPRODUCE
_Steps to reproduce the behavior: try to build `tasmota-ir`

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

barbudor, 2023-10-29:
You did wrong, so happens as expected

[image]

deleonio, 2023-10-29:
Is there any way to make the default smaller. My device has only 1MB space and say `not enogh space`.

Thank you.

barbudor, 2023-10-29:
Not related to your initial question

You should first upgrade to tasmota-minimal.bin.gz
The to your newly compiled tasmota.bin.gz

Take care of using directly the gz binaries

deleonio, 2023-10-29:
I disabled some features in the `Default` variant. Ohjee 🙁 - but I have now a small artifact with SML.

Edit: `tasmota_configurations.h` -> https://github.com/deleonio/Tasmota/commit/595854b8714798308457d83bc3ed5a8e702d672b#diff-00f571142d9f9ecce398802637a0fe4e073f1b2b6b36cd4712aba1a475c19567

barbudor, 2023-10-29:
When you start to modify Tasmota internal files instead of following the recommended way of working only with user_config_override.h and building tasmota/tasmota32, you are on your own