arendst/Tasmota #19873: warning: unrecognized command line option '-Wno-volatile' when compiling Tasmota/lib/libesp32_lvgl/lvgl/src/widgets/lv_checkbox.c

gneiss15, 2023-10-30:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
When compiling Tasmota (using VSCode & Platformio) I get the following "warning":
Retrieved `.pio/build/tasmota32-DE/lib193/lvgl/widgets/lv_checkbox.c.o' from cache
cc1plus: warning: unrecognized command line option '-Wno-volatile'

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Any ESP32
- [x] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [x] Self-compiled
- [x] Flashing tools used: not relevant
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
not relevant

```
- [x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
not relevant

```
- [x] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
not relevant

```
- [x] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

Use "build" inside VSCode

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

No Waring

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
Unfortunally I couldn't figure out why the build-process uses the CXXFLAGS when compiling (only this) file..
I figured out that this compiler option was only set by "pio-tools/add_c_flags.py", but I don't find why this was applied when compiling 
"Tasmota/lib/libesp32_lvgl/lvgl/src/widgets/lv_checkbox.c". It seams that this file was compiled in C++ Mode even it is a c file ??

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2023-10-30:
I'm seeing the warning all the time when building for ESP32, and haven't worried about it :smile_cat:

Jason2866, 2023-11-12:
Where do you see this warning? Can't reproduce. The GH Actions run building the LVGL variant does not show this warning
https://github.com/arendst/Tasmota/actions/runs/6840128964/job/18599096305

closed 2023-11-12 by Jason2866