arendst/Tasmota #19874: Webui Timer - Output and Action selector not available

JHurk, 2023-10-30:
### PROBLEM DESCRIPTION
In the timer Webui I don't have an option to select 'Output' (it is not visible) and the 'Action' is not a selector, just text 'Rule'.
So for configuring a timer I am unable to select an Output and Action.  

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x ] Device used (e.g., Sonoff Basic): Atom Lite with Tail485 (M5stack)
- [ ] Tasmota binary firmware version number used: 13.1.0
  - [ ] Pre-compiled
  - [ x] Self-compiled; ESP32 with NeoPool enabled
- [ x] Flashing tools used: Tasmocompiler and ESPhome (web.esphome.io)
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
08:17:49.757 MQT: stat/SmartPool/TEMPLATE = {"NAME":"ESP32-DevKit","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,0,1,1,1,0,1,1,1,0,0,0,0,1,1,1,1,1,0,0,1],"FLAG":0,"BASE":1}
08:17:49.975 MQT: stat/SmartPool/MODULE = {"Module":{"0":"ESP32-DevKit"}}
08:17:50.230 MQT: stat/SmartPool/GPIO = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"0":"None"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"0":"None"},"GPIO22":{"0":"None"},"GPIO23":{"0":"None"},"GPIO24":{"0":"None"},"GPIO25":{"0":"None"},"GPIO26":{"6976":"NeoPool Tx"},"GPIO27":{"0":"None"},"GPIO32":{"7008":"NeoPool Rx"},"GPIO33":{"0":"None"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
08:19:01.808 MQT: stat/SmartPool/RULE = {"Rule1":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
08:19:02.044 MQT: stat/SmartPool/RULE = {"Rule2":{"State":"ON","Once":"OFF","StopOnError":"OFF","Length":35,"Free":476,"Rules":"ON Clock#Timer=10 DO NPTime 0 ENDON"}}
08:19:02.293 MQT: stat/SmartPool/RULE = {"Rule3":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
08:19:24.156 MQT: stat/SmartPool/STATUS = {"Status":{"Module":0,"DeviceName":"SmartPool","FriendlyName":["SmartPool"],"Topic":"SmartPool","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
08:19:24.169 MQT: stat/SmartPool/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"","RestartReason":"Software reset CPU","Uptime":"0T00:35:49","StartupUTC":"2023-10-30T06:43:35","Sleep":50,"CfgHolder":4617,"BootCount":31,"BCResetTime":"2023-10-18T11:01:01","SaveCount":64}}
08:19:24.180 MQT: stat/SmartPool/STATUS2 = {"StatusFWR":{"Version":"13.1.0(TasmoCompiler-esp32generic)","BuildDateTime":"2023-10-18T09:32:33","Core":"2_0_11","SDK":"4.4.5.230722","CpuFrequency":160,"Hardware":"ESP32-PICO-D4 rev.1","CR":"327/699"}}
08:19:24.188 MQT: stat/SmartPool/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Diva",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008019","2805C80001000600003C5A0A192800000000","00000081","00006000","00004000","00000000"]}}
08:19:24.204 MQT: stat/SmartPool/STATUS4 = {"StatusMEM":{"ProgramSize":1217,"Free":1662,"Heap":124,"StackLowMark":4,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"1640C8","FlashFrequency":40,"FlashMode":"DIO","Features":["00000809","0FC045CA","00040001","00000003","00000000","C0000881","00000080","08200000","54000008","00000080"],"Drivers":"1,2,4,8,9,10,12,24,26,29,35,38,50,52,62,68","Sensors":"1,2,83,127"}}
08:19:24.220 MQT: stat/SmartPool/STATUS5 = {"StatusNET":{"Hostname":"SmartPool-3416","IPAddress":"192.168.1.24","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.1.1","DNSServer2":"0.0.0.0","Mac":"D4:D4:DA:84:ED:58","IP6Global":"","IP6Local":"fe80::d6d4:daff:fe84:ed58","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":16.0}}
08:19:24.233 MQT: stat/SmartPool/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.***","MqttPort":1883,"MqttClientMask":"SmartPool","MqttClient":"SmartPool","MqttUser":"****","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
08:19:24.245 MQT: stat/SmartPool/STATUS7 = {"StatusTIM":{"UTC":"2023-10-30T07:19:24","Local":"2023-10-30T08:19:24","StartDST":"2023-03-26T01:00:00","EndDST":"2023-10-29T01:00:00","Timezone":99,"Sunrise":"07:29","Sunset":"17:16"}}
08:19:24.272 MQT: stat/SmartPool/STATUS10 = {"StatusSNS":{"Time":"2023-10-30T08:19:24","NeoPool":{"Time":"2023-10-30T08:19:22","Type":"Oxilife","Modules":{"pH":1,"Redox":1,"Hydrolysis":1,"Chlorine":0,"Conductivity":0,"Ionization":0},"Temperature":13.2,"Power":{"Version":"V5.0","NodeID":"0059 002F 4143 5017 2038 3130","5V":5,"12V":13,"24-30V":33,"4-20mA":0.050},"pH":{"Data":7.3,"Min":7.0,"Max":7.5,"State":0,"Pump":0,"FL1":0,"Tank":1},"Redox":{"Data":691,"Setpoint":700},"Hydrolysis":{"Data":0,"Unit":"%","Runtime":{"Total":"1T12:36:26","Part":"1T12:36:26","Pol1":"0T19:25:49","Pol2":"0T17:10:37","Changes":21},"State":"OFF","Cover":1,"Boost":0,"Low":0},"Filtration":{"State":0,"Speed":1,"Mode":1},"Light":0,"Relay":{"State":[0,0,0,0,0,0,0],"Aux":[0,0,0,0],"Acid":0}},"TempUnit":"C"}}
08:19:24.289 MQT: stat/SmartPool/STATUS11 = {"StatusSTS":{"Time":"2023-10-30T08:19:24","Uptime":"0T00:35:49","UptimeSec":2149,"Heap":125,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Berry":{"HeapUsed":3,"Objects":28},"Wifi":{"AP":1,"SSId":"Diva","BSSId":"24:5A:4C:7E:7E:FF","Channel":1,"Mode":"11n","RSSI":74,"Signal":-63,"LinkCount":1,"Downtime":"0T00:00:04"}}}

```

### TO REPRODUCE
Go to Webui --> Configuration --> Configure Timer and select timer 10 (which is the one enabled).

I am unable to select the output or the action, this is also the case with the other timers.

### EXPECTED BEHAVIOUR
I thought I would be able to select the output and/or action.
And I experienced strange behavior after I followed the this tutorial for synching the time of my Sugar Valley (pool device) with Tasmota time. 

I did exactly what is stated in the tutorial (https://tasmota.github.io/docs/NeoPool/#daily-sync-device-to-tasmota-time), but for some reason at 4:01 in the morning (time when rule is triggered) the time of my Sugar Valley was set to 8:16 in the morning (so 4 hours and 15 minutes off). After seeing this this morning I looked in the Tasmota webui and saw that the time of the Tasmota device was set to 4 hours and 15 minutes later than the real time. After looking at the Timezone, NTPserver and Time the time in Tasmota got set straight again and a NPTime 0 command did the sync with the Neopool (Sugar Valley).
Do not know if this is due to the missing selectors (Output and Rule) or that because I can not set these, the syncing goes wrong?

### SCREENSHOTS
Current situation:
[image]

Expected situatie:

[image]

### ADDITIONAL CONTEXT
I tried with Safari and Chrome, in both cases the behavior is the same.

**(Please, remember to close the issue when the problem has been addressed)**

barbudor, 2023-10-30:
Your configuration includes no Relay or light, so no output that Tasmota Timer can control.

JHurk, 2023-10-30:
Thanks for the quick reply. But looking at my setup and goal of the timer (sync time between two devices) I do not need to configure any relays/lights. 
So this is probably not why the sync went wrong last night, correct?
Any guesses on why Tasmota thought it was 8:16 instead of 4:01 this morning?

sfromis, 2023-10-30:
Well you cannot configure an output without having an output to configure. The only available action without relays (or similar) is to make the timer event available to a rule. And then you need a rule to actually react to the event, like in the example you linked. For that you need no change of output or action, as the defaults works for that.

When it comes to what local time Tasmota has, you can configure time zone related settings. For full options, including automatic of control of normal vs daylight savings time, this tool is an easy way: https://tasmotatimezone.com/

JHurk, 2023-10-30:
This is the log;
09:35:00.703 RUL: CLOCK#TIMER=10 performs "NPTime 0"
09:35:00.705 SRC: Rule
09:35:00.707 CMD: Grp 0, Cmd 'NPTIME', Idx 1, Len 1, Pld 0, Data '0'
09:35:00.782 MQT: stat/SmartPool/NPTIME = {"NPTime":"2023-10-30T09:34:59"}

Seems like the sync went correct this time, will wait this night again to see if it was a 'one off' or if something is indeed wrong.
Maybe I fixed it with the tasmotatimezone.com, which I found this morning already and gave the backlog command which was generated when filling my details.