arendst/Tasmota #19896: PID compile error

krishy2, 2023-11-01:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

Custom build of the firmware fails if PID is defined.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ x] Tasmota binary firmware version number used: _____13.2.0.1 development
  - [ ] Pre-compiled
  - [ x] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

xdrv_49_pid.ino line 469:

const char HTTP_PID_HL[] /*PROGMEM*/ = "{s}<hr>{m}<hr>{e}";
  const char HTTP_PID_INFO[] /*PROGMEM*/ = "{s}" D_PID_DISPLAY_NAME "{m}%s{e}";
  const char HTTP_PID_SP_FORMAT[] /*PROGMEM*/ = "{s}%s " "{m}%*_f ";
  const char HTTP_PID_PV_FORMAT[] /*PROGMEM*/ = "{s}%s " "{m}%*_f ";
  const char HTTP_PID_POWER_FORMAT[] /*PROGMEM*/ = "{s}%s " "{m}%*_f " D_UNIT_PERCENT;

I've commented out the PROGMEMs and it compiled and working.

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2023-11-01:
A compile error around those lines was fixed 15 hours ago. Did you use the very newest source?
https://github.com/arendst/Tasmota/pull/19890

Jason2866, 2023-11-01:
Closing since none of the Infos asked in the Template is filled.

closed 2023-11-01 by Jason2866