arendst/Tasmota #19911: Tilt of shutter does not work in one command for first time (regression)

JosefRypacek, 2023-11-02:
### PROBLEM DESCRIPTION
_The following scenario worked on some previous 13.1.x.x version._
I have shutters with tilt in range [-90; 0].
Let's suppose current position is 100 (open) and tilt is 0 (horizontal).
Command `shutterposition1 0,-45` will set shutter to position 0 (closed) with tilt -90 (vertical - closed).
Second submission of the same command when shutter is in position 0 will set the tilt correctly.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Sonoff Dual R3
- [x] Tasmota binary firmware version number used: 13.2.0.1
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: OTA
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

17:26:27.604 MQT: stat/tasmota_4B34F0/RESULT = {"NAME":"Sonoff Dual R3","GPIO":[34,0,0,0,0,0,0,0,0,576,225,0,0,0,0,0,0,0,0,0,0,7296,7328,224,0,0,0,0,32,33,0,0,0,0,0,0],"FLAG":0,"BASE":1}
17:26:27.807 MQT: stat/tasmota_4B34F0/RESULT = {"Module":{"0":"Sonoff Dual R3"}}
17:26:28.008 MQT: stat/tasmota_4B34F0/RESULT = {"GPIO0":{"34":"Button3"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"576":"LedLink_i"},"GPIO14":{"225":"Relay2"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"0":"None"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"0":"None"},"GPIO22":{"0":"None"},"GPIO23":{"0":"None"},"GPIO24":{"0":"None"},"GPIO25":{"7296":"CSE7761 Tx"},"GPIO26":{"7328":"CSE7761 Rx"},"GPIO27":{"224":"Relay1"},"GPIO32":{"32":"Button1"},"GPIO33":{"33":"Button2"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

17:28:12.714 MQT: stat/tasmota_4B34F0/STATUS = {"Status":{"Module":0,"DeviceName":"zaluzie-schody-dole","FriendlyName":["Tasmota",""],"Topic":"tasmota_4B34F0","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
17:28:12.720 MQT: stat/tasmota_4B34F0/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/tasmota32.bin","RestartReason":"Software reset CPU","Uptime":"0T00:18:20","StartupUTC":"2023-11-02T16:09:52","Sleep":50,"CfgHolder":4617,"BootCount":16,"BCResetTime":"2023-07-09T22:59:13","SaveCount":430}}
17:28:12.728 MQT: stat/tasmota_4B34F0/STATUS2 = {"StatusFWR":{"Version":"13.2.0.1(tasmota32)","BuildDateTime":"2023-11-02T13:00:58","Core":"2_0_14","SDK":"4.4.6.231024","CpuFrequency":160,"Hardware":"ESP32-D0WD-V3 v3.0","CR":"400/699"}}
17:28:12.733 MQT: stat/tasmota_4B34F0/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["RtIOT",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["0000C009","2805C80001000600003C5A0A192800000000","40800080","00006000","00004000","00000000"]}}
17:28:12.759 MQT: stat/tasmota_4B34F0/STATUS4 = {"StatusMEM":{"ProgramSize":1871,"Free":1008,"Heap":142,"StackLowMark":4,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"1640EF","FlashFrequency":40,"FlashMode":"DIO","Features":["00000809","9F9AD7DF","0015A001","B7F7BFCF","05DA9BC4","E0360DC7","480840D2","20200000","D4BC482D","010A80B1"],"Drivers":"1,2,3,4,5,7,8,9,10,11,12,14,16,17,20,21,24,26,27,29,34,35,38,50,52,59,60,62,63,66,67,68,82,86,87,88","Sensors":"1,2,3,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,31,34,37,39,40,42,43,45,51,52,55,56,58,59,64,66,67,74,85,92,95,98,103,105,109,127","I2CDriver":"7,8,9,10,11,12,13,14,15,17,18,20,24,29,31,36,41,42,44,46,48,58,62,65,69,76,77,82"}}
17:28:12.776 MQT: stat/tasmota_4B34F0/STATUS5 = {"StatusNET":{"Hostname":"tasmota-4B34F0-5360","IPAddress":"10.139.103.48","Gateway":"10.139.103.1","Subnetmask":"255.255.255.0","DNSServer1":"10.139.103.1","DNSServer2":"0.0.0.0","Mac":"3C:61:05:4B:34:F0","IP6Global":"","IP6Local":"fe80::3e61:5ff:fe4b:34f0","Ethernet":{"Hostname":"","IPAddress":"0.0.0.0","Gateway":"0.0.0.0","Subnetmask":"0.0.0.0","DNSServer1":"10.139.103.1","DNSServer2":"0.0.0.0","Mac":"00:00:00:00:00:00","IP6Global":"","IP6Local":""},"Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":16.0}}
17:28:12.804 MQT: stat/tasmota_4B34F0/STATUS6 = {"StatusMQT":{"MqttHost":"mqtt.rypi.net","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_4B34F0","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
17:28:12.816 MQT: stat/tasmota_4B34F0/STATUS7 = {"StatusTIM":{"UTC":"2023-11-02T16:28:12","Local":"2023-11-02T17:28:12","StartDST":"2023-03-26T02:00:00","EndDST":"2023-10-29T03:00:00","Timezone":"+01:00","Sunrise":"07:38","Sunset":"17:28"}}
17:28:12.826 MQT: stat/tasmota_4B34F0/STATUS9 = {"StatusPTH":{"PowerDelta":[0,0,0,0,0,0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
17:28:12.846 MQT: stat/tasmota_4B34F0/STATUS10 = {"StatusSNS":{"Time":"2023-11-02T17:28:12","ENERGY":{"TotalStartTime":"2023-07-09T22:02:04","Total":0.424,"Yesterday":0.002,"Today":0.003,"Power":[0,0],"ApparentPower":[0,0],"ReactivePower":[0,0],"Factor":[0.00,0.00],"Frequency":0,"Voltage":0,"Current":[0.000,0.000]},"Shutter1":{"Position":0,"Direction":0,"Target":0,"Tilt":-90}}}
17:28:12.857 MQT: stat/tasmota_4B34F0/STATUS11 = {"StatusSTS":{"Time":"2023-11-02T17:28:12","Uptime":"0T00:18:20","UptimeSec":1100,"Heap":145,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":24,"MqttCount":1,"Berry":{"HeapUsed":4,"Objects":47},"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"RtIOT","BSSId":"3A:60:F9:54:03:6E","Channel":1,"Mode":"11n","RSSI":86,"Signal":-57,"LinkCount":1,"Downtime":"0T00:00:04"}}}
17:28:12.872 MQT: stat/tasmota_4B34F0/STATUS13 = {"StatusSHT":{"SHT0":{"Relay1":1,"Relay2":2,"Open":340,"Close":340,"50perc":50,"Delay":2,"Opt":"0100","Calib":[300,500,700,900,1000],"Mode":"1","TiltConfig":[-90,0,19,0,-90]}}}
```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

# used weblog 3 to be able past the output here

17:30:12.452 CMD: shutterposition
17:30:12.453 SRC: WebConsole from 10.139.101.113
17:30:12.456 CMD: Grp 0, Cmd 'SHUTTERPOSITION', Idx 1, Len 0, Pld -99, Data ''
17:30:12.474 MQT: stat/tasmota_4B34F0/RESULT = {"Shutter1":{"Position":10,"Direction":0,"Target":10,"Tilt":0}}
17:30:23.161 CMD: shutterposition 0,-45
17:30:23.164 SRC: WebConsole from 10.139.101.113
17:30:23.167 CMD: Grp 0, Cmd 'SHUTTERPOSITION', Idx 1, Len 5, Pld 0, Data '0,-45'
17:30:23.171 SRC: Shutter
17:30:23.207 MQT: stat/tasmota_4B34F0/RESULT = {"POWER2":"ON"}
17:30:23.215 MQT: stat/tasmota_4B34F0/POWER2 = ON
17:30:23.219 SHT: Shtr1 Real 68000, Start 68000, Stop -20000, Dir -1, Delay 2, Rtc 0.00 [s], Freq 0, PWM 0, Tilt 0
17:30:23.228 MQT: stat/tasmota_4B34F0/RESULT = {"Shutter1":{"Position":10,"Direction":-1,"Target":0,"Tilt":0}}
17:30:23.767 SHT: Shtr1 Real 68000, Start 68000, Stop -20000, Dir -1, Delay 2, Rtc 0.55 [s], Freq 0, PWM 0, Tilt -37
17:30:23.804 MQT: stat/tasmota_4B34F0/RESULT = {"Shutter1":{"Position":10,"Direction":-1,"Target":0,"Tilt":-37}}
[... 33 lines cut ...]
17:30:37.548 SHT: Shtr1 Real 0, Start 0, Stop 0, Dir 0, Delay 2, Rtc 0.65 [s], Freq 0, PWM 0, Tilt -43
[... cut ...]
17:30:41.178 MQT: stat/tasmota_4B34F0/RESULT = {"Shutter1":{"Position":0,"Direction":0,"Target":0,"Tilt":-43}}
```

### TO REPRODUCE
_Steps to reproduce the behavior:_
As mentioned in problem description and in console output above

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Set correct tilt on first execution of the command.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

stefanbode, 2023-11-03:
I do agree that if you send the tilt WITH the position the tilt should be maintained. If you only send the position then the behavior is intended. Let me check.

stefanbode, 2023-11-03:
Please wait until the changed got merged into the DEVELOPMENT branch. Then you can recheck with a new binary. Please be aware that on the info page the BUILD time must be AFTER the MERGE time. Just if you want to be sure your build has the fix inside.

closed 2023-11-03 by arendst

JosefRypacek, 2023-11-03:
It works, thanks!

commit b8351697 (2023-11-03): fix #19911 tilt position is not restored with command: shutterposition 0,-45 (#19916)

* fix shutterposition 0,-45 does not keep tilt

allow , on different config settings

* fix shutterposition 0,-45 does not maintain tilt

allow , on different config settings.