arendst/Tasmota #19913: HTTP and ping stops responding for 13.2.0

bilogic, 2023-11-03:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

The HTTP and ping of my wemos D1 ESP8266 running the `tasmota-ir.bin` stops responding every few days and does not self resolve the issue until I power cycle it. It has become quite a chore to need to always do that.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): wemos D1 mini
- [x] Tasmota binary firmware version number used: 13.2.0
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: tasmotizer
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
06:50:42.682 MQT: stat/tasmota_rfbridge/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
06:50:42.893 MQT: stat/tasmota_rfbridge/RESULT = {"Module":{"18":"Generic"}}
06:50:43.146 MQT: stat/tasmota_rfbridge/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"1056":"IRsend1"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
06:52:05.275 MQT: stat/tasmota_rfbridge/RESULT = {"Rule1":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
06:52:05.485 MQT: stat/tasmota_rfbridge/RESULT = {"Rule2":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
06:52:05.736 MQT: stat/tasmota_rfbridge/RESULT = {"Rule3":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
06:51:46.155 MQT: stat/tasmota_rfbridge/STATUS = {"Status":{"Module":18,"DeviceName":"RF bridge","FriendlyName":["Tasmota"],"Topic":"tasmota_rfbridge","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
06:51:46.193 MQT: stat/tasmota_rfbridge/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota-4M.bin.gz","RestartReason":"External System","Uptime":"0T00:05:36","StartupUTC":"2023-11-03T05:46:10","Sleep":50,"CfgHolder":4617,"BootCount":12,"BCResetTime":"2023-10-26T13:14:34","SaveCount":28,"SaveAddress":"FA000"}}
06:51:46.228 MQT: stat/tasmota_rfbridge/STATUS2 = {"StatusFWR":{"Version":"13.2.0(ir)","BuildDateTime":"2023-10-19T09:02:08","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"374/699"}}
06:51:46.248 MQT: stat/tasmota_rfbridge/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["rtfckut","Recovery"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A192800000000","00000080","00006000","00004000","00000000"]}}
06:51:46.280 MQT: stat/tasmota_rfbridge/STATUS4 = {"StatusMEM":{"ProgramSize":648,"Free":352,"Heap":26,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"16405E","FlashFrequency":40,"FlashMode":"DOUT","Features":["00000809","0F1A0582","04000001","00000003","00000000","00000000","00000000","00000000","04000000","00000080"],"Drivers":"1,2,4,5,7,9,10,12,62,68","Sensors":"1"}}
06:51:46.312 MQT: stat/tasmota_rfbridge/STATUS5 = {"StatusNET":{"Hostname":"tasmota-rfbridge-2862","IPAddress":"192.168.9.194","Gateway":"192.168.9.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.9.1","DNSServer2":"0.0.0.0","Mac":"5C:CF:7F:AB:6B:2E","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
06:51:46.337 MQT: stat/tasmota_rfbridge/STATUS6 = {"StatusMQT":{"MqttHost":"mqttbroker","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_AB6B2E","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
06:51:46.361 MQT: stat/tasmota_rfbridge/STATUS7 = {"StatusTIM":{"UTC":"2023-11-03T05:51:46","Local":"2023-11-03T06:51:46","StartDST":"2023-03-26T02:00:00","EndDST":"2023-10-29T03:00:00","Timezone":"+01:00","Sunrise":"07:40","Sunset":"17:26"}}
06:51:46.382 MQT: stat/tasmota_rfbridge/STATUS10 = {"StatusSNS":{"Time":"2023-11-03T06:51:46"}}
06:51:46.393 MQT: stat/tasmota_rfbridge/STATUS11 = {"StatusSTS":{"Time":"2023-11-03T06:51:46","Uptime":"0T00:05:36","UptimeSec":336,"Vcc":3.050,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"rtfckut","BSSId":"78:24:AF:E4:F8:F0","Channel":1,"Mode":"11n","RSSI":80,"Signal":-60,"LinkCount":1,"Downtime":"0T00:00:06"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
None, it happens by itself

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
HTTP and ping should respond again after some time. It should not require a human power cycle to resolve.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
I used a custom compiled Tasmota (v8 til v12) for years, and it had the same hanging. But I recently decided I had enough and switch to using a precompiled Tasmota, but was quite surprised that it still had the same problem. I have tried on 2 units of D1 mini, so I don't think it is the hardware.

**(Please, remember to close the issue when the problem has been addressed)**

bilogic, 2023-11-03:
@barbudor 

Good point, yes MQTT too. 

Initially (years ago), I realized the issue because my A/C which is controlled by IR failed to turn on/off. That was when I began pings to know immediately when this occured.

Earlier today, I realized that although HTTP/MQTT/pings had no response, my router still shows it was receiving signals from the d1 mini.

barbudor, 2023-11-03:
My guess is a problem on managing interrupts
I think the IR code somewhere needs to disable interrupts to achieve tight code timing and it may happens that under unknown conditions the interrupts remains disabled.
So the base code loop still works, which prevent the watchdog to trigger. But the rest is stuck.

Maybe @s-hadinger has some ideas

I may be able to have a look next week

bilogic, 2023-11-03:
Ok, for a lack of better word, let's use the word `hang`.

The hanging occurs randomly, it does not occur when I'm trying to send IR via MQTT to the A/C.

sfromis, 2023-11-03:
One risk factor with D1 Mini is that clone boards very often have a too small voltage regulator for 3.3V, and having IR stuff wired up may starve the ESP8266 of power for reliable operation.

bilogic, 2023-11-03:
> One risk factor with D1 Mini is that clone boards very often have a too small voltage regulator for 3.3V, and having IR stuff wired up may starve the ESP8266 of power for reliable operation.

Sure, but I never had it hang on me during IR operations. The way my pings work is that, my PC starts beeping the moment the pings fail and it happens at really random times when I'm not sending commands to the D1.

sfromis, 2023-11-03:
What I also meant is that any time of being thrown into lacking power might throw off internal state affecting stability thereafter. Stable 3.3V power is very important to the ESP8266, which does not have mechanisms to detect this.

bilogic, 2023-11-03:
I have an old nodemcu, is its power more stable?

sfromis, 2023-11-03:
They normally have a 500 mA voltage regulator for the 3.3V power, which non-clone D1 Mini also has. Actually, I was lucky to get clones with 500 mA too (I measured that), but 150 mA is very common penny-pinching, which is not really workable when wiring up more than very low power stuff.

Jason2866, 2023-11-03:
The symptoms you see are very typical for an ESP which has a borderline power supply.