arendst/Tasmota #19930: Support for MAX17043 fuel-gauge systems Lipo batteries for Tasmota

peteeee, 2023-11-04:
Max17043 purchased from Amazon.

MAX17043 Monitor Battery Hardware Alert Pin Fuel Gauge System Connected In Circuit Lithium Battery Detection Module for Fuel Gauge Single Li-Ion Batteries

[image]

Using a Wemos D1 with current release of Tasmota-Sensors.bin.  Tasmota 13.2.0
This is the only device connected to the Wemos D1.

Connectivity GPIO 4 - I2C - SDA and GPIO 5 - I2C - SCL

13:01:36.195 CMD: i2cscan
13:01:36.217 MQT: stat/Battery/RESULT = {"I2CScan":"Device(s) found at 0x32"}

Which is different than what is documented.  This is for an outdoor weather station.  Two other I2C devices are connected to two other Wemos D1's and working fine (Lightning sensor and Light sensor).

barbudor, 2023-11-04:
Is there a question here?

peteeee, 2023-11-04:
Can you fix this?  

The current release of Tasmota-Sensors does not work with the Max17043 Fuel Guage board sold on Amazon, Ebay, et al.

barbudor, 2023-11-04:
It's not that it doesn't work but that it's not included
Max17043 is not included in any pre-built variant of Tasmota
You have to [compile your own binary](https://tasmota.github.io/docs/Compile-your-build/)

barbudor, 2023-11-04:
And by the way, the i2c address of the MAX17043 is 0x36 and cannot be changed
So if i2cscan gives you a different value, you have a problem...

peteeee, 2023-11-04:
Thank you @barbudor.  I assumed that the Max17043 was included in the pre-built variant of Tasmota per this document page:

https://tasmota.github.io/docs/I2CDEVICES/#supported-i2c-devices

Wondering here then it may be a clone Max17043.

barbudor, 2023-11-05:
The supported i2c devices pages is listing all drivers available in the code repo.
What is included is each variant is here: https://tasmota.github.io/docs/BUILDS/

When a specific feature/sensor is not listed, you have to assume it is not in any pre-built variant

If you want to avoid building yourself, you may want to try one of the unofficial builds available here:
https://github.com/tasmota/install/tree/firmware/firmware/unofficial
tasmota-allsensors or tasmota-mega may include what you want
Beware that some of those extensive builds are so large that once installed you cannot upgrade them through OTA, only serial flashing

Don't forget to close the issue
Thanks

closed 2023-11-05 by peteeee