arendst/Tasmota #19976: sonoff pow origin/elite powr320D spontaneous boot loop when flashed

bryanb334, 2023-11-11:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
After successfully flashing sonoff pow origin/elite powr320D it enters a boot loop, just constantly rebooting before it gets a chance to start the AP.
I have flashed with both espflasher and the web flasher, both with the same results.
I have tried the current version of Tasmota32.bin (13.2.0) and an older version (12.5.0) with the same result.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
It should start an access point so I can enter wifi details etc.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

bryanb334, 2023-11-11:
unable to run commands ....

sfromis, 2023-11-11:
You posted no details about what happens, like the log. Still, typical issue is that it will not boot to Wifi without using mains power instead of the setup for flashing.

bryanb334, 2023-11-11:
What I am getting is constant rebooting so not time for the wifi access
point to establish.  When I connect to mains, nothing happens.  There are
no LEDs, no AP.  Here is a few seconds of the logs:

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13132
load:0x40080400,len:3036
entry 0x400805e4

00:00:00.002-220/51 HDW: ESP32-D0WD-V3
00:00:00.037-218/51 UFS: FlashFS mounted with 308 kB free
00:00:00.059 CFG: Loaded from File, Count 2
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13132
load:0x40080400,len:3036
entry 0x400805e4

00:00:00.002 HDW: ESP32-D0WD-V3
00:00:00.040 UFS: FlashFS mounted with 312 kB free
00:00:00.045 TFS: File '.settings' not found
00:00:00.046 CFG: Loaded from NVS, Count 2
00:00:00.048 FRC: Some settings have been reset (2)
00:00:00.420 BRY: Berry initialized, RAM used=4711 bytes
00:00:00.441 TFS: File '.drvset003' not found
00:00:00.441 CFG: Energy use defaults as file system not ready or file not
found
00:00:00.449 Project tasmota - Tasmota Version
12.5.0(tasmota32)-2_0_7(2023-04-17T08:07:16)
00:00:00.786 WIF: WifiManager active for 3 minutes
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13132
load:0x40080400,len:3036
entry 0x400805e4

00:00:00.002-220/52 HDW: ESP32-D0WD-V3
00:00:00.040-218/51 UFS: FlashFS mounted with 312 kB free
00:00:00.045-218/51 TFS: File '.settings' not found
00:00:00.051-218/51 TFS: File '.settings.lkg' not found
00:00:00.051-218/51 CFG: Use defaults
00:00:00.240 QPC: Reset
00:00:00.625 BRY: Berry initialized, RAM used=4711 bytes
00:00:00.647 TFS: File '.drvset003' not found
00:00:00.647 CFG: Energy use defaults as file system not ready or file not
found
00:00:00.659 Project tasmota - Tasmota Version
12.5.0(tasmota32)-2_0_7(2023-04-17T08:07:16)
00:00:00.136 WIF: WifiManager active for 3 minutes
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
[... 250 lines cut ...]
12.5.0(tasmota32)-2_0_7(2023-04-17T08:07:16)
00:00:00.521 WIF: WifiManager active for 3 minutes
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13132
load:0x40080400,len:3036
entry 0x400805e4

00:00:00.002-220/51 HDW: ESP32-D0WD-V3
00:00:00.017-218/51 UFS: FlashFS mounted with 308 kB free
00:00:00.030 CFG: Loaded from File, Count 2
00:00:00.040 QPC: Count 4
00:00:00.392 BRY: Berry initialized, RAM used=4711 bytes
00:00:00.405 TFS: File '.drvset003' not found
00:00:00.406 CFG: Energy use defaults as file system not ready or file not
found
00:00:00.413 Project tasmota - Tasmota Version
12.5.0(tasmota32)-2_0_7(2023-04-17T08:07:16)
00:00:00.521 WIF: WifiManager active for 3 minutes
ets Jul 29 2019 12:21:46

sfromis, 2023-11-11:
Try doing as I suggested. Mains power instead of having the flashing setup connected.